liquidctl/liquidctl #1: Get current values in percentage of fan and pump speed

leinardi, 2018-10-03:
Hi, is it possible to get the current values in percentage of fan and pump speed? The status just report the rpm value, I would like to get the percentage value that is currently used.

jonasmalacofilho, 2018-10-03:
It's possible (to implement) but will probably require a initial training step, where we go through the range of PWM duty and read the resulting speeds.

This is necessary to account for:

 - PWM duty translation to actual speed is hardly ever linear
 - specific device revisions and firmware versions (might use different translations)
 - non-stock fans
 - manufacturing differences
 - resistance to airflow, average supplied voltage etc.

leinardi, 2018-10-03:
Thank you for your answer. I thought that the CAM software was able to do this, but after checking better I noticed that they just reset the values to a default value when the software starts (I set the fan speed to 100% on Linux, rebooted on Windows and as soon as the CAM started, it reset the value to 25%).

jonasmalacofilho, 2018-10-03:
Yeap, I think they just show the last speed that they've set.

In particular, this wouldn't work at all with our use of the standalone profiles based on liquid temperature.

jonasmalacofilho, 2018-12-15:
The proof of concept will be kept in the sources – see `extra/krakenduty-poc` – but I'm otherwise closing this: the training required is probably too much hassle for most users.

Maybe we can revisit this in the future; software based curves (#8) could support this with very little effort.

closed 2018-12-15 by jonasmalacofilho