liquidctl/liquidctl #2: Collaboration for GUI application for Linux

leinardi, 2018-10-03:
Hi, I started developing a GUI application that should allow to monitor and configure the NZXT Kraken X on Linux.

I have already a minimal front-end in place that can show the information about speed and temp and also a profile section that can set and show the curve for the selected profile, I just need to feed it with real data:

[image]

In the past days I was checking the available projects that provide Linux supports for the Kraken and I really liked yours.

Would you be interested in a collaboration for the realization of such application?

From your side I think I need just minimal adjustment to the current driver. For example I would like to have exposed the driver value of the `critx` normalization parameter (currently [hardcoded to 60](https://github.com/jonasmalacofilho/liquidctl/blob/master/liquidctl/driver/kraken_two.py#L162)) and an easy way to parse the status and setter verbose data (for example an option to get the output in json format).

Currently I am targeting only Linux with no plans to support other OS since the GUI toolkit I want to use is GTK3. The app will be release under GPL v3 license.

jonasmalacofilho, 2018-10-03:
That's a pretty cool looking app!  I'm very interested in collaborating on it.

How will you be consuming the data?  Externally or through Python?

leinardi, 2018-10-03:
I am very glad to hear that :+1: 
It's a Python 3 app so I should be able to use directly the driver.

jonasmalacofilho, 2018-10-03:
:+1: 

And I can easily expose `critx`, of course.

---

For the status information, maybe some `LIQUID_TEMPERATURE`, `FAN_SPEED`, `PUMP_SPEED`, `FIRMWARE_VERSION` constants would already allow you interpret and manipulate the results from `get_status()` more reliably.

```python
status = { k: v for k, v, u device.get_status() }
fan_speed = status[KrakenTwoDriver.FAN_SPEED]
```

Do you think this would be enough?

---

What do you want to extract from the verbosely logged data?  Just the final profiles or the warnings as well (e.g. too many colors for a given RGB mode)?

leinardi, 2018-10-03:
Oh, those constants should work perfectly, thanks!

Regarding the verbose, I just saw you can get more info with it, like in this case:
```
detaching currently active kernel driver
for liquid temperature >= 20°C, setting pump PWM duty to 60%
write 02 4d c0 14 3c
for liquid temperature >= 22°C, setting pump PWM duty to 60%
write 02 4d c1 16 3c
for liquid temperature >= 24°C, setting pump PWM duty to 60%
write 02 4d c2 18 3c
for liquid temperature >= 26°C, setting pump PWM duty to 60%
write 02 4d c3 1a 3c
for liquid temperature >= 28°C, setting pump PWM duty to 60%
write 02 4d c4 1c 3c
for liquid temperature >= 30°C, setting pump PWM duty to 60%
write 02 4d c5 1e 3c
for liquid temperature >= 32°C, setting pump PWM duty to 60%
write 02 4d c6 20 3c
for liquid temperature >= 34°C, setting pump PWM duty to 60%
write 02 4d c7 22 3c
for liquid temperature >= 36°C, setting pump PWM duty to 60%
write 02 4d c8 24 3c
for liquid temperature >= 38°C, setting pump PWM duty to 60%
write 02 4d c9 26 3c
for liquid temperature >= 40°C, setting pump PWM duty to 60%
write 02 4d ca 28 3c
for liquid temperature >= 42°C, setting pump PWM duty to 60%
write 02 4d cb 2a 3c
for liquid temperature >= 44°C, setting pump PWM duty to 60%
write 02 4d cc 2c 3c
for liquid temperature >= 46°C, setting pump PWM duty to 60%
write 02 4d cd 2e 3c
for liquid temperature >= 48°C, setting pump PWM duty to 60%
write 02 4d ce 30 3c
for liquid temperature >= 50°C, setting pump PWM duty to 60%
write 02 4d cf 32 3c
for liquid temperature >= 52°C, setting pump PWM duty to 60%
write 02 4d d0 34 3c
for liquid temperature >= 54°C, setting pump PWM duty to 60%
write 02 4d d1 36 3c
for liquid temperature >= 56°C, setting pump PWM duty to 60%
write 02 4d d2 38 3c
for liquid temperature >= 58°C, setting pump PWM duty to 60%
write 02 4d d3 3a 3c
for liquid temperature >= 60°C, setting pump PWM duty to 100%
write 02 4d d4 3c 64
reattaching previously active kernel driver
```
but currently I don't have any specific use case where I need it.

So far I just played with the CLI tool and not with the python driver directly. With the app I would like to poll for a status update every 1-2 seconds and, when the user clicks the apply button, set fan/pump profile. Could you please give me some advice if would be better to establish a single connection to the usb device and use it for all the interactions (statuses and sets) or if would be better to create a new connection every time? Also, is it safe to run multiple commands in parallel (e.g. get status and set a channel at the same time) or should I implement a serialization of the commands?

jonasmalacofilho, 2018-10-03:
Well, you can already access this information by providing a custom handler to the `logger` module.  Was for this reason that I changed to a standard logging solution.

However, the profile normalization and interpolation might be particularly useful for a more realistic graph.  If this is indeed confirmed, we might want a better API.  Maybe splitting `set_speed_profile` into two functions (retaining the old API, of course), or adding a hook that you can customize to get this info...

But for warnings and other verbose or debug information I imagine it's sufficient to use the `logger` handlers.

---

I honestly don't know.  Also, I just found out my `connect()` implementation calls `set_configuration` on the device way too much.  So it's something we will need to test.

leinardi, 2018-10-03:
Hey good news, I have managed to get the data from your python library:

```python
import logging
from typing import Optional

from injector import singleton, inject
from liquidctl.driver.kraken_two import KrakenTwoDriver

from gkraken.model import Status
from gkraken.di import INJECTOR

LOG = logging.getLogger(__name__)

LIQUID_TEMPERATURE = 0
FAN_RPM = 1
PUMP_RPM = 2
FIRMWARE_VERSION = 3

@singleton
class KrakenRepository:
    @inject
    def __init__(self) -> None:
        self.__driver: Optional[KrakenTwoDriver] = None
        self.__load_driver()

    def cleanup(self):
        LOG.debug("KrakenRepository cleanup")
        if self.__driver:
            self.__driver.finalize()

    def get_status(self) -> Optional[Status]:
        self.__load_driver()
        if self.__driver:
            status = [v for k, v, u in self.__driver.get_status()]
            return Status(
                status[LIQUID_TEMPERATURE],
                status[FAN_RPM],
                status[PUMP_RPM],
                status[FIRMWARE_VERSION]
            )
        else:
            return None

    def __load_driver(self) -> None:
        if not self.__driver:
            self.__driver = INJECTOR.get(Optional[KrakenTwoDriver])

            if self.__driver:
                self.__driver.initialize()
            else:
                LOG.warning("KrakenTwoDriver not found!")
```

I am calling `initialize()` the first time a driver is found, then I'm holding the instance and reusing it and when the application is closed I am calling `finalize()`.

When the app is open, if I try to run `liquidctl` from a terminal I get `usb.core.USBError: [Errno 16] Resource busy` so, I guess, only one connection at the time is allowed.

Currently I prefer this approach because I don't think there should be more than one application changing the configuration of the Kraken at the same time.

Please let me know if you think this is ok or not.

The code of the app is not ready yet to go public but if you want to have a look and try it, I can add you to the private GitLab repository (at least I think is possible).

My next step will be handling the profiles. If I understood it correctly a user is allowed to add from 1 to 21 different steps for one profile. Is that right?

jonasmalacofilho, 2018-10-04:
> I am calling initialize() the first time a driver is found, then I'm holding the instance and reusing it and when the application is closed I am calling finalize().

While these APIs will remain (for compatibility), you should switch to `connect()` and `disconnect()`.  As of the next release, they will replace `initialize()` and `finalize()`.

(`initialize()` has been clarified to mean actual device initialization, for devices that need it, or a NOOP for the rest)

You should also try to make sure to `disconnect()` even in case of failures.

---

> When the app is open, if I try to run liquidctl from a terminal I get usb.core.USBError: [Errno 16] Resource busy so, I guess, only one connection at the time is allowed.
>
> Currently I prefer this [...]

I haven't given this much thought yet, but I think I agree.  At least in principle, having two unsynchronized processes talking to the same cooler seems like a recipe for disaster.  There might be cases where two different programs need to talk to it, but then it would be nice to have to some concurrency primitives (e.g. locks).

Either way, I'll still see if I can improve concurrency, and better explore our hard limits in this regard.

---

> The code of the app is not ready yet to go public but if you want to have a look and try it, I can add you to the private GitLab repository (at least I think is possible).

Please add me, I'm `jonasmalacofilho` there as well.

---

> My next step will be handling the profiles. If I understood it correctly a user is allowed to add from 1 to 21 different steps for one profile. Is that right?

The cooler (firmware > 2.1.8) accepts 21 different steps per channel.

There is however a big quirk: from my testing the temperature lookup will only be performed once. 
 Practically, this means that both profiles must share the same set of temperatures.

Initially I implemented a method that would generate, in an (more or less) optimal way, intermediate steps between whatever points the user had supplied.  For example, if the user specified a profile like `(30, 60), (36, 90), (45, 100)`, the algorithm would generate the a quasi-optimal profile that would respect this restrictions while keeping step size minimal, in particular in the regions where the profile drastically changes.

However, because of the quirk that makes both profiles share the same set of temperatures, and how I wanted the CLI to work, I had to go back and change this.   So I currently generate profiles using 2°C increments in the range from 20°C to 60°C, with a simple interpolation.

leinardi, 2018-10-04:
Ah I forgot to ask: is there another way that we can use to talk about this project? For me email and/or Google Hangouts would be fine.

jonasmalacofilho, 2018-10-05:
> I wanted to use them but they are not available on the PIP version 1.0.0. Any ETA for the next release?

Had originally planned for the end of September, but I still have some minor details to finalize on the new APIs (mostly for the Smart Device).  Should happen in a couple of weeks.

> Do you know exactly which exceptions could be raised? I don't like to catch a generic exception and also PyLint complains about it.

I use Python standard exceptions in the driver.  I'm not sure what PyUsb might raise.

By the way, I personally think that (in general) a catch all/anything is perfectly fine for things like running (or trying to) finalizers or other clean up code.

> Let me know if you try it.

I'll definitely look at the code!

> Ah I forgot to ask: is there another way that we can use to talk about this project? For me email and/or Google Hangouts would be fine.

Sure, you can email me at the address on my GitHub profile.  And I can send you my Hangouts handle later.

(closing this issue as we'll continue the discussion elsewhere)

closed 2018-10-05 by jonasmalacofilho

Slikkit, 2021-04-24:
Hey. Non-programmer here. 
Would this software work for Corsair Commander Pro?

jonasmalacofilho, 2021-04-24:
Hi,

It's not clear which software you were asking about, but that's ok.

liquidctl *does* support the Commander Pro.  See our list of [supported devices](https://github.com/liquidctl/liquidctl#supported-devices) for a link to the appropriate use  guide.

On the other hand, [GKkraken](https://gitlab.com/leinardi/gkraken), which is what @leinardi was discussing here, does *not* currently support the Commander Pro.

On a final note, please don't revive very old and only marginally related issues, as a lot of people may have followed the original discussion and have now been pinged as a result of your question.