liquidctl/liquidctl #3: NZXT Kraken X61

xsuchy, 2018-11-10:
I have NZXT Kraken X61, which is not supported. Can I help somehow to get this one on the list?

xsuchy, 2018-11-10:
The reverse engineered protocol is here:
  https://github.com/jaksi/leviathan/blob/master/PROTOCOL.md
The usb id for this device is: ID 2433:b200

jonasmalacofilho, 2018-11-10:
That helps, I'll take a first look this weekend.

I might also need some more information from your device.  I've noticed that seem to be some differences in the protocol for each of the many variants of the base ` 2433:b200` device from Asetek.

Are you on Linux?  If so, can you send me the entire output for `sudo lsusb -v -d 2433:b200`?  This will show your `iSerial` value, but in an EVGA cooler this contained some sort of version information (e.g. `CCVI_1.0`) that looked relevant to differentiate between the different models.

jonasmalacofilho, 2018-11-10:
Can you take a look at c4a6627c8d11149c3b663aa5df148091f0d23202?

It's just a starting point.  Let me know if anything works, and if you see any obvious fixes.

There are a few useful options for debugging:

 - `-v`/`--verbose`: enables some helpful messages
 - `-g`/`--debug`: enables debug messages in liquidctl and sets up tracing on PyUSB

```
# liquidctl list -v
# liquidctl status -v
# liquidctl set fan speed 100 -v
# liquidctl set pump speed 100 -v
# liquidctl status -v

jonasmalacofilho, 2018-11-10:
It has just occurred to me that on Windows it will probably be necessary to replace the kernel driver, because this generation might use normally use a proprietary driver from Asetek, instead of the generic Microsoft Human Interface Device (HID) one.

If this is indeed the case – and please let me know so we can mention this in the documentation – it will be necessary to follow the [libusb driver installation instructions for Windows](https://github.com/libusb/libusb/wiki/Windows#driver-installation).  From the suggested methods, [Zadig](http://zadig.akeo.ie/) is probably the easiest:

> Select "Options -> List All Devices", select your NZXT device, change target driver to libusb-win32 or libusbK and install the driver.  CAM software will not detect the device after this.

—_from krakenx/README_

xsuchy, 2018-11-11:
> Are you on Linux?

Yes.

```
$ sudo lsusb -v -d 2433:b200

Bus 003 Device 004: ID 2433:b200  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2433 
  idProduct          0xb200 
  bcdDevice            1.00
  iManufacturer           1 ASETEK
  iProduct                2 690LC
  iSerial                 3 CCVI_1.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               50mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
[m
```

xsuchy, 2018-11-11:
I reseted --hard to your commit and run `sudo pip3 install --editable .` and then:
```
$ liquidctl list -v
Traceback (most recent call last):
  File "/usr/local/bin/liquidctl", line 11, in <module>
    load_entry_point('liquidctl', 'console_scripts', 'liquidctl')()
  File "/home/mirek/projects/liquidctl/liquidctl/cli.py", line 160, in main
    all_devices = list(enumerate(find_all_supported_devices()))
  File "/home/mirek/projects/liquidctl/liquidctl/cli.py", line 77, in find_all_supported_devices
    return itertools.chain(*res)
  File "/home/mirek/projects/liquidctl/liquidctl/cli.py", line 76, in <lambda>
    res = map(lambda driver: driver.find_supported_devices(), DRIVERS)
  File "/home/mirek/projects/liquidctl/liquidctl/driver/base_usb.py", line 61, in find_supported_devices
    drivers.append(cls(dev, description, **kwargs))
TypeError: __init__() missing 2 required positional arguments: 'speed_channel_count' and 'color_channel_count'
```

xsuchy, 2018-11-11:
After:
```
    def __init__(self, device, description, speed_channel_count=None, color_channel_count=None):
```
and running the command again:

```
$ liquidctl list -v
Device 0, NZXT Kraken X (X31, X41 or X61; experimental)
  Vendor: 0x2433
  Product: 0xb200
  Port number: 7
  Serial number: CCVI_1.0
  Hierarchy: AsetekDriver, BaseUsbDriver
```

```
$ liquidctl initialize -v
$
```

```
$ liquidctl status -v
Device 0, NZXT Kraken X (X31, X41 or X61; experimental)
Traceback (most recent call last):
  File "/usr/local/bin/liquidctl", line 11, in <module>
    load_entry_point('liquidctl', 'console_scripts', 'liquidctl')()
  File "/home/mirek/projects/liquidctl/liquidctl/cli.py", line 171, in main
    _device_get_status(dev, i)
  File "/home/mirek/projects/liquidctl/liquidctl/cli.py", line 118, in _device_get_status
    status = dev.get_status()
  File "/home/mirek/projects/liquidctl/liquidctl/driver/asetek.py", line 90, in get_status
    msg = self._end_transaction_and_read()
  File "/home/mirek/projects/liquidctl/liquidctl/driver/asetek.py", line 117, in _end_transaction_and_read
    msg = self.device.read(_READ_ENDPOINT, _READ_LENGTH, _READ_TIMEOUT)
  File "/usr/lib/python3.7/site-packages/usb/core.py", line 988, in read
    self.__get_timeout(timeout))
  File "/usr/lib/python3.7/site-packages/usb/backend/libusb1.py", line 833, in bulk_read
    timeout)
  File "/usr/lib/python3.7/site-packages/usb/backend/libusb1.py", line 936, in __read
    _check(retval)
  File "/usr/lib/python3.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out

```

```
$ liquidctl set fan speed 100 -v
INFO: setting fan PWM duty to 100%
```
And the fan went crazy to full speed. Setting it to 25 slowed down the fan back to normal:

```
$ liquidctl set fan speed 25 -v
INFO: setting fan PWM duty to 25%
```

I did not tried the pump as this is critical, I do not want to demage the cooling and I would prefer to run it after the `status` is operating.

xsuchy, 2018-11-11:
I tried to set
```
_READ_TIMEOUT = 200000
```
but it did not helped.

jonasmalacofilho, 2018-11-11:
I'll sit down a bit later to look into this, but I am wondering if _not_ starting a transaction in `get_status` could help.

xsuchy, 2018-11-11:
Negative sir. It does not help. Still the same timeout.

jonasmalacofilho, 2018-11-13:
See #7 for updates.

jonasmalacofilho, 2018-12-15:
I've figured out a lot more about the USBXpress driver used on Windows.  Based on this, I'll improve the low level implementation, and eventually merge #7 to the main tree, marking it as _experimental._

That said, the underlying protocol is only part of the issue: these bare bones Asetek devices seem to like very specific – and limiting – groupings of commands, and our only reference – leviathan – only shows how to simultaneously change both channels and/or lighting.

I have a few ideas to solve this for liquidctl (some of them already implemented, like the `cooler` pseudo-channel), but more work and experimentation is necessary.  And with no access to a compatible device, there's only so much I can do.

So here's a call to action...

If you have a second generation Kraken (X31, X41, X61), an EVGA CLC with USB (120 CL12, 240, 280) or another device that identifies itself as a "ASETEK 690LC",[1] consider contributing:

 - test the pseudo `cooler` mode
 - try different orders or groupings of the (internal) commands
 - consider tracing the USB communication
 - _if the EULA does not restrict it,[2]_ you can try to reverse engineer the software

You can see the latest developments on #7, and you can chat with me on [gitter](https://gitter.im/liquidctl/Lobby).

---

[1] Preferably, but not necessarily with serial  CCVI_1.0.  
[2] Or if such clause is not legal in your country.

RBlafford, 2019-01-11:
maybe this can help? 
https://github.com/RBlafford/levd

jonasmalacofilho, 2019-01-17:
Thanks, @RBlafford.

It looks like you followed the same approach as leviathan.  I started from that as well, but the main issue I have is how to apply one setting at a time, because of how this CLI is designed: `liquidctl set fan speed` and `liquidctl set pump speed` are two independent invocations, that could come in any order or at any time.

Also, according to a user report, even successfully set fan speed still is automatically reverted in a couple of seconds.  Since you're much more familiar than I with this device: is it able to run without software, once properly configured?  As far as I knew it was, but I'm starting to have some doubts...

RBlafford, 2019-01-17:
@jonasmalacofilho I am not home this week, i will check to see if i have a second gen x61 or not at home and give it a shot. 

AFAIK when I send a command to the device, it stays in that state until power is turned off for the computer. If its changing and then ignoring the command then maybe the protocol is not followed correctly or an incorrect command is being sent.

jonasmalacofilho, 2019-01-18:
@RBlafford ok, thanks.

In the mean time, I'll also see if I can find a device of this generation for cheap.  I would love to support these in liquidctl, but working on the driver without access to the coolers is quite ~~difficult~~ boring: I can't really explore a different approach or test potential new features (e.g. the fan speed profile).

jonasmalacofilho, 2019-01-30:
@xsuchy, can you test/debug the current state of #7?

 - status: `liquidctl status`
 - fixed combined speed: `liquidctl set sync speed <fixed value>`
 - fixed independent speeds: `liquidctl set <fan|pump> speed <fixed value>`
 - variable fan speed: `liquidctl set fan speed <temp> <value> [...]`

@RBlafford, if you can, please test it too.

P.S. `--debug` will dump a bunch of useful (or less so) information, including all libusb calls.

RBlafford, 2019-01-30:
@jonasmalacofilho sure will do!

KeithMyers, 2019-02-03:
I just installed commit 8db7f7f with the repressed pipe error.  The output of:
liquidctl status --debug
root@Serenity:/home/keith/Downloads/liquidctl-8db7f7f24e6b539c53d80bf46ac77f9fd794cf00# liquidctl status --debug
[DEBUG] usb.backend.libusb1: _LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle e36a70 at 0x7fbfbbb56e80>)
[INFO] usb.core: find(): using backend "usb.backend.libusb1"
[DEBUG] usb.backend.libusb1: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbb9f5898>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13a20>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbc7d1e48>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13a58>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13a90>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbb9f5898>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13a20>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13be0>)
[DEBUG] usb.backend.libusb1: _LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle e36a70 at 0x7fbfbbb56e80>)
[INFO] usb.core: find(): using backend "usb.backend.libusb1"
[DEBUG] usb.backend.libusb1: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13be0>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13b70>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13cc0>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13be0>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13b70>)
[DEBUG] usb.backend.libusb1: _LibUSB._finalize_object()
[DEBUG] usb.backend.libusb1: _LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle e36a70 at 0x7fbfbbb56e80>)
[INFO] usb.core: find(): using backend "usb.backend.libusb1"
[DEBUG] usb.backend.libusb1: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13b70>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13c50>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13c18>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13b70>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13c50>)
[DEBUG] usb.backend.libusb1: _LibUSB._finalize_object()
[DEBUG] usb.backend.libusb1: _LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle e36a70 at 0x7fbfbbb56e80>)
[INFO] usb.core: find(): using backend "usb.backend.libusb1"
[DEBUG] usb.backend.libusb1: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13c50>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13be0>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbba13cc0>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fbfbee77e10>)
[... cut ...]
The output of liquidctl status without debug:

root@Serenity:/home/keith/Downloads/liquidctl-8db7f7f24e6b539c53d80bf46ac77f9fd794cf00# liquidctl status
Device 0, Asetek 690LC (NZXT, EVGA or other) (experimental)
report: failed to (pre) open (control: open), ignoring
Liquid temperature          30.0  °C 
Fan speed                   1860  rpm
Pump speed                  2850  rpm
Firmware version        2.10.0.0     

root@Serenity:/home/keith/Downloads/liquidctl-8db7f7f24e6b539c53d80bf46ac77f9fd794cf00#

jonasmalacofilho, 2019-02-04:
@KeithMyers thank you!

jonasmalacofilho, 2019-04-08:
From now on the driver for these second generation Krakens will be improved and maintained directly in the master branch.  @xsuchy, @RBlafford and @cdgriffith, thanks for all the help so far.

Bellow is a quick enumeration of the corresponding features and options.  These have _not_ been tested on a real Kraken, but only on a EVGA CLC.  Because of this, I still expect you'll find a few bugs...

Please test and open specific issues for any problems you notice.

---

First, it is really important to remember that the `--legacy-690lc` option must be used at all times when dealing with these Krakens.  We have yet to find a way to automatically differentiate these devices from more recent generic Asetek 690LC coolers.

_Note: that's a lowercase L in `--legacy-690lc`, as in 690LC.  Perhaps this name is not very good...  I might adjust it before the final 1.2.0 release if it causes too many issues._

The device really must also be initialized after every boot.

```
liquidctl --legacy-690lc initialize
```

After that you can query the device status and perform other operations.

```
liquidctl --legacy-690lc status
```

Both fan and pump speeds can be configured by fixed duty values (i.e. values between 0 and 100, where the % sign is implicit).

```
liquidctl --legacy-690lc set fan speed 50
liquidctl --legacy-690lc set pump speed 75
```

There's a single lighting channel; as no particular name is enforced yet, I'll use 'logo' in these examples. 

The 'fading' mode supports specifying the `--time-per-color` in seconds (the default is 1 second per color).

```
liquidctl --legacy-690lc set logo color fading ff8000 00ff80
liquidctl --legacy-690lc set logo color fading ff8000 00ff80 --time-per-color 2
```

The 'blinking' mode accepts both `--time-per-color` and `--time-off` (also in seconds).  The default is 1 second for each, and whenever unspecified `--time-off` will equal `--time-per-color`.

```
liquidctl --legacy-690lc set logo color blinking 8000ff
liquidctl --legacy-690lc set logo color blinking 8000ff --time-off 2
liquidctl --legacy-690lc set logo color blinking 8000ff --time-per-color 2
liquidctl --legacy-690lc set logo color blinking 8000ff --time-per-color 2 --time-off 1
```

Finally, the cooler supports two more lighting modes: 'fixed' and 'blackout'.  The latter is the only one to completely turn off the LED; however, it also inhibits the visual high-temperature alert.

```
liquidctl --legacy-690lc set logo color fixed 00ff00
liquidctl --legacy-690lc set logo color blackout
```

closed 2019-04-08 by jonasmalacofilho