liquidctl/liquidctl #6: NZXT Smart Device - spectrum-wave mode turns LEDs off

hst12, 2018-11-12:
Please see below the output for liquidctl status and  liquidctl set led color spectrum-wave -g. Other options such as fading or super-fixed work correctly.

Microsoft Windows [Version 10.0.17763.55]
(c) 2018 Microsoft Corporation. All rights reserved.

C:\WINDOWS\system32>liquidctl status
Device 0, NZXT Smart Device
Traceback (most recent call last):
  File "C:\Program Files\Python37\Scripts\liquidctl-script.py", line 11, in <module>
    load_entry_point('liquidctl==1.1.0', 'console_scripts', 'liquidctl')()
  File "c:\program files\python37\lib\site-packages\liquidctl\cli.py", line 169, in main
    _device_get_status(dev, i)
  File "c:\program files\python37\lib\site-packages\liquidctl\cli.py", line 116, in _device_get_status
    status = dev.get_status()
  File "c:\program files\python37\lib\site-packages\liquidctl\driver\nzxt_smart_device.py", line 171, in get_status
    msg = self.device.read(_READ_ENDPOINT, _READ_LENGTH, _READ_TIMEOUT)
  File "c:\program files\python37\lib\site-packages\usb\core.py", line 988, in read
    self.__get_timeout(timeout))
  File "c:\program files\python37\lib\site-packages\usb\backend\libusb1.py", line 851, in intr_read
    timeout)
  File "c:\program files\python37\lib\site-packages\usb\backend\libusb1.py", line 936, in __read
    _check(retval)
  File "c:\program files\python37\lib\site-packages\usb\backend\libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 10060] Operation timed out

C:\WINDOWS\system32>liquidctl set led color spectrum-wave -g
[DEBUG] usb.backend.libusb1: _LibUSB.__init__(<WinDLL 'C:\Program Files\Python37\libusb-1.0.dll', handle 7ffbe6040000 at 0x17e7f81af60>)
[INFO] usb.core: find(): using backend "usb.backend.libusb1"
[DEBUG] usb.backend.libusb1: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69DD8>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69EF0>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69CF8>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69F28>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69E10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69F60>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69DD8>)
[DEBUG] usb.backend.libusb1: _LibUSB._finalize_object()
[DEBUG] usb.backend.libusb1: _LibUSB.__init__(<WinDLL 'C:\Program Files\Python37\libusb-1.0.dll', handle 7ffbe6040000 at 0x17e7f81af60>)
[INFO] usb.core: find(): using backend "usb.backend.libusb1"
[DEBUG] usb.backend.libusb1: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69CF8>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69F60>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69E10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69DD8>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69EF0>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69F28>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC5A160>)
[DEBUG] usb.backend.libusb1: _LibUSB._finalize_object()
[DEBUG] usb.backend.libusb1: _LibUSB.__init__(<WinDLL 'C:\Program Files\Python37\libusb-1.0.dll', handle 7ffbe6040000 at 0x17e7f81af60>)
[INFO] usb.core: find(): using backend "usb.backend.libusb1"
[DEBUG] usb.backend.libusb1: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7F5D8390>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69CC0>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69CF8>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69E10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7F5D8160>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69DD8>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7F5CE9B0>)
[DEBUG] usb.backend.libusb1: _LibUSB.__init__(<WinDLL 'C:\Program Files\Python37\libusb-1.0.dll', handle 7ffbe6040000 at 0x17e7f81af60>)
[INFO] usb.core: find(): using backend "usb.backend.libusb1"
[DEBUG] usb.backend.libusb1: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7F81ABA8>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69F60>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7F5CE9B0>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69E10>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69DD8>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7F81ABA8>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7F219518>)
[DEBUG] usb.backend.libusb1: _LibUSB._finalize_object()
[... 10 lines cut ...]
[DEBUG] usb.backend.libusb1: _LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69CC0>, 1, 0, 0, 0)
[DEBUG] usb.backend.libusb1: _LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69CC0>, 0, 0, 0)
[DEBUG] usb.backend.libusb1: _LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x0000017E7FC69CC0>, 0)
[DEBUG] usb.backend.libusb1: _LibUSB.claim_interface(<usb.backend.libusb1._DeviceHandle object at 0x0000017E7FC5A160>, 0)
[DEBUG] usb.backend.libusb1: _LibUSB.intr_write(<usb.backend.libusb1._DeviceHandle object at 0x0000017E7FC5A160>, 1, 0, array('B', [2, 75, 2, 0, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 2000)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (and 1 padding bytes)
[DEBUG] usb.backend.libusb1: _LibUSB.intr_write(<usb.backend.libusb1._DeviceHandle object at 0x0000017E7FC5A160>, 1, 0, array('B', [3, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 2000)
[DEBUG] usb.backend.libusb1: _LibUSB.release_interface(<usb.backend.libusb1._DeviceHandle object at 0x0000017E7FC5A160>, 0)
[DEBUG] usb.backend.libusb1: _LibUSB.close_device(<usb.backend.libusb1._DeviceHandle object at 0x0000017E7FC5A160>)
[DEBUG] usb.backend.libusb1: _LibUSB._finalize_object()
Exception ignored in: <function _AutoFinalizedObjectBase.__del__ at 0x0000017E7F810158>
Traceback (most recent call last):
  File "c:\program files\python37\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "c:\program files\python37\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "c:\program files\python37\lib\weakref.py", line 548, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "c:\program files\python37\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "c:\program files\python37\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "c:\program files\python37\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000024

hst12, 2018-11-12:
It works when doing liquidctl initialize first! I didn’t realise that the initialize was necessary since other modes worked okay without it.

Here is the output:

C:\WINDOWS\system32>liquidctl initialize
Exception ignored in: <function _AutoFinalizedObjectBase.__del__ at 0x0000027B0E790158>
Traceback (most recent call last):
  File "c:\program files\python37\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "c:\program files\python37\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "c:\program files\python37\lib\weakref.py", line 548, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "c:\program files\python37\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "c:\program files\python37\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "c:\program files\python37\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000024

C:\WINDOWS\system32>liquidctl set led color spectrum-wave
Exception ignored in: <function _AutoFinalizedObjectBase.__del__ at 0x0000024CAC0C0158>
Traceback (most recent call last):
  File "c:\program files\python37\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "c:\program files\python37\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "c:\program files\python37\lib\weakref.py", line 548, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "c:\program files\python37\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "c:\program files\python37\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "c:\program files\python37\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000024

jonasmalacofilho, 2018-11-12:
> It works when doing liquidctl initialize first! I didn’t realize that the initialize was necessary since other modes worked okay without it.

I did not know that either!  I mostly run `initialize` anyway to use (or test) the fan channels.

The requirement probably has something to do with the device needing to know the exact length of the strip to render this effect; the type and number of LED accessories is also learned during the initialization.

I will add to the documentation that some effects require the device to be fully initialized.

---

The `access violation` errors from `_lib.libusb_unref_device(self.devid)` are _probably_ caused by a [know issue](https://github.com/pyusb/pyusb/issues/203) between PyUSB and libusb.  I assume you are running libusb 1.0.22 or later, am I correct?

hst12, 2018-11-12:
That makes sense. Where the number of LEDs are specified for example:
liquidctl set led color super-fixed 000000 000000 000000 000000 000000 000000 000000 000000 000000 000000 202020 202020 202020 202020 202020 202020 202020 202020 202020 202020
This works without needing the initialize and I'm using it in a startup script.

I'm using libusb 1.0.22