liquidctl/liquidctl #9: MacOS Support

ctso, 2018-12-08:
Have you managed to get this working on MacOS?  I'm getting the following error:

```
$ liquidctl status
Device 0, NZXT Kraken X (X42, X52, X62 or X72)
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)
```

Apparently this is an issue with libusb and the fact that MacOS loads a HID driver.  I don't know much about USB, though.  I've also tried running this as root and get the same error.

jonasmalacofilho, 2018-12-08:
Enabling `--debug` could point to some other issue, but after checking the libusb FAQ I think you're right that there's a conflict with the kernel HID driver.

The FAQ lists a [few ways](https://github.com/libusb/libusb/wiki/FAQ#how-can-i-run-libusb-applications-under-mac-os-x-if-there-is-already-a-kernel-extension-installed-for-the-device) to get it to work.  Can you try some of them?

---

I expected it to work on MacOS based on the main documentation of libusb, but I hadn't given much attention to the FAQ : \

ctso, 2018-12-08:
Unloading the HID kext, `com.apple.driver.usb.IOUSBHostHIDDevice`, definitely fixed it.  Downside?  Can't use your HID devices, like your keyboard and mouse.

It sounds like using something like [hidapi](https://github.com/signal11/hidapi) instead of `libusb` will work just fine, though.  I may take a stab at it this weekend and see how it goes.

ctso, 2018-12-09:
I've got a working proof of concept over at https://github.com/ctso/liquidctl/tree/macos.

Seems like if we want macOS support, switching to `hidapi` over `libusb` is the way to go -- it should also work in Linux/Windows.

jonasmalacofilho, 2018-12-09:
Thanks a lot!

I made a few comments directly to your commit, and you can also check my [work on top of it](https://github.com/jonasmalacofilho/liquidctl/tree/macos-extra).  I was able to get both drivers to work on Linux without many issues.

You will notice I tried to change the `BaseUsbDevice.device` field back to being the underlying USB device handle, but ended up with a very ugly hack on `find_supported_devices`; just ignore that, it's actually a symptom of not having yet figured out how to organize the different backends.

At its core, the issue is that not all devices I plan to support are HIDs (e.g. work in #7 for older Krakens and the EVGA CLCs) and, though unlikely, we might still need to stick with libusb on some platforms, even for HIDs.

jonasmalacofilho, 2019-03-11:
The 'master' branch already automatically uses hidapi for HIDs on Mac OS.

On the other platforms, the preferred HID API can be selected with `--hid <module>` (see `liquidctl --help`).

Related: 7300f70a6222 ("Merge PR #19 with modifications: add documentation for Mac OS")

closed 2019-03-11 by jonasmalacofilho