liquidctl/liquidctl #10: NZXT Smart Device doesnot recognize

Jriandono, 2018-12-19:
Hi there I had some problem while trying to controlling the smart device 

I did liquidctl initialize and liquidctl list only my kraken that shown not the smart device 

I look up from the lsusb:
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 046d:c248 Logitech, Inc. G105 Gaming Keyboard
Bus 001 Device 002: ID 1e7d:2e22 ROCCAT 
Bus 001 Device 010: ID 0b05:1872 ASUSTek Computer, Inc. 
Bus 001 Device 007: ID 0b05:185c ASUSTek Computer, Inc. 
Bus 001 Device 009: ID 1e71:170e NZXT 
Bus 001 Device 005: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 008: ID 28de:1142  
Bus 001 Device 006: ID 1e71:1714 NZXT 
Bus 001 Device 004: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

it showed over there, anybody knows why ?

Jriandono, 2018-12-19:
I currently running v.1.1.0

here the outputs for : 

lsusb -d 1e71:1714 -v

Bus 001 Device 006: ID 1e71:1714 NZXT 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x1714 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      91
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

liquidctl list -v -g

Usage:
  liquidctl [options] status
  liquidctl [options] set <channel> speed (<temperature> <percentage>) ...
  liquidctl [options] set <channel> speed <percentage>
  liquidctl [options] set <channel> color <mode> [<color>] ...
  liquidctl [options] list
  liquidctl --help
  liquidctl --version

should I running v1.1.0rc1 ?

jonasmalacofilho, 2018-12-19:
>```
>Usage:
>liquidctl [options] status
>liquidctl [options] set speed ( ) ...
>liquidctl [options] set speed 
>liquidctl [options] set color [] ...
>liquidctl [options] list
>liquidctl --help
>liquidctl --version
>```

This output suggests the command entered was not recognized.  Can you double check the command you're trying to execute for typos?

> should I running v1.1.0rc1

No, v1.1.0 should be fine.

jonasmalacofilho, 2018-12-19:
Actually I just noticed: that help message is not what v1.1.0 would output; for example, it's missing the `initialize` command.  Maybe you simply have multiple instances of the package installed, or the package is not in a consistent state.

jonasmalacofilho, 2018-12-20:
I'm closing this since it's an issue with the incorrect version being used.[1]  This also explains why `liquidctl list -v -g` failed, back then `-g, --debug` didn't exist either.

Since you expected to have v1.1.0 installed, checking `which liquidctl` might help you debug any path, pip, virtual env or other issues your system may have.

And while unlikely – since all installation is taken care of by standard python facilities – please let me know if the issue was somehow caused by anything here.

---

[1] The line `liquidctl [options] initialize` is missing in the usage message you showed.  It was [added on Sep 29](https://github.com/jonasmalacofilho/liquidctl/commit/8908e5740fce7b123fdf4de0b6b65b612d784142), and would be present in v1.1.0.

closed 2018-12-20 by jonasmalacofilho

Jriandono, 2018-12-21:
Yes, it turns out, I'm running v1.0.0, 
Thank you

jonasmalacofilho, 2018-12-21:
> Yes, it turns out, I'm running v1.0.0,

No worries!

> usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

This is a pretty specific error concerning device permissions.  Are you running with root privileges  – or `sudo`?

Handling USB devices is usually restricted to the root user; on most distributions you would need a udev rule to overrule that.

Jriandono, 2018-12-21:
you mean running it with sudo ? or do I have root privileges

Jriandono, 2018-12-21:
> The first one.
I tried to run this :
$ sudo liquidctl initialize

it give me : 
sudo: liquictl: command not found

jonasmalacofilho, 2018-12-21:
> $ sudo **liquidctl** initialize
> sudo: **liquictl**: command not found

Which one was it, liquictl or liquidctl?

Jriandono, 2018-12-21:
I run this, there is a typo last night : 

~$ sudo liquidctl initialize
sudo: liquidctl: command not found

jonasmalacofilho, 2018-12-21:
Try to run the following commands (the first as regular user, the second escalating from regular user to root with `sudo`):

```
$ which liquidctl
$ sudo which liquidctl
```

Jriandono, 2018-12-23:
~$ which liquidctl
/home/wmods/.local/bin/liquidctl

~$ sudo which liquidctl
~$

Jriandono, 2018-12-23:
Fixed the problem by adding groups and set a rule to /etc/udev/rules 
here some step that I did : 

make a new group :
$ sudo groupadd usbusers
$ sudo usermod -G usbuser -a USERNAME 
#change USERNAME into your username 

then make a new file at /etc/udev/rules.d 
$sudo gedit /etc/udev/rules.d/42-usb.rules 

#From what i read from another forum the number is important so make sure your number is not the same with another rules,

then write :
SUBSYSTEM=="usb", MODE="0666", GROUP="usbusers"

save, exit, reboot . then It works for me

jonasmalacofilho, 2018-12-23:
Glad to hear you managed to sort it out!

---

FYI only:

> ```
> $ which liquidctl
> /home/wmods/.local/bin/liquidctl
> ```

The issue you had essentially boiled down to needing to have `/home/wmods/. local/bin` in the PATH environment variable (at the right time).

When you `sudo`, the callee's environment is not exported by the default.  Instead, the environment of root is used, which doesn't know about that path and thus can't find the executable.