liquidctl/liquidctl #13: NotImplementedError for all operations?

flobernd, 2018-12-31:
Hi there,

today CAM stopped recognizing my X62 and as I was sick of this low quality software anyways, I decided to try your script. These are the results:
```
C:\Users\user>liquidctl list
Device 0, NZXT Kraken X (X42, X52, X62 or X72)

C:\Users\user>liquidctl status
Device 0, NZXT Kraken X (X42, X52, X62 or X72)
NotImplementedError: Operation not supported or unimplemented on this platform

C:\Users\user>liquidctl initialize
NotImplementedError: Operation not supported or unimplemented on this platform
```

Not sure if this is a problem on your side, but pretty much every operation gives me an `NotImplementedError` error. Is there anything I can try?

Best regards

flobernd, 2018-12-31:
Looks like my hardware is faulty. Your script is unable to retrieve information like serial-number, firmware-version, etc on my main machine. Works fine on another computer tho. Closing this for now :)

closed 2018-12-31 by flobernd

jonasmalacofilho, 2018-12-31:
Hi there,

Try enabling debug output with `-g`, `--debug`; it might give some insight.

```
liquidctl status -g
```

You can also try power cycling the device, by turning off all power (even standby USB power) or by disconnecting it from the motherboard and the PSU.

flobernd, 2018-12-31:
> You can also try power cycling the device, by turning off all power (even standby USB power)

Already tried this and some other things like reinstalling the driver. Completely removing the device from in the device-manager and re-adding it again resolved the problem for me. Thanks for your help. The script does now work as well.