liquidctl/liquidctl #14: Can I autostart liquidctl on windows?

RaspberryPi5, 2019-01-22:
I found autostart instruction for Linux, but not for windows.

Should I make batch file for this?  If so, can you plz advice me how to write it?

jonasmalacofilho, 2019-01-22:
I create a batch file and then set up a new scheduled task for (every) log on.

The batch file could be really simple, just listing the various invocations of liquidctl you would otherwise do by hand.

But it could also do a bit more, for instance using the LEDs to convey progress or eventual errors.  Mine looks a lot like this:

```bat
rem Use VID/PID to filter the device we want; this way, if we add a new device,
rem or if a new version of liquidctl includes new drivers, our script remains stable
set KRAKEN=--vendor 0x1e71 --product 0x170e

rem Configure the cooler (1/3)
liquidctl %KRAKEN% set sync color off || goto :error

rem Configure the cooler (2/3)
liquidctl %KRAKEN% set fan speed 35 60 45 100 || goto :error
liquidctl %KRAKEN% set ring color super-fixed 000000 8C5A24 8C5A24 8C5A24 || goto :error

rem Configure the cooler (3/3)
liquidctl %KRAKEN% set pump speed 30 50 40 100 || goto :error
liquidctl %KRAKEN% set sync color fixed 8C5A24 || goto :error

rem To finalize wait a second and use a single color covering marquee for dramatic effect
timeout /t 1 /nobreak > NUL
liquidctl %KRAKEN% set ring color covering-marquee 8C5A24 || goto :error

goto :EOF

rem If a liquidctl invocation has failed we'll have jumped here; to communicate that
rem visually, we (try to) set the cooler to pulse red
:error
set code=%errorlevel%
liquidctl %KRAKEN% set ring color pulse ff2608 --speed faster
exit /b %code%
```

Make sure that Python and executables from its packages are available in the context where the batch file will run: in short, `python --version` and `liquidctl --version` should work within a normal Command Prompt window.

If necessary, try installing Python with the option to set the PATH variable enabled, or manually add the necessary folders to the PATH.  Or if you're using Anaconda, try adding the following line to the beginning of the file:

```bat
call %homepath%\Anaconda3\Scripts\activate.bat
```

---

I'll leave the issue open to remind me to add a short section about this to the documentation.

RaspberryPi5, 2019-01-26:
I made batch script like this.

`set KRAKEN=--vendor 0x1e71 --product 0x170e

liquidctl %KRAKEN% set fan speed 35 50 40 60 45 70 50 80 55 90 60 100

liquidctl %KRAKEN% set pump speed 100

goto :EOF`

It works like a charm! Thanks for your support.

commit 8ed288fd (2019-02-25): Make the automation section a first class citizen and extend it to Windows

Closes #14
Closes #17
[skip ci]
closed 2019-02-25 by jonasmalacofilho