liquidctl/liquidctl #15: Odd lighting issue with Smart Device

Dids, 2019-01-23:
When I set a fixed/solid color to all the LED strips on the Smart Device (I have 4), there are 2-3 individual LEDs per strip that ignore the color and usually get assigned a seemingly random color, but there's a pattern to it.

I've previously RE'd the Smart Device protocol for a different project, and it looks like I get the same results with both your method and mine.

I'm hoping that someone else is also seeing this behavior, and that it's not local to my current setup/code. It does work fine with CAM.

jonasmalacofilho, 2019-01-23:
Hi!

Can you run the same command with `--debug` and post the results (and the command itself) here?

And – since you already have experience with this – what does CAM send to Smart Device for a similar setting?

---

Also, do _all_ strips present this issue?  I don't have four strips to test this right now, but if the issue manifests on all of them, I could try replicating the same thing here with only two real strips (and two emulated ones).

jonasmalacofilho, 2019-01-30:
@Dids can you provide more information, like command-line parameters, `--debug` output and firmware/software stack?

> _Also, do all strips present this issue?_  [...] I could try replicating the same thing here with only two real strips (and two emulated ones).

So far I was unable to replicate the issue this way.

Dids, 2019-02-06:
Unfortunately I'm primarily on macOS, where it's been a bit of a hit and miss to get `liquidctl` running correctly, especially with the Smart Device. I eventually had to resort to writing a quick proof-of-concept utility that uses the same logic as `liquidctl`, but it directly uses the Objective-C APIs for USB HID devices.

You did bring up an excellent point regarding the strips, because I've now just realized that this issue is _only_ present in the _last two_ strips, while the first two strips are fine and working exactly as expected.

Regarding the data dumps from CAM, while I do have the raw data somewhere (don't have access to them right now), I believe the logic still matches with what `liquidctl` is doing. I should probably double check though.

jonasmalacofilho, 2019-02-06:
Have you tried branch [macos-extra](https://github.com/jonasmalacofilho/liquidctl/tree/macos-extra) ([diff](https://github.com/ctso/liquidctl/compare/macos...jonasmalacofilho:macos-extra?expand=1))?

As I understand it, it should (tm) be more reliable on Mac OS because it uses the HID driver through hidapi, instead of libusb.  Please let me know which issues remain #9 ("MacOS Support").

---

> this issue is _only_ present in the _last two_ strips

I'll get some extra WS2812B LEDs to reproduce the issue.

icedterminal, 2019-02-16:
> Unfortunately I'm primarily on macOS, where it's been a bit of a hit and miss

I stumbled upon this building a hackintosh. My X62 died two years in and I got an RMA replacement last month. During the one month RMA wait, I used an H7 Plus. I installed the master bulid (required pip3 since macOS runs really outdated python) and it just wouldn't communicate with the device. I later found out my USB SSDT layout excluded the USB port on the board where the kraken was connected. I made the SSDT during the waiting period. After fixing that, it was seen but didn't communicate reliably. I restarted from scratch with macos-extra and indeed it works. I didn't even bother to check the issues tab at first (should have tbh, would have saved time). You should definitely give that version a try.

@[jonasmalacofilho](/jonasmalacofilho): I'm not sure why you started with libusb from the start. From what I have seen and read, hidapi is superior as it's already built on top of libusb. Either way, kudos for getting this off the ground as this is the only repo that has worked for me. I really didn't want to construct some hacky method to make it work.

Dids, 2019-02-18:
@jonasmalacofilho The `macos-extra` branch seems to work:

```
❯ liquidctl status
Device 0, NZXT Kraken X (X42, X52, X62 or X72)
Liquid temperature          26.9  °C 
Fan speed                    487  rpm
Pump speed                  1777  rpm
Firmware version           2.1.8     

Device 1, NZXT Kraken X (X42, X52, X62 or X72)
Liquid temperature          28.5  °C 
Fan speed                    482  rpm
Pump speed                  1759  rpm
Firmware version           2.1.8     

Device 2, NZXT Smart Device
Fan 1                          —     
Fan 2                          —     
Fan 3                          —     
Firmware version           1.0.7     
LED accessories                4     
LED accessory type    Hue+ Strip     
LED count (total)             40     
Noise level                   53  dB
```

What's more, this actually seems to work for _all four LED strips_:  
```
❯ liquidctl --device 2 set led color fixed 000000
```

I guess this means that his issue could be closed, since everything seems to work exactly as expected, although I have no idea why my "replicated logic" still exhibits the issue, while the `macos-extra` branch version of `liquidctl` does not. Still, I'm happy to report that everything works. :)

jonasmalacofilho, 2019-02-18:
@Dids did the original issue manifest itself even when setting all LEDs to black/0x000000?

```
❯ liquidctl --device 2 set led color fixed 000000
```

I ask because padding bytes are also `0x00`, making black a very "easy" color to get right.

jonasmalacofilho, 2019-02-18:
@icedterminal, I wasn't aware of hidapi when I started the project.  Still, after experimenting a bit with it, I think I'll stick with libusb on Linux (possibly Windows as well).

Libusb gives me a few more nobs, is more centrally maintained and, more importantly, not all devices that I want to support are HIDs.

icedterminal, 2019-02-18:
@[jonasmalacofilho](/jonasmalacofilho) Good points. 

Going forward, if you're sticking with libusb, does that mean the macos branch is going to be maintained separate? I can see there are countless forks of hidapi across multiple platforms. So there is no true current master source. Two years of no updates.

If we can work around the libusb bugs that would be great. Not sure if this would help, but this is some documentation on gaining access to devices. https://developer.apple.com/documentation/iokit

I would much rather use libusb so the repo isn't split into two branches.

jonasmalacofilho, 2019-02-18:
I'm not really aware of the bugs libusb might have on Mac OS; though just getting it to bind to the device seems unreasonably error prone.  We could certainly explore fixing libusb and/or adding a option for it to directly use kernel HID drivers on Mac, but I would need some help with that.

For now, my plan is to merge the macos-extra branch by adding an abstract HID driver beneath the specific device drivers.  Its implementation will support both hidapi and libusb, but on Mac OS only the former will be available.  Other platforms will default to libusb, but a `--use-hidapi` will be added mostly to allow convenient testing of the hidapi path under Linux (or Windows).

jonasmalacofilho, 2019-02-18:
@icedterminal

icedterminal, 2019-02-19:
[jonasmalacofilho](/jonasmalacofilho) Your plan for now seems like the best solution. In all honestly, people are using NZXT devices for a hackintosh.

I created a second install to play around. It appears when Apple revamped USB in 10.11 they have a heavy reliance on ACPI. 

> *One of the serious issues most users will face with 10.11.x (and later, including 10.12.x) is the new USB stack in OS X/macOS. It has a much heavier reliance on ACPI*

– https://www.tonymacx86.com/threads/guide-creating-a-custom-ssdt-for-usbinjectall-kext.211311/

Devices and drivers only operate in exclusive mode. Unlike Linux and Windows which can operate in shared mode. As a previous issue pointed out, it requires unloading the kext which then renders all keyboards, mice, etc from working. You can reload the kext if you use a script. Unload kext, run liquidctl, reload kext. But this came with issues. Sometimes I got a kernel panic, other times ports stopped working.

Dids, 2019-02-26:
@jonasmalacofilho Sorry for the delayed responses. Yes, I've been primarily testing/debugging with black, in an attempt to turn off all the LEDs.

I'm still not entirely sure what the issue is/was, but my best guess is a timing issue in my own POC code, or a bug with the native Objective-C APIs/frameworks.

jonasmalacofilho, 2019-02-26:
@Dids thanks for the feedback.

As the issue is related to the libusb-based stack only, which is unusable on Mac OS anyway (at least with HIDs), I'll close this.  Mac users will want to use branch [macos-extra](https://github.com/jonasmalacofilho/liquidctl/tree/macos-extra) for now.

By the way, I'll be merging both backends very soon ([latest diff](https://github.com/jonasmalacofilho/liquidctl/compare/wip-0621?expand=1)) : )

closed 2019-02-26 by jonasmalacofilho