liquidctl/liquidctl #17: Windows Tutorial

cdgriffith, 2019-02-11:
I think it might be nice if the github had a more in depth guide for getting this software working on Windows from the ground up (as most CAM users may not have a lot of Python experience.)

[I wrote a guide here](https://codecalamity.com/replacing-nzxts-cam-software-on-windows-for-kraken/), that you make take any or all of to include for documentation.

jonasmalacofilho, 2019-02-11:
I saw that on r/NZXT a bit earlier, great stuff!

I would pick some parts of it to include here and there, but if you're up for it, I would very much welcome a pull request for a proper page under the `/docs` folder.

Another thing: I want to set up a CI system to build packages for Windows automatically.  Is [build.bat](https://github.com/cdgriffith/liquidctl/blob/0b6845e540c41e518658421d6d09bbc4442f0fef/build.bat#L1) all I need?  Do you know if I can easily cross-package it from Linux/Travis CI, or I'm probably better off with Windows/Appveyor?

---

P.S. On the post you mention how you had to look up the available modes in the code.  I guess the docs aren't as visible as I had hoped for, but you can find (links to) them in the [_Supported devices_](https://github.com/jonasmalacofilho/liquidctl#supported-devices) section.

cdgriffith, 2019-02-11:
Will do! 

PyInstaller was honestly quite the pain on Windows with Python 3.7. Good to document how I got it working though: 

1. Download latest development version (as standard is broken for 3.7) `pip install https://github.com/pyinstaller/pyinstaller/tarball/develop`
2. Modify the `venv\Lib\site-packages\PyInstaller\bindepend.py` file at line 881  (based on https://github.com/pyinstaller/pyinstaller/pull/3956 )  to include: 
```
    if is_win and 'VERSION.dll' in dlls:
        pydll = 'python%d%d.dll' % sys.version_info[:2]
        return getfullnameof(pydll)
```
3. Download the Windows 10 SDK https://developer.microsoft.com/en-us/windows/downloads/windows-10-sdk  (based off their notes here: https://pyinstaller.readthedocs.io/en/stable/usage.html#windows) 
4. Download libusb and add the 32 bit dll in same folder as `build.bat`
5. Update `pip` and `setuptools` via `pip install pip setuptools --upgrade -
-ignore-installed`
5. Then should be able to run `build.bat` 

(For 64-bit Python 3.7 need to update the build.bat to point to x64 folder and grab libusb 64 I believe, not tested.)

 I just tried on linux subsystem Ubuntu 18.04 with the built in 3.6 and am getting an error I had on windows before updating pip and setuptools (even after updating to latest on that system.) Barring figuring that out, should be pretty simple to do in Travis CI. (As it doesn't require linking any DLLs)

---

Ah HA, I totally missed that support page. Thanks for pointing that out!

commit 8ed288fd (2019-02-25): Make the automation section a first class citizen and extend it to Windows

Closes #14
Closes #17
[skip ci]
closed 2019-02-25 by jonasmalacofilho