liquidctl/liquidctl #21: Kraken x62 - minimum pump speed

vecinohk, 2019-03-06:
Hi there,

many thank for project liquidctl - is amazing. I complete uninstalled CAM.

Please is there any change set minimum pump speed to 30-40%? Now is minimum 50%. In my case is 50% = 1940 rpm. I want to less. On my own risk.

I have Kraken x62 with firmware 6.01.

Thanks

Vecíno

`Another important note: while allowed by the firmware, pump speeds bellow 60% are not currently exposed in CAM. Presumably, there are scenarios when such low speeds are not be suitable at all.`

jonasmalacofilho, 2019-03-06:
Up to version 4 the firmware would simply ignore duty values bellow 50%; I'm not sure if that has changed with v6.

You grab the sources with git

```
$ git clone https://github.com/jonasmalacofilho/liquidctl
$ cd liquidctl
# pip install --editable .
```

and try modifying the [minimum pump duty](https://github.com/jonasmalacofilho/liquidctl/blob/14a1b00284e1547f756cb9d90d257283e706bb4f/liquidctl/driver/kraken_two.py#L52) liquidctl expects to parse to 0%, and test how the new firmware responds to values between 0% and 50%.

---

By the way, are fan and pump profiles still functional in v6?  And what changes did you notice with v6, when compared with v4?

vecinohk, 2019-03-06:
I will try it tonight. But I am noob Python user ... is posibile easy compile exe binaries file?

NZXT do not show changelog. Version 6.01 is from 27.02.2019.
Download here: https://sta3-nzxtcorporation.netdna-ssl.com/uploads/download/attachment/753/KrakenXFWUpdate__02.27.19_.zip

Yes fan and pump profiles are still functional in v6.xx.

jonasmalacofilho, 2019-03-06:
> I will try it tonight. But I am noob Python user ... is possibile easy compile exe binaries file?

You don't need to compile an `.exe`.

First, install Python 3.7 (or whatever is the latest) with the option to add Python to your PATH.  Then, manually install libusb, as outlined in [README/System Dependencies](https://github.com/jonasmalacofilho/liquidctl#system-dependencies).

Afterwards, executing

```
pip install --editable .
```

from within the sources directory will install the remaining dependencies (PyUSB, docopt) and expose a `liquidctl` command.

Note: if you placed the v1.1.0 `liquidctl.exe` someplace where it can be found by everything in your system (i.e. somewhere in you PATH, like `C:\Windows`), you might want to remove it, so you'll be sure that you're testing the correct liquidctl.  Otherwise, when you type `liquidctl ` in the command line, Windows will search for it in all locations in PATH in the order they appear in, which might or might not be what you want/expect.

---

> NZXT do not show changelog

Yeap : \

Which is particularly unfortunate since downgrading is not supposed to be possible.

> Yes fan and pump profiles are still functional in v6.xx.

Good to know, thank you so much!

vecinohk, 2019-03-06:
Please help ... I installed Python, install liquidctl, copy files libusb-1.0.pdb, libusb-1.0.lib and libusb-1.0.dll to Windows\System32

But a have this error message: "usb.core.NoBackendError: No backend available"

```
C:\Users\Vecíno\Desktop\liquidctl-master>pip install --editable .
Obtaining file:///C:/Users/Vec%C3%ADno/Desktop/liquidctl-master
Requirement already satisfied: docopt in c:\users\vecíno\appdata\local\programs\python\python37-32\lib\site-packages (from liquidctl==1.1.0) (0.6.2)
Requirement already satisfied: pyusb in c:\users\vecíno\appdata\local\programs\python\python37-32\lib\site-packages (from liquidctl==1.1.0) (1.0.2)
Installing collected packages: liquidctl
  Found existing installation: liquidctl 1.1.0
    Uninstalling liquidctl-1.1.0:
      Successfully uninstalled liquidctl-1.1.0
  Running setup.py develop for liquidctl
Successfully installed liquidctl

C:\Users\Vecíno\Desktop\liquidctl-master>liquidctl list
usb.core.NoBackendError: No backend available
```

jonasmalacofilho, 2019-03-06:
That's probably because you installed 32-bit Python.  In that case, and if I remember correctly, the corresponding (32-bit) libraries must be placed in `\Windows\SysWoW64\`... oh Windows =D

jonasmalacofilho, 2019-03-06:
You can also put the library files in the same folder as the `python.exe` executable.

vecinohk, 2019-03-06:
```
C:\Users\Vecíno\Desktop\liquidctl-master>liquidctl status
Device 0, NZXT Kraken X (X42, X52, X62 or X72)
Liquid temperature          26.5  °C
Fan speed                      0  rpm
Pump speed                  1926  rpm
Firmware version           6.0.1

Exception ignored in: <function _AutoFinalizedObjectBase.__del__ at 0x03331BB8>
OSError: exception: access violation writing 0x00000014
```

vecinohk, 2019-03-06:
Pump is still working on 50%.

```
C:\Users\Vecíno\Desktop\liquidctl-master>liquidctl set pump speed 30 -v
INFO: setting pump PWM duty to 30% for liquid temperature >= 20°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 22°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 24°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 26°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 28°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 30°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 32°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 34°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 36°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 38°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 40°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 42°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 44°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 46°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 48°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 50°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 52°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 54°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 56°C
INFO: setting pump PWM duty to 30% for liquid temperature >= 58°C
INFO: setting pump PWM duty to 100% for liquid temperature >= 60°C
Exception ignored in: <function _AutoFinalizedObjectBase.__del__ at 0x03331BB8>
Traceback (most recent call last):
  File "c:\users\vecíno\appdata\local\programs\python\python37-32\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "c:\users\vecíno\appdata\local\programs\python\python37-32\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "c:\users\vecíno\appdata\local\programs\python\python37-32\lib\weakref.py", line 548, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "c:\users\vecíno\appdata\local\programs\python\python37-32\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "c:\users\vecíno\appdata\local\programs\python\python37-32\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "c:\users\vecíno\appdata\local\programs\python\python37-32\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x00000014

C:\Users\Vecíno\Desktop\liquidctl-master>liquidctl status -v
Device 0, NZXT Kraken X (X42, X52, X62 or X72)
Liquid temperature          26.4  °C
Fan speed                      0  rpm
Pump speed                  1926  rpm
Firmware version           6.0.1

```

jonasmalacofilho, 2019-03-06:
Thanks for the reporting back.  If seems as there is still no way in software to force it to run bellow ~1900 rpm : \

The only other thing I can think of is a hardware mod... (it's something I would like to explore someday, but for now I need to keep my units stock to maintain liquidctl and finish the Linux hwmon drivers)

---

As for the finalizer errors... did you install libusb v1.0.22?

vecinohk, 2019-03-06:
Yes I did. Now I am on v1.0.21 and finalizer errors are gone.

vecinohk, 2019-03-06:
Some people have Kraken x62 with 1400 rpm in silent mode.

jonasmalacofilho, 2019-03-06:
As far as I'm aware (see [thread on krakenx](https://github.com/KsenijaS/krakenx/issues/15#issuecomment-415775501)) they usually have rev 1 units, with slightly different hardware and running v2 (now v5 firmware).

jonasmalacofilho, 2019-03-06:
Note: and while the older revision allows a wider range for the pump speed, it [does _not_ support profiles](https://github.com/jonasmalacofilho/liquidctl/issues/8).

vecinohk, 2019-03-06:
Ok thank you for assistance. That is a pity - I would like have cca 1700 rpm. Now with 2060 rpm is Kraken more noise.

closed 2019-03-07 by jonasmalacofilho

CKGrafico, 2021-01-22:
> That's probably because you installed 32-bit Python. In that case, and if I remember correctly, the corresponding (32-bit) libraries must be placed in `\Windows\SysWoW64\`... oh Windows =D
I've tried with 32 and 64 versions and always same error.

liquidctl list
usb.core.NoBackendError: No backend available

jonasmalacofilho, 2021-01-22:
@CKGrafico

First, is there a particular reason why you can't use the `liquidctl.exe` we supply that already bundles libusb?

Still, if you really need to install libusb manually, you should probably first remove all copies of libusb DLLs from your filesystem, and then re-add them only to corresponding 32 and 64-bit system directories.  You definitively don't want programs finding a libusb DLL that's outdated or that was built for the wrong architecture.

CKGrafico, 2021-01-23:
Yes @jonasmalacofilho, the reason is that I'm a little bit stupid (or sleepy xD) sorry and thanks