liquidctl/liquidctl #22: Build dev11 breaks usb backend

icedterminal, 2019-03-14:
Using dev10 it works flawlessly. I uninstalled dev10, installed dev11 and Terminal reports No back-end. I rolled back to dev10 and it works again.

Here's a demo for you https://youtu.be/EUKUuAwrGRo

If this is just a temporary thing, ignore me and close this. If not, let me know what I can do to help.

jonasmalacofilho, 2019-03-14:
If you tell me the exact commit I can double check that nothing broke _accident,_ but I'm pretty sure the issue is that you hadn't installed libusb, but on dev10 it wasn't yet used on Mac (by default).

However, on dev11 the support for EVGA CLCs landed on the master branch, and as those are not HIDs, probing for them started to use libusb.  HID coolers like the Kraken X62 continue to use hidapi by default, so libusb should not cause issues with the kernel HID driver.

By the way, what is the best way to install libusb on Mac?  I would like to add that to the README...

icedterminal, 2019-03-14:
I merged from your master 15 hours ago. So my fork is level with yours. I cloned that and installed. Which was 15 hours ago.

> but I'm pretty sure the issue is that you hadn't installed libusb

You were not wrong. Added libusb and it worked. Good catch. I need new glasses.

> By the way, what is the best way to install libusb on Mac?

Install [homebrew](https://brew.sh/) then run`brew install libusb` – This one isn't sketchy and maintained by the homebrew project. It automatically adds to path.

jonasmalacofilho, 2019-03-15:
Glad to hear, and thanks for the info about installing libusb; already added to README.

commit 1181a371 (2019-03-15): Note that on Mac libusb can easily be installed through homebrew

Closes #22 ("Build dev11 breaks usb backend"; missing libusb on Mac OS).
closed 2019-03-15 by jonasmalacofilho