liquidctl/liquidctl #23: Build dev11 breaks EVGA CLC 280 support

KeithMyers, 2019-03-14:
Since the add-second-gen-krakens branch comments indicated that support for EVGA CLC 280 coolers was now merged into master, I tried to install the master dev 11 branch.  Everything looked fine until the end of the installation.
`sudo python3 setup.py install
[sudo] password for keith: 
/usr/lib/python3.6/distutils/dist.py:261: UserWarning: Unknown distribution option: 'long_description_content_type'
  warnings.warn(msg)
running install
running bdist_egg
running egg_info
creating liquidctl.egg-info
writing liquidctl.egg-info/PKG-INFO
writing dependency_links to liquidctl.egg-info/dependency_links.txt
writing entry points to liquidctl.egg-info/entry_points.txt
writing requirements to liquidctl.egg-info/requires.txt
writing top-level names to liquidctl.egg-info/top_level.txt
writing manifest file 'liquidctl.egg-info/SOURCES.txt'
reading manifest file 'liquidctl.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'liquidctl.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib
creating build/lib/liquidctl
copying liquidctl/version.py -> build/lib/liquidctl
copying liquidctl/__init__.py -> build/lib/liquidctl
copying liquidctl/cli.py -> build/lib/liquidctl
copying liquidctl/util.py -> build/lib/liquidctl
creating build/lib/liquidctl/driver
copying liquidctl/driver/__init__.py -> build/lib/liquidctl/driver
copying liquidctl/driver/kraken_two.py -> build/lib/liquidctl/driver
copying liquidctl/driver/usb.py -> build/lib/liquidctl/driver
copying liquidctl/driver/base.py -> build/lib/liquidctl/driver
copying liquidctl/driver/asetek.py -> build/lib/liquidctl/driver
copying liquidctl/driver/nzxt_smart_device.py -> build/lib/liquidctl/driver
creating build/bdist.linux-x86_64
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/liquidctl
copying build/lib/liquidctl/version.py -> build/bdist.linux-x86_64/egg/liquidctl
copying build/lib/liquidctl/__init__.py -> build/bdist.linux-x86_64/egg/liquidctl
copying build/lib/liquidctl/cli.py -> build/bdist.linux-x86_64/egg/liquidctl
creating build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/__init__.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/kraken_two.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/usb.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/base.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/asetek.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/nzxt_smart_device.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/util.py -> build/bdist.linux-x86_64/egg/liquidctl
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/version.py to version.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/cli.py to cli.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/kraken_two.py to kraken_two.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/usb.py to usb.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/base.py to base.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/asetek.py to asetek.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/nzxt_smart_device.py to nzxt_smart_device.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/util.py to util.cpython-36.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/entry_points.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/requires.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating dist
creating 'dist/liquidctl-1.2.0.dev11-py3.6.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing liquidctl-1.2.0.dev11-py3.6.egg
Copying liquidctl-1.2.0.dev11-py3.6.egg to /usr/local/lib/python3.6/dist-packages
Removing liquidctl 1.2.0.dev1 from easy-install.pth file
Adding liquidctl 1.2.0.dev11 to easy-install.pth file
Installing liquidctl script to /usr/local/bin

Installed /usr/local/lib/python3.6/dist-packages/liquidctl-1.2.0.dev11-py3.6.egg
Processing dependencies for liquidctl==1.2.0.dev11
Searching for hidapi
Reading https://pypi.python.org/simple/hidapi/
Download error on https://pypi.python.org/simple/hidapi/: [Errno 104] Connection reset by peer -- Some packages may not be found!
Couldn't find index page for 'hidapi' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.python.org/simple/
No local packages or working download links found for hidapi
error: Could not find suitable distribution for Requirement.parse('hidapi')
`

It looks like it is looking to install hidapi which is not found.  I didn't see or I missed any instructions to install that beforehand.  From the output it looks like the installer is supposed to retrieve that on its own but fails.

jonasmalacofilho, 2019-03-14:
I'll take a look, but https://pypi.python.org/simple/hidapi/ seems to load fine here.

Maybe it was an isolated network issue?

KeithMyers, 2019-03-14:
https://pypi.python.org/simple/hidapi/

Loads fine here.  The install script did not find it for some reason.

KeithMyers, 2019-03-14:
Tried again.  Got something different this time.

```
root@Serenity:/home/keith/Downloads/liquidctl-master# python3 setup.py install
/usr/lib/python3.6/distutils/dist.py:261: UserWarning: Unknown distribution option: 'long_description_content_type'
  warnings.warn(msg)
running install
running bdist_egg
running egg_info
creating liquidctl.egg-info
writing liquidctl.egg-info/PKG-INFO
writing dependency_links to liquidctl.egg-info/dependency_links.txt
writing entry points to liquidctl.egg-info/entry_points.txt
writing requirements to liquidctl.egg-info/requires.txt
writing top-level names to liquidctl.egg-info/top_level.txt
writing manifest file 'liquidctl.egg-info/SOURCES.txt'
reading manifest file 'liquidctl.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'liquidctl.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib
creating build/lib/liquidctl
copying liquidctl/version.py -> build/lib/liquidctl
copying liquidctl/__init__.py -> build/lib/liquidctl
copying liquidctl/cli.py -> build/lib/liquidctl
copying liquidctl/util.py -> build/lib/liquidctl
creating build/lib/liquidctl/driver
copying liquidctl/driver/__init__.py -> build/lib/liquidctl/driver
copying liquidctl/driver/kraken_two.py -> build/lib/liquidctl/driver
copying liquidctl/driver/usb.py -> build/lib/liquidctl/driver
copying liquidctl/driver/base.py -> build/lib/liquidctl/driver
copying liquidctl/driver/asetek.py -> build/lib/liquidctl/driver
copying liquidctl/driver/nzxt_smart_device.py -> build/lib/liquidctl/driver
creating build/bdist.linux-x86_64
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/liquidctl
copying build/lib/liquidctl/version.py -> build/bdist.linux-x86_64/egg/liquidctl
copying build/lib/liquidctl/__init__.py -> build/bdist.linux-x86_64/egg/liquidctl
copying build/lib/liquidctl/cli.py -> build/bdist.linux-x86_64/egg/liquidctl
creating build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/__init__.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/kraken_two.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/usb.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/base.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/asetek.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/nzxt_smart_device.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/util.py -> build/bdist.linux-x86_64/egg/liquidctl
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/version.py to version.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/cli.py to cli.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/kraken_two.py to kraken_two.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/usb.py to usb.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/base.py to base.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/asetek.py to asetek.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/nzxt_smart_device.py to nzxt_smart_device.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/util.py to util.cpython-36.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/entry_points.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/requires.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating dist
creating 'dist/liquidctl-1.2.0.dev11-py3.6.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing liquidctl-1.2.0.dev11-py3.6.egg
Removing /usr/local/lib/python3.6/dist-packages/liquidctl-1.2.0.dev11-py3.6.egg
Copying liquidctl-1.2.0.dev11-py3.6.egg to /usr/local/lib/python3.6/dist-packages
liquidctl 1.2.0.dev11 is already the active version in easy-install.pth
Installing liquidctl script to /usr/local/bin

Installed /usr/local/lib/python3.6/dist-packages/liquidctl-1.2.0.dev11-py3.6.egg
Processing dependencies for liquidctl==1.2.0.dev11
Searching for hidapi
Reading https://pypi.python.org/simple/hidapi/
Downloading https://files.pythonhosted.org/packages/c1/86/89df0e8890f96eeb5fb68d4ccb14cb38e2c2d2cfd7601ba972206acd9015/hidapi-0.7.99.post21.tar.gz#sha256=e0be1aa6566979266a8fc845ab0e18613f4918cf2c977fe67050f5dc7e2a9a97
Best match: hidapi 0.7.99.post21
Processing hidapi-0.7.99.post21.tar.gz
Writing /tmp/easy_install-_sb4ohr1/hidapi-0.7.99.post21/setup.cfg
Running hidapi-0.7.99.post21/setup.py -q bdist_egg --dist-dir /tmp/easy_install-_sb4ohr1/hidapi-0.7.99.post21/egg-dist-tmp-apvi9c69
[... cut ...] thon/Compiler/Main.py:367: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /tmp/easy_install-_sb4ohr1/hidapi-0.7.99.post21/hid.pyx
  tree = Parsing.p_module(s, pxd, full_module_name)
hidapi/libusb/hid.c:47:10: fatal error: libusb.h: No such file or directory
 #include <libusb.h>
          ^~~~~~~~~~
compilation terminated.
error: Setup script exited with error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```

jonasmalacofilho, 2019-03-14:
This is something I was afraid of...

I'll look into it.

jonasmalacofilho, 2019-03-14:
In the mean time (I'll be away for work for a few days), the 3dce63b8a232bf8518f7304899b58d53820176bd is probably the best commit from the old branch (for this device).

Though feature wise it's quite a bit far behind what's on master : \

KeithMyers, 2019-03-14:
OK, installed the 3dce63b commit of the old-second-gen-krakens branch. Back in business.  Will watch this issue for when you think the master branch works with EVGA CLC 280's.

jonasmalacofilho, 2019-03-15:
What Linux distro/version are you on, and what versions of pip and setuptools do you have?

KeithMyers, 2019-03-15:
I'm on Ubuntu 18.04.2. 4.15.0-46-generic #49-Ubuntu SMP Wed Feb 6 09:33:07 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.6)

python3-setuptools version 39.0.1-2

jonasmalacofilho, 2019-03-15:
Hi @KeithMyers,

Apparently cython-hidapi is _always_ built locally on Linux.

I was so concerned about how reliably its packages for Windows were (I didn't want to deal with that), and it builds so quickly and smoothly on my main/Linux system that I completely missed that fact.

Try first installing

```
sudo apt-get install python3-dev libusb-1.0-0-dev libudev1
```

and let me know if this fixes building hidapi and installing liquidctl.

KeithMyers, 2019-03-15:
I already had python3-dev and libudev1 installed.  I also already had libusb-1.0.0 installed.  But I did not have libusb-1.0.0-dev installed.  Downloaded and installed the https://github.com/jonasmalacofilho/liquidctl/commit/261493b35a26699ede176e8d09346ed170d35a07
commit.  Still ran into issues where hidpapi is not being compiled correctly.
```
root@Serenity:/home/keith/Downloads/liquidctl-master# python3 setup.py install
/usr/lib/python3.6/distutils/dist.py:261: UserWarning: Unknown distribution option: 'long_description_content_type'
  warnings.warn(msg)
running install
running bdist_egg
running egg_info
creating liquidctl.egg-info
writing liquidctl.egg-info/PKG-INFO
writing dependency_links to liquidctl.egg-info/dependency_links.txt
writing entry points to liquidctl.egg-info/entry_points.txt
writing requirements to liquidctl.egg-info/requires.txt
writing top-level names to liquidctl.egg-info/top_level.txt
writing manifest file 'liquidctl.egg-info/SOURCES.txt'
reading manifest file 'liquidctl.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'liquidctl.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib
creating build/lib/liquidctl
copying liquidctl/version.py -> build/lib/liquidctl
copying liquidctl/__init__.py -> build/lib/liquidctl
copying liquidctl/cli.py -> build/lib/liquidctl
copying liquidctl/util.py -> build/lib/liquidctl
creating build/lib/liquidctl/driver
copying liquidctl/driver/__init__.py -> build/lib/liquidctl/driver
copying liquidctl/driver/kraken_two.py -> build/lib/liquidctl/driver
copying liquidctl/driver/usb.py -> build/lib/liquidctl/driver
copying liquidctl/driver/base.py -> build/lib/liquidctl/driver
copying liquidctl/driver/asetek.py -> build/lib/liquidctl/driver
copying liquidctl/driver/nzxt_smart_device.py -> build/lib/liquidctl/driver
creating build/bdist.linux-x86_64
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/liquidctl
copying build/lib/liquidctl/version.py -> build/bdist.linux-x86_64/egg/liquidctl
copying build/lib/liquidctl/__init__.py -> build/bdist.linux-x86_64/egg/liquidctl
copying build/lib/liquidctl/cli.py -> build/bdist.linux-x86_64/egg/liquidctl
creating build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/__init__.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/kraken_two.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/usb.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/base.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/asetek.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/driver/nzxt_smart_device.py -> build/bdist.linux-x86_64/egg/liquidctl/driver
copying build/lib/liquidctl/util.py -> build/bdist.linux-x86_64/egg/liquidctl
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/version.py to version.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/cli.py to cli.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/kraken_two.py to kraken_two.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/usb.py to usb.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/base.py to base.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/asetek.py to asetek.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/driver/nzxt_smart_device.py to nzxt_smart_device.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/liquidctl/util.py to util.cpython-36.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/entry_points.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/requires.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying liquidctl.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating dist
creating 'dist/liquidctl-1.2.0.dev11-py3.6.egg' and adding 'build/bdist.linux-x86_64/egg' to it
[... 15 lines cut ...]
Writing /tmp/easy_install-pz28lr05/hidapi-0.7.99.post21/setup.cfg
Running hidapi-0.7.99.post21/setup.py -q bdist_egg --dist-dir /tmp/easy_install-pz28lr05/hidapi-0.7.99.post21/egg-dist-tmp-g1_12xt3
/tmp/easy_install-pz28lr05/hidapi-0.7.99.post21/.eggs/Cython-0.29.6-py3.6-linux-x86_64.egg/Cython/Compiler/Main.py:367: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /tmp/easy_install-pz28lr05/hidapi-0.7.99.post21/hid.pyx
  tree = Parsing.p_module(s, pxd, full_module_name)
/usr/bin/ld: cannot find -ludev
collect2: error: ld returned 1 exit status
error: Setup script exited with error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
root@Serenity:/home/keith/Downloads/liquidctl-master# liquidctl status
Traceback (most recent call last):
[... cut ...]
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 574, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 892, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 778, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'hidapi' distribution was not found and is required by liquidctl
```

jonasmalacofilho, 2019-03-15:
Ops, `libudev-dev` is also necessary, for the linker.

KeithMyers, 2019-03-15:
OK.  Success.  The libudev-dev library was the last bit needed.  Successful installation with no errors during compilation.

Thanks.

jonasmalacofilho, 2019-03-15:
Great, but reopening because I still have to fix the documentation ; )

Thanks for letting me know about the issue.

commit f00f280b (2019-04-07): Fix missing dependency to libudev-dev on Ubuntu

Closes: #23 ("Build dev11 breaks EVGA CLC 280 support")
Reported-by: @KeithMyers
closed 2019-04-07 by jonasmalacofilho