liquidctl/liquidctl #24: Adding HUE+

icedterminal, 2019-04-01:
This isn't so much an issue as it is a feature request

I noticed that there is a smart device listed in the code. But those only come with the cases. I have an older NZXT case with the HUE+ (leds) and GRID+ (fan control) devices. I don't use the fan controller as I have no interest in it. I do use the hue for the fan lights. I've obtained the device specific information and added it in. I'm not very good with Python. An amateur really.

What would it take to get the lights controlled through terminal? 

```
Device 0, NZXT Kraken X (X42, X52, X62 or X72)
  Vendor ID: 0x1e71
  Product ID: 0x170e
  Release number: 0x0200
  Serial number: X
  Bus: hid
  Hierarchy: KrakenTwoDriver, UsbHidDriver, UsbDeviceDriver, BaseDriver; HidapiDevice=hid

Device 1, NZXT HUE+ (MCP2200)
  Vendor ID: 0x04d8
  Product ID: 0x00df
  Release number: 0x0101
  Serial number: X
  Bus: hid
  Hierarchy: NzxtSmartDeviceDriver, UsbHidDriver, UsbDeviceDriver, BaseDriver; HidapiDevice=hid
```

akmadian, 2019-04-01:
It took me about 1.25 months to reverse engineer, document, and implement the Hue+ protocol in C#. It's no small thing, but if @jonasmalacofilho wants to work on it, I would be happy to help with the protocol!

It also depends on the scope of an implementation. If it's just lighting control, it would (of course) be easier than 100% support.

icedterminal, 2019-04-01:
Literally just lighting control. That's why I haven't bothered using the nzxt smart device because it's all in one. I would rather let bios handle the fans. I have the rgb fans and the led strips.

jonasmalacofilho, 2019-04-01:
Hi @icedterminal ,

> I'm not very good with Python. An amateur really.
> What would it take to get the lights controlled through terminal?

I can't really work on this right now, but I encourage you to try to contribute a driver yourself.  I can give you pointers along the way and do periodic code reviews.

As a basis (or an alternative) you might want to check out kusti8/hue-plus, specifically the [CLI in hue.py](https://github.com/kusti8/hue-plus/blob/master/hue_plus/hue.py).

@akmadian's (thanks for the offer to help, btw) [NZXTSharp](akmadian/NZXTSharp) is also a good reference.

icedterminal, 2019-04-01:
Well, I'll get started. lol.

akmadian, 2019-04-01:
I was gonna start working on it too lol, @icedterminal if you're attached to it, I'd be happy to let you work on it. 

But, if you're just looking for Hue+ support, I can probably get some basic lighting support in within a couple weeks, and full support in (most likely) a month or less.

icedterminal, 2019-04-02:
You could teach me something. I can sort of make out what is what, but a large amount of it is unknown to me *exactly* what it does and why it's like that.

jonasmalacofilho, 2019-04-02:
Hi @icedterminal,

I think you should start with the connection/disconnection life cycle.  Some references:

 - [akmadian's docs on the protocol](https://nzxtsharp.readthedocs.io/en/latest/Protocols/Hue+.html)
 - [`hue-plus: hue.init` and its callers](https://github.com/kusti8/hue-plus/blob/7ce7c4603c6d0ab1da29b0d4080aa05f57bd1760/hue_plus/hue.py#L396-L403)

On the liquidctl side...

Create a `serial.py` module in `liquidctl/driver/` and implement a base `SerialDeviceDriver` similar to `UsbDeviceDriver` and `UsbHidDriver`.  However, at this level, you will probably need to consider any serial port a potential Hue device.  More on this later...

Create a module for your driver (perhaps `hue-plus`), again in `liquidctl/driver`, and start your driver.  It should extend the `SerialDeviceDriver` you created before.

 - if there's something to be returned by `get_status`, implement that first; it's useful to know if you're in the right path
 - afterwards, you can also start working on the colors; use the existing Kraken and Smart Device drivers as a starting point (from the two, the latter is easier to follow)

---

_Deciding what is the correct serial port_

First, both hue-plus and NZXTSharp (please correct me if I'm wrong, @akmadian) rely on the user to supply the correct serial port to use.  This is not ideal, both in general (IMO, of course) and in the particular case of running `liquidctl get status`.

Hue-plus does something even worse (again, IMO), which is supplying a _default_ port: not only are there still real serial devices out there (particular for expensive and technical things), but, just as with the Hue+, a lot of devices with USB ports might actually just use a bridge similar to the MCP2200.

If there's no way to accurately get the correct port from the OS, we should at least present _all_ ports in `liquidctl list` as "Unknown serial device", and add a option (similar to how `--hid` works) to allow the user to select a specific driver (i.e. the Hue+) for a given serial device.  However, unlike the HID devices, `<driver>.get_status` should be a no-op unless the driver has been set (i.e. the option has been used).

My hope (from reading some of the MCP2200 datasheet) is that there might be a way to query the USB device (the MCP2200 bridge itself) for descriptors of the underlying device, the one that goes to the "virtual" serial port.  @akmadian do you know something about this?

Still, this can be solved a bit later in the development process.  For now, it should suffice to

 - list all serial ports as separate devices; the usual liquidctl device selection methods will apply
 - assume that any connected serial port is a Hue+ (warning, warning, warning)

Do note that this cannot be merged like that, as it could damage hardware far more expensive (or critical) than a $100-$200 LED controller.  If you ask someone to test the patches, make sure they understand the risks and know what they are doing.  And always run `liquidctl list` first, and stop if there's more than one serial device.

jonasmalacofilho, 2019-04-02:
Ops... missed one last bit:

> I see that kusti8/hue-plus has attached the audio sync. Like the Grid, not interested in that.

For completeness, I would like you to consider implementing any features supported in the hardware (this doesn't look to be the case, by the way).  If you really can't implement one, at least account for it when designing your (internal and external) APIs.

akmadian, 2019-04-02:
NZXTSharp's system isn't great, but it essentially tests each active COM port with the "hello" handshake code, if a response is received, then it's found the correct port. If 5 handshake attempts fail, go to the next port. Again, this is not a good method (could potentially wreck hardware if `c0` is a dangerous code on a device), but it's the best I've come up with so far. 

I have not tested querying the MCP2200 itself, but that's a good idea, and I think it should at least be tested. (let me know if it works so I can get it into NZXTSharp!)

icedterminal, 2019-04-02:
Funnily enough I did start picking apart smart device driver. Analyzing the communication and responses from the hue in Windows (using CAM) I was able to see how different the chipset communicates versus the later NZXT devices. Querying the status returned an empty result. It is recognised with list. However, in macOS, status hung and I was unable to terminate with Ctrl C. I had to force close Terminal. This broke the Hue. All the lights went white. It's now COM3 in Windows and unresponsive. As well as undetectable by CAM. Luckily I do have an extra one that was unused. Due to it being boxed for so long it had a firmware update and asked me to restart. The lights returned a saved profile.

I put it all down for the night. As I don't want to cause something else to break. Unfortunately I don't know what or where the error happened. I lost track. n00b mistakes.

jonasmalacofilho, 2019-04-02:
> Funnily enough I did start picking apart smart device driver. Analyzing the communication and responses from the hue in Windows (using CAM) I was able to see how different the chipset communicates versus the later NZXT devices.

Were you analyzing serial port or USB traffic?

> Querying the status returned an empty result. It is recognised with list. However, in macOS, status hung and I was unable to terminate with Ctrl C. I had to force close Terminal.

Both hue-plus and NZXTSharp interact with the "virtual" serial port that the MCP2200 bridge creates, instead of talking to the bridge (i.e. the USB device) itself.  That seems easier and safer.

In theory the bridge doesn't do anything special, and just forwards the raw bytes from the serial port that it exposes to the OS to the on-board controller.<sup>1</sup>  However, it can itself be programmed, and certain USB packets might cause it to misbehave.<sup>2,3</sup>

Additionally, on MacOS the Smart Device uses Hidapi, instead of libusb and raw URBs; that adds another layer for you do deal with, and could explain why the crash only happened there.

> This broke the Hue. All the lights went white. It's now COM3 in Windows and unresponsive. As well as undetectable by CAM. Luckily I do have an extra one that was unused. Due to it being boxed for so long it had a firmware update and asked me to restart. The lights returned a saved profile.

Which device returned to a saved profile (or just a "working" state)?  The first, the second, or both?

On newer devices from NZXT completely powering off the unit (including removing any source of standby power) resets the controller and clears all settings.  I would test that this works on the Hue+ too.

> I put it all down for the night. As I don't want to cause something else to break. Unfortunately I don't know what or where the error happened. I lost track. n00b mistakes.

Hopefully nothing can be bricked _easily,_ but taking a break is a always good approach.  Luckily you have two (presumably) working implementations to use as references, plus CAM and whatever traffic you intercept.<sup>4</sup>

Once you figure out how to correctly connect/disconnect and do basic read/writes to the device, the rest of the work usually goes very smoothly.

---

<sup>1</sup> [Two ways UART over USB devices commonly present themselves](https://rfc1149.net/blog/2013/03/05/what-is-the-difference-between-devttyusbx-and-devttyacmx/)

<sup>2</sup> The [datasheet for the MCP2200](http://ww1.microchip.com/downloads/en/DeviceDoc/200022228D.pdf) says:

> The USB controller in the MCP2200 is full-speed USB 2.0 compliant.
> - Composite device (CDC + HID):
>    + CDC: USB-to-UART communications
>    + HID: I/O control, EEPROM access and initial configuration

<sup>3</sup> [Example MCP2200 control program](https://github.com/thezbyg/mcp2200ctl)

<sup>4</sup> You can still use a USB packet analyzer to see the bytes being read/written (as the USB bridge will forward them from the virtual port to the actual device through USB).  Just don't try to emulate that on the USB level, but take a step back and go through the virtual port.

icedterminal, 2019-04-02:
* I was analysing USB traffic on a specific port while changing settings with CAM.
* Knowing that HID is preferred for macOS, It was foolish of me to try and communicate with it without having that support implemented or even really developed/learned with the basic Libusb.
* The first device was the one that was bricked. It somehow sent an initialise signal to all the RGB devices connected, resulting in all of them going white. It has yet to be restored to a working state. The second device, once CAM updated it, works as normal. Before opening up the second one, I did disconnect the first one to ran it through multiple power cycles, system reboots and driver re-installs. Nothing worked. It's just flat out unresponsive. 
* My lack knowledge didn't know either program was using a virtual port created by the hardware. It's good to know that because I was going about it all the wrong way.
* Previously, done with a laptop battery, I have reset the EEPROM with a light voltage charge to the negative terminal. I replaced the worn out cells with new ones. To prevent BIOS from halting the laptop because of modified hardware saying I need a "genuine battery." The data sheet says user EEPROM. If possible I could always try the method above or somehow rewrite the firmware with the new one. I found the location where CAM downloaded the newer firmwares. These are the first generation models, from 2016. So they have no warranty.

jonasmalacofilho, 2019-04-02:
The MCP2200 device exposes both HID and CDC devices, _each for a different purpose;_ on Mac OS you were talking to the HID device, which apparently is only meant for controlling the bridge itself.

The commands you mentioned should only generate reads (apart from standard HID stuff), but maybe there was something else you tried that inadvertently caused a garbage write?  If you can remember what it was (maybe it's in your bash history), you could try figuring out what it did from reading the code for the [MCP2200 control program](https://github.com/thezbyg/mcp2200ctl) (or, alternatively, corresponding official docs), and from there figure out how to reverse that.

The fact that it continues to appear as "COM3 in Windows" is a good sign, I think.  Looks like the bridge is still working.  Do you also still have a MCP2200 HID?

> The data sheet says user EEPROM.

I'm not sure what's generally saved in the bridge's EEPROM.  I would imagine the "Hue+" firmware resides in the microcontroller itself ([a STM8S207, apparently](https://forum.level1techs.com/t/nzxt-hue-a-look-inside/104836)).

icedterminal, 2019-04-03:
My bash history was mostly disk management, permission repairs and cache rebuilds.

> You probably know more about the lower level stuff than me. I suppose you're right and it should be possible to flash the firmware externally somehow.

It's got to be. CAM does it. You really like links. Every comment you find another resource for something. It's amazing. That link to the control program looks like it might allow me to complete such a task. 

> The fact that it continues to appear as "COM3 in Windows" is a good sign, I think. Looks like the bridge is still working. Do you also still have a MCP2200 HID in the device tree?

Yes. But doesn't explain much why it went from 4 to 3. It's still under the same ACPI plane according to device manager. `/_SB/PCI0/XHC1/XHC14/HS14`

----

OT, neat findings: I've used a tool to access USB flash drive firmware to discover _some_ generic USB devices are all one model and just have firmware caps. An 8, 16 and 32gb generic flash drive all have a max of 32gb. The 8 and 16 have limitations placed on them by the firmware that reports to the OS how much space is there. Reset the firmware and enter the value from the 32gb model and you got three 32gb for a lot less. Most name brand ones are honest though. I can't think of the name at the moment. [MPTool](http://www.flashdrive-repair.com/2015/11/SSS-MP-utility-v2.178.html) might be it, but it's GUI. The one I used was terminal based.

akmadian, 2019-04-03:
In testing, I think I fried a Hue+ by sending too many lighting updates/ second. Iced, I would recommend not exceeding ~90-100 updates/ second. It still showed a COM port on the system, but was not showing in CAM, and was completely unresponsive to handshakes. I haven't tested it recently, but I assume that it's in the same state now.

RE what Jonas said, you definitely want to be sending data to the virtual COM port, and not the MCP2200 device itself. I have already documented the protocol in detail, that can be found [here](https://nzxtsharp.readthedocs.io/en/latest/Protocols/Hue+.html). I would recommend that you do your own tinkering to familiarize yourself with the protocol, but most/ all of the hard work has been done for you already! Let me know if you have any questions RE the protocol docs, I imagine that some parts may be confusing.

icedterminal, 2019-04-04:
>     After powering on from Mechanical Off, or if there have been hardware changes,
>     the devices must be manually initialized by calling `initialize()`.  This will
>     cause all connected fans and LED accessories to be detected, and enable status
>     updates.  It is recommended to initialize the devices at every boot.

Why is this required? This actually isn't necessary on macOS. They are ready to go the moment the system boots from any state. I found out this *was* the problem with the first hue.

With the first hue
* `krx initilize` > `krx list` > `krx status` Terminal hung and status never completed for the hue

With the second hue
* `krx list` > `krx status` Terminal returned an empty result just like Windows PS.

----

Also looked into audio. I left a [note for myself](https://github.com/icedterminal/krx/issues/1) after I discovered a separate dependency for macOS is required to get PyAudio working.

jonasmalacofilho, 2019-04-04:
> Why is this required? This actually isn't necessary on macOS. They are ready to go the moment the system boots from any state.

That call is required _for the Smart Device,_ for detection of the connected accessories.  It causes _that_ device to: (a) test each fan channel response to PWM; and (b) measure the resistance between the FD pin and ground to figure out the type and count of LED accessories.<sup>1</sup>

I don't know how the Hue+ device works, maybe it does the detection automatically when powering on.

> I found out this was the problem with the first hue.
> `krx list` > `krx status` Terminal returned an empty result just like Windows PS.

I still think the issue was, technically, making two "random" writes to the HID control-only endpoint.

This is (IMO) worth discussing to save you from bricking the other device as well.  Of course hindsight is 20-20, but I think you should avoid repeating any of the following:

 - (a) reusing code for a known different protocol (i.e. Smart Device ≠ Hue+)
 - (b) talking to the USB bridge instead of the serial port it exposes to the device controller
 - (b) writing to the HID control-only endpoint of the USB bridge (because on Mac OS we have to disable raw USB packets when we see a HID)

I could very well have made those same mistakes, but by now you should really stop trying to force the Smart Device driver to talk to the Hue+ unit (which is what I understood you did with `krx list; krx status`).

Please, don't brick your backup device! : )

> Also looked into audio. I left a note for myself after I discovered a separate dependency for macOS is required to get PyAudio working.

Can you confirm if the audio mode is a software or hardware/firmware feature?  Requiring PyAudio suggests the former.

I do care about having all features supported in hardware/firmware implemented, but if audio mode is just a software daemon that constantly changes the lighting settings (using the same APIs that will already be exposed to the user), then from the point of view of liquidctl there's else to be implemented.

All actual lighting modes (i.e. APIs that wrap specific messages and commands to the device) should be implemented by liquidctl.  Anything else should not be part of the driver and, at most, would live in [`extra/`](https://github.com/jonasmalacofilho/liquidctl/tree/master/extra) as an example or proof-of-concept.

---

<sup>1</sup> [How Hue+ accessories are detected](https://gist.github.com/jonasmalacofilho/262f382ccca8f414da64d161b9bc081f)

icedterminal, 2019-04-04:
> but by now you should really stop trying to force the Smart Device driver to talk to the Hue+ unit

I  was only using the smart device driver to detect the device. When I called the init that was what what made it stop working. I am working from a blank start reading from multiple references. With a very basic script that only works as far as `list` and includes the logger feature you have implemented. Other things like color and status are not added. Currently looking at channels and colors.

I was only asking a question in regards to the init command you suggest to be done every boot. All of the devices respond and report a status without passing the init command.

It is a software feature. I have had this hue for a long time and it didn't support audio response at first. It was a feature added into CAM much later down the line. Much like how they added CS:GO player health color. I have the Philips Hue kit with Hue Sync installed and it too just reads the audio stream and changes color. Both CAM and Hue Sync have some latency. You can increase the responsiveness based on how quickly you want it to react to the audio. This increases CPU load. macOS seems to restrict more from an audio aspect. For one, to capture or read audio, each program that wants to do that must have a driver to attach to Core Audio. Then you get prompted to allow access in the form of "control your computer". You can run CAM in a VM, but because you cannot pass permissions that way, the audio feature is useless.

jonasmalacofilho, 2019-04-04:
> I was only using the smart device driver to detect the device.

I see, but what you want to detect the is the virtual serial port created by the bridge, not the bridge itself.

If you can get the correct virtual serial port by talking to the bridge, that will be very cool (as we discussed previously).  And, in theory, it should be possible to simply skip the virtual serial port and talk directly through the "modem" endpoint of the bridge.  But in both cases you have to be really careful: you don't want to send color stuff to the HID endpoint, which is only meant to be used for control/reprogramming of the bridge itself.

> I am working from a blank start reading from multiple references.

:+1:

> I was only asking a question in regards to the init command you suggest to be done every boot. All of the devices respond and report a status without passing the init command.

To present a consistent API (both for the CLI and for external callers) liquidctl does require that all drivers implement a `initialize()` method (see:  `liquidctl/drivers/base.py`).  It can be a no-op though, if there's nothing to be done for a particular device.

> [audio] It is a software feature.

Ok, then (from my/liquidctl's point of view) you should ignore it in the driver.

> macOS seems to restrict more from an audio aspect. For one, to capture or read audio, each program that wants to do that must have a driver to attach to Core Audio.

Interesting to know.  Apparently, like USB, audio is another thing where Windows is a lot more permissive (or promiscuous) than other operating systems.

---

By the way, I hope you don't mind that I made an unsolicited code-review of your fork.

akmadian, 2019-04-04:
The Hue+ detects what devices are connected as part of the “hello” handshake process. 

If you sniff the data sent, after the Hue+ responds to the handshake, you can see `8d 01` and `8d 02` sent to the device, and a response for each sent back. These responses contain the data on what is connected where. You can find out more about this and what bytes in the responses mean what in my protocol documentation.

jonasmalacofilho, 2019-05-24:
Closing for now.

@icedterminal feel free to tag me on any commit where you need some help, and I look forward to receiving your pull request.

closed 2019-05-24 by jonasmalacofilho