liquidctl/liquidctl #25: Latest master removing experimental tag from Asetek 690LC coolers fails

KeithMyers, 2019-04-08:
Latest master removing experimental tag from Asetek 690LC coolers fails.  Just tried to compile the latest master to get rid of the !!very experimental!! tag when using the earlier commit.

Hit errors on connecting to https://pypi.python.org/simple/appdirs/:

`Installed /usr/local/lib/python3.6/dist-packages/liquidctl-1.2.0.dev12-py3.6.egg
Processing dependencies for liquidctl==1.2.0.dev12
Searching for appdirs
Reading https://pypi.python.org/simple/appdirs/
Download error on https://pypi.python.org/simple/appdirs/: [Errno 104] Connection reset by peer -- Some packages may not be found!
Couldn't find index page for 'appdirs' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.python.org/simple/
No local packages or working download links found for appdirs
error: Could not find suitable distribution for Requirement.parse('appdirs')
root@Serenity:/home/keith/Downloads/Utils/liquidctl-master# `

jonasmalacofilho, 2019-04-08:
Can you check that it isn't another transient/local network issue?

https://pypi.org/simple/appdirs/ looks fine here and 'appdirs' was already required last night by the time you tested the code in the master branch.

KeithMyers, 2019-04-08:
Closed.  Must have been some transient.  Downloaded the latest master again and compiled.  Found the appdirs this time.

closed 2019-04-08 by KeithMyers

jonasmalacofilho, 2019-04-08:
Thanks for testing things so promptly though! : )

Just to double check (as it should not have been there for a while): is the "!!very experimental!!" tag finally gone for you?

KeithMyers, 2019-04-08:
Yes, the tag is gone now.  Guess the repo must have had glitches or something.  I tried a couple of times in the space of five minutes. Everything working normally again.