liquidctl/liquidctl #26: High reported RPM variation

ltkAlpha, 2019-05-15:
Hi and thanks for all your work!

I'm using an NZXT X52 to cool an 1080 Ti w/ a g12 bracket on my Windows 10 Home. I followed Chris Griffith's guide here https://codecalamity.com/replacing-nzxts-cam-software-on-windows-for-kraken/#using-liquidctl to install and I'm using a startup script to set the following values:

`liquidctl --device 0 set pump speed 60
liquidctl --device 0 set fan speed 30 36  38 40  45 49  53 66  60 100`

The issue is I'm seeing quite a bit of variation in fan and pump speeds as reported by HWInfo 6.04 even under no load. The pump speed is 2060 rpm one moment and 2180 rpm the next (min. 2052, max. 2348 over about 10 minutes) . The fan speed varies less, but for a fixed fan speed a variation between 720-785 (min. 722, max. 778 over the same period) is more than I'd expect. Again, under 0 load. For comparison, over the same period the fans of the other AIO in my system have varied by a single rpm and the pump has a range of 8 rpm. Is this expected behavior?

jonasmalacofilho, 2019-05-17:
Hi!

That pump speed variation is a bit higher than what I personally see on my device, but maybe it's still within acceptable range.

On my device I have also seen some color modes seemly interfering with the (reported) pump and fan speeds, both on their average and deltas.  My hypothesis so far is that more demanding lighting modes (that require more cycles from the embedded controller) "starve" the functions responsible for counting pump/fan pulses.

That said, the fan speed delta seems totally normal to be (either using CAM or liquidctl).

Can you show me the outputs of

```
liquidctl --version
liquidctl list --verbose
```

for my reference?

(you can redact the serial number if you want)

ltkAlpha, 2019-05-17:
`(venv) C:\venv\Scripts>liquidctl --version
liquidctl v1.1.0

(venv) C:\venv\Scripts>liquidctl list --verbose
Device 0, NZXT Kraken X (X42, X52, X62 or X72)
  Vendor: 0x1e71
  Product: 0x170e
  Port number: 11
  Serial number: 6D80*******
  Hierarchy: KrakenTwoDriver, BaseUsbDriver`

ltkAlpha, 2019-05-17:
Hi! Re. your theory - I'm doing nothing w/ lighting via liquidctl. I don't remember touching the relevant setting in CAM either (during the 15 minutes it was installed).

jonasmalacofilho, 2019-05-20:
Well, the default animation – loading – is not the lightest.

But I guess what you can do is try to check if liquidctl itself shows the pump speed varying this much as well.  Because HWiNFO is closed sourced there's not much I can do there.  Anyway, this is _probably_ just a firmware/controller issue (btw, what's your firmware version?).

To rule out a problem your specific device, I would also compare what the cooler reports to the motherboard via the fan-style connector.

ltkAlpha, 2019-05-20:
Ah, I hadn't considered the default animation.  To hopefully minimize the impact of animation I've now added the following to my startup script:
```
liquidctl --device 0 set ring color off
liquidctl --device 0 set logo color off
```

Doing this shows no significant impact on reported rpm, as read by liquidctl status or HWInfo (i.e. it still varies quite a bit), which, by the way, appear to report the same range of values. My firmware is reported by liquidctl as 4.0.4
The chassis channel shows less variation:

[image]
[image]
[image]

jonasmalacofilho, 2019-05-20:
I guess there's not much we can do then.

Looking at the graphs, it does look like bad sampling of the tach pulses.  You could try reporting this to NZXT, so they might fix it in a future firmware release.

closed 2019-05-20 by jonasmalacofilho