liquidctl/liquidctl #27: Kraken x61 'fading' mode blinks instead

AlecVercruysse, 2019-05-17:
Hello, its been a while since I've interfaced with my Kraken x61 (I used to use CAM on windows), so I am not sure if this is a hardware bug or something, but I see no discernible difference between running the command:
`sudo python3 cli.py --legacy-690lc --verbose set logo color blinking FFFFFF 0F0F0F`
and 
`sudo python3 cli.py --legacy-690lc --verbose set logo color fading FFFFFF 0F0F0F`
Both throw no errors, and only print
```
INFO: setting pump duty to 50%
INFO: setting fan duty to 0%
```
The reason I invoke `sudo python3` is that without running as root when I tried to run the `initialize`, I got `usb.core.USBError: [Errno 13] Access denied (insufficient permissions)`. liquidctl was somehow not in my sudo path either.
I'm running Ubuntu 18.04.

Thank you very much for this tool. Its been a while since I've had Windows on my computer and I was just checking around to see if it was possible to light up my x61 on Ubuntu and lo and behold!

jonasmalacofilho, 2019-05-17:
Hi!

I'm glad that you're testing this with some success, as I don't have that particular device myself.

> I see no discernible difference between running the command:
> `sudo python3 cli.py --legacy-690lc --verbose set logo color blinking FFFFFF 0F0F0F`
> and
> `sudo python3 cli.py --legacy-690lc --verbose set logo color fading FFFFFF 0F0F0F`

I can't see anything obviously wrong in [that section](https://github.com/jonasmalacofilho/liquidctl/blob/3a849a578412a4a5f54e812e8df9b559e40e3007/liquidctl/driver/asetek.py#L386-L394) of the code, but I remember reading something about a firmware change related to these modes.

You could try increasing `--time-per-color` (and for 'blinking' only, there's `--time-off` as well) to double check visually that there isn't a more subtle difference between the two modes; maybe CAM defaulted to higher values, or this device doesn't interpret these as being in seconds (as later devices do).

Another possible issue is that 'blinking' (as far as I know) only takes one color... so liquidctl is actually just sending 'blinking FFFFFF' to the device, and the Kraken might interpret this differently than the (similar but newer) device I was able to test on.

Let me know if you somehow manage to get it to work, and once I'm back from vacation I'll also try to search for mentions of similar issues with this and other devices of the same family.

> The reason I invoke `sudo python3` is that without running as root when I tried to run the initialize, I got `usb.core.USBError: [Errno 13] Access denied (insufficient permissions)`. liquidctl was somehow not in my sudo path either.
I'm running Ubuntu 18.04.

Yeap, low level USB access typically requires root privileges.  And most distros by default set sudo to _not_ copy the environment (or aliases) of the user that calls it.  It's something I personally always set up, but then forget how to do it soon after.

You can also make this particular USB device accessible by your user with an `udev` rule.  Though the location I chose for the temporary file required by `--legacy-690lc` is currently also in a privileged path, as I didn't want non privileged users to be able to interfere in what a (normally) privileged executable would do.

> Thank you very much for this tool. Its been a while since I've had Windows on my computer and I was just checking around to see if it was possible to light up my x61 on Ubuntu and lo and behold!

Thank you for the kind words, and for testing and using this particular driver : )

AlecVercruysse, 2019-05-18:
Looks like `--time-per-color` was the issue!
Setting that to 1 gives me the blinking effect I mentioned, but as I increase it to 2 it already starts to look like a fading effect. After further testing, it looks like what's happening is that the hardware samples the "expected color" around every 1 second, and the `--time-per-color` flag translates to 1/2 the period of the cycle (with two colors, essentially what the name implies). So it was only sampling at fully on and fully off.

As for a fix, may I suggest a larger default for that flag (at least for legacy devices), so that the fading effect is shown with a regular call? I can go ahead and make a pull request if you'd like.

jonasmalacofilho, 2019-05-20:
Please do make a pull request (legacy devices only).  I would appreciate if you also took a look at the other default values for the legacy devices.

commit a0a0b5a7 (2019-05-26): Differentiate legacy fading and blinking intervals

This adjusts 20cfb6770ee5 ("Change default time-per-color on legacy
drivers") making the fading effect stand out more (on legacy devices),
while at the same time reverting the blinking mode intervals back to 1
second, as that is plenty.

Closes: #27 ("Kraken x61 'fading' mode blinks instead")
closed 2019-05-26 by jonasmalacofilho