liquidctl/liquidctl #30: Liquidctl doesn't work after reboot sometimes

Untalia, 2019-05-24:
I'm not really sure why but sometimes after a PC shutdown it won't work.  
Any ideas why / how we can fix it?

jonasmalacofilho, 2019-05-24:
These devices do take a while to fully initialize, and might fail if either kernel or user space is too quick.  I've seen these on both Linux and Windows.

Let me know what OS and liquidctl versions you're running, and how you've set them up to initialize, and  can give you some tips on how to mitigate these issues.

jonasmalacofilho, 2019-05-24:
Err, sorry, I missed your edit.

> The colors don't respond to any commands but the fans and the pump do.

This I have _not_ seen yet.  Either way, let me know how your system is set up and we'll try to debug this.

Note for myself: if this is about Windows, I read somewhere about a firmware update to "resolve a [recent] Windows 10 USB issue".

Untalia, 2019-05-25:
OS: Windows 10
LiquidCTL: 4.0.4
Let me know if you need anything else.

jonasmalacofilho, 2019-05-26:
> LiquidCTL: 4.0.4

This is (probably) the firmware version.  Please add the output of `liquidctl --version`.

---

When you experience the issue, do you see any driver/device warnings on Device Manager?

Expanding on my previous note (at the time just for myself), this could be related to (some) Windows USB stack update:

> The recent KrakenX firmware is necessary to fix a USB issue related to recent Windows update. 
–[NZXT's Jason Chang, comment on r/NZXT](https://www.reddit.com/r/NZXT/comments/bp9eaf/is_there_anyway_i_can_downgrade_my_firmware_for/enqaz8b?utm_source=share&utm_medium=web2x)

jonasmalacofilho, 2019-05-26:
I can't seem to reproduce the issue here.  I have firmware 4.0.2 and Windows 10 1809.

In addition to my previous questions:

 - are you on 1905 already?
 - do you have Fast Boot enabled?
 - can you send me the exact commands you use, that work on a cold boot but fail after the system reboots?

Untalia, 2019-05-26:
I have Windows 10 1809, Fast Boot disabled.
Color can't be controlled on both cold boot and restart. (used to work when I first installed the software)
Fan/Pump speed works always.

`liquidctl set pump speed 35 60 38 70 40 80 42 100
liquidctl set fan speed  35 25 38 40 40 60 49 100
liquidctl set ring color fading FFFFFF 4400CC
liquidctl set logo color fixed FFFFFF
`

jonasmalacofilho, 2019-05-26:
Still can't reproduce.  This + "used to work when I first installed the software" + the comment from Jason Chang suggest that it might be a firmware vs. Windows issue.

Please show me the outputs of:

```
liquidctl --version
liquidctl list --verbose
liquidctl set pump speed 35 60 38 70 40 80 42 100 --debug
liquidctl set fan speed 35 25 38 40 40 60 49 100 --debug
liquidctl set ring color fading FFFFFF 4400CC --debug
liquidctl set logo color fixed FFFFFF --debug
```

The following markup may be useful to avoid making the reply too messy with that much information:

```
<details>
<summary>liquidctl ...parameters...</summary>
...output...
</details>
```

Untalia, 2019-05-26:
<details>
<summary>liquidctl (parameters ...)</summary>

```
((venv) C:\Users\Alexander>liquidctl --version
liquidctl v1.1.0

(venv) C:\Users\Alexander>liquidctl list --verbose
Device 0, NZXT Kraken X (X42, X52, X62 or X72)
  Vendor: 0x1e71
  Product: 0x170e
  Port number: None
  Serial number: 496E4F77353
  Hierarchy: KrakenTwoDriver, BaseUsbDriver

(venv) C:\Users\Alexander>liquidctl set pump speed 35 60 38 70 40 80 42 100 --debug
[ERROR] usb.libloader: Libusb 1 (C:\Users\Alexander\venv\Scripts\libusb-1.0.dll) could not be loaded
Traceback (most recent call last):
  File "c:\users\alexander\venv\lib\site-packages\usb\libloader.py", line 114, in load_library
    return lib_cls(lib)
  File "C:\Users\Alexander\AppData\Local\Programs\Python\Python37-32\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application
[ERROR] usb.backend.libusb1: Error loading libusb 1.0 backend
[ERROR] usb.libloader: 'OpenUSB library' could not be found
[ERROR] usb.backend.openusb: Error loading OpenUSB backend
[INFO] usb.core: find(): using backend "usb.backend.libusb0"
[DEBUG] usb.backend.libusb0: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb0: _LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x03034530>)
[ERROR] usb.libloader: Libusb 1 (C:\Users\Alexander\venv\Scripts\libusb-1.0.dll) could not be loaded
Traceback (most recent call last):
  File "c:\users\alexander\venv\lib\site-packages\usb\libloader.py", line 114, in load_library
    return lib_cls(lib)
  File "C:\Users\Alexander\AppData\Local\Programs\Python\Python37-32\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application
[ERROR] usb.backend.libusb1: Error loading libusb 1.0 backend
[ERROR] usb.libloader: 'OpenUSB library' could not be found
[ERROR] usb.backend.openusb: Error loading OpenUSB backend
[INFO] usb.core: find(): using backend "usb.backend.libusb0"
[DEBUG] usb.backend.libusb0: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb0: _LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x03034620>)
[ERROR] usb.libloader: Libusb 1 (C:\Users\Alexander\venv\Scripts\libusb-1.0.dll) could not be loaded
Traceback (most recent call last):
  File "c:\users\alexander\venv\lib\site-packages\usb\libloader.py", line 114, in load_library
    return lib_cls(lib)
  File "C:\Users\Alexander\AppData\Local\Programs\Python\Python37-32\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application
[ERROR] usb.backend.libusb1: Error loading libusb 1.0 backend
[ERROR] usb.libloader: 'OpenUSB library' could not be found
[ERROR] usb.backend.openusb: Error loading OpenUSB backend
[INFO] usb.core: find(): using backend "usb.backend.libusb0"
[DEBUG] usb.backend.libusb0: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb0: _LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x030346C0>)
[ERROR] usb.libloader: Libusb 1 (C:\Users\Alexander\venv\Scripts\libusb-1.0.dll) could not be loaded
Traceback (most recent call last):
  File "c:\users\alexander\venv\lib\site-packages\usb\libloader.py", line 114, in load_library
    return lib_cls(lib)
  File "C:\Users\Alexander\AppData\Local\Programs\Python\Python37-32\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application
[ERROR] usb.backend.libusb1: Error loading libusb 1.0 backend
[ERROR] usb.libloader: 'OpenUSB library' could not be found
[ERROR] usb.backend.openusb: Error loading OpenUSB backend
[INFO] usb.core: find(): using backend "usb.backend.libusb0"
[DEBUG] usb.backend.libusb0: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb0: _LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x03034710>)
[DEBUG] usb.backend.libusb0: _LibUSB.open_device(<usb.backend.libusb0._usb_device object at 0x03034530>)
[DEBUG] usb.backend.libusb0: _LibUSB.get_configuration(8989168)
[... 290 lines cut ...]
[DEBUG] usb.backend.libusb0: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb0: _LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x03A54710>)
[DEBUG] usb.backend.libusb0: _LibUSB.open_device(<usb.backend.libusb0._usb_device object at 0x03A54530>)
[DEBUG] usb.backend.libusb0: _LibUSB.get_configuration(28125680)
[DEBUG] usb.backend.libusb0: _LibUSB.ctrl_transfer(28125680, 128, 8, 0, 0, array('B', [0]), 100)
[DEBUG] usb.backend.libusb0: _LibUSB.get_configuration_descriptor(<usb.backend.libusb0._usb_device object at 0x03A54530>, 0)
[DEBUG] liquidctl.driver.kraken_two: write 02 4c 02 01 02 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff (and 33 padding bytes)
[DEBUG] usb.backend.libusb0: _LibUSB.get_configuration_descriptor(<usb.backend.libusb0._usb_device object at 0x03A54530>, 0)
[DEBUG] usb.backend.libusb0: _LibUSB.get_interface_descriptor(<usb.backend.libusb0._usb_device object at 0x03A54530>, 0, 0, 0)
[DEBUG] usb.backend.libusb0: _LibUSB.get_configuration_descriptor(<usb.backend.libusb0._usb_device object at 0x03A54530>, 0)
[DEBUG] usb.backend.libusb0: _LibUSB.get_endpoint_descriptor(<usb.backend.libusb0._usb_device object at 0x03A54530>, 0, 0, 0, 0)
[... cut ...]
[DEBUG] liquidctl.driver.kraken_two: write 02 4c 02 01 22 00 44 cc 44 00 cc 44 00 cc 44 00 cc 44 00 cc 44 00 cc 44 00 cc 44 00 cc 44 00 cc (and 33 padding bytes)
[DEBUG] usb.backend.libusb0: _LibUSB.intr_write(28125680, 1, 0, array('B', [2, 76, 2, 1, 34, 0, 68, 204, 68, 0, 204, 68, 0, 204, 68, 0, 204, 68, 0, 204, 68, 0, 204, 68, 0, 204, 68, 0, 204, 68, 0, 204, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 2000)
[DEBUG] usb.backend.libusb0: _LibUSB.release_interface(28125680, 0)
[DEBUG] usb.backend.libusb0: _LibUSB.close_device(28125680)
 ...)
```
</details>

jonasmalacofilho, 2019-05-27:
Try to replace libusb0 (I'm guessing from `libusb-win32`) with libusb-1.0.21 ([as recommended in the README](https://github.com/jonasmalacofilho/liquidctl#additional-requirements-on-windows)), or use a [pre-build binary for Windows](https://github.com/jonasmalacofilho/liquidctl/releases).

You might also try appending `--hid hid` to the commands, after updating to liquidctl 1.2.0rc1; this will make it use the Microsoft HID Driver directly, instead of libusb, and might reveal if the problem is related to our side of the USB stack.

Untalia, 2019-05-28:
I can't find any information about replacing libusb0 after I downloaded libusb-1.0.21.
I extracted the MS64 folder but I don't know where to move on from there.

jonasmalacofilho, 2019-05-28:
Assuming I guessed correctly, first uninstall libusb-win32.

Then, place the contents of the MS64 folder either next to the python executable you're using (since you're using a venv, it should be in there), or globally inside `C:\Windows\System32`.  The dynamic linker should check those locations automatically, without the need of further configuration.  You can also check where else you have dlls required by this venv's python, and place the files there.

Or just grab the pre-build liquidctl 1.2.0rc1 binary for Windows from the ["Releases"](https://github.com/jonasmalacofilho/liquidctl/releases) tab (file: liquidctl-1.2.0rc1-bin-windows-x86_64.zip); it already bundles libusb-1.0.21.

jonasmalacofilho, 2019-05-29:
In the mean time, I tried reproducing the issue by setting my own device to use libusb0 (with Zadig).  No luck...

It could be an issue with firmware v4.0.4, I have received one other report of inconsistencies with it.  But I'm not ruling out a bug in liquidctl yet.  As there is no official spec, it's possible that the protocol simply changed on firmware 4.0.4, and that some adjustments to liquidctl are necessary.

I suggest you upgrade the firmware to v6.x first.  If the issue remains (and you can confirm that it is specific to liquidctl [i.e. that CAM works]), then I'll ask you for some USB traffic captures to determinate what needs to be fixed.

Untalia, 2019-06-02:
Do you use Discord?
I'd prefer we use it instead of the replies here.

>

jonasmalacofilho, 2019-06-04:
I generally prefer to track issues on open-source projects publicly and next to their code/repository.  This keeps as much information as possible available for other users or contributors.

But if you think we really need to talk synchronously to debug (or discuss) something specific, let me know what that is and we can try to set something up.

jonasmalacofilho, 2019-06-06:
Most relevant points so far:
 - everything used to respond to commands; now colors don't, while pump and fan speeds still do
 - device runs firmare v4.0.4
 - liquidctl has been set up to use libusb 0.1
 - probably not using the default Microsoft HID Driver (I couldn't get libusb 0.1 to work at all without using a custom driver)

_Assuming the issue lies in our implementation of the (unspecified) protocol..._
 - **test: reset the device (turn off the PSU for 15 seconds and ensure no USB type-C devices are connected[1]); do the colors respond to at least one invocation of liquidctl (before other commands for pump/fan/status)?**
 - if they still don't respond at all, something _else_ must have changed, as neither liquidctl nor the firmware has the capability to update itself autonomously; the bug must be subtle (maybe a timing issue, maybe something else) and, as this can be hard to debug, we should probably rule out all the other possible causes for the problem first

_What if the issue lies with libusb 0.1 or the (I'm guessing) custom driver in use?_
 -  **test: revert back to libusb 1.0.21 and the Microsoft HID Driver; reset the device; do the colors respond to liquidctl invocations?**

_What if this is an issue between the firmware and Windows/the kernel driver?_
 - NZXT has stated that a firmware update is needed on Windows; maybe the HID drivers is more strict nowadays
 - **test: upgrade the firmware; reset the device; do the colors respond to liquidctl invocations?**

_What if this is an issue with your specific device?_
 - **visually, what are the ring and logo stuck on? the default white "loading" animation or something else?**
 - **does CAM work?**

[1] Apparently the USB bus in some boards can use power from devices connected through Type-C ports; at least one user has reported that this can keep the controller in the Kraken powered on even if the PSU has been turned off.

jonasmalacofilho, 2019-06-13:
This is a common issue, and apparently completely unrelated to liquidctl.  In both reports I saw recently, upgrading the firmware solved the problem.¹²

¹ _"When I start up my pc the lights just stay like this (pic) and i cant chang it in CAM app. I can change the fan&pump tho":_ user also had firmware 4.0.4 ([see on r/NZXT](https://www.reddit.com/r/NZXT/comments/c04g7a/so_ive_got_a_problem_with_kraken_x62_when_i_start/))  
² _"Kraken x62 lighting locked":_ the issue also affects rev1 devices ([see on r/NZXT](https://www.reddit.com/r/NZXT/comments/bxejhp/kraken_x62_lighting_locked/))

closed 2019-06-13 by jonasmalacofilho