liquidctl/liquidctl #31: Support for nzxt e series digital power supplies

jnettlet, 2019-06-08:
I have started reverse-engineering the pmbus protocol used by CAM to talk with the NZXT e500 and associated digital power supplies.  Is this something that would you would be interested in adding to liquidctl?

jonasmalacofilho, 2019-06-08:
Actually, I would be very interested in having support for those power supplies (and others) in liquidctl!

So it follows the PMBus spec?  I had no idea...  Can you share what you have figured out so far with me?

jnettlet, 2019-06-08:
I have some dumps I can share, and so far this is the format I have worked out.  One some functions there appears to be some crc formula being used for data integrity.   So far I have focused on reading the data and not writing.

Host command
byte 1 = command 0xad / 173
byte 2 = 0x00 always for a read command - most likely used for writes
byte 3 seems to be datalength  0x03 read word 0x02 read  byte 
byte 4 seems to be write length command because it is generally 0x01 for simple commands  There appear to be other methods that are larger perhaps for pmbus process call support
byte 5 = Address Byte
byte 6 = Command Byte
byte 7+ appears to be some sort of checksumming

Data response
byte 1 = 0xaa / 170  - I assume this corresponds to valid response
byte 2 = I believe this is the length of the data to parse.  0x03 seems to be a word etc.  byte 3 of the host command matches byte 2 of the data response so I assume that data length
following bytes seem to be data, and a checksum.

So
ad00030160fc02048c00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
is command
read 0xad
word 0x030x01
address 0x60 - This is my assumption because for most read commands it is consistent
command 0xFC which appears to be TI's MFR command for firmware information.  The last 3 bytes I have no idea about, although they tend to repeat with only slight variations. 

The e500 is device 3.2.0 and 3.2.1

[e500_init.capture.zip](https://github.com/jonasmalacofilho/liquidctl/files/3268279/e500_init.capture.zip)

jonasmalacofilho, 2019-06-09:
So it's encapsulating PMBus in USB (using the Mass Storage class).  Any idea what chip is used for this USB to PMBus bridge?

---

Just so you know, I don't have the device, so I won't be able to help a lot with the protocol.  But don't hesitate to ask me if you have any questions about writing the liquidctl driver.

jnettlet, 2019-06-09:
It is a usbhid interface just like the smart controller etc.  The DSP
it is using is a TI  UCD3138.  This is actually a seasonic IP, that is
what it identifies as.  seasonic usb2i2c

jonasmalacofilho, 2019-09-08:
@jnettlet did you figure out any more of the protocol?

jnettlet, 2019-09-08:
@jonasmalacofilho Unfortunately, that project has been on the back burner for the last couple of months.  Although just the last couple of weeks I have pulled that machine back out and was going to start looking into this again.

jonasmalacofilho, 2019-09-27:
Replacing with #46 ("Improve support for Seasonic/NZXT E-series PSUs") since some initial support has already been merged.

closed 2019-09-27 by jonasmalacofilho