liquidctl/liquidctl #32: NZXT Smart Device v2?

jahluwalia, 2019-06-24:
The new cases will be shipping with Smart Device v2. Any idea if liquidctl will be able to control that device?

Thanks,
Jas

jonasmalacofilho, 2019-06-24:
Someone would need to contribute a new driver for it.  But it should be fairly easy to implement one.

Otherwise I would need to get my hands on the device (and/or the protocol spec).

jahluwalia, 2019-07-01:
I ended up getting the older version of the case that just had Smart Device v1. I think you can close this issue unless you want to keep it open for when the new versions are released.

jonasmalacofilho, 2019-07-02:
I'll leave it open for now, in case someone else wants to contribute a driver once the new version hits the market.

rileyweber13, 2019-08-24:
I have a new case with the Smart Device v2. I have plenty of programming experience but have never written a driver. How can I help? Do you have any resources on learning to write a driver?

Also, I'd be happy to provide the 'protocol spec' if having the device enables me to do so.

olielvewen, 2019-08-31:
Great information.

If I was aware about PyusbTutorial, HID Specifications ( I have some difficulties to find them),  I do not know about nzxt_smart_device.py and also Linux Kernel USB Documentation. Very interesting.

People use Wireshartk on a virtual machine with Windows for making Reverse Engineering. A great [blog entry (in French)](https://blog.flozz.fr/2016/03/27/steelseries-rival-100-reverse-engineering-dun-peripherique-usb/) of my friend [Flozz](https://blog.flozz.fr/) who is too the author of [Rivalcfg](https://github.com/flozz/rivalcfg) thanks to his method.

CaseyJRS, 2019-09-30:
I've posted an early experimental version of the modified driver [here](https://www.tonymacx86.com/threads/success-gigabyte-designare-z390-thunderbolt-3-i7-9700k-amd-rx-580.267551/page-814#post-2010636) along with a bit of detail about what works and what does not yet work. Briefly:

1. Fan channels are named fan1, fan2, fan3.
2. Lighting channels are named sync1, sync2.
3. "set color" is mostly working.
4. "get status" is partially working. It reports fan RPMs and noise level.
5. "set fan" is not yet started.

Having recently built a Hackintosh around the NZXT H510 Elite, I thought I would jump in and attempt the driver modification myself. So I'm really doing it for my own selfish reasons, but hopefully it can be submitted here once it's a bit more mature.

Jonas' August 26 post above was very helpful.

jonasmalacofilho, 2019-10-01:
@CaseyJRS

Very nice!  I'll try to follow your updates there.

I'm assuming you are capturing the USB traffic... can you share the raw captures?

jonasmalacofilho, 2019-10-01:
Also...

> So I'm really doing it for my own selfish reasons, but hopefully it can be submitted here once it's a bit more mature.

Lots of things happen faster because they scratch someone's particular itch.

And, @CaseyJRS, please don't worry about submitting your contributions here.  I'll carefully review your code and I might suggest or even ask for some changes before merging your patches, but I'll also be there to help you in the process.

Also, consider that you can open _draft_ pull requests when you think some changes are not quite ready yet for prime time.

By the way I have only glanced at the code so far... let me know when you want me to do a real code review.  But from this quick glance:
 - (a) I'm surprised with the similarities, this might mean we're not far from supporting the Hue 2 as well;
 - (b) I think you were right to keep the new device under the same module;
 - but (c), depending on how many conditional branches we ended up with, I might suggest using subclasses for the different families of devices.

CaseyJRS, 2019-10-01:
@jonasmalacofilho 

This is my first foray into Python coding and I'm surprised by the richness of the language. Coming from a C, C++, C# background, it's refreshing to see all of STL and more seamlessly integrated into the Python syntax. I mention this just in case you spot rookie coding mistakes!

I'm using **Free USB Analyzer** on Windows to capture raw byte streams. I tried **Wireshark**, but immediately gave it up in favor of Free USB Analyzer which has, in my opinion, an amazing set of user-friendly ways of selecting how to view a USB conversation.

I'll capture and post a number of raw byte streams this evening, and provide some comments on the context of those streams. I can certainly use some help to determine such things as:

1. Where is the firmware version info?
2. Where is the LED accessories info?
3. How to parse DC vs PWM for the fan channels?
4. Is fan voltage and fan current information available?

There seem to be 2 status message streams (each 64 bytes) returned by the Smart Device V2. I've been able to parse a fair bit of information from the first stream of 64 bytes, but not from the second. Some of the above information may (or may not) be in the second stream.

Anyway, I'll copy-and-paste a number of relevant raw streams hopefully starting this evening.

CaseySJ, 2019-10-02:
@jonasmalacofilho 

As promised, attached are two files in the ZIP:

1. NZXT USB Log Raw Data.rtf (Rich Text Document -- WordPad) with my current understanding of the protocol.

2. Raw USB.htm (can be opened in web browser). This file shows a more lengthy USB bidirectional conversation. It includes "Setup Packet".

Hope this helps...

[NZXT USB Log.zip](https://github.com/jonasmalacofilho/liquidctl/files/3679216/NZXT.USB.Log.zip)

jonasmalacofilho, 2019-10-02:
@CaseySJ

Again, very nice work!

> 1. Where is the firmware version info?

What version is reported by CAM?

> 2. Where is the LED accessories info?
> 3. How to parse DC vs PWM for the fan channels?
> (from RTF file) I thought they also represented lighting accessory types and fan types, but when I disconnect the HUE 2 LED Strip, these values still do not change.

I also expect these flags to be easy to spot when comparing packets from different combinations of fans and accessories.

Did you reinitialize the device after disconnecting the strip, either by power cycling it or by restarting CAM?  At least on v1 devices that's the only time when the device checks what's connected to it.

On this topic, `initialize` probably need some work too.  It seems that on v2 devices the commands for that are different (probably `\x10\x01` and `\x20\x03`).  The next messages could also be part of this (`\x60\x03` and `\x60\x02..`).

> It includes "Setup Packet".

Those control messages can (probably) be ignored; everything that's important to us seems to happen within the HID layer.

> 4. Is fan voltage and fan current information available?

I couldn't easily spot them either.  If they are there, the encoding must have changed from v1, maybe to some type of short float.

Still, if voltage measurements are there, they should be easy to spot once you have some 3-pin fans detected: vary the duty and observe what changes in the incoming messages (apart from rpm and duty).  But I don't think we'll find current values...

---

Thanks for working on this driver, and I'm excited to see it finished and integrated.

CaseySJ, 2019-10-03:
@jonasmalacofilho 

Good hints! A little more exploration yielded more information that has been added to the experimental driver (v3) at the [same site](https://www.tonymacx86.com/threads/success-gigabyte-designare-z390-thunderbolt-3-i7-9700k-amd-rx-580.267551/post-2010636). Specifically:

- We need to issue a new command string (0x10, 0x01) and read the reply to obtain firmware version. This has been added to the "status" command.
- We need to issue another new command (0x20, 0x03) and read the reply to obtain info about attached HUE2 LED accessories. This has also been added to the "status" command. Because I have only the HUE2 LED Strip and HUE2 Cable Comb, the driver supports just these two devices. I'll order the HUE2 Underglow and add that as well.
- When issuing a device.read() command to get the reply, we may not get the expected reply in the first call, so the code loops several times until it gets the correct reply.
- For command (0x10, 0x01) the expected reply must begin with (0x11, 0x01).
- For command (0x20, 0x03) the expected reply must begin with (0x21, 0x03).

Will try to add some more information to the "status" command, but what we have now is fairly close to sufficient. Anyway, working on this is quite a bit of fun!

jonasmalacofilho, 2019-10-04:
@CaseySJ 

> We need to issue a new command string (0x10, 0x01) and read the reply to obtain firmware version. [...]
> We need to issue another new command (0x20, 0x03) and read the reply to obtain info about attached HUE2 LED accessories. [...]

Do you know which one of these, if any, actually starts a new accessory detection process?  On v1 devices this was easy to spot because it made the fans ramp up to 100%.

This is important because we will probably need some way for the user to explicitly control when to start the detection (usually the role of `initialize`).

> When issuing a device.read() command to get the reply, we may not get the expected reply in the first call, so the code loops several times until it gets the correct reply.

Hm, I think we might need to have a look at the HID Report Descriptors for this device.  Can you try to get those (no idea how to do that in macOS though).  If this device uses them correctly then there should be a better way to get the one we want.

> Anyway, working on this is quite a bit of fun!

I think so too! : )

CaseySJ, 2019-10-04:
> Do you know which one of these, if any, actually starts a new accessory detection process? On v1 devices this was easy to spot because it made the fans ramp up to 100%.

When I examine the USB conversation in Windows immediately after launching NZXT CAM (i.e. after manually killing any preexisting CAM processes in *Task Manager*) I see that CAM issues these two commands in a row: `\0x10\0x01` and `\0x20\0x03`.

The Smart Device V2 then returns a series of replies that include not only `\0x11\0x01` and `\0x21\0x03` but also the "standard" status replies of `\0x67\0x02` and `\0x67\0x04`. This is actually quite nice because it reduces two `for` loops into just a single loop. This also explains why the *expected* replies `\0x11\0x01` and `\0x21\0x03` seem to come out of order.

The HUE 2 Underglow LED accessory is arriving today so the next update to the driver will include the reworked `for` loops and support for the Underglow.

I will also check whether new accessory detection occurs with just `\0x11\0x01` and `\0x21\0x03`.

CaseySJ, 2019-10-05:
Looks like my assumption was correct. Namely:

- We just need to issue '\0x10\0x01' and '\0x20\0x03' to trigger a series of replies that include everything we're looking for.
- These commands also perform new accessory detection. 
- We will get about 10 reply messages, of which the current driver parses 3. 
- I'll tinker with the reply messages a little more to see what else can be extracted.

Meanwhile, **V4** of the driver has been posted to the [same location](https://www.tonymacx86.com/threads/success-gigabyte-designare-z390-thunderbolt-3-i7-9700k-amd-rx-580.267551/post-2010636).

jonasmalacofilho, 2019-10-05:
> We just need to issue \0x10\0x01 and \0x20\0x03 to trigger a series of replies that include everything we're looking for.

I might be wrong, but I imagine accessory and fan detection can be disruptive for both fan speeds and animations.  If that's indeed the case we probably don't want to do that for every `get_status` call.

Ideally, we would put accessory and fan detection code in `initialize`, so it's explicitly triggered by the user.  But this would make this driver different from the rest, as it would not return the firmware version or the accessory types in a status call.

To solve that we could: (a) add additional output to `initialize` (but this is hackish, because output should not live in the drivers, only in cli.py); or (b) add an option called `--detect`, `--initialize` or something like that to `get_status`, that when passed makes it first call `initialize` so that the extra data is included.

What do you think?

---

As usual, good job on the v4!  I'm very happy to see all the progress you're making!

Let me know – simply by opening a pull request – when you think it's functionally complete and that you want me to start reviewing it line by line.

So far, I don't have any major comments: like I mentioned before, I might suggest that we split the v2 device driver into a subclass (to clean up some conditionals), and I would also like to clean up the read loops, if possible, by requesting specific reports from the device (but this could be left for later).

CaseySJ, 2019-10-06:
@jonasmalacofilho 

Your hunch was right! The various `status` and `set` commands of this driver were working only because I had booted into Windows first and launched NZXT CAM. 

Fortunately, it was easy enough to find the initialization command sequence and add it to the `initialize` function. Now the driver works even from a dead-cold start! :)

The `initialize` function does not ramp up the fans or even report the various accessories connected to the unit. The previously mentioned commands, `\0x10\0x01` and `\0x20\0x03`, always obtain the accurate and most current set of connected accessories without needing to issue the `initialize` command each time.

If I hot-remove an accessory or hot-plug an accessory and just invoke `status`, I immediately get the correct list of connected accessories! All preexisting settings remain in effect.

So it seems the V2 device has really been improved!

I'm almost ready to open a Pull Request -- just need to test all of the color modes. The latest driver (**nzxt_smart_device.py-v5.zip**) has been posted to the [same location](https://www.tonymacx86.com/threads/success-gigabyte-designare-z390-thunderbolt-3-i7-9700k-amd-rx-580.267551/post-2010636).

David-Frick, 2019-10-17:
Is this in reference to the Hue 2? The program does not seem to detect the fans connected to my Hue2 but it does detect my Kraken X62. I think it might not detect the Hue2 because it is connected to an internal USB Hub (NZXT sells one), which has space for other USB devices as well. Would this confuse the signals the program is receiving?

Thanks to the creator and contributors so much for this software! I really liked the NZXT parts, but was pretty angered/upset when I saw the reviews on CAM software mining data, etc etc. I did not want to spend another couple hundred dollars on parts so when I recently found this, I was ecstatic. Thanks so much!

jonasmalacofilho, 2019-10-17:
The HUE 2 and the Smart Device V2 are not the same.  The latter comes in the recent H[257](i|Elite) cases and roughly is a hybrid between a HUE 2 and a Grid+.

Still, I would very much like to support the HUE 2 as well.  There's even a chance that some of what has been developed for the Smart Device V2 (see #52) can be shared with the HUE 2.

So can you open another issue specifically for the HUE 2.  Please already include its USB vendor and product IDs.

If you can, also append the USB and HID descriptors.  And if you can provide some USB traffic captures that would be even better: we would be able to answer straight away how much we can borrow from the Smart Device V2 driver.

Regarding the use of a USB hub, as long as both devices are visible to the OS, it shouldn't matter.  But I have indeed seen many reports of problems with that hub.

CaseySJ, 2019-10-17:
@David-Frick 

In addition to the reply and suggestions from @jonasmalacofilho, I'd like to know what HUE 2 and HUE+ devices you currently own.

I've been able to find IDs for most HUE 2 accessories, but I'm missing the following:

- HUE 2 Underglow 300mm (I have the 200mm version already)
- HUE 2 Underglow 250mm (this might be available only in select geographic regions)
- HUE+ devices

Do you own Aer RGB fans (original HUE+) or Aer RGB 2 (HUE 2) fans?
Do you own any HUE+ accessories?

David-Frick, 2019-10-17:
Can you walk me through getting the USB vendor and product IDs? I will gladly supply them. For NXZT products, I have the 

- Kraken X 62 (already supported, and working)
- HueV2 
- Internal USB hub
- Six 120mm AER RGB 2 fans 

I hope to get underglow, and the Hue2 Ambient lighting (goes behind the monitor) around Black Friday potentially.

jonasmalacofilho, 2019-10-18:
@David-Frick 

[image]

To be easier to spot the NZXT products in Device Manager, change View → Devices by container. 

The highlighted line shows VID (vendor ID), PID (product ID) and REV (device revision); you can include all three values in the new issue.

Like me, you'll probably have two "NZXT USB Devices"; the Kraken X62 has PID 0x170E, the other one will be the HUE 2.

---

> Side note: Is this any consideration of adding support for Logitech RGB equipment by chance?

Ultimately it comes to the specific device, whether our CLI can be made to (sanely) work with it, how much interest there is for it, and if there is someone willing to help create and maintain the new driver.

David-Frick, 2019-10-18:
Ah, so sorry. I forgot to mention that I am using Debian with KDE Plasma
apps.

jonasmalacofilho, 2019-10-18:
@David-Frick 

That's even better.

```
# lsusb -d 1e71: -v
```

---

There are several ways to dump the report descriptors.  One of them is through lsusb, but you first need to unbind the usbhid driver from the device.

1. Take not of the PID of interest from the previous lsusb run (e.g. `170e`).
2. `# lsusb -tv` and take note of the corresponding bus and USB ports for that device  
(e.g. `Bus 01.Port 1` → `Port 9`)
3. `# ls -l /sys/bus/usb/drivers/usbhid` and and take note of the corresponding bus-ports:interface id  
(e.g. `1-9:1.0`)
4. unbind  
`# echo '1-9:1.0' > /sys/bus/usb/drivers/usbhid/unbind`
5. rerun lsusb for that pid  
`# lsusb -d 1e71:<pid> -v`
6. rebind  
`# echo '1-9:1.0' > /sys/bus/usb/drivers/usbhid/bind`

David-Frick, 2019-10-18:
1) sudo lsusb -d 1e71:
```
Bus 001 Device 002: ID 1e71:170e NZXT 
Bus 001 Device 005: ID 1e71:2001 NZXT
```

2) The command ` sudo lsusb -d 1e71:  --verbose` returned a bunch of info: the PIDs being `1e71:170e` on `Bus 001 Device 002` which has Port 9  and `1e71:2001` on `Bus 001 Device 005.` which has Port 3 (however, it is nested beneath device 4 at port 13)

3) returns:
```
lrwxrwxrwx 1 root root    0 Oct 17 23:41 1-13.3:1.0 -> ../../../../devices/pci0000:00/0000:00:01.1/0000:01:00.0/usb1/1-13/1-13.3/1-13.3:1.0
lrwxrwxrwx 1 root root    0 Oct 17 23:41 1-9:1.0 -> ../../../../devices/pci0000:00/0000:00:01.1/0000:01:00.0/usb1/1-9/1-9:1.0
lrwxrwxrwx 1 root root    0 Oct 17 23:41 5-4:1.3 -> ../../../../devices/pci0000:00/0000:00:07.1/0000:0a:00.3/usb5/5-4/5-4:1.3
lrwxrwxrwx 1 root root    0 Oct 17 23:41 9-3:1.0 -> ../../../../devices/pci0000:40/0000:40:07.1/0000:42:00.3/usb9/9-3/9-3:1.0
lrwxrwxrwx 1 root root    0 Oct 17 23:41 9-3:1.1 -> ../../../../devices/pci0000:40/0000:40:07.1/0000:42:00.3/usb9/9-3/9-3:1.1
lrwxrwxrwx 1 root root    0 Oct 17 23:41 9-4:1.0 -> ../../../../devices/pci0000:40/0000:40:07.1/0000:42:00.3/usb9/9-4/9-4:1.0
lrwxrwxrwx 1 root root    0 Oct 17 23:41 9-4:1.1 -> ../../../../devices/pci0000:40/0000:40:07.1/0000:42:00.3/usb9/9-4/9-4:1.1
lrwxrwxrwx 1 root root    0 Oct 17 23:41 9-4:1.2 -> ../../../../devices/pci0000:40/0000:40:07.1
```

which I am not sure how to intrepret. Like you showed, I believe it is 1-9:1.0
Does this sound correct so far?

jonasmalacofilho, 2019-10-18:
@David-Freck

The correct bus id will be `1-13.3:1.0`: bus 1, port 13, "sub" port 3, config(?) 1, interface(?) 0.  As you can see from the question marks, I can't completely interpret these either.

David-Frick, 2019-10-18:
Okay, I will continue from there tomorrow morning!

jonasmalacofilho, 2019-10-22:
I'm happy to say that as of 7f0d6a251f2a2e4865844609f0b9bfe09d6c3521 support for the Smart Device V2 has been merged.

There's a new [documentation section](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/nzxt-smart-device-v2.md) for it, and the `liquidctl(8)` man page was also updated.

Thank you @CaseySJ for developing the driver and being patient during my code reviews.

---

@rileyweber13, @darek,

Please test and report any issues or missing features (e.g. unknown accessories in `status`).

@David-Frick,

The protocol for the V2 is quite a lot messier than on previous NZXT devices.  I'm not sure we'll be able to support the HUE 2 without checking some USB captures.

So it's a long shot, but if you want to give this driver a try, just add a new entry for your device to `SmartDeviceDriverV2.SUPPORTED_DEVICES`.

closed 2019-10-22 by jonasmalacofilho

CaseySJ, 2019-10-22:
@David-Frick,

To make it easier to support your HUE 2 lighting controller, here's an example code:

- File: drivers/nzxt_smart_device.py
- Line: 303

Current code:
```Python
    SUPPORTED_DEVICES = [
        (0x1e71, 0x2006, None, 'NZXT Smart Device V2', {
            'speed_channel_count': 3,
            'color_channel_count': 2
        }),
    ]
```
To add the HUE 2 lighting controller, try replacing those lines with these lines:
```Python
    SUPPORTED_DEVICES = [
        (0x1e71, 0x2006, None, 'NZXT Smart Device V2', {
            'speed_channel_count': 3,
            'color_channel_count': 2
        }),
        (0x1e71, 0x2001, None, 'NZXT HUE 2 Lighting Controller', {
            'speed_channel_count': 0,
            'color_channel_count': 4
        }),
    ]
```
Then type `liquidctl list` to see whether your Kraken and HUE 2 devices are listed. If they are, and if the HUE 2 is device #2 (index 1) in the list, you may then try these commands:
- `liquidctl -d 1 initialize`
- `liquidctl -d 1 status`
- `liquidctl -d 1 set led1 color fixed ff0000` (this will set all LEDs on channel 1 to red)

David-Frick, 2019-10-22:
I am out of town until Monday. When I am bqco, I will gove it my full
attention!

flyweight, 2019-10-24:
@jonasmalacofilho Would it be possible to get a new windows binary release with this driver included please?

jonasmalacofilho, 2019-10-24:
@flyweight 

Now that the driver has been merged (and also in general) you can simply use the latest build [from the 'master' branch](https://ci.appveyor.com/project/jonasmalacofilho/liquidctl/history/branch/master).

flyweight, 2019-10-24:
@jonasmalacofilho Thanks for pointing me in the right direction.

Running `liquidctl status`, I get:

```
Device 0, NZXT Smart Device V2 (experimental)
ERROR: Unexpected error
Traceback (most recent call last):
usb.core.USBError: [Errno 5] Input/Output Error
```

I'm able to change the colors for my `led1` hardware but cannot change any of my fan speeds currently.

jonasmalacofilho, 2019-10-24:
@flyweight 

Try running `liquidctl initialize` first, and let me know if that makes a difference.

CaseySJ, 2019-10-24:
And then (after the `initialize` command) please repeat your original command as follows:

`liquidctl --debug status`

flyweight, 2019-10-24:
@jonasmalacofilho 

```
(venv) C:\Users\me\venv\Scripts>liquidctl initialize

(venv) C:\Users\me\venv\Scripts>liquidctl set fan1 speed 0

(venv) C:\Users\me\venv\Scripts>liquidctl --debug status
[DEBUG] __main__: liquidctl v1.2.0 (AppVeyor; liquidctl-75df4346227f-bin-windows-x86_64)
Device 0, NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[ERROR] __main__: Unexpected error
Traceback (most recent call last):
  File "liquidctl\cli.py", line 185, in _device_get_status
  File "liquidctl\driver\nzxt_smart_device.py", line 406, in get_status
  File "liquidctl\driver\usb.py", line 334, in read
  File "site-packages\pyusb-1.0.2-py3.6.egg\usb\core.py", line 988, in read
  File "site-packages\pyusb-1.0.2-py3.6.egg\usb\backend\libusb1.py", line 851, in intr_read
  File "site-packages\pyusb-1.0.2-py3.6.egg\usb\backend\libusb1.py", line 936, in __read
  File "site-packages\pyusb-1.0.2-py3.6.egg\usb\backend\libusb1.py", line 595, in _check
usb.core.USBError: [Errno 5] Input/Output Error
[DEBUG] liquidctl.driver.usb: ensure interface is released
```

jonasmalacofilho, 2019-10-24:
Move this to a new issue: #58 ("Input/Output error when reading from Smart Device V2").