liquidctl/liquidctl #33: liquidctl list is empty on macOS

jahluwalia, 2019-07-01:
Hi,

I have a x62 and smart device both connected to a nzxt usb hub. The hub is connected to the motherboard f_usb connector. `liquidctl list` is empty. Below is the verbose output. Any pointers? 

```bash
$ liquidctl list -v -g
[DEBUG] usb.backend.libusb1: _LibUSB.__init__(<CDLL '/usr/local/lib/libusb-1.0.dylib', handle 7fa5afeab170 at 0x10204bda0>)
[INFO] usb.core: find(): using backend "usb.backend.libusb1"
[DEBUG] usb.backend.libusb1: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x1020979e8>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x102186550>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x1020979e8>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x102186588>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x1020979e8>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x1021865c0>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x1020979e8>)
[DEBUG] usb.backend.libusb1: _LibUSB._finalize_object()
[DEBUG] usb.backend.libusb1: _LibUSB.__init__(<CDLL '/usr/local/lib/libusb-1.0.dylib', handle 7fa5afeab170 at 0x10204bda0>)
[INFO] usb.core: find(): using backend "usb.backend.libusb1"
[DEBUG] usb.backend.libusb1: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x10204bc50>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x102186550>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x10204bc50>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x102186588>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x10204bc50>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x1021865c0>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x10204bc50>)
[DEBUG] usb.backend.libusb1: _LibUSB._finalize_object()
[DEBUG] usb.backend.libusb1: _LibUSB.__init__(<CDLL '/usr/local/lib/libusb-1.0.dylib', handle 7fa5afeab170 at 0x10204bda0>)
[INFO] usb.core: find(): using backend "usb.backend.libusb1"
[DEBUG] usb.backend.libusb1: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x102134208>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x102186550>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x102134208>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x102186588>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x102134208>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x1021865c0>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x102134208>)
[DEBUG] usb.backend.libusb1: _LibUSB._finalize_object()
[DEBUG] usb.backend.libusb1: _LibUSB.__init__(<CDLL '/usr/local/lib/libusb-1.0.dylib', handle 7fa5afeab170 at 0x10204bda0>)
[INFO] usb.core: find(): using backend "usb.backend.libusb1"
[DEBUG] usb.backend.libusb1: _LibUSB.enumerate_devices()
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x10204bdd8>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x102186550>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x10204bdd8>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x102186588>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x10204bdd8>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x1021865c0>)
[DEBUG] usb.backend.libusb1: _LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x10204bdd8>)
[DEBUG] usb.backend.libusb1: _LibUSB._finalize_object()
```

Thanks, 
Jas

jonasmalacofilho, 2019-07-02:
What version of liquidctl are you running?

jahluwalia, 2019-07-02:
liquidctl v1.1.0

jahluwalia, 2019-07-02:
I took the hub out of the equation and connected the x62 directly to the motherboard. liquidctl list shows x62. I think I might have a faulty hub. 

Will get a replacement or splitter and keep you posted here.

jonasmalacofilho, 2019-07-02:
I do remember seeing some other reports of problems with the NZXT USB Hub.

Still, you should upgrade to [v1.2.0rc1](https://github.com/jonasmalacofilho/liquidctl/releases/tag/v1.2.0rc1), as both X62 and Smart Devices are HIDs and, on Mac OS, accessing them through HIDAPI [is a lot more reliable](https://github.com/jonasmalacofilho/liquidctl/issues/9).

v1.2.0rc1 is available on PyPI but, since it's a pre-release, you need to explicitly request it when using pip:

```
# pip install liquidctl==v1.2.0rc1
```

jonasmalacofilho, 2019-07-02:
> Will get a replacement or splitter and keep you posted here.

Also, you can fashion your own splitter for a USB 2.0 header with some jumper cables, like those for Arduinos/Raspberry PIs.  I personally do this, since decent simple splitters are not always easy to find, and hubs are overkill when a single header already provides two "native" USB ports.

jahluwalia, 2019-07-04:
Turns out it has to do with the USB port mappings in the EFI folder. for now, I'm not going to mess with it. I just disconnected the led strips and connected the kraken 62 directly to the motherboard. Liquidctl is working great. I think this issue can be closed. 

thanks,
jas

jonasmalacofilho, 2019-07-04:
I see... I'll close the issue then.  And thanks for letting me know.

Just for future reference, on what version of Mac OS are you?

closed 2019-07-04 by jonasmalacofilho