liquidctl/liquidctl #34: help/docs isn't thorough on options

jcfrosty, 2019-07-07:
I can see the ability to set to fading, but what are the other options I can set from CLI? It would be nice to see a list. I'm also assuming the code afterwards is hex for colors?

jonasmalacofilho, 2019-07-07:
The set of supported features and parameters varies from model to model.  The existing documentation for each device is linked in the [Supported devices](https://github.com/jonasmalacofilho/liquidctl#supported-devices) section.

What device(s) do you own?

jcfrosty, 2019-07-07:
I have a Kraken X52. I was able to comb through the code to find the
different color modes, but the documentation should probably list them or
have a way for the --help flag to list them. Tai-chi works nicely BTW. Some
of the options though, strangely result in what appears like it's trying to
perform the animation but the LEDs are flickering other colors during the
animation.

Very impressed and pleased with your work man. Thank you so much. Beats
using a VM and passing through USB any day! You're a godsend!

jonasmalacofilho, 2019-07-07:
> I have a Kraken X52. I was able to comb through the code to find the different color modes, but the documentation should probably list them or have a way for the --help flag to list them.

Well, the [docs for that family of devices](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/nzxt-kraken-x-3rd-generation.md#rgb-lighting) do list the modes, colors and options they support, **but I'll take the hint and make that document more easy to find.**

I did originally planned to make all documentation available directly in the CLI, but as the number of supported devices grew that became infeasible (or more complex to maintain than I would like).

> Some of the options though, strangely result in what appears like it's trying to perform the animation but the LEDs are flickering other colors during the animation.

Can you open another issue, listing which modes and/or options are needed to reproduce the problem, as well as both liquidctl and Kraken firmware versions (the latter is reported by the device as part of the `status` call)?

> Very impressed and pleased with your work man. Thank you so much. Beats using a VM and passing through USB any day! You're a godsend!

Thank you as well for the feedback!

jonasmalacofilho, 2019-07-31:
Fixed with 525cb42a30c76be6a6097d6febd0ffcfea633f26 ("Simplify the list of supported devices and related documentation").

closed 2019-07-31 by jonasmalacofilho