liquidctl/liquidctl #35: Some Kraken X61 functionality still missing.

Firehawke, 2019-08-21:
Fan profiles aren't implemented yet, and it'd be nice to see some of the missing color functionality return (e.g. set color based on temperature)-- this functionality disappeared around the time they did a rewrite of Cam, along with a lot of the color profiles.

I've got an X61 and I'm running Windows 10. I can reinstall Cam if needed for data capture; just tell me what steps you'd need me to take.

jonasmalacofilho, 2019-08-22:
You're right.  I totally forgot that I didn't enable the fan profiles on the legacy devices.

I imagine the command byte and message structure for setting up a profile will be fairly similar, if not identical, to how the EVGA CLCs work.  However, legacy 690LC coolers like the X61 require _unrelated_ messages to follow a particular order (see description of 52a55247196d6ed252e3f991be99786a296a679f).  So, more than confirming the message structure, it's necessary to understand (and test) how configuring a profile interacts with the other commands.

On the other hand, the color change based on temperature, as far as I know, is simply an override of any lighting mode (apart from blackout) when the liquid temperature reaches a chosen limit.  This is already [implemented and in use in the asetek.py](https://github.com/jonasmalacofilho/liquidctl/blob/61110b41de07a1fe01b6349002e61b0d3d39b50c/liquidctl/driver/asetek.py#L144-L149) driver, it's just its configuration that hasn't been exposed to the user yet.  So, given your interest in it, I'll just go ahead and add the necessary CLI options.

Can you also let me know if there's any other lighting mode that's missing or broken on the X61?  Currently `fading`, `blinking`, `fixed` and `blackout` [should work](https://github.com/jonasmalacofilho/liquidctl/issues/3#issuecomment-480667136).

---

To capture and analyze USB traffic on Windows I think the most reliable software tool is probably [Wireshark](https://wiki.wireshark.org/CaptureSetup/USB#Windows), though the [Free USB Analyzer](https://freeusbanalyzer.com/) from HHD should work well too (even if it's not open-source software and might even seem a bit sketchy).

Also, you'll probably want to familiarize yourselft with the [asetek.py](https://github.com/jonasmalacofilho/liquidctl/blob/master/liquidctl/driver/asetek.py) driver, in particular:

 - the relationship between `CommonAsetekDriver`, (modern) `AsetekDriver` and `LegacyAsetekDriver`;
 - the fan profile setup implemented in `AsetekDriver.set_speed_profile` and `._prepare_profile` (for EVGA CLCs);
 - how `LegacyAsetekDriver` deals with persisting previous messages to reissue then when necessary (`._store_integer`, `._load_integer`), and how this is already used in that driver

Firehawke, 2019-08-25:
Going to be at least a few days before I can get to anything related to this. The entire room my PC is in is literally half-disassembled at the moment. I do plan to at least look into what I can do when the deconstruction and construction is complete.

jonasmalacofilho, 2019-09-09:
I ended up working a bit on the fan profiles already.  Can you give commit 3f004ee3b68bba4b72de947b021bedf2bd320695 ("Add support for setting fan profiles on all legacy 690LC devices") a try?

Since it's a first attempt I imagine there might be some issues left.¹  Hopefully you'll be able to easily fix them or, least, figure out more about what's going on with the `--debug` flag.

Besides an obvious failure, I would also like to ask you to check for more subtle issues:

 - fan speed profiles work (i.e. the fan speed changes) but result in errors or warnings (e.g. timeouts);
 - or they appear to fully work but the everything else starts to behave badly afterwards (e.g. lighting modes, pump speed, or even checking the status)

In case anything goes bad, reverting to 'master' and re-initializing the device should clear most problems.  As a last resort, power cycling it should completely reset it.

---

¹ Just so you know I don't have access to these exact devices, so I'm testing this on a similar (but only to a point) cooler.

Firehawke, 2019-09-09:
Sorry about that, I just got my system back up and running and I'll be pulling and testing now.

Firehawke, 2019-09-10:
Results of liquidctl --debug --legacy-690lc initialize
```
[DEBUG] liquidctl.driver.asetek: data directory for device is C:\ProgramData\jonasmalacofilho\liquidctl\2433_b200_0100\usb2_11
[DEBUG] liquidctl.driver.asetek: open device
[DEBUG] liquidctl.driver.asetek: begin transaction
[DEBUG] liquidctl.driver.asetek: write 10 00 00 00 00 00 00 ff 00 00 2d 00 00 01 00 00 01 00 01
[DEBUG] liquidctl.driver.asetek: received 08 70 7e bd 10 be c0 e7 06 18 21 10 3b 32 01 00 6d 84 77 e9 03 de 00 00 07 00 00 2c fc 00 1c a1
[DEBUG] liquidctl.driver.asetek: stored pump_duty=100
[DEBUG] liquidctl.driver.asetek: stored fan_temp0=10
[DEBUG] liquidctl.driver.asetek: stored fan_duty0=25
[DEBUG] liquidctl.driver.asetek: stored fan_temp1=20
[DEBUG] liquidctl.driver.asetek: stored fan_duty1=40
[DEBUG] liquidctl.driver.asetek: stored fan_temp2=30
[DEBUG] liquidctl.driver.asetek: stored fan_duty2=55
[DEBUG] liquidctl.driver.asetek: stored fan_temp3=40
[DEBUG] liquidctl.driver.asetek: stored fan_duty3=70
[DEBUG] liquidctl.driver.asetek: stored fan_temp4=50
[DEBUG] liquidctl.driver.asetek: stored fan_duty4=85
[DEBUG] liquidctl.driver.asetek: stored fan_temp5=60
[DEBUG] liquidctl.driver.asetek: stored fan_duty5=100
[DEBUG] liquidctl.driver.asetek: begin transaction
[DEBUG] liquidctl.driver.asetek: loaded fan_temp0=10 (from cache)
[DEBUG] liquidctl.driver.asetek: loaded fan_duty0=25 (from cache)
[INFO] liquidctl.driver.asetek: setting fan PWM point: (10°C, 25%), device interpolated
[DEBUG] liquidctl.driver.asetek: loaded fan_temp1=20 (from cache)
[DEBUG] liquidctl.driver.asetek: loaded fan_duty1=40 (from cache)
[INFO] liquidctl.driver.asetek: setting fan PWM point: (20°C, 40%), device interpolated
[DEBUG] liquidctl.driver.asetek: loaded fan_temp2=30 (from cache)
[DEBUG] liquidctl.driver.asetek: loaded fan_duty2=55 (from cache)
[INFO] liquidctl.driver.asetek: setting fan PWM point: (30°C, 55%), device interpolated
[DEBUG] liquidctl.driver.asetek: loaded fan_temp3=40 (from cache)
[DEBUG] liquidctl.driver.asetek: loaded fan_duty3=70 (from cache)
[INFO] liquidctl.driver.asetek: setting fan PWM point: (40°C, 70%), device interpolated
[DEBUG] liquidctl.driver.asetek: loaded fan_temp4=50 (from cache)
[DEBUG] liquidctl.driver.asetek: loaded fan_duty4=85 (from cache)
[INFO] liquidctl.driver.asetek: setting fan PWM point: (50°C, 85%), device interpolated
[DEBUG] liquidctl.driver.asetek: loaded fan_temp5=60 (from cache)
[DEBUG] liquidctl.driver.asetek: loaded fan_duty5=100 (from cache)
[INFO] liquidctl.driver.asetek: setting fan PWM point: (60°C, 100%), device interpolated
[DEBUG] liquidctl.driver.asetek: write 11 00 0a 14 1e 28 32 3c 19 28 37 46 55 64
[DEBUG] liquidctl.driver.asetek: loaded pump_duty=100 (from cache)
[INFO] liquidctl.driver.asetek: setting pump duty to 100%
[DEBUG] liquidctl.driver.asetek: write 13 64
[DEBUG] liquidctl.driver.asetek: close device
Traceback (most recent call last):
  File "liquidctl\cli.py", line 240, in <module>
  File "liquidctl\cli.py", line 228, in main
  File "liquidctl\driver\asetek.py", line 385, in initialize
  File "liquidctl\driver\asetek.py", line 377, in _set_all_speeds
  File "liquidctl\driver\asetek.py", line 134, in _end_transaction_and_read
  File "liquidctl\driver\usb.py", line 260, in read
  File "site-packages\pyusb-1.0.2-py3.6.egg\usb\core.py", line 988, in read
  File "site-packages\pyusb-1.0.2-py3.6.egg\usb\backend\libusb1.py", line 833, in bulk_read
  File "site-packages\pyusb-1.0.2-py3.6.egg\usb\backend\libusb1.py", line 936, in __read
  File "site-packages\pyusb-1.0.2-py3.6.egg\usb\backend\libusb1.py", line 595, in _check
usb.core.USBError: [Errno 10060] Operation timed out
[856] Failed to execute script cli
```

So, it looks like it definitely doesn't like that format. I've side-by-side installed the beta with the previous known-working install, so it's not like this is putting my system at risk testing it.

jonasmalacofilho, 2019-09-10:
Thanks for testing this...

The problem appears to be limited to the next read just after setting the fan profile and pump speed (this is in _end_trasaction_and_read).  So I think there's still hope we can get this feature to work.

Do we know of a specific version of CAM that does set up fan profiles _on the device?_  As you mentioned, the recent one I have (3.7.8) does everything in software.¹

Apart from that, I have a few ideas... but not a device where this happens.  Can you give some of these a look and/or try?

---

First, it's possible we just need to nail the correct order of operations in _set_all_speeds.  [Currently it can be simplified to](https://github.com/jonasmalacofilho/liquidctl/blob/3f004ee3b68bba4b72de947b021bedf2bd320695/liquidctl/driver/asetek.py#L357-L375):

```
_set_all_speeds:
    _begin_transaction
    "write fan profile"
    "write pump speed"
    return _end_transaction_and_read ("read" then "release the interface" for other processes)
```

It's possible (though not very likely) that reordering the two writes is enough.  We can also experiment with more _begin_transaction and/or _end_transaction_and_read calls.

---

Another possibility is that we simply waited enough time for the device to have something to return us.

In fact, according to some [documentation for the serial to USB bridge](https://www.silabs.com/documents/public/application-notes/AN169.pdf) used in the cooler, we should have a read queue, get the data from the device in chunks and check the size of queue before reading from it.

Assuming the problem is indeed related to this, we could try adding delays ~~or figuring out this receive queue thing~~.²

---

Finally, it's possible that neither the fan profile nor the pump speed writes cause the device to return any status data.

Before going on further, the only reason in the legacy code why we save all speed settings on the filesystem and then have a _set_all_speeds function that reapplies them is so we can use those writes (used to be a "write _fixed_ fan speed") to trigger a read, when executing get_status.

So it seems to me that another potential solution could be to rewrite the legacy code to use  _configure_device as this "dummy" write that triggers the read for get_status.

This might be a bit annoying to implement, so it would probably be wise to first test if something like this (or maybe slightly reordered) works:³

```
to set fan profiles and/or fixed pump speeds:
    _begin_transaction
    "write fan profile"
    "write pump speed"
    _configure_device
    _end_transaction_and_read

to get the device status:
    _begin_transaction
    _configure_device
    return _end_transaction_and_read
```

---

Notes:

¹ You could give EVGA Flow Control a try, because it does configure the device with profiles, and old Krakens and EVGA CLCs are indistinguishable in software.  But I think it will probably just crash (itself and/or the cooler), because I'm almost sure the write it uses before a read is not supported on these Krakens.

² [libSiUSBXp](https://github.com/craigshelley/SiUSBXp/blob/master/SiUSBXp.c) implements this receive queue.  But I personally would leave looking into this for after everything else has failed I already looked into it in the past).

³ On the [captures of an EVGA CLC](https://github.com/jonasmalacofilho/liquidctl/blob/3f004ee3b68bba4b72de947b021bedf2bd320695/docs/development/asetek690lc/evga-clc-capture-1.txt#L135-L151) the fan profile configuration did show the use of more calls that what's needed for that device; perhaps FlowControl has some legacy code as well?

jonasmalacofilho, 2019-09-27:
@Firehawke did you manage to give any of the ideas above a try?

closed 2019-10-24 by jonasmalacofilho