liquidctl/liquidctl #36: CoolerMaster MasterLiquid ML240 RGB support ?

olielvewen, 2019-08-25:


olielvewen, 2019-08-25:
Hi,

I would like to know if you will implement the support for the [CoolerMaster MasterLiquid ML240](https://www.coolermaster.com/catalog/coolers/cpu-liquid-coolers/masterliquid-ml240r-rgb/) RGB or for CoolerMaster product in the future ?  
And if it is a positive answer, When ? 

BTW Thanks a lot.

olielvewen, 2019-08-27:
Thanks jonasmalacofilho for your quick answer and all this explanations. I was not thinking that everything was playing at the level of the motherboard. 

Unfortunately, I can not give you all this information (it will be with a great pleasure to help you in this task) for the simple reason I have not yet bought this AIO. And I was asking the question for a future purchase.

What about the Corsair Hydro Series H115i Platinium (2x140) or Corsair Hydro Series H100i Platinium (2x120) ? Are they manage by liquidctl ? 
Personally, my preference goes to the  H115 i Platinium but you are more expert than me, Which one will you advise me ?

Another thing. My motherboard is an Asus Rog Strix B450 Gaming F. My CPU is a Ryzen R7 2700X with 32 Go of G.Skils Trident Z RGB CAS16. If I can help you, it will be with pleasure. 

After, If you are sure to implement it (the MasterLiquid Ml240 RGB) on Liquidctl I 'm ready to purchase it in September (end) or in October. Nobody seems to have one and ask you until now.

BTW Thanks a lot for your work your time to port on Linux this features for this kind of product too much forgotten by this manufacturers.

olielvewen, 2019-08-29:
Sorry for the delay, I'm not every day behind my computer. Especially, at this season......^_^

Here, I give you the output of lsusb, lspci, Inxi under different forms.
##lsusb
`[olivier@thorielle ~]$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 046d:c517 Logitech, Inc. LX710 Cordless Desktop Laser
Bus 003 Device 003: ID 03f0:6411 HP, Inc PhotoSmart C8100 series
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 1b1c:1b51 Corsair 
Bus 001 Device 002: ID 1038:12ad SteelSeries ApS 
Bus 001 Device 004: ID 1b1c:1c0a Corsair 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
##lspci
`[olivier@thorielle ~]$ lspci
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Root Complex
00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) I/O Memory Management Unit
00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge
00:01.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge
00:01.3 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge
00:02.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge
00:03.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge
00:03.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge
00:04.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge
00:07.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge
00:07.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Internal PCIe GPP Bridge 0 to Bus B
00:08.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge
00:08.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Internal PCIe GPP Bridge 0 to Bus B
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller (rev 59)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge (rev 51)
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 0
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 1
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 2
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 3
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 4
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 5
00:18.6 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 6
00:18.7 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 7
01:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981
02:00.0 USB controller: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset USB 3.1 XHCI Controller (rev 01)
02:00.1 SATA controller: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset SATA Controller (rev 01)
02:00.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Bridge (rev 01)
03:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port (rev 01)
03:01.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port (rev 01)
03:04.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port (rev 01)
03:06.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port (rev 01)
03:07.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] 400 Series Chipset PCIe Port (rev 01)
04:00.0 Ethernet controller: Intel Corporation I211 Gigabit Network Connection (rev 03)
08:00.0 Network controller: Broadcom Inc. and subsidiaries BCM4360 802.11ac Wireless Network Adapter (rev 03)
09:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] RV770 [Radeon HD 4850]
09:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] RV770 HDMI Audio [Radeon HD 4850/4870]
0a:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Zeppelin/Raven/Raven2 PCIe Dummy Function
0a:00.2 Encryption controller: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Platform Security Processor
0a:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Zeppelin USB 3.0 Host controller
0b:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Zeppelin/Renoir PCIe Dummy Function
0b:00.2 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 51)
0b:00.3 Audio device: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) HD Audio Controller
`
`[olivier@thorielle ~]$ inxi -Flxxxz
System:    Host: thorielle Kernel: 5.2.8-1-MANJARO x86_64 bits: 64 compiler: gcc v: 9.1.0 
           Desktop: Cinnamon 4.2.3 info: cairo-dock wm: muffin dm: LightDM 1.30.0 
           Distro: Manjaro Linux 
[... cut ...]
           ID-2: /home size: 381.51 GiB used: 130.11 GiB (34.1%) fs: ext4 dev: /dev/nvme0n1p4 
           label: N/A 
           ID-3: swap-1 size: 15.62 GiB used: 0 KiB (0.0%) fs: swap dev: /dev/nvme0n1p3 
           label: N/A 
Sensors:   System Temperatures: cpu: 48.0 C mobo: N/A gpu: radeon temp: 94 C 
           Fan Speeds (RPM): cpu: 0 
Info:      Processes: 351 Uptime: 1h 01m Memory: 31.40 GiB used: 5.84 GiB (18.6%) 
           Init: systemd v: 242 Compilers: gcc: 9.1.0 clang: 8.0.1 Shell: bash v: 5.0.7 
           running in: qterminal inxi: 3.0.35 

`

olielvewen, 2019-08-29:
I get an issue with OpenCorsairLink and I'm not able to run it. 
See this bug report : [https://github.com/audiohacked/OpenCorsairLink/issues/198](url)
And nobody for helping me at first look....
As you could see in the lsusb and inxi -xxx --usb outputs, I've got an Headset Steelseris Artic7, an usb mouse corsair Dark Core SE RGB (**1b51**) and an Corsair RM650I (**1c0a**)

|Considering both of these points, have you taken a look at the EVGA CLC 280, or the recently announced CLC 360?

No. I was on these models because they are not too expensive, performance and EVGA seems to be a brand not very well known here in France. But, I will learn as soon as I have a sufficiently important moment of free. In addition, I thought that corsair was well supported, at least for this kind of product.

I already know this project and I know that she is not yet supported unfortunately. As you could see on my blog, I'm a little geek and fan of Hardware..... And I'm like you, I prefer software for desktop and you can avoid to pass to the UEFI for this task. Not frankly funny by the way.

jonasmalacofilho, 2019-08-29:
> I get an issue with OpenCorsairLink and I'm not able to run it.
See this bug report : https://github.com/audiohacked/OpenCorsairLink/issues/198
And nobody for helping me at first look....

Left a comment there, hope it helps.

> No. I was on these models because they are not too expensive, performance and EVGA seems to be a brand not very well known here in France.

From a quick look on Amazon, the EVGA 280 seems to sit just in the middle between the prices on those listings you linked for the ML240R and H115i Platinum.

And both Corsair and NZXT have models using the same (H115i, X61) or very similar (H115i Pro, X62) pump/block design from Asetek that's used in these EVGA CLCs.

Of course, pick whatever makes more sense for you : )

> I already know this project and I know that she is not yet supported unfortunately.

Weird, [this commit](https://github.com/electrified/asus-wmi-sensors/commit/c8bb47e6dfd6c8502a2c3dbfca676aee48034a38) claims to add support for it.

> As you could see on my blog, I'm a little geek and fan of Hardware

Oh, I missed that you had a blog.  I'll take a look!

> Did you use Hidapi or udev ? Is it at a more low level ? Did are you done Reverse Engineering ? How did you process ?

liquidctl uses both libusb (through PyUSB) and HIDAPI (through cython-hidapi).  On Linux HIDAPI is available for use with HID devices but libusb is still the default.  And not all supported devices are HIDs anyway.

I wrote a quick intro into what can be done for a new USB device as [comment on #32](https://github.com/jonasmalacofilho/liquidctl/issues/32#issuecomment-525150079).

jonasmalacofilho, 2019-08-29:
// closing this issue as there's nothing actionable here, but I'll still reply normally

closed 2019-08-29 by jonasmalacofilho

olielvewen, 2019-08-30:
|Left a comment there, hope it helps.
Your intervention was the small stone that triggered the avalanche
|From a quick look on Amazon, the EVGA 280 seems to sit just in the middle between the prices on those listings you linked for the ML240R and H115i Platinum.

And both Corsair and NZXT have models using the same (H115i, X61) or very similar (H115i Pro, X62) pump/block design from Asetek that's used in these EVGA CLCs.

Of course, pick whatever makes more sense for you : 
Well evga is only present on amazon in limited model but not in my two usual suppliers. For now, I still have the choice. Nothing is decided.
Will you support them : 

- Enermax RGB LiQMax RGB 120/240 mm // AQuaFusion 120/240 // LiqFusion 240/360 mm

- ThermalRight Turbo Right RGB 240/360 mm

| Weird, this commit claims to add support for it.
Great news but the last time that I've take a look on the Readme she was not.

 | Oh, I missed that you had a blog. I'll take a look!
I have a part 7 and probably a part 8 to do. And several others little articles about GVE, Tuxclocker,...

|I wrote a quick intro into what can be done for a new USB device as comment on #32.
I'll take a look more deeply tomorrow or in the next days.

jonasmalacofilho, 2019-08-30:
> Will you support them:
> 
>  - Enermax RGB LiQMax RGB 120/240 mm // AQuaFusion 120/240 // LiqFusion 240/360 mm
>  - ThermalRight Turbo Right RGB 240/360 mm

They also seem to rely on the motherboard for software fan and LED control, meaning that it's about supporting the motherboard, not the devices themselves.  So, just as with the ML240R, the short answer is "no".

In fact, I don't have specific plans for any new devices.  This doesn't mean that it wont happen or that I'm not interested in or willing to help supporting more devices.  But right now I'm just not actively working on any new models.

For your purchase this means that, unless the device you ask for is already supported, my answer will be "no specific plans, sorry".  Thus, I think your best bet is to look for siblings of devices for which we already have drivers for (as those will _likely_ work too).  Specifically, take a look at [other vendors using Asetek coolers](https://www.asetek.com/gaming/diy-partners), but keep in mind that we don't support their 6-th devices yet (e.g. the Corsair H115i Pro).

If you still need to pick a device that's unsupported by this or any other project right now, I suggest you look for:

 - use of USB for communication and software control
 - software control of all features (fans, pump and LEDs); this isn't always the case (e.g. NZXT M22)
 - good track record of open-source support for other models of the same vendor or manufacturer

If the model you pick matches all of these criteria, you will have a higher chance of being able to quickly write the necessary code for it, or to get lucky that someone else does that for you.  You can also try to contact the manufacturer or vendor: in principle I'd be willing to work with them to get the device supported on liquidctl and Linux in general.

olielvewen, 2019-08-31:
| They also seem to rely on the motherboard for software fan and LED control, meaning that it's       about supporting the motherboard, not the devices themselves. So, just as with the ML240R, the     short answer is "no".

This decision is due to a technical impossibility or the fact that it is necessary to write from scratch a new software (not to mention the time and the study necessary to achieve this task) ?

     |For your purchase this means that, unless the device you ask for is already supported, my answer will be "no specific plans, sorry". Thus, I think your best bet is to look for siblings of devices for which we already have drivers for (as those will likely work too). Specifically, take a look at other vendors using Asetek coolers, but keep in mind that we don't support their 6-th devices yet (e.g. the Corsair H115i Pro).

I hope that will be done in a near future. 

    | If you still need to pick a device that's unsupported by this or any other project right now, I suggest you look for:

    use of USB for communication and software control
    software control of all features (fans, pump and LEDs); this isn't always the case (e.g. NZXT M22)
    good track record of open-source support for other models of the same vendor or manufacturer

If the model you pick matches all of these criteria, you will have a higher chance of being able to quickly write the necessary code for it, or to get lucky that someone else does that for you. You can also try to contact the manufacturer or vendor: in principle I'd be willing to work with them to get the device supported on liquidctl and Linux in general.

Thanks a lot for all this good advices.

How did you know that the pump was manufactured by Asatek or by coolit ? This information like controllers for towers fan is hidden.

Anyway, I'll take my decision in October probably and following your advices. An yet a big thanks for taking the time to explain me in details . Our conversation is very informative.

jonasmalacofilho, 2019-08-31:
> This decision is due to a technical impossibility or the fact that it is necessary to write from scratch a new software (not to mention the time and the study necessary to achieve this task) ?

Think of these non-USB devices as having three controllable bits:

 - the pump: controlled only by varying the voltage supplied to it
 - the fans: controlled either by varying the voltage or with a [25 kHz pulse-width modulation (PWM) signal](https://en.wikipedia.org/wiki/Computer_fan_control)
 - the LEDs: controlled with pulses of different lengths in a single wire ([datasheet for the WS2812B LED IC](https://cdn-shop.adafruit.com/datasheets/WS2812B.pdf), used by most [if not all] _addressable_ LED accessories for PCs)

These devices, unlike the USB ones liquidctl supports, don't have any communication port.  They directly expose their hardware and expect some external controller to take care of it all.  So, in a way, there's a technical impossibility here: they don't support software control, which is what this project is about.

Your motherboard is a different story, and already has controllers for all three bits.  _In theory_ we could support it, but I'm not willing to spend the huge amount of effort and time needed to understand how to talk to its controllers.

You could also build your own controller board with an Arduino, Raspberry PI or similar, and liquidctl could be made capable to talking to this DYI controller.  I would be willing to implement this, provided the controller was open-source (so other's could benefit from it) and that you provided the specs to the protocol you designed.

> > we don't support their 6-th devices yet (e.g. the Corsair H115i Pro).
> I hope that will be done in a near future.

Me too!  Let's see if someone contributes a driver here or to OpenCorsairLink.

> How did you know that the pump was manufactured by Asatek or by coolit ? This information like controllers for towers fan is hidden.

Some reviewers do know and report this (e.g. GamersNexus).

For the rest, Asetek pumps have very characteristic cold-plate format and bayonet-style mounting hardware (take a closer look at the notches on the pump/block).

[image]

Of course this doesn't always mean that's specifically a 5-th gen. version, or that it's software controllable at all, but it's hint.

CoolIT I don't know much about, only that some Corsair models are made by them.

olielvewen, 2019-09-02:
|

This conversation is very interesting and I allow myself to present projects that could help you or give suggestions for other people who would read us.

``` You could also build your own controller board with an Arduino,1, 2 Raspberry Pi3, 4 or similar, and liquidctl could be made capable to talking to this DYI controller. I would be willing to implement this, provided the controller was open-source (so other's could benefit from it) and that you provided the specs to the protocol you designed.```

Like this project by sample : [SP120 Controller](https://github.com/petealves/Corsair-SP120-RGB-ESP8266-Custom-Controller). 
It is exactly what you said with a HAT Opensource NodeMCU 8266 (in fact the ESP-12) and you can communicate by WIfi and obviously you take the place of the "officiel" controler.

For the leds this one is too interesting : [FastLeds](https://github.com/FastLED/FastLED)
After that's could be The Solution. 

|Your motherboard is a different story, and already has controllers for all three bits. In theory we could support it, but I'm not willing to spend the huge amount of effort and time needed to understand how to talk to its controllers.

[Asus-wmi-sensors](https://github.com/electrified/asus-wmi-sensors) seems to be a good place for testing that. And I understand why you have started this other project. You are in the good direction.

For the moment, I don't know which one I could choose (Corsair or TT) but one things is sure : I 'll don't take a CoolerMaster.

A few set of projects about WS2810 , perhaps they could help you even you know a lot about this subject.

- [micropython-ws2812.](https://github.com/JanBednarik/micropython-ws2812)

- [libws2810](https://gitlab.com/NexAdn/libws2801)

- a last but abandoned : [adressable-leds-controller](https://github.com/obsunal/Adressable-LED-Controller)
Even they are for Raspberry, Arduinio, there are a lot of informations that I'm sure that she could be used in others projects.

|we don't support their 6-th devices yet (e.g. the Corsair H115i Pro).
| Me too! Let's see if someone contributes a driver here or to OpenCorsairLink.

About OpenCorsairLink, like you seems to know well this project. It seems that the display of the values ​​of the power supply is only done when we place the order. We do not have a continuous display when we launch OCL. I missed something ? 
I found [this project](https://github.com/definelicht/powermeter) that seems to complete this . situation. Damage that he is not integrated in OpenCorsairLInk.

jonasmalacofilho, 2019-09-03:
> About OpenCorsairLink, like you seems to know well this project. It seems that the display of the values ​​of the power supply is only done when we place the order. We do not have a continuous display when we launch OCL. I missed something ?

That's right.

> I found this project that seems to complete this . situation. Damage that he is not integrated in OpenCorsairLInk.

While not super efficient, you could write a script that calls OpenCorsairLink every second or so, parses the results, and outputs the status info as a continuous stream.

But, since you are on Linux, I think the best solution is to write a kernel driver for the Hardware Monitoring subsystem, which will allow lm-sensors, and other tools based on these (e.g. Freon extension for Gnome) to seamless display the data for that device.

In fact, this is the reason liquidctl-linux-drivers exists.

olielvewen, 2019-09-05:
Hi again,

Sorry for the delay.

```While not super efficient, you could write a script that calls OpenCorsairLink every second or so, parses the results, and outputs the status info as a continuous stream.```

You have reason but before I'll try another project called [CorsairRmi](https://github.com/notaz/corsairmi) which done only one thing: read PSUs. I 'll must just create a pkg, if I can remember how to do that. ^_^. After, I have allready done for HeadSetControl. I just need some free times.
But before, I must make a virtual machine for seeing what ICue do it with that.

 ```But, since you are on Linux, I think the best solution is to write a kernel driver for the Hardware Monitoring subsystem, which will allow lm-sensors, and other tools based on these (e.g. Freon extension for Gnome) to seamless display the data for that device. In fact, this is the reason liquidctl-linux-drivers exists.```

Indeed but this task ask a lot of work and skills that I can't provide. I don't know Freon. Interesting application. It is near what I should get.

 A little bit earlier I was talking about TT, and I forgot to talk about [this project](https://github.com/chestm007/linux_thermaltake_riing) which is based on this [other project](https://github.com/MoshiMoshi0/TTController) but for Windows only. Perhaps, it could be interesting for you.

In any cases, I will keep you informed of what I will order.

jonasmalacofilho, 2019-09-05:
It should be fairly easy to port CorsairRmi into a Linux hwmon driver.  I might look into that in a few weeks.

olielvewen, 2019-09-06:
Great news.

In a similar vein, there is also [psumon](https://github.com/ka87/cpsumon) a bit more evolved. He has been designed for the AX1 Series.

I've just found this project called [OpenAuraSDK](https://gitlab.com/CalcProgrammer1/OpenAuraSDK) And I think that he could be interesting for you. This one provides  at low level a open-source implementation of the Asus Aura RGB controller software (memory RGB and Fans ?).
Perhaps it could be help you in your task (i.e  liquidctl-linux-drivers). My two cents..........

jonasmalacofilho, 2019-09-06:
It's nice to see OpenAuraSDK: people are figuring out how Aura works and how it's connected to the system.

When I have some time to spare I'll try to give it a better look.  I have a compatible board and some things (the secondary SMBus controllers in particular) are very interesting.

olielvewen, 2019-09-07:
|It's nice to see OpenAuraSDK: people are figuring out how Aura works and how it's connected to the system.

Yes I'm agree. And like it is open-source we could hope that will be re-taken by others. Crossed fingers.
Calcprogrammer1 aka Adam Honse is not a beginner. Here a set of his projects and I think that the last could be interesting for you even he is old (if my memory is still good.)

- [CorsairKeyboardSpectograph](https://github.com/CalcProgrammer1/CorsairKeyboardSpectrograph)
- [KeyboardVisualiser](https://gitlab.com/CalcProgrammer1/KeyboardVisualizer)
- [Fan Bus - Fan Controler](https://github.com/CalcProgrammer1/FanBus-Fan-Controller)

In the same way (ie at low level) here a collection of project which can be interesting for your projects.

- [nvidia-led](https://github.com/Madh93/nvidia-led)
- [amdgpu-tweakd](https://github.com/amezin/amdgpu-tweakd)
- [asus-fan](https://github.com/daringer/asus-fan)
- [msi-rgb](https://github.com/nagisa/msi-rgb)
- [AsusZenbook-Ambient-Light-Sensor-Controller](https://github.com/danieleds/Asus-Zenbook-Ambient-Light-Sensor-Controller) with [ALS](https://github.com/danieleds/als)
 
I'm going back to what you said earlier about the fact that you want to include the code of Corsairmi in liquidctl and I checked out the readme on the fact that you wanted to implant it in liquidctl. I thought that liquidctl was only for AIO. Hum I have missed something. Could I pilot my RM650i with liquidctl ? If yes (certainly not now if you want to implement the code in) but how ? 

|and some things (the secondary SMBus controllers in particular) are very interesting.

Indeed. It is the only project that  I know which is speaking about SMBus devices. And for me it is interesting about my 32 GB of G.SkillsTrident Z RGB. Hovewer my board is not (yet ?) compatible. 

I crossed fingers that you could understand and implement it quickly in  liquidctl-linux-drivers.

jonasmalacofilho, 2019-09-08:
> I'm going back to what you said earlier about the fact that you want to include the code of Corsairmi in liquidctl

Actually I just mentioned I might look into making a _kernel_ driver for it, but I haven't really given much thought on where I'd host it.  It'd probably be liquidctl-linux-drivers or a separate repository.

> I thought that liquidctl was only for AIO.

The description is: "Cross-platform CLI and Python drivers for AIO liquid coolers _and other devices."_   Anything with sensors or fan controllers that's related to hardware monitoring can fit that "other" category; in fact, neither the NZXT Smart Device nor the Grid+ V3 are AIOs.  Also, on #31 I state that I'm ok with liquidctl having support for PSUs.

> Could I pilot my RM650i with liquidctl ? If yes (certainly not now if you want to implement the code in) but how ?

Sure, if support is added for it.  Both arbitrary sensors and fan control are already things liquidctl is able to handle.

liquidctl is also designed so that's easy to new options for specific devices; for example, creating a `--rail-mode <single|multi>` would be very straightforward, if the RM650i supports/requires this feature.

**In fact, I strongly encourage you to start porting the code from corsarRMi to liquidctl!**

> Indeed. It is the only project that I know which is speaking about SMBus devices. And for me it is interesting about my 32 GB of G.SkillsTrident Z RGB. Hovewer my board is not (yet ?) compatible.
> I crossed fingers that you could understand and implement it quickly in liquidctl-linux-drivers.

Honestly, I'm not super interested in working on RGB kernel drivers.  In part because these controllers don't _usually_  expose the LEDs themselves, but only a small set of presets.  So for me that's something more suited to userspace.

In the development of OpenAuraSdk the kernel drivers only exist to expose a _secondary_ SMBus controller that's still lacking in the mainline drivers.  And that's stuff not even technically part of the project, and will hopefully be mainlined (=included in "official" kernels).

olielvewen, 2019-09-08:
> The description is: "Cross-platform CLI and Python drivers for AIO liquid coolers and other devices." Anything with sensors or fan controllers that's related to hardware monitoring can fit that [...]

Okay. I keep this link under the elbow.

>Sure, if support is added for it. Both arbitrary sensors and fan control are already things liquidctl is able to handle.

Not sure because when I run `sudo liquidctl list` I got nothing. My device is not recognize.

>liquidctl is also designed so that's easy to new options for specific devices; for example, creating a --rail-mode <single|multi> would be very straightforward, if the RM650i supports/requires this feature.

Indeed, this example is logical and will be welcome. And the RM650i supports this feature.
Here the features of the RMi series power supplies :  

- read monitoring of PSUs (in real time)
- pilot fan but you must know that he is off at less of 50% of charge. After he is active
- Rail +12V configurable multirail +12 (by default) and you can switch in single mode

>In fact, I strongly encourage you to start porting the code from corsarRMi to liquidctl!

- Firstly, I want to test corsairmi, I have asked on the french section of Manjaro to check if my pkbuild is good !!
- Secondly, I must make a virtual machine and see what corsair can provide about this device(in progress since 3 days that I want to run it!!!)
- Thirsty, I'll try to make reverse engineering 
- Latellyl, I'll study deeply the code of Corsairmi (after all he is pretty short)
and finally, perhaps I'll see if I can have some free time to do that and obviously if I can do it. 
Anyway, I am a turtle........ and initially it was not I ve been planned. Well we  'll see. I have some things to finish and the good news is that the good season for programming is coming.

>Honestly, I'm not super interested in working on RGB kernel drivers. In part because these controllers don't usually expose the LEDs themselves, but only a small set of presets. So for me that's something more suited to userspace.
In the development of OpenAuraSdk the kernel drivers only exist to expose a secondary SMBus controller that's still lacking in the mainline drivers. And that's stuff not even technically part of the project, and will hopefully be mainlined (=included in "official" kernels).

it could be complete liquidctl-linux-drivers as you have said that now more and more manufacturers are using the motherboard (i.e CoolerMaster). I think that at a moment you should come to this solution.

jonasmalacofilho, 2019-09-11:
> Not sure because when I run sudo liquidctl list I got nothing. My device is not recognize.

Of course, support for the _device_ still has to be added...

... which is what I just started in #39!  :smile: 

Please give it a try and report back (on the PR, to keep things organized).