liquidctl/liquidctl #37: Strange behaviour running "status" in userspace ( non sudo/root )

source83, 2019-09-05:
Hi, I'm loving the fact that this is Python based (my favorite programming language).

My AIO: Corsair H115i
Detected correctly as well: `sudo liquidctl list
Device 0, Corsair Hydro H115i (experimental)`

More of a strange bug or strange behavior:
when using `liquidctl status` as normal user, you get the readout/status just fine (might be because of my changes to udev so the aio would be accessible in userspace).
But, the fan revs up to max.
As soon as I run the same command with sudo, the fan revs back down again.

jonasmalacofilho, 2019-09-05:
Hi!

> More of a strange bug or strange behavior: when using liquidctl status as normal user, you get the readout/status just fine. [...] But, the fan revs up to max.  As soon as I run the same command with sudo, the fan revs back down again.

This is most likely an issue related to how we work around a limitation of these older Asetek devices: they require sending some sort of configuration change just to read the current status, so we have to store those values somewhere.

My first guesses were either a permission issue or that I had set the path differently for root and regular users, but reviewing the code it doesn't seem that either case fits.

Still, since you haven't reported getting an error message, and the fans are maxing out, I still think the problem is liquidctl not finding the data for the fan speed configuration and thus [defaulting to 100%](https://github.com/jonasmalacofilho/liquidctl/blob/f4ec138b03dbee39231da6db643b33a3e6445a01/liquidctl/driver/asetek.py#L362).

Could you post the outputs of both invocations (root and regular user) but with the addition of the `--debug` flag?

---

```
$ liquidctl set fan speed  20 30  30 50  34 80  40 90  50 100
NotImplementedError
```

I haven't enabled the configuration of fan profiles on the "legacy" Asetek 690LC devices, because it further complicates the quirk I mentioned above (possible ordering issues).  In fact, enabling this could be particularly tricky with old Krakens, as mentioned in #35 ("Some Kraken X61 functionality still missing").

But I'm taking a second look at this with more focus on the Corsair coolers, which "quirk-wise" could sit somewhere in between the old Krakens and the new EVGA units.  I'll let you know if I come up with anything.

source83, 2019-09-05:
[liquidctl_sudo.log](https://github.com/jonasmalacofilho/liquidctl/files/3581441/liquidctl_sudo.log)
[liquidctl_userspace.log](https://github.com/jonasmalacofilho/liquidctl/files/3581442/liquidctl_userspace.log)

source83, 2019-09-05:
Btw, it might be a good idea to add to the default output when you don't have any args that to get the full options etc list, you need to use `--help`

source83, 2019-09-05:
I almost posted a bug post on it after finding the full list in the cli.py file and not testing the `--help` option first.

jonasmalacofilho, 2019-09-06:
> liquidctl_sudo.log
> liquidctl_userspace.log

You forgot to add the `--debug` flag during the run without sudo.

Can you also check if regular users can read the files in `/usr/local/share/liquidctl/1b1c_0c0a_0100/usb1_9/`?

> Also, here is my UDEV rule for the AIO's USB ctrl:

Thanks!  I was going to ask for that too but forgot, so I'm glad that you remembered.

---

> Btw, it might be a good idea to add to the default output when you don't have any args that to get the full options etc list, you need to use --help.
> I almost posted a bug post on it after finding the full list in the cli.py file and not testing the --help option first.

If you can help me with this and other documentation issues (e.g. new docs for the Corsair devices; converting the issue comments for the EVGA CLCs and old Krakens into proper `/docs/*.md` files) I would very much appreciate it.

jonasmalacofilho, 2019-09-09:
Can you give commit 3f004ee3b68bba4b72de947b021bedf2bd320695 ("Add support for setting fan profiles on all legacy 690LC devices") a try?

Since it's a first attempt I imagine there might be some issues left.¹  Hopefully you'll be able to easily fix them or, least, figure out more about what's going on with the `--debug` flag.

Besides an obvious failure, I would also like to ask you to check for more subtle issues:

 - fan speed profiles work (i.e. the fan speed changes) but result in errors or warnings (e.g. timeouts);
 - or they appear to fully work but the everything else starts to behave badly afterwards (e.g. lighting modes, pump speed, or even checking the status)

In case anything goes bad, reverting to 'master' and re-initializing the device should clear most problems.  As a last resort, power cycling it should completely reset it.

¹ You should know I don't have access to these exact devices, so I'm testing this on a similar (but only to a point) cooler.

source83, 2019-09-11:
okay, before doing a pull, here is the missing userspace status with --debug
[liquidctl_userspace-2.log](https://github.com/jonasmalacofilho/liquidctl/files/3599181/liquidctl_userspace-2.log)

source83, 2019-09-11:
after a pull:
```source@megumi:liquidctl$ ls -l /usr/local/share/liquidctl/1b1c_0c0a_0100/usb1_9/
total 8
-rw-r--r-- 1 root root 2 sep.  11 07:55 fan_duty
-rw-r--r-- 1 root root 3 sep.   5 21:04 pump_duty
source@megumi:liquidctl$ ls -l /usr/local/share/liquidctl/1b1c_0c0a_0100/
total 4
drwxr-xr-x 2 root root 4096 sep.   3 10:22 usb1_9
```

jonasmalacofilho, 2019-09-11:
> Is the path mismatch the reason the fans go ballistic in userspace ? :D

Yeap!  I just pushed a change to the master branch that should solve this (though you'll need to reconfigure the device).

Another thing: did you get a chance to test the fan profiles from the commit I mentioned before?

> Can you give commit 3f004ee ("Add support for setting fan profiles on all legacy 690LC devices") a try?

---

P.S. Programs running as root are still running in user space.  The only things _not_ running in user space are the kernel and its loaded modules.

source83, 2019-09-13:
Okay, I just tried the commit you mentioned, still gettin the NotImplementedError message:
```
sudo liquidctl set fan speed  20 30  30 50  34 80  40 90  50 100
NotImplementedError
sudo liquidctl --legacy-690lc set fan speed  20 30  30 50  34 80  40 90  50 100
NotImplementedError
sudo liquidctl --legacy-690lc --device 0 set fan speed  20 30  30 50  34 80  40 90  50 100
NotImplementedError
```

source83, 2019-09-13:
```
source@megumi:~$ sudo liquidctl set fan speed  20 30  30 50  34 80  40 90  50 100
NotImplementedError
source@megumi:~$ sudo liquidctl --legacy-690lc set fan speed  20 30  30 50  34 80  40 90  50 100
NotImplementedError
source@megumi:~$ sudo liquidctl --legacy-690lc --device 0 set fan speed  20 30  30 50  34 80  40 90  50 100
NotImplementedError
```

jonasmalacofilho, 2019-09-13:
I don't understand what's going on:

 - `CorsairAsetekDriver` is a pretty dump subclass of `LegacyAsetekDriver`.  
 - before `LegacyAsetekDriver` did not implement `set_speed_profile`, which caused `NotImplementedError` to be raised... but now it does implement it!

Just as sanity check, can I ask you to confirm that `liquidctl` is pointing to the correct place (maybe you have multiple installations?) and version (3f004ee)?

If the mystery persists, can you show the output with `--debug`?

source83, 2019-09-13:
Okay, sorry, I was not on that commit, I am now. there is no error, but the fan will not change its speed at all.

source83, 2019-09-13:
```
source@megumi:~$ liquidctl --version
liquidctl v1.2.0rc2.dev0
source@megumi:~$ liquidctl status
Device 0, Corsair Hydro H115i (experimental)
Liquid temperature          29.4  °C 
Fan speed                    720  rpm
Pump speed                  3090  rpm
Firmware version         2.8.0.0     

source@megumi:~$ sudo liquidctl --legacy-690lc --device 0 set fan speed  20 30  30 50  34 80  40 90  50 100
[sudo] password for source: 
source@megumi:~$ sudo liquidctl --legacy-690lc --device 0 set fan speed  50^C0  30 50  34 80  55 90  50 100
source@megumi:~$ sudo liquidctl --legacy-690lc --device 0 set fan speed  20 50  30 55  34 80  40 90  50 100
source@megumi:~$ liquidctl status
Device 0, Corsair Hydro H115i (experimental)
Liquid temperature          29.4  °C 
Fan speed                    720  rpm
Pump speed                  3060  rpm
Firmware version         2.8.0.0     

source@megumi:~$ sudo liquidctl --legacy-690lc --device 0 set fan speed  20 100  30 100  34 100  40 90  50 100
source@megumi:~$ sudo liquidctl --device 0 set fan speed  20 100  30 100  34 100  40 90  50 100
source@megumi:~$ liquidctl status
Device 0, Corsair Hydro H115i (experimental)
Liquid temperature          29.4  °C 
Fan speed                    720  rpm
Pump speed                  3030  rpm
Firmware version         2.8.0.0     

source@megumi:~$ sudo liquidctl --device 0 set fan speed  50 50  30 100  34 100  40 90  50 100
source@megumi:~$ liquidctl status
Device 0, Corsair Hydro H115i (experimental)
Liquid temperature          29.3  °C 
Fan speed                    660  rpm
Pump speed                  3090  rpm
Firmware version         2.8.0.0     

source@megumi:~$ sudo liquidctl --device 0 set fan speed  20 50  25 55  30 60  35 65  40 70
source@megumi:~$ liquidctl status
Device 0, Corsair Hydro H115i (experimental)
Liquid temperature          29.3  °C 
Fan speed                    720  rpm
Pump speed                  3090  rpm
Firmware version         2.8.0.0     

source@megumi:~$ sudo liquidctl --device 0 set fan speed  100
source@megumi:~$ sudo liquidctl --legacy-690lc --device 0 set fan speed 100
source@megumi:~$ liquidctl status
Device 0, Corsair Hydro H115i (experimental)
Liquid temperature          29.3  °C 
Fan speed                    720  rpm
Pump speed                  3000  rpm
Firmware version         2.8.0.0     
```

source83, 2019-09-13:
sorry, forgot the --debug output:
```
sudo liquidctl --legacy-690lc --device 0 set fan speed 100 --debug
[DEBUG] liquidctl.driver.asetek: data directory for device is /usr/local/share/liquidctl/1b1c_0c0a_0100/usb1_9
[DEBUG] liquidctl.driver.asetek: open device
[INFO] liquidctl.driver.asetek: using a flat profile to set fan to a fixed duty
[INFO] liquidctl.driver.asetek: filling missing 4 PWM points with (60°C, 100%)
[DEBUG] liquidctl.driver.asetek: stored fan_temp0=0
[DEBUG] liquidctl.driver.asetek: stored fan_duty0=100
[DEBUG] liquidctl.driver.asetek: stored fan_temp1=59
[DEBUG] liquidctl.driver.asetek: stored fan_duty1=100
[DEBUG] liquidctl.driver.asetek: stored fan_temp2=60
[DEBUG] liquidctl.driver.asetek: stored fan_duty2=100
[DEBUG] liquidctl.driver.asetek: stored fan_temp3=60
[DEBUG] liquidctl.driver.asetek: stored fan_duty3=100
[DEBUG] liquidctl.driver.asetek: stored fan_temp4=60
[DEBUG] liquidctl.driver.asetek: stored fan_duty4=100
[DEBUG] liquidctl.driver.asetek: stored fan_temp5=60
[DEBUG] liquidctl.driver.asetek: stored fan_duty5=100
[DEBUG] liquidctl.driver.asetek: begin transaction
[DEBUG] liquidctl.driver.asetek: loaded fan_temp0=0 (from cache)
[DEBUG] liquidctl.driver.asetek: loaded fan_duty0=100 (from cache)
[INFO] liquidctl.driver.asetek: setting fan PWM point: (0°C, 100%), device interpolated
[DEBUG] liquidctl.driver.asetek: loaded fan_temp1=59 (from cache)
[DEBUG] liquidctl.driver.asetek: loaded fan_duty1=100 (from cache)
[INFO] liquidctl.driver.asetek: setting fan PWM point: (59°C, 100%), device interpolated
[DEBUG] liquidctl.driver.asetek: loaded fan_temp2=60 (from cache)
[DEBUG] liquidctl.driver.asetek: loaded fan_duty2=100 (from cache)
[INFO] liquidctl.driver.asetek: setting fan PWM point: (60°C, 100%), device interpolated
[DEBUG] liquidctl.driver.asetek: loaded fan_temp3=60 (from cache)
[DEBUG] liquidctl.driver.asetek: loaded fan_duty3=100 (from cache)
[INFO] liquidctl.driver.asetek: setting fan PWM point: (60°C, 100%), device interpolated
[DEBUG] liquidctl.driver.asetek: loaded fan_temp4=60 (from cache)
[DEBUG] liquidctl.driver.asetek: loaded fan_duty4=100 (from cache)
[INFO] liquidctl.driver.asetek: setting fan PWM point: (60°C, 100%), device interpolated
[DEBUG] liquidctl.driver.asetek: loaded fan_temp5=60 (from cache)
[DEBUG] liquidctl.driver.asetek: loaded fan_duty5=100 (from cache)
[INFO] liquidctl.driver.asetek: setting fan PWM point: (60°C, 100%), device interpolated
[DEBUG] liquidctl.driver.asetek: write 11 00 00 3b 3c 3c 3c 3c 64 64 64 64 64 64
[DEBUG] liquidctl.driver.asetek: loaded pump_duty=100
[INFO] liquidctl.driver.asetek: setting pump duty to 100%
[DEBUG] liquidctl.driver.asetek: write 13 64
[DEBUG] liquidctl.driver.asetek: received 02 94 01 2f 00 06 00 da 0c 12 1d 11 00 00 08 f9 00 22 7b 07 1c be 98 02 08 00 00 00 01 29 64 15
[DEBUG] liquidctl.driver.asetek: close device
```

jonasmalacofilho, 2019-09-13:
Oh well, I guess the device doesn't like this approach.

Hm, I see that the firmware version is pretty recent... my EVGA CLC is only on 2.10.0.0, and I'm guessing Asetek uses a global versioning scheme for the 690LC family.  You could probably give the "modern" `AsetekDriver` a try.

First, revert back to the latest version on 'master', and make sure fan and pump speeds are working again.  For this you might need to powercycle the device (i.e. the system).

Then, edit `liquidctl/driver/asetek.py`, moving your [device id](https://github.com/jonasmalacofilho/liquidctl/blob/8de7d64dcd2bd0b69c51bfbc1b61d1af0be3dc10/liquidctl/driver/asetek.py#L456) to the [`AsetekDriver`](https://github.com/jonasmalacofilho/liquidctl/blob/8de7d64dcd2bd0b69c51bfbc1b61d1af0be3dc10/liquidctl/driver/asetek.py#L200) class.  And please let me know how it goes.

source83, 2019-09-13:
My pump and correct description is already in the list:
`(0x1b1c, 0x0c0a, None, 'Corsair Hydro H115i (experimental)', {}),`

jonasmalacofilho, 2019-09-13:
Oh, did you `initialize` the device before testing the new code?

The command used specifically for fixed fan speeds overrules any profiles until the device is initialized again (actually I even had to powercycle it once, but maybe that was my fault).

The same will be necessary if you change your device to bind to the modern `AsetekDriver`, instead of `CorsairAsetekDriver`.

---

> My pump and correct description is already in the list: (0x1b1c, 0x0c0a, None, 'Corsair Hydro H115i (experimental)', {}),

It's a member of `CorsairAsetekDriver` (which subclasses the `LegacyAsetekDriver`), not `AsetekDriver`.

source83, 2019-09-13:
dooooh...... no I didn't run init :headpalm:

source83, 2019-09-13:
yepp, that worked

source83, 2019-09-13:
btw, is there a discord server I could chat with you in ?

jonasmalacofilho, 2019-09-13:
:smiley: 

3f004ee or the custom edit?

---

> btw, is there a discord server I could chat with you in ?

Sorry, I stopped using that.  We could do Telegram, it that's a thing you use.

commit 0c2dc045 (2019-09-13): Merge pull request #41 from source83/use-modern-driver-for-h115i

Move device identifier from CorsairAsetekDriver to AsetekDriver.

@source83 also reports that while `rainbow` doesn't work, it also doesn't cause any issues.

Closes: #37 ("Strange behavior running "status" as regular user")
Related: 3f004ee ("Add support for setting fan profiles on all legacy 690LC devices") (supersedes it for the H115i)
closed 2019-09-13 by jonasmalacofilho