liquidctl/liquidctl #38: Control Two different sets of fans.

Juveb, 2019-09-07:
Hi, 

First, a thank you: 
I recently built a Hackintosh and love liquidctl! Without it I would have been nervous about inadequate cooling, so a big thank you for your hard work!!! It's especially impressive considering the Hackintosh driver FakeSMC doesn't support or even see any fans, yet you guys still have control over the fans on my device! 

Issue/Feature Request: 
I have two different sets of fans installed, one set attached to my x72 rad, and another set from a different brand for case fans. The case fans are connected to a second fan header on my MB. All of my fans are "supported" and spool up correctly, for instance with **_liquidctl set fan speed 100_** both sets of fans ramp up. I assume this is a bit of a happy accident since **_liquidctl list_** only shows the x72 installed. Since they are different brands and spin at different min/max I would like to have individual control over the two sets. Right now it seems at idle case fans spin at ~1100 while the other set on the x72 is at ~700.  

Is this at all possible? I'm happy to even work on this myself if you could provide some guidance on how your code is working. I have some programming knowledge of python, C++ and random bits of other languages. I'm not very knowledgable about architecture level stuff though,  like which bus on the mobo and such. Thanks!!

jonasmalacofilho, 2019-09-08:
Thanks!

> I have two different sets of fans installed, one set attached to my x72 rad, and another set from a different brand for case fans. The case fans are connected to a second fan header on my MB. 

liquidctl isn't controlling the fans connected to the motherboard header.  They must be running according to the UEFI/BIOS settings, since they aren't visible on the OS level.

> Is this at all possible? I'm happy to even work on this myself if you could provide some guidance on how your code is working. I have some programming knowledge of python, C++ and random bits of [...]

Technically it is possible, but I don't know much (or at all) about how hard it is to write kernel drivers for Mac OS.  Assuming that's not too much of a pain, the first step will be to figure out exactly the chip the header is connected to.

Generally this will be the Super I/O chip, but some headers can also be directly connected to something called the board's "embedded controller" (e.g. CPU_OPT on my ASUS PRIME Z370-A).  Windows + HWiNFO64 should help.

You can also just assume _something_ good will come from having access to the controllers and sensors from the Super I/O chip, and look at the board (or check online) to figure out what chip it uses.

jonasmalacofilho, 2019-09-11:
Closing...

@Juveb feel free to ask more questions, I'll try to answer them as best as I can.

closed 2019-09-11 by jonasmalacofilho