liquidctl/liquidctl #43: Maybe this is a stupid question but: which temperate?

camh, 2019-09-18:
First off, thanks for liquidctl! I had a couple issues at first, but then noticed you said that rc1 had better compatibility so I got rc3 up and running without a hitch.

This may be a stupid question, but when setting temperature/percentage profiles, what's temperate is this based on? I assume the profiles for pump speed is based on the temperate of the liquid. Is that correct?

Are the temperates for fan speed the CPU temperature or the liquid temperature?

A fan speed setting of 

```liquidctl set fan speed 25 35  30 45  35 60  40 70  45 80  50 90  60 100```

really doesn't make sense if the temperate is the liquid temperate, cause as far as I can tell my liquid temp never gets above ~38° C. So it would only ever get to 60% fan speed.

On the other hand, if this is based on CPU temperate, I'm not seeing RPMs at percentages relative to the temp/percentage for the current temperate of the CPU. Example: using the example above, my CPU at 80° C doesn't ramp up to 1800 RPM (the highest RPM of the X62 Kraken) but it should be at 100% since it's over 60° C.

jonasmalacofilho, 2019-09-18:
Hi there!

The liquid temperature is used on both channels, because that's all that the device¹ has access to, and liquidctl configures a profile that will run on the device, and not on the host CPU.

Personally I'm fine with that because the coolant mass acts as buffer and prevents sudden fan speed bursts.  But there's also an experimental script in [`extra/krakencurve-poc`](https://github.com/jonasmalacofilho/liquidctl/blob/master/extra/krakencurve-poc#L1) that can run on your host system and control the pump and fan speeds according to arbitrary sensors.   

```
# extra/krakencurve-poc --use-psutil show-sensors
kraken:Liquid temperature                                                 29.1°C
coretemp:Package id 0                                                     39.0°C
coretemp:Core 0                                                           39.0°C
coretemp:Core 1                                                           30.0°C
coretemp:Core 2                                                           30.0°C
coretemp:Core 3                                                           30.0°C
coretemp:Core 4                                                           36.0°C
coretemp:Core 5                                                           29.0°C
nct6793:SYSTIN                                                            28.0°C
nct6793:PECI Agent 0                                                      31.0°C
nct6793:Virtual_TEMP                                                      34.0°C
nct6793:PECI Agent 0 Calibration                                          28.0°C

# extra/krakencurve-poc control --use-psutil --fan-sensor 'coretemp:Package id 0' '(25,50),(35,100)' '(25,35),(35,60),(60,100)' --verbose
```

**Well, sort of...**

The thing is, what operating system are you running?  Linux has a standard hardware monitoring API, and krakencurve-poc can access those sensors through psutil (it could even do it directly from sysfs, but it's just a proof of concept).

Some Unixes have standard sensor APIs too, but I'm not sure if psutil can handle them, or if MacOS is not an exception.  And running Windows you'd certainly need to adapt the script and integrate it with something like OpenHardwareMonitor or, maybe a RTSS server (e.g. HWiNFO).

---

Notes:

¹ I'm assuming yours is a X62.

camh, 2019-09-18:
In hindsight, it makes sense the liquid cooling AIO unit only has access to the liquid temperate. Thanks for clearing that up! It'll be much easier to create profiles knowing for sure/

I'm on macOS, but I'm not sure I want to mess with experimental scripts for something as important as my CPU cooling. I'm not proficient with Python enough to know how to fix it myself.

jonasmalacofilho, 2019-09-18:
I get that, and the proof of concept script really isn't robust enough for unsupervised use.

On the other hand, running

```
liquidctl set pump speed 100
liquidctl set fan speed 100
```

before you starting krakencurve would probably make you safe _with any v4 or later¹ firmware versions._

Krakencurve uses a specific "instantaneous/override speed change" command that only lasts a few seconds, and these devices should revert back to the configured profile if the script crashes and no new updates are sent.

Then again, you'd still need to figure out how to make it access generic sensor data Mac OS...

¹ As far as I know.

---

From a theoretical standpoint the best would be:

 - the pump takes into account both CPU and coolant temperatures, as increased flow helps both CPU-to-coolant and coolant-to-environment heat exchanges;
 - the fans only depend on the coolant temperature, as they only need to keep it cool (they don't directly affect the CPU temperature).

So using the coolant temperature for the fans is actually ok, just make it more aggressive as that parameter doesn't vary a lot.

And in practice whatever you use to control the pump will be fine: these AIO pumps are not very powerful or flexible and the differences in CPU temperatures between 50% duty (~2000 rpm) and 100% duty (~2800 rpm) are quite small.

closed 2019-09-18 by jonasmalacofilho