liquidctl/liquidctl #45: Improve support for Corsair HXi/RMi PSUs

jonasmalacofilho, 2019-09-27:
Improve [CorsairHidPsuDriver](https://github.com/jonasmalacofilho/liquidctl/blob/master/liquidctl/driver/corsair_hid_psu.py) adding support for:

 - [ ] fan control
 - [ ] 12V multirail OCP configuration

Both features probably require some USB traffic captures, but it might be possible to implement fan control just from the PMBus spec and some trial and error.

jonasmalacofilho, 2019-09-27:
@olielvewen, can you help?

olielvewen, 2019-10-02:
Hi Jonas,

Sorry for the delay, I was far away from my computer and I have not been notified by your message and others. Fortunately, Jonathan sent me an email at last the second because I also did not see the first. I hate phones and I like computers.

Okay. Tell me what do you want. 

I have done some search and I have not seen the Firmware. And not only that even the idvendor name is not present. I have take a look how do you (i.e liquidctl) works and I just flew the PMBus specification and I confirm I still hate electricity.
Well I take all ideas for fan control but it will be not an easy task as he works only at 40/50 % of charge. I'm curious what do you want to proceed.

olielvewen, 2019-10-02:
Did we are "working" on this branch ("45) as the last time with the #39 ?
I think that will be that but i would like to be clear. 
Anyway, I will remove my actual branch to update to this one for helping you.

olielvewen, 2019-10-02:
BTW. Did you have seen this news ? 

- [https://www.overclockingmadeinfrance.com/enermax-liqmax-iii-argb/](url)

- [https://www.tomshardware.fr/enermax-lance-ses-kits-de-watercooling-aio-liqmax-iii-en-version-argb/](url)

- [https://www.tomshardware.fr/thermaltake-lance-des-configs-de-watercooling-completes-bourrees-de-rgb/](url)

A lot of work in perspective..

jonasmalacofilho, 2019-10-02:
Hi @olielvewen!

> Sorry for the delay, I was far away from my computer and I have not been notified by your message and others

Don't worry about it.  And not getting GitHub notifications is, unfortunately, not unheard of.  Try double checking your GitHub notification settings.  And if there's nothing obviously wrong there try contacting GitHub support.

> Okay. Tell me what do you want.

Try installing CorsairLink in a virtual machine, pass the PSU through to it, and send me USB traffic captures during which you change fan speeds, the multirail configuration and any other available settings.  And also some screenshots...

> I have done some search and I have not seen the Firmware. 

I think there's supposed to be a "devices" screen in CorsairLink that should list the firmware version.  Let me know what it shows.

> Did we are "working" on this branch ("45) as the last time with the #39 ?

From your feedback I gathered that #39 was already functional, and so I have since merged it into the 'master' branch.

Note: there is no corresponding specific branch for this/#45, as this is an issue, not a pull request.  The initial HXi/RMi driver now lives in 'master', and the new features we're discussing here will probably go through their own pull requests (unless they are trivial).

olielvewen, 2019-10-03:
>Don't worry about it. And not getting GitHub notifications is, unfortunately, not unheard of. Try double checking your GitHub notification settings. And if there's nothing obviously wrong there try contacting GitHub support.

Well, it doesn't matter that's not a priority for the moment. The most important is that I've seen your message and that I could get some free time for helping you.

>Try installing CorsairLink in a virtual machine, pass the PSU through to it, and send me USB traffic captures during which you change fan speeds, the multirail configuration and any other available settings. And also some screenshots...

That's already done but I've not still played with ICue but just one time with Steelseries Engine. Not funny for my HeadSet Artic Silver 7.

>I think there's supposed to be a "devices" screen in CorsairLink that should list the firmware version. Let me know what it shows.

Okay

>From your feedback I gathered that #39 was already functional, and so I have since merged it into the 'master' branch.
Yes I 've seen that yesterday. Great news. 

> Note: there is no corresponding specific branch for this/#45, as this is an issue, not a pull request. The initial HXi/RMi driver now lives in 'master', and the new features we're discussing here will probably go through their own pull requests (unless they are trivial).

Okay.

olielvewen, 2019-10-05:
I come back with a lot of stuff that I've done yesterday afternoon.

Here how I have proceded :
Test Protocol:

- Test of mutirail (default) to switch monorail

- Test of modification fan - choose between custom curve (default) and switch to custom curve at  %fixed (default 40%)  at 53%

- Test of curve (pre-enregistred) called Quiet

- Test of curve (pre-enregistred) called Balanced (seems to be by default when we re-initialize preset)

- Test of curve (pre-enregistred) called Extreme (note this time we can see fan to 975 RPM...)

- Test of fan in graphic tab by value( 1minute, .....1day) (display fan use in a graph ...)

I have all download on my personal and open source cloud and I will put at the end the link. You could watch, download them for 30 days (more or other if you want just told me.)

I have put screenshots in a general folder with pcapg.gz file and a text file. You could open it in Wireshark.

I have lost a lot of time because this dirty ICue doesn't run but SteelSeries Engine3 yes. After 10 go to run it, 3 re-install I have been oblige to remove it and install the previous version 3.11.114 (instead of the 3.19.120).

I think that I have all done. Tell me if : 

- it is okay for you (like it is the first time I was using Wireshark and the interface was a bit different of Tutorial's Flozz)

- you need more test

---

## Note:

1. Impossible to get the firmware in Icue. This feature seems not to be available.

2. not sure about protocol used usb.device_adress and not usbhid.date or usbhid.item.date
---

## Link file : 
[CorsairRm650iReverseEngineering](https://my.pcloud.com/publink/show?code=kZSWfLkZksTtBQ9gioHWsb8POY3b075abcD7)

jonasmalacofilho, 2019-10-05:
That's awesome!

I haven't checked the contents yet... just looked them over, I'm out now and writing this from my phone.

I guess the device release number (bcdDevice) could be used for firmware revisions, yes, though it's normally more used to indicate hardware revisions.  But just to check: iCue/CorsairLink doesn't list the firmware version for the PSU, only for another device, and for that device the firmware version it (iCue/CorsairLink) lists is the same as the bcdDevice?

As for the address on the USB bus, it indeed isn't persistent.  It's simply incremented once a new device is bound to the bus, and initialization order and driver binding/unbinding can change it.

USB port numbers are the physical way to identify a device.  But even those are not cross platform, so the best way to uniquely identify a device usually is with (vendor, product, serial number).  

Since the RM650i apparently doesn't expose its serial number through the USB descriptors, it's lucky that having two PSUs in a system is not very common.  But if that were the case then port numbers (or just connecting them in a way that places each of different buses) would be the way to go.

jonasmalacofilho, 2019-10-12:
_After having a good look at the captured traffic, here are my notes on fan control._

There seem to be only two real settings: _mode_ – automatic or manual – and, for the latter, _speed._  Note that "automatic" and "manual" should be interpreted from the perspective of the device: whenever iCue/CorsairLink is controlling the fan speeds, the fan mode is manual, even it the user has selected one of the preset curves.¹

The automatic mode presumably follows the curve displayed in the manual.  In manual mode the fan is (apparently) always set to a constant speed; fan curves in iCue result in speed changes being issued as needed.

Fan mode is (very likely) set with the MFR_SPECIFIC_F0 command.  The parameter appears to be a single byte constant, either 0x00 or 0x01.  These (probably) mean, respectively, auto or manual.²

Fan speed in manual mode is controlled with the standard PMBus FAN_COMMAND_1 (0x3b) command.  Here the parameter is (very likely) the desired duty cycle, (likely) encoded in LINEAR11.

---

Notes:

¹ According to the product specifications and manual, and also corroborated by the captured USB traffic.

² Inferred from the captures: extremecurve.pcapng.gz lists MFR_SPECIFIC_F0 twice, both with parameter 0x01; on the other hand, 1firstcapture.pcapng.gz shows the use of that command with parameter 0x00, which would make sense since the default is "auto" and that was indeed the very first capture file.

olielvewen, 2019-10-18:
> I guess the device release number (bcdDevice) could be used for firmware revisions, yes, though it's normally more used to indicate hardware revisions. But just to check: iCue/CorsairLink doesn't [...]

I just come back on this. I'm nearly sure that the firmware revisions. Check [here]([url=http://pix.toile-libre.org/?img=1571419915.jpg][img]http://pix.toile-libre.org/upload/img/1571419915.jpg[/img][/url])
And even I've not the screenshot about my headset Steelseries Artic, i can said that his firmware version 1.19 is the same that the number bcdDevice shows. 
It will be confirm the day that I could get another one after updating it. and check again.

closed 2019-10-20 by jonasmalacofilho