liquidctl/liquidctl #46: Improve support for Seasonic/NZXT E-series PSUs

jonasmalacofilho, 2019-09-27:
Basic monitoring support has been added with #44, but several features are still missing in the [SeasonicEDriver](https://github.com/jonasmalacofilho/liquidctl/blob/master/liquidctl/driver/seasonic.py):

 - [ ] device IDs for the E650 and E850 models
 - [ ] monitor total uptime
 - [ ] read firmware version
 - [ ] fan control
 - [ ] 12V multirail OCP configuration

Replaces: #31 ("Support for nzxt e series digital power supplies")

Also see:

 - [First capture of USB traffic (e500-capture-1.zip)](https://github.com/jonasmalacofilho/liquidctl/files/3664436/e500-capture-1.zip)
 - [Additional USB traffic (e500-capture-2.zip)](https://github.com/jonasmalacofilho/liquidctl/files/3664437/e500-capture-2.zip)
 - [A typical update cycle (comment on #31; possibly outdated)](https://github.com/jonasmalacofilho/liquidctl/issues/31#issuecomment-530712828)

Note: captures provide by Jon in #31 and #44.

---

Device IDs
---------------

Use `lsusb` (Linux) or search for hardware IDs in Device Manager (Windows).

The Vendor ID is probably the same as the E500: `0x7793`.  Searching for it online yields `0x5912` and `0x2500` as possible product IDs for the remaining SKUs.

_Trivia: NZXT's official vendor ID is `0x1e71`, which in **decimal** is 7793._

Total uptime
------------------

Seems to part of the the response for the later writes in a update cycle.  Since this depends on writes, it's important to first figure out what those do.

```
command bytes                    | response bytes           | decoded (r|w: command: response)
-------------------------------- | ------------------------ | ------------------------------------------------------------------
af 00 ef 01 60 dc 02 04 8c 00... | aa ef 00 00 00 00 00...  | r: MFR_SPECIFIC(???): 0xaaef_...
ae 01 ef 01 60 dc 02 04 8c 00... | aa 2f ed 00 00 00 00...¹ | w: MFR_SPECIFIC(???): 0xaa2f_0000d0de0100_...
ae 02 ef 01 60 dc 02 04 8c 00... | aa 26 00 00 00 00 00...¹ | w: MFR_SPECIFIC(???): 0xaa26_0000d0de0100_...
ae 03 ef 01 60 dc 02 04 8c 00... | aa 26 00 00 00 00 00...¹ | w: MFR_SPECIFIC(???): 0xaa26_0000d0de0100_...
ae 04 ef 01 60 dc 02 04 8c 00... | aa 26 00 00 00 00 00...¹ | w: MFR_SPECIFIC(???): 0xaa26_0000d0de0100_...
ae 05 ef 01 60 dc 02 04 8c 00... | aa 26 00 00 00 00 00...¹ | w: MFR_SPECIFIC(???): 0xaa26_0000d0de0100_...
ae 06 ef 01 60 dc 02 04 8c 00... | aa 28 00 00 00 00 00...¹ | w: MFR_SPECIFIC(???): 0xaa28_2e00d0de0100_...
                                                                                               ^^^^^^^^
                                                                                             offset: 0x2b
¹ Some non-zero bytes appear later.
```

Current hypothesis: minutes, encoded as a 4-byte big endian integer from offset 0x2b.

Firmware version
-------------------------

Likely related to command 0xFC:

```
command bytes                    | response bytes           | decoded (r|w: command: response)
-------------------------------- | ------------------------ | ------------------------------------------------------------------
ad 00 03 01 60 fc 02 04 8c 00... | aa 03 11 41 d0 00 00...  | r: MFR_SPECIFIC(firmware???): ???
               ^^                        ^^ ^^
```

Figure out how `0x1141` translates into version number "40983.00".

Fan control
-----------------

Unknown.

12V multirail OCP configuration
----------------------------------------------

Unknown.

jonasmalacofilho, 2019-09-27:
@jnettlet hopefully you can continue to help with this : )

jnettlet, 2019-09-28:
> @jnettlet hopefully you can continue to help with this : )

Sure will start with some more usb dumps for fan control and 12V multirail.

jonasmalacofilho, 2019-10-12:
I was working on fan control for Corsair PSUs and realized that I already had the information necessary for also implementing it for NZXT units.  A video confirming that the default curve has a [plateau at 23%](https://youtu.be/V7uAPjb41Nk?t=212) also helped.

Already updated the issue description above with the relevant details... will try to implement and push it to a test PR later.

@jnettlet, did you manage to figure out anything else?

jnettlet, 2019-10-12:
@jonasmalacofilho Unfortunately I have not.  Doing more USB dumps requires a bit of time to move USB connectors around, and time has not permitted.  If you have something to test I can gladly give that a try.  Hopefully tomorrow I will have some time to get the rest of the dumps.

jonasmalacofilho, 2019-10-12:
@jnettlet

Right, while searching for device IDs for the other SKUs I read you mention that you're using the E500 to power a HoneyComb LX2K test system.

By the way, how's the support for ECC memory on that board?

jnettlet, 2019-10-12:
Well besides the fact that ECC so-dimms are a bit hard to find, and ddr4
3200 not existent, good.  I am running dual crucial 16gb 2600 ECC sticks
without issues.

jonasmalacofilho, 2019-12-23:
It seems that interest on the missing features is low.

Anyone out there still able and willing to help with this?

jnettlet, 2019-12-23:
I am mostly just waiting until after the holiday season.  I will have
new cases coming in which means that I can free up my PSU for
development.  Right now it is integrated into my ARM64 workstation, so
no Windows ECAM support yet.   This is a pet project for me and my
time has been limiting working on higher priority items right now.

jonasmalacofilho, 2019-12-23:
Hi Jon!

That's fine.  It's just good to know that you're still interested.

I'll the issue open, let me know when you start working on this.

Evils-Devils, 2020-03-29:
@jonasmalacofilho since my unit is suspect i'm inclined not to open a separate issue for this

but i've been seeing 2 additional issues with my instance
these occur fairly often and apparently also when this is the only instance running (had one in my status bar and another saving to a log (can share that as well if you want an idea of frequency (seems to be about every ~20th)))
1. bad readings
[bad-reading.2020-03-29.txt](https://github.com/jonasmalacofilho/liquidctl/files/4398257/bad-reading.2020-03-29.txt)
[bad-reading-master.2020-03-29.1.txt](https://github.com/jonasmalacofilho/liquidctl/files/4398258/bad-reading-master.2020-03-29.1.txt)
[bad-reading-master.2020-03-29.2.txt](https://github.com/jonasmalacofilho/liquidctl/files/4398259/bad-reading-master.2020-03-29.2.txt)
[bad-reading-master.2020-03-29.txt](https://github.com/jonasmalacofilho/liquidctl/files/4398260/bad-reading-master.2020-03-29.txt)
2. failures (at output?) with traceback
[traceback.2020-03-29.txt](https://github.com/jonasmalacofilho/liquidctl/files/4398261/traceback.2020-03-29.txt)
[traceback-master.2020-03-29.txt](https://github.com/jonasmalacofilho/liquidctl/files/4398262/traceback-master.2020-03-29.txt)

jonasmalacofilho, 2020-03-29:
If I'm not mistake, your specific unit doesn't ever produce sensible results for these fields (even in CAM).  Thus I'm inclined to think that these other failures are just different manifestations of the same problem.

That said, given the very limited testing of this driver (and possible firmware limitations), I expect there to be a least some issues to fix when there are multiple processes simultaneously accessing the same PSU.

One thing that worried me is your report that "the first part of the firmware version get mangled".  That's weird, can you show me an example?

jnettlet, 2020-03-29:
Note that we had one unit and there are multiple reports online of
others with the same issue, where the monitoring DSP does not return
correct, or any values.  Definitely verify your unit with NZXTs CAM
software if you are seeing oddities with the hardware output.

Evils-Devils, 2020-03-29:
@jonasmalacofilho 
expected
`Firmware version                         A017/40983`
get
`Firmware version                         001/40983`

this is on both 1.3.3 and master
it seems to consistently be `001` if it's not `A017`

jonasmalacofilho, 2020-03-29:
@jnettlet 

> multiple reports online of others with the same issue, where the monitoring DSP does not return correct, or any values.

Indeed, I've seen a few myself on r/NZXT (failing in CAM alone).

---

@Evils-Devils 

I had left a constant test string in there and that's why only one of the firmware version representations was changing.  Now both should change when the PSU responds with different (in this case invalid) data.

Evils-Devils, 2020-03-29:
@jnettlet i don't recall seeing the out of range values on CAM, but A and W both read 0 there
just ran my windows VM again and kept an eye on it for enough time that i'd expect to see several bad readings, none have occurred (except for the null current readings)
[image]

jonasmalacofilho, 2020-03-29:
@Evils-Devils it's possible that CAM nowadays just filters out (or replaces with zero) negative or too large values.  This is why I think it's the same issue, just with a slightly different result.

Evils-Devils, 2020-03-29:
built from 70f6b899b82466a9c744e894f76b8916cd62b729, new failure
[traceback-master-_get_fw_versions-2020-03-29.txt](https://github.com/jonasmalacofilho/liquidctl/files/4398386/traceback-master-_get_fw_versions-2020-03-29.txt)

jonasmalacofilho, 2020-03-29:
Yeah, it seems like one of the cases where the version string would get mangled.  I also see that the PSU seems to be sending responses to commands other than those just sent.  It's possible that it somehow keeps a queue, and things can easily get out of sync.

jnettlet, 2020-03-29:
I would have to agree they are just returning zero for non-positive
results.  Have they rolled out a new firmware to fix this issue?  When I
had the problem originally they just wanted to swap the unit out, and
couldn't give a response if this issue could be fixed in a future firmware
release.

jonasmalacofilho, 2020-03-29:
> When I had the problem originally

Did this also happen with liquidctl?

In one of your output samples there's a 12 V rail with 0 A, but that's probably just because your LX2K board doesn't use a supplemental EPS/ATX12V 8/4-pin cable for CPU power (as far as I could see in pictures I found).

jnettlet, 2020-03-29:
yes it happened with both cam and liquidctl.  Correct, we don't need
supplemental CPU power as it is a 30 Watt chip.

closed 2020-12-25 by jonasmalacofilho