liquidctl/liquidctl #48: pip install -U liquidctl==1.2.0 fails on Ubuntu

leinardi, 2019-09-28:
I'm trying to update on one venv liquidctl from 1.1.0 to 1.2.0 but it fails with this error:
```
$ pip install -U liquidctl==1.2.0
Collecting liquidctl==1.2.0
  Using cached https://files.pythonhosted.org/packages/dd/fb/7dedead18957a8fa1d937055fb435c85a611a7eacaad11818caa716ef86d/liquidctl-1.2.0-py3-none-any.whl
Collecting hidapi (from liquidctl==1.2.0)
  Using cached https://files.pythonhosted.org/packages/c1/86/89df0e8890f96eeb5fb68d4ccb14cb38e2c2d2cfd7601ba972206acd9015/hidapi-0.7.99.post21.tar.gz
Requirement already satisfied, skipping upgrade: pyusb in ./venv/lib/python3.7/site-packages (from liquidctl==1.2.0) (1.0.2)
Requirement already satisfied, skipping upgrade: appdirs in ./venv/lib/python3.7/site-packages (from liquidctl==1.2.0) (1.4.3)
Requirement already satisfied, skipping upgrade: docopt in ./venv/lib/python3.7/site-packages (from liquidctl==1.2.0) (0.6.2)
Requirement already satisfied, skipping upgrade: setuptools>=19.0 in ./venv/lib/python3.7/site-packages/setuptools-40.8.0-py3.7.egg (from hidapi->liquidctl==1.2.0) (40.8.0)
Installing collected packages: hidapi, liquidctl
  Running setup.py install for hidapi ... error
    Complete output from command /home/leinardi/Workspace/gitlab/gkraken/venv/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-ihfxsw5r/hidapi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-vnjfgh9k/install-record.txt --single-version-externally-managed --compile --install-headers /home/leinardi/Workspace/gitlab/gkraken/venv/include/site/python3.7/hidapi:
    running install
    running build
    running build_ext
    cythoning hid.pyx to hid.c
    /tmp/pip-install-ihfxsw5r/hidapi/.eggs/Cython-0.29.13-py3.7-linux-x86_64.egg/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /tmp/pip-install-ihfxsw5r/hidapi/hid.pyx
      tree = Parsing.p_module(s, pxd, full_module_name)
    building 'hid' extension
    creating build
    creating build/temp.linux-x86_64-3.7
    creating build/temp.linux-x86_64-3.7/hidapi
    creating build/temp.linux-x86_64-3.7/hidapi/libusb
    x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Ihidapi/hidapi -I/usr/include/libusb-1.0 -I/home/leinardi/Workspace/gitlab/gkraken/venv/include -I/usr/include/python3.7m -c hid.c -o build/temp.linux-x86_64-3.7/hid.o
    x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Ihidapi/hidapi -I/usr/include/libusb-1.0 -I/home/leinardi/Workspace/gitlab/gkraken/venv/include -I/usr/include/python3.7m -c hidapi/libusb/hid.c -o build/temp.linux-x86_64-3.7/hidapi/libusb/hid.o
    hidapi/libusb/hid.c:47:10: fatal error: libusb.h: File o directory non esistente
       47 | #include <libusb.h>
          |          ^~~~~~~~~~
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/home/leinardi/Workspace/gitlab/gkraken/venv/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-ihfxsw5r/hidapi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-vnjfgh9k/install-record.txt --single-version-externally-managed --compile --install-headers /home/leinardi/Workspace/gitlab/gkraken/venv/include/site/python3.7/hidapi" failed with error code 1 in /tmp/pip-install-ihfxsw5r/hidapi/

```

System:
  Host: c7h Kernel: 5.3.0-12-generic x86_64 bits: 64 Desktop: Gnome 3.34.0 
  Distro: Ubuntu 19.10 (Eoan Ermine)

leinardi, 2019-09-28:
Solved installing the following dependencies:
```
sudo apt install libudev-dev libusb-1.0-0-dev
```

closed 2019-09-28 by leinardi

jonasmalacofilho, 2019-09-28:
This is a common issue when installing cython-hidapi, and a big part of why the [Additional requirements on Linux](https://github.com/jonasmalacofilho/liquidctl#additional-requirements-on-linux) section of the README exists.

leinardi, 2019-09-28:
Ops, I missed that section.