liquidctl/liquidctl #50: Error message in syslog every time a command is used: "unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0"

acoutts, 2019-09-29:
More information in this ticket: https://gitlab.com/leinardi/gkraken/issues/14

Xubuntu 18.04
Kraken X62
Firmware version 6.0.2
liquidctl v1.2.0 (05248012be46)

Every time I issue a command in liquidctl these 2 lines are repeated in syslog:
```
Sep 29 15:29:29 xff0c upowerd[2668]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0/0003:1E71:170E.0025
Sep 29 15:29:29 xff0c upowerd[2668]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0
Sep 29 15:29:29 xff0c upowerd[2668]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0/0003:1E71:170E.0025
Sep 29 15:29:29 xff0c upowerd[2668]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0
Sep 29 15:29:29 xff0c upowerd[2668]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0
Sep 29 15:29:29 xff0c upowerd[2668]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0
```

jonasmalacofilho, 2019-09-29:
As I understood from the issue you linked, the problem only reproduces when you have setup udev rules for this device; can you include those here?

This might be a kernel vs. udev issue.  Are you still on kernel 5.0.0-29?  Can you also include the full output of `uname -a`, including any build/patch numbers?

Can you also check if manually {un,}binding causes the same errors, with the udev rules in place?

```
$ echo '1-11' | sudo tee /sys/bus/usb/drivers/usb/unbind
$ echo '1-11' | sudo tee /sys/bus/usb/drivers/usb/bind
```

acoutts, 2019-09-29:
Linux xff0c 5.0.0-29-generic #31~18.04.1-Ubuntu SMP Thu Sep 12 18:29:21 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

udev rules.d file:
```rules.d
SUBSYSTEM=="usb", ATTRS{idVendor}=="1e71", ATTRS{idProduct}=="170e", MODE="0666"
```

When I execute those commands above, it does produce the error output in syslog.

jonasmalacofilho, 2019-09-29:
> When I execute those commands above, it does produce the error output in syslog.

Ok.  And and can't reproduce it on my systems.

What version of upower do you have?

acoutts, 2019-09-29:
```
$ upower --version
UPower client version 0.99.7
UPower daemon version 0.99.7
```

jonasmalacofilho, 2019-09-29:
Ok, I think I figured it out.

You're probably missing this patch:

> **linux: lower severity of "unhandled action" messages")**
> 
> Newer kernels emit bind/unbind uevents that are not of interest to powerd. To avoid littering logs with scary messages, let's lower their severity to "debug".
— [upower/upower@40e525edbde4](https://gitlab.freedesktop.org/upower/upower/commit/40e525edbde41f24f53a3a5255fbace71c79261a) _(see also: freedesktop bug [#106019](https://bugs.freedesktop.org/show_bug.cgi?id=106019), duplicate of [#106018](https://bugs.freedesktop.org/show_bug.cgi?id=106018)</sup> )_

Unfortunately it was only released on 0.99.8.  You could try opening a bug on Ubuntu for it to be backported to 18.04.  But the warning is harmless after all.

---

Specifically for liquidctl a workaround you can try is passing `--hid hidraw`: with that backend replacing the kernel driver is no longer necessary and thus should prevent the warnings.

closed 2019-09-29 by jonasmalacofilho

jonasmalacofilho, 2019-09-30:
I closed the issue but please let me know if passing `--hid hidraw` to liquidctl helps.

If it does, maybe @leinardi can add a switch to enable its use in Gkraken too.

acoutts, 2019-09-30:
Passing that flag fixes it - I'll ask @icedterminal if he can enable it in gkraken too.

icedterminal, 2019-09-30:
I don't use Gkraken. However, I did install it per the README in the repo. I use Ubuntu 19.04. All sources are up-to-date. I was unable to reproduce the issue you describe.

I'm going to agree with Jonas on this one. A complete solution would be to have a the patch back-ported to 18.xx.

acoutts, 2019-09-30:
Thanks! That explains it then.