liquidctl/liquidctl #51: Install liquidctl witch Mac OS

SkyWar-design, 2019-10-03:
Hello, i get installation error

1. pip3 install --upgrade pip
2. git clone URL
3. pip3 install liquidctl

```
Collecting liquidctl
  Using cached https://files.pythonhosted.org/packages/dd/fb/7dedead18957a8fa1d937055fb435c85a611a7eacaad11818caa716ef86d/liquidctl-1.2.0-py3-none-any.whl
Requirement already satisfied: docopt in /usr/local/lib/python3.7/site-packages (from liquidctl) (0.6.2)
Requirement already satisfied: pyusb in /usr/local/lib/python3.7/site-packages (from liquidctl) (1.0.2)
Collecting hidapi (from liquidctl)
  Using cached https://files.pythonhosted.org/packages/c1/86/89df0e8890f96eeb5fb68d4ccb14cb38e2c2d2cfd7601ba972206acd9015/hidapi-0.7.99.post21.tar.gz
Requirement already satisfied: appdirs in /usr/local/lib/python3.7/site-packages (from liquidctl) (1.4.3)
Requirement already satisfied: setuptools>=19.0 in /usr/local/lib/python3.7/site-packages (from hidapi->liquidctl) (41.0.1)
Building wheels for collected packages: hidapi
  Building wheel for hidapi (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/local/opt/python/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-install-i_l0b0g_/hidapi/setup.py'"'"'; __file__='"'"'/private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-install-i_l0b0g_/hidapi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-wheel-1w8lqxkn --python-tag cp37
       cwd: /private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-install-i_l0b0g_/hidapi/
  Complete output (23 lines):
  running bdist_wheel
  running build
  running build_ext
  cythoning hid.pyx to hid.c
  /private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-install-i_l0b0g_/hidapi/.eggs/Cython-0.29.13-py3.7-macosx-10.14-x86_64.egg/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-install-i_l0b0g_/hidapi/hid.pyx
    tree = Parsing.p_module(s, pxd, full_module_name)
  building 'hid' extension
  creating build
  creating build/temp.macosx-10.14-x86_64-3.7
  creating build/temp.macosx-10.14-x86_64-3.7/hidapi
  creating build/temp.macosx-10.14-x86_64-3.7/hidapi/mac
  Compiling with an SDK that doesn't seem to exist: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk
  Please check your Xcode installation
  clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Tk.framework/Versions/8.5/Headers -framework IOKit -framework CoreFoundation -Ihidapi/hidapi -I/usr/local/include -I/usr/local/opt/openssl@1.1/include -I/usr/local/opt/sqlite/include -I/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/include/python3.7m -c hid.c -o build/temp.macosx-10.14-x86_64-3.7/hid.o
  clang: warning: -framework IOKit: 'linker' input unused [-Wunused-command-line-argument]
  clang: warning: -framework CoreFoundation: 'linker' input unused [-Wunused-command-line-argument]
  clang: warning: no such sysroot directory: '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk' [-Wmissing-sysroot]
  In file included from hid.c:4:
  /usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/include/python3.7m/Python.h:25:10: fatal error: 'stdio.h' file not found
  #include <stdio.h>
           ^~~~~~~~~
  1 error generated.
  error: command 'clang' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for hidapi
  Running setup.py clean for hidapi
Failed to build hidapi
Installing collected packages: hidapi, liquidctl
  Running setup.py install for hidapi ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/local/opt/python/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-install-i_l0b0g_/hidapi/setup.py'"'"'; __file__='"'"'/private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-install-i_l0b0g_/hidapi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-record-39_hw9i9/install-record.txt --single-version-externally-managed --compile
         cwd: /private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-install-i_l0b0g_/hidapi/
    Complete output (21 lines):
    running install
    running build
    running build_ext
    skipping 'hid.c' Cython extension (up-to-date)
    building 'hid' extension
    creating build
    creating build/temp.macosx-10.14-x86_64-3.7
    creating build/temp.macosx-10.14-x86_64-3.7/hidapi
    creating build/temp.macosx-10.14-x86_64-3.7/hidapi/mac
    Compiling with an SDK that doesn't seem to exist: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk
    Please check your Xcode installation
    clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Tk.framework/Versions/8.5/Headers -framework IOKit -framework CoreFoundation -Ihidapi/hidapi -I/usr/local/include -I/usr/local/opt/openssl@1.1/include -I/usr/local/opt/sqlite/include -I/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/include/python3.7m -c hid.c -o build/temp.macosx-10.14-x86_64-3.7/hid.o
    clang: warning: -framework IOKit: 'linker' input unused [-Wunused-command-line-argument]
    clang: warning: -framework CoreFoundation: 'linker' input unused [-Wunused-command-line-argument]
    clang: warning: no such sysroot directory: '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk' [-Wmissing-sysroot]
    In file included from hid.c:4:
    /usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/include/python3.7m/Python.h:25:10: fatal error: 'stdio.h' file not found
    #include <stdio.h>
             ^~~~~~~~~
    1 error generated.
    error: command 'clang' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/local/opt/python/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-install-i_l0b0g_/hidapi/setup.py'"'"'; __file__='"'"'/private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-install-i_l0b0g_/hidapi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-record-39_hw9i9/install-record.txt --single-version-externally-managed --compile Check the logs for full command output.
```

CaseySJ, 2019-10-03:
Try installing **hidapi** and **libusb** ahead of time as follows (all within a *Terminal* window):

* Download the "homebrew" package installer:
  * `/usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"`
* Install libusb and hidapi:
  * `brew install libusb`
  * `brew install hidapi`
* If needed, PyUSB can also be installed:
  * `pip install pyusb`

And then finally:
* `pip install liquidctl==1.2.0`

SkyWar-design, 2019-10-03:
Hello, all libraries are installed, thanks for helping, but it didn’t help me, I still get an error

1. hidapi 0.9.0 is already installed
2. libusb 1.0.23 is already installed
3. pyusb already satisfied in /usr/local/lib/python3.7/site-packages (1.0.2)

```
Installing collected packages: hidapi, liquidctl
  Running setup.py install for hidapi ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/local/opt/python/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-install-6cvfzja7/hidapi/setup.py'"'"'; __file__='"'"'/private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-install-6cvfzja7/hidapi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-record-how0uwlj/install-record.txt --single-version-externally-managed --compile
         cwd: /private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-install-6cvfzja7/hidapi/
    Complete output (21 lines):
    running install
    running build
    running build_ext
    skipping 'hid.c' Cython extension (up-to-date)
    building 'hid' extension
    creating build
    creating build/temp.macosx-10.14-x86_64-3.7
    creating build/temp.macosx-10.14-x86_64-3.7/hidapi
    creating build/temp.macosx-10.14-x86_64-3.7/hidapi/mac
    Compiling with an SDK that doesn't seem to exist: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk
    Please check your Xcode installation
    clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Tk.framework/Versions/8.5/Headers -framework IOKit -framework CoreFoundation -Ihidapi/hidapi -I/usr/local/include -I/usr/local/opt/openssl@1.1/include -I/usr/local/opt/sqlite/include -I/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/include/python3.7m -c hid.c -o build/temp.macosx-10.14-x86_64-3.7/hid.o
    clang: warning: -framework IOKit: 'linker' input unused [-Wunused-command-line-argument]
    clang: warning: -framework CoreFoundation: 'linker' input unused [-Wunused-command-line-argument]
    clang: warning: no such sysroot directory: '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk' [-Wmissing-sysroot]
    In file included from hid.c:4:
    /usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/include/python3.7m/Python.h:25:10: fatal error: 'stdio.h' file not found
    #include <stdio.h>
             ^~~~~~~~~
    1 error generated.
    error: command 'clang' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/local/opt/python/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-install-6cvfzja7/hidapi/setup.py'"'"'; __file__='"'"'/private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-install-6cvfzja7/hidapi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/2r/k1g7ls797cn6nl18dl2xvjp40000gn/T/pip-record-how0uwlj/install-record.txt --single-version-externally-managed --compile Check the logs for full command output.

```

CaseySJ, 2019-10-03:
I had the same errors so I reinstalled *hidapi* and *libusb* with `brew` and tried again. I'm not sure whether I needed to add `sudo` in front of some of these commands, so it's good to try without `sudo` and with.

jonasmalacofilho, 2019-10-03:
The Pythin hidapi module (cython-hidapi) compiles some native/cython extensions during its setup.py.

From the error log it seems that you need to stall Xcode (and back in my day I remember something about having to install Xcode command line tools separately).

SkyWar-design, 2019-10-03:
> The Pythin hidapi module (cython-hidapi) compiles some native/cython extensions during its setup.py.
> 
> From the error log it seems that you need to install Xcode (and back in my day I remember something about having to install Xcode command line tools separately).

Thanks for the suggestion, xcode and its command line tools are installed so far nothing helps (

```
$ xcode-select --install
line tools are already installed, use "Software Update" to install update
```

SkyWar-design, 2019-10-03:
For fun, I tried installing version 1.1.0 and it worked, are there any key differences that will interfere with the work in this version?

jonasmalacofilho, 2019-10-03:
In liquidctl v1.1.0 hidapi & cython-hidapi were not yet required.  But you'll have a terrible time controlling any of HID devices (e.g. Kraken X62) on a mac without them.

Maybe your Xcode SDK is missing something or corrupted.  Clang, while installed, can't even find `<stdio.h>`>.  Or maybe you also need to (re)install a C specific SDK?

Have you tried checking for macOS specific instructions in the project page for [cython-hidapi](https://github.com/trevor/cython-hidapi)?

_(sorry, I don't have access to a mac right now)_

SkyWar-design, 2019-10-03:
I think this is due to the version of my sdk

Error:
```
    Compiling with an SDK that doesn't seem to exist: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk
    Please check your Xcode installation
```

I checked the version
```
xcrun --sdk macosx --show-sdk-version
10.15
```

jonasmalacofilho, 2019-10-03:
Can you install multiple SDKs side by side?  Because it seems that you're still running 10.14, and if that's the case then it makes sense that you need the SDK for it.

jonasmalacofilho, 2019-10-04:
I found a possible explanation and solution [on StackOverflow](https://stackoverflow.com/a/58192406/2801377):

>So, let's go to that first directory clang is complaining about and look:
>
>[...]
>
>Hah, we have a symlink for MacOSX10.15.sdk, but none for MacOSX10.14.sdk. Here's my first clang error again:
>```
>clang: warning: no such sysroot directory: > 
>'/Applications/Xcode.app/.../Developer/SDKs/MacOSX10.14.sdk' [-Wmissing-sysroot]
>```
>My guess is Apple jumped the gun on their xcode config and are already thinking they're on Catalina. Since it's a new Mac, the old config for 10.14 is not in place.
>
>THE FIX:
>
>Let's symlink 10.14 the same way as 10.15:
>```
>ln -s MacOSX.sdk/ MacOSX10.14.sdk
>```
>[...]

SkyWar-design, 2019-10-04:
> I found a possible explanation and solution [on StackOverflow](https://stackoverflow.com/a/58192406/2801377):

Thank you very much, it helped me

jonasmalacofilho, 2019-10-04:
I'm glad to hear!

---

@CaseySJ, @SkyWar-design 

Since you're both on macOS, could you contribute a a brew recipe for liquidctl?  Let me know if this is something either (or both) of you would be willing to help out with, and I can give some pointers.

And as to why I can't do it myself: well, first, I don't have a mac on hand right now to test any of this; additionally, homebrew [frowns on authors submitting their own work](https://github.com/Homebrew/brew/blob/master/docs/Acceptable-Formulae.md#niche-or-self-submitted-stuff).

CaseySJ, 2019-10-04:
Yes I can give it a try. Some pointers will certainly help as this would be my first such recipe.

jonasmalacofilho, 2019-10-05:
@CaseySJ 

I never made one, but I did some reading about it last night...

The relevant docs are in [brew/docs](https://github.com/Homebrew/brew/tree/master/docs).  From those, [Python for Formula Authors](https://github.com/Homebrew/brew/blob/master/docs/Python-for-Formula-Authors.md) should be specially useful.

You can fetch the latest liquidctl sdist directly from PyPI:

```
url "https://files.pythonhosted.org/packages/f2/08/7e56efece1ed5083e82826a1734f6e3c452401602ca211448c4b2c9e4cc9/liquidctl-1.2.0.tar.gz"
sha256 "ad8c03c0695620fedaec11e7a8286bb5d4da18ba0c71e55888bfa06f8f7d7529"
```

Next, I think the following list sums up all dependencies you will need to handle, including any indirect ones:

 * System: python, [libusb (brew)](https://formulae.brew.sh/formula/libusb), build tools & Xcode SDK¹ (to build cython-hidapi)
 * Python: [setuptools (pip)](https://pypi.org/project/setuptools/), [docopt (pip)](https://pypi.org/project/docopt/), [appdirs (pip)](https://pypi.org/project/appdirs/), [pyusb](https://pypi.org/project/pyusb/), [(cython-)hidapi (pip)](https://pypi.org/project/hidapi/), [cython¹ (brew; to build cython-hidapi)](https://formulae.brew.sh/formula/cython)

_¹ Possibly or even probably already implicitly required/available._

I would check existing formulas for packages with similar dependencies.  For example:

 - [trezor-agent](https://github.com/Homebrew/homebrew-core/blob/9360c630337641d34c3e208ae805f764fe215541/Formula/trezor-agent.rb): depends on cython-hidapi, and apparently that's as easy as depending on libusb and python and adding the resource block for cython-hidapi

liquidctl has no issues with `pip install`, so I don't think you'll have any surprises using `virtualenv_install_with_resources` (from the documentation I linked above).   The only detail is that liquidctl's setup² can honor DIST_NAME and DIST_PACKAGE environment variables to customize the output of `--version`, making it more informative.

DIST_NAME can be set to something simple like `Homebrew` (if you can add the MacOS version without breaking bottles, great), and DIST_PACKAGE can mimic the final package+version naming used by brew.  For example, this is the output [from the Arch Linux package](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=python-liquidctl):

```
$ liquidctl --version
liquidctl v1.2.0 (Arch Linux; python-liquidctl 1.2.0-1)
```

_² I actually need to check that DIST_NAME|PACKAGE works with pip, only been using it with setup.py; I'll do that later, and let me know if you spot any issues._

closed 2019-10-20 by jonasmalacofilho