liquidctl/liquidctl #56: Liquidctl x72 won't install on Windows 10

optimumbox, 2019-10-18:
The automated installer wouldn't do anything. When trying to follow this guide here: https://codecalamity.com/replacing-nzxts-cam-software-on-windows-for-kraken/. I get errors after attempting the install. So far I haven't found another user who has encountered this and threads pertaining this install method are archived leaving me unable to comment.

https://imgur.com/a/9hYmLaF

jonasmalacofilho, 2019-10-18:
The pre-built binary requires no installation.  It's a portable executable: just extract it and call it from a Windows Command Prompt or Power Shell.

For additional convenience, you can place it somewhere already in your PATH variable, or add it's folder to that variable.  This will allow you to call it from any location (e.g. no need to `cd` into the correct directory first).

---

The error when installing through pip comes from cython-hidapi.  It seems that's you're having to recompile it (probably using a recent Python release for which there's no pre-compiled cython-hidapi yet), but are missing Cython and/or a compiler.

If you manage to use the pre-built liquidctl.exe you don't have to worry about this, as it already bundles its own copy of that dependency.

optimumbox, 2019-10-18:
Thank you. I was able to get it working properly through cmd prompt.

closed 2019-10-18 by optimumbox