liquidctl/liquidctl #58: Input/Output error when reading from Smart Device V2

jonasmalacofilho, 2019-10-24:
```
(venv) C:\Users\me\venv\Scripts>liquidctl initialize

(venv) C:\Users\me\venv\Scripts>liquidctl set fan1 speed 0

(venv) C:\Users\me\venv\Scripts>liquidctl --debug status
[DEBUG] __main__: liquidctl v1.2.0 (AppVeyor; liquidctl-75df4346227f-bin-windows-x86_64)
Device 0, NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[ERROR] __main__: Unexpected error
Traceback (most recent call last):
  File "liquidctl\cli.py", line 185, in _device_get_status
  File "liquidctl\driver\nzxt_smart_device.py", line 406, in get_status
  File "liquidctl\driver\usb.py", line 334, in read
  File "site-packages\pyusb-1.0.2-py3.6.egg\usb\core.py", line 988, in read
  File "site-packages\pyusb-1.0.2-py3.6.egg\usb\backend\libusb1.py", line 851, in intr_read
  File "site-packages\pyusb-1.0.2-py3.6.egg\usb\backend\libusb1.py", line 936, in __read
  File "site-packages\pyusb-1.0.2-py3.6.egg\usb\backend\libusb1.py", line 595, in _check
usb.core.USBError: [Errno 5] Input/Output Error
[DEBUG] liquidctl.driver.usb: ensure interface is released
```

Reported by: @flyweight

jonasmalacofilho, 2019-10-24:
@CaseySJ

Can you see if you can reproduce the problem with the same commit (75df4346227f) and order of commands?

jonasmalacofilho, 2019-10-24:
@flyweight

Can you run the same commands again but passing `--hid hid`?

```
> liquidctl --hid hid --debug initialize
> liquidctl --hid hid --debug set fan1 speed 0
> liquidctl --hid hid --debug status
```

CaseySJ, 2019-10-24:
@jonasmalacofilho 

Certainly -- currently at the office so it will be a few hours.

flyweight, 2019-10-24:
@jonasmalacofilho 
```
(venv) C:\Users\me\venv\Scripts>liquidctl --hid hid --debug initialize
[DEBUG] __main__: liquidctl v1.2.0 (AppVeyor; liquidctl-75df4346227f-bin-windows-x86_64)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 60 02 01 e8 03 01 e8 03 (and 56 padding bytes)

(venv) C:\Users\me\venv\Scripts>liquidctl --hid hid --debug set fan1 speed 0
[DEBUG] __main__: liquidctl v1.2.0 (AppVeyor; liquidctl-75df4346227f-bin-windows-x86_64)
[INFO] liquidctl.driver.nzxt_smart_device: setting fan1 duty to 0%
[DEBUG] liquidctl.driver.nzxt_smart_device: write 62 01 01 00 00 00 (and 58 padding bytes)

(venv) C:\Users\me\venv\Scripts>liquidctl --hid hid --debug status
[DEBUG] __main__: liquidctl v1.2.0 (AppVeyor; liquidctl-75df4346227f-bin-windows-x86_64)
Device 0, NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 20 00 24 00 15 57 41 47 35 31 33 20 06 20 01 01 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 1a 07 00 00 b0 05 00 00 00 00 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 46 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 fa 2e f7 2e 9d 2e 00 00 00 00 00 00 00 00 00 00 84 00 00 00 c8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 1a 07 00 00 b3 05 00 00 00 00 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 46 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 ea 2e ea 2e 7f 2e 00 00 00 00 00 00 00 00 00 00 85 00 00 00 ac 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 1a 07 00 00 b3 05 00 00 00 00 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 46 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 f0 2e f0 2e 4c 2e 00 00 00 00 00 00 00 00 00 00 92 00 00 00 c5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 1a 07 00 00 b3 05 00 00 00 00 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 46 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 f7 2e ed 2e 9d 2e 00 00 00 00 00 00 00 00 00 00 8c 00 00 00 a1 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 20 00 24 00 15 57 41 47 35 31 33 20 02 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Fan 1 duty                         0  %
Fan 1 speed                     1818  rpm
Fan 3 duty                         0  %
Fan 3 speed                     1456  rpm
Firmware version               1.5.0
LED 1 accessory 1    HUE 2 LED Strip
Noise level                       70  dB
```

jonasmalacofilho, 2019-10-24:
@flyweight thank you, that's very helpful.

@CaseySJ this means that the problem is specific to the PyUSB/libusb backend.  I think we should start by checking the endpoint numbers and (again) the read/write lengths.

jonasmalacofilho, 2019-10-24:
@flyweight can you give d6b3ed1a054b185c98d3d0d0aa328a1a9424fc39 ([build #309](https://ci.appveyor.com/project/jonasmalacofilho/liquidctl/builds/28361509)) a go, this time without passing `--hid hid`?

CaseySJ, 2019-10-24:
> @flyweight thank you, that's very helpful.
> 
> @CaseySJ this means that the problem is specific to the PyUSB/libusb backend. I think we should start by checking the endpoint numbers and (again) the read/write lengths.

Yes, `_READ_LENGTH = 60` probably needs to change to 64.

Also, **Fan duty** should be 10% (not 0%) so I'll take a look at that as well.

jonasmalacofilho, 2019-10-24:
Well, the previous command tried to set it to 0%.  So that's weird as well.

@flyweight are you running DC only (i.e. 3-pin only) fans?

CaseySJ, 2019-10-24:
Maybe the fans were in the process of spinning down, so RPM was still high. That might make sense if the `set fan1 speed 0` command was immediately followed by `status`. 

It would be good to set fan speed to 0, visually verify that fan has stopped, and then issue `liquidctl status`.

jonasmalacofilho, 2019-10-24:
> My device, of course, is the Smart Device V2 instead of HUE 2 controller

His is a Smart Device V2 as well.  Otherwise no fan data [should have been reported](https://github.com/jonasmalacofilho/liquidctl/blob/75df4346227f397c016960e1504ffbb3f022d83b/liquidctl/driver/nzxt_smart_device.py#L432).

flyweight, 2019-10-24:
@jonasmalacofilho @CaseySJ 

Fans never spin down after time.  Fans are 3-pin.  The controller should be able to regulate with voltage, right?  Otherwise, why would the manufacturer have included them with the case.

If I set to 10%, the fan 1 duty is reported correctly but the RPM remains the same after some time.

```
(venv) C:\Users\me\venv\Scripts>liquidctl --debug initialize
[DEBUG] __main__: liquidctl v1.2.0 (AppVeyor; liquidctl-d6b3ed1a054b-bin-windows-x86_64)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 60 02 01 e8 03 01 e8 03 (and 56 padding bytes)
[DEBUG] liquidctl.driver.usb: ensure interface is released
[DEBUG] liquidctl.driver.usb: ensure interface is released

(venv) C:\Users\me\venv\Scripts>liquidctl --debug set fan1 speed 0
[DEBUG] __main__: liquidctl v1.2.0 (AppVeyor; liquidctl-d6b3ed1a054b-bin-windows-x86_64)
[INFO] liquidctl.driver.nzxt_smart_device: setting fan1 duty to 0%
[DEBUG] liquidctl.driver.nzxt_smart_device: write 62 01 01 00 00 00 (and 58 padding bytes)
[DEBUG] liquidctl.driver.usb: ensure interface is released
[DEBUG] liquidctl.driver.usb: ensure interface is released

(venv) C:\Users\me\venv\Scripts>liquidctl --debug status
[DEBUG] __main__: liquidctl v1.2.0 (AppVeyor; liquidctl-d6b3ed1a054b-bin-windows-x86_64)
Device 0, NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 20 00 24 00 15 57 41 47 35 31 33 20 06 20 01 01 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 1a 07 00 00 c5 05 00 00 00 00 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 48 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 ed 2e f0 2e a7 2e 00 00 00 00 00 00 00 00 00 00 8b 00 00 00 a5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 1a 07 00 00 c5 05 00 00 00 00 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 48 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 ea 2e ed 2e 96 2e 00 00 00 00 00 00 00 00 00 00 8c 00 00 00 9c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 1f 07 00 00 c5 05 00 00 00 00 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 46 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 f0 2e ed 2e 85 2e 00 00 00 00 00 00 00 00 00 00 8f 00 00 00 9b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 1a 07 00 00 c5 05 00 00 00 00 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 46 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 20 00 24 00 15 57 41 47 35 31 33 20 03 00 00 00 00 00 00 00 00 00 ea 2e ed 2e 8c 2e 00 00 00 00 00 00 00 00 00 00 82 00 00 00 c2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 20 00 24 00 15 57 41 47 35 31 33 20 02 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.usb: ensure interface is released
Fan 1 duty                         0  %
Fan 1 speed                     1818  rpm
Fan 3 duty                         0  %
Fan 3 speed                     1477  rpm
Firmware version               1.5.0
LED 1 accessory 1    HUE 2 LED Strip
Noise level                       72  dB
[DEBUG] liquidctl.driver.usb: ensure interface is released
```

[image]

CaseySJ, 2019-10-24:
> > My device, of course, is the Smart Device V2 instead of HUE 2 controller
> 
> His is a Smart Device V2 as well. Otherwise no fan data [should have been reported](https://github.com/jonasmalacofilho/liquidctl/blob/75df4346227f397c016960e1504ffbb3f022d83b/liquidctl/driver/nzxt_smart_device.py#L432).

Oh yeah! Wires crossed with Mr. David Frick...

CaseySJ, 2019-10-24:
@flyweight @jonasmalacofilho 

The current Smart Device V2 driver was developed with only PWM fans attached. However, I have some generic 3-pin fans that I can connect to this device.

jonasmalacofilho, 2019-10-24:
Merged d6b3ed1 already since it [fixes the main issue](https://github.com/jonasmalacofilho/liquidctl/issues/58#issuecomment-546097054).

Leaving the issue open so @CaseySJ can take a look at the DC control part.

CaseySJ, 2019-10-25:
I'll post some observations here first. Analysis and solutions in subsequent post. 

**These tests are based on code in my fork of liquidctl.**  I am now downloading latest code from Jonas' repository. Results to be posted in next reply.

3-Pin DC fan must connect to **top 3 pins** of the 4-pin header on Smart Device V2.

**Test 1:**
- 3-Pin DC NZXT fan on `fan1` with `set fan1 speed=60`
- 4-Pin PWM Aer RGB 2 fan on `fan2` with `set fan2 speed=50`

**Result: Good**
```
Caseys-iMac-Pro:~ casey_sj$ liquidctl --debug -d 1 status
[DEBUG] liquidctl.cli: liquidctl v1.2.0rc4 (7569a02e3e43)
Device 1, NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 28 00 2e 00 16 57 41 47 35 31 33 20 06 20 01 01 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 ef 04 03 04 00 00 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 40 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 7d 2f 73 2f c3 2f 00 00 00 00 00 00 00 00 00 00 c2 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 ef 04 01 04 00 00 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 40 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 7d 2f 65 2f c0 2f 00 00 00 00 00 00 00 00 00 00 cb 00 24 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 ec 04 ff 03 00 00 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 40 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 7d 2f 73 2f b2 2f 00 00 00 00 00 00 00 00 00 00 c5 00 44 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 28 00 2e 00 16 57 41 47 35 31 33 20 02 04 0a 0a 00 00 00 0c 0c 0c 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Fan 1 duty                              60  %
Fan 1 speed                           1263  rpm
Fan 2 duty                              50  %
Fan 2 speed                           1027  rpm
Firmware version                     1.5.0  
LED 1 accessory 1          HUE 2 LED Strip  
LED 1 accessory 2    HUE 2 Underglow 200mm  
LED 1 accessory 3    HUE 2 Underglow 200mm  
LED 2 accessory 1          AER RGB 2 140mm  
LED 2 accessory 2          AER RGB 2 140mm  
LED 2 accessory 3          AER RGB 2 140mm  
LED 2 accessory 4          AER RGB 2 120mm  
Noise level                             64  dB
```

**Test 2:**
- 3-Pin DC NZXT fan on `fan3` with `set fan3 speed=50`
- 4-Pin PWM Aer RGB 2 fan on `fan2` with `set fan2 speed=50`

**Result: Good**
```
Caseys-iMac-Pro:~ casey_sj$ liquidctl --debug -d 1 status
[DEBUG] liquidctl.cli: liquidctl v1.2.0rc4 (7569a02e3e43)
Device 1, NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 28 00 2e 00 16 57 41 47 35 31 33 20 06 20 01 01 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 03 04 d2 04 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 40 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 c6 2f 76 2f 73 2f 00 00 00 00 00 00 00 00 00 00 00 00 41 00 d5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 01 04 d7 04 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 40 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 cd 2f 62 2f 73 2f 00 00 00 00 00 00 00 00 00 00 00 00 22 00 c2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 ff 03 d5 04 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 3e 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 cd 2f 73 2f 55 2f 00 00 00 00 00 00 00 00 00 00 00 00 37 00 cc 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
Fan 3 duty                              50  %
Fan 3 speed                           1234  rpm
Firmware version                     1.5.0  
LED 1 accessory 1          HUE 2 LED Strip  
LED 1 accessory 2    HUE 2 Underglow 200mm  
LED 1 accessory 3    HUE 2 Underglow 200mm  
LED 2 accessory 1          AER RGB 2 140mm  
LED 2 accessory 2          AER RGB 2 140mm  
LED 2 accessory 3          AER RGB 2 140mm  
LED 2 accessory 4          AER RGB 2 120mm  
Noise level                             64  dB
```

CaseySJ, 2019-10-25:
**These results are from latest code from Jonas' repository.** Looks like they are identical to those in previous post. All tests done on **macOS Catalina**.

**Test 1:**

- 3-Pin DC NZXT fan on `fan1` with `set fan1 speed=60` (does not have any RGB)
- 4-Pin PWM Aer RGB 2 fan on `fan2` with `set fan2 speed=50`

**Result: Good**
```
Caseys-iMac-Pro:~ casey_sj$ liquidctl --debug -d 1 status
[DEBUG] liquidctl.cli: liquidctl v1.2.0
Device 1, NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 28 00 2e 00 16 57 41 47 35 31 33 20 06 20 01 01 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 cd 04 03 04 00 00 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 40 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 9e 2f 7d 2f c6 2f 00 00 00 00 00 00 00 00 00 00 c8 00 5b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 c8 04 05 04 00 00 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 40 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 80 2f 6c 2f c0 2f 00 00 00 00 00 00 00 00 00 00 be 00 25 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 ca 04 03 04 00 00 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 40 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 80 2f 76 2f bc 2f 00 00 00 00 00 00 00 00 00 00 b3 00 69 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 28 00 2e 00 16 57 41 47 35 31 33 20 02 04 0a 0a 00 00 00 0c 0c 0c 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Fan 1 duty                              60  %
Fan 1 speed                           1229  rpm
Fan 2 duty                              50  %
Fan 2 speed                           1027  rpm
Firmware version                     1.5.0  
LED 1 accessory 1          HUE 2 LED Strip  
LED 1 accessory 2    HUE 2 Underglow 200mm  
LED 1 accessory 3    HUE 2 Underglow 200mm  
LED 2 accessory 1          AER RGB 2 140mm  
LED 2 accessory 2          AER RGB 2 140mm  
LED 2 accessory 3          AER RGB 2 140mm  
LED 2 accessory 4          AER RGB 2 120mm  
Noise level                             64  dB
```

**Test 2:**

- 3-Pin DC NZXT fan on `fan3` with `set fan1 speed=50`
- 4-Pin PWM Aer RGB 2 fan on `fan2` with `set fan2 speed=50`

**Result: Good**
```
Caseys-iMac-Pro:~ casey_sj$ liquidctl --debug -d 1 status
[DEBUG] liquidctl.cli: liquidctl v1.2.0
Device 1, NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 28 00 2e 00 16 57 41 47 35 31 33 20 06 20 01 01 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 03 04 c3 04 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 42 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 d7 2f 6f 2f 6c 2f 00 00 00 00 00 00 00 00 00 00 00 00 22 00 a2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 01 04 c5 04 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 42 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 da 2f 76 2f 8d 2f 00 00 00 00 00 00 00 00 00 00 00 00 61 00 99 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 03 04 c5 04 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 42 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 c0 2f 79 2f 8a 2f 00 00 00 00 00 00 00 00 00 00 00 00 12 00 a1 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
Fan 3 duty                              50  %
Fan 3 speed                           1219  rpm
Firmware version                     1.5.0  
LED 1 accessory 1          HUE 2 LED Strip  
LED 1 accessory 2    HUE 2 Underglow 200mm  
LED 1 accessory 3    HUE 2 Underglow 200mm  
LED 2 accessory 1          AER RGB 2 140mm  
LED 2 accessory 2          AER RGB 2 140mm  
LED 2 accessory 3          AER RGB 2 140mm  
LED 2 accessory 4          AER RGB 2 120mm  
Noise level                             66  dB
```

CaseySJ, 2019-10-25:
Just found another 3-pin DC fan (generic brand). Fan connections for this test are:

- Fan 1: Generic 3-pin @ 60%
- Fan 2: NZXT 4-pin Aer RGB 2 @ 50%
- Fan 3: NZXT 3-pin @ 50%

**Result:** Good.

```
Caseys-iMac-Pro:~ casey_sj$ liquidctl --debug -d 1 status
[DEBUG] liquidctl.cli: liquidctl v1.2.0
Device 1, NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 28 00 2e 00 16 57 41 47 35 31 33 20 06 20 01 01 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 b9 04 27 04 ab 04 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 42 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 76 2f 6c 2f 58 2f 00 00 00 00 00 00 00 00 00 00 94 00 28 00 b9 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 bc 04 27 04 ad 04 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 42 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 7d 2f 76 2f 58 2f 00 00 00 00 00 00 00 00 00 00 89 00 01 00 c5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 be 04 29 04 a8 04 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 42 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 8a 2f 55 2f 5b 2f 00 00 00 00 00 00 00 00 00 00 86 00 34 00 9e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 28 00 2e 00 16 57 41 47 35 31 33 20 02 04 0a 0a 00 00 00 0c 0c 0c 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Fan 1 duty                              60  %
Fan 1 speed                           1209  rpm
Fan 2 duty                              50  %
Fan 2 speed                           1063  rpm
Fan 3 duty                              50  %
Fan 3 speed                           1195  rpm
Firmware version                     1.5.0  
LED 1 accessory 1          HUE 2 LED Strip  
LED 1 accessory 2    HUE 2 Underglow 200mm  
LED 1 accessory 3    HUE 2 Underglow 200mm  
LED 2 accessory 1          AER RGB 2 140mm  
LED 2 accessory 2          AER RGB 2 140mm  
LED 2 accessory 3          AER RGB 2 140mm  
LED 2 accessory 4          AER RGB 2 120mm  
Noise level                             66  dB
```

CaseySJ, 2019-10-25:
Speculation on cause of problem:

- Perhaps the problem reported in Windows is in fact due to `_READ_LENGTH=60` instead of `64`.

jonasmalacofilho, 2019-10-25:
@CaseySJ

In the tests from the first comment it appears that setting the DC fan to 50% or 60% made no difference.

The description for the second test from the second comment appears to have a mistake, but assuming it's the same second test as before, we can observe the same behavior: the speed of the DC fan does not change when it's changed to 50%.

Can you also check the extremes (i.e. 0%, 100%) and compare the maximum reported rpm with the specs for each of the DC fans?

CaseySJ, 2019-10-25:
@jonasmalacofilho 

The output is a bit confusing. We should ignore `Fan 2` and just look at `Fan 1` and `Fan 3` as those are the 3-pin fans. `Fan 2` is just the Aer RGB 2 PWM fan, so it is always set to 50%.

jonasmalacofilho, 2019-10-25:
I noticed.  Still, the follow

```
# comment 2, test 1
Fan 1 duty                              60  %
Fan 1 speed                           1229  rpm
# comment 2, test 2
Fan 3 duty                              50  %
Fan 3 speed                           1219  rpm
```

This change looks could be normal voltage and/or speed variance.  DC controlled fans should vary their speed linearly with the supplied voltage until stalling so I expect a much larger change here.

Try to see how those fans behave under more extreme changes (e.g. 0%, 50%, 75%, 100%).  Also, what's their maximum specified speed (i.e. at 12 V)?

CaseySJ, 2019-10-25:
And here's the same NZXT 3-pin fan on `Fan 2` in Windows, using Jonas' AppVeyor build:

```
C:\Users\casey\Downloads\liquidctl-d6b3ed1a054b-bin-windows-x86_64>liquidctl.exe --debug -d 1 status
[DEBUG] __main__: liquidctl v1.2.0 (AppVeyor; liquidctl-d6b3ed1a054b-bin-windows-x86_64)
Device 1, NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 28 00 2e 00 16 57 41 47 35 31 33 20 06 20 01 01 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 25 04 bc 04 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 42 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 c0 2f 65 2f 58 2f 00 00 00 00 00 00 00 00 00 00 00 00 37 00 d2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 24 04 b7 04 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 42 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 c0 2f 4b 2f 44 2f 00 00 00 00 00 00 00 00 00 00 00 00 57 00 d6 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 25 04 bc 04 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 42 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 c3 2f 5b 2f 62 2f 00 00 00 00 00 00 00 00 00 00 00 00 38 00 c3 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 22 04 b4 04 00 00 00 00 00 00 00 00 00 00 3c 32 32 00 00 00 00 00 3c 32 32 00 00 00 00 00 40 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 d0 2f 8a 2f 6c 2f 00 00 00 00 00 00 00 00 00 00 00 00 54 00 c8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 28 00 2e 00 16 57 41 47 35 31 33 20 02 04 0a 0a 00 00 00 0c 0c 0c 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.usb: ensure interface is released
Fan 2 duty                              50  %
Fan 2 speed                           1061  rpm
Fan 3 duty                              50  %
Fan 3 speed                           1212  rpm
Firmware version                     1.5.0
LED 1 accessory 1          HUE 2 LED Strip
LED 1 accessory 2    HUE 2 Underglow 200mm
LED 1 accessory 3    HUE 2 Underglow 200mm
LED 2 accessory 1          AER RGB 2 140mm
LED 2 accessory 2          AER RGB 2 140mm
LED 2 accessory 3          AER RGB 2 140mm
LED 2 accessory 4          AER RGB 2 120mm
Noise level                             66  dB
[DEBUG] liquidctl.driver.usb: ensure interface is released
```

And now we increase speed to 60% on `Fan 2` only:

```
C:\Users\casey\Downloads\liquidctl-d6b3ed1a054b-bin-windows-x86_64>liquidctl.exe --debug -d 1 set fan2 speed 60
[DEBUG] __main__: liquidctl v1.2.0 (AppVeyor; liquidctl-d6b3ed1a054b-bin-windows-x86_64)
[INFO] liquidctl.driver.nzxt_smart_device: setting fan2 duty to 60%
[DEBUG] liquidctl.driver.nzxt_smart_device: write 62 01 02 00 3c 00 (and 58 padding bytes)
[DEBUG] liquidctl.driver.usb: ensure interface is released
[DEBUG] liquidctl.driver.usb: ensure interface is released
```

I can hear that the fan is louder. Now let's see the `status`:

```
C:\Users\casey\Downloads\liquidctl-d6b3ed1a054b-bin-windows-x86_64>liquidctl.exe --debug -d 1 status
[DEBUG] __main__: liquidctl v1.2.0 (AppVeyor; liquidctl-d6b3ed1a054b-bin-windows-x86_64)
Device 1, NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 28 00 2e 00 16 57 41 47 35 31 33 20 06 20 01 01 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 9f 04 cd 04 00 00 00 00 00 00 00 00 00 00 3c 3c 32 00 00 00 00 00 3c 3c 32 00 00 00 00 00 44 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 da 2f 58 2f 7d 2f 00 00 00 00 00 00 00 00 00 00 00 00 3e 00 c1 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
LED 1 accessory 1          HUE 2 LED Strip
LED 1 accessory 2    HUE 2 Underglow 200mm
LED 1 accessory 3    HUE 2 Underglow 200mm
LED 2 accessory 1          AER RGB 2 140mm
LED 2 accessory 2          AER RGB 2 140mm
LED 2 accessory 3          AER RGB 2 140mm
LED 2 accessory 4          AER RGB 2 120mm
Noise level                             68  dB
[DEBUG] liquidctl.driver.usb: ensure interface is released
```

**Result:** RPM increased from 1061 to 1183 going from 50% to 60%. Functionality looks okay.

CaseySJ, 2019-10-25:
After setting `Fan 2` speed to 100% in Windows, RPM increased to 1587 and is much louder...

```
C:\Users\casey\Downloads\liquidctl-d6b3ed1a054b-bin-windows-x86_64>liquidctl.exe --debug -d 1 set fan2 speed 100
[DEBUG] __main__: liquidctl v1.2.0 (AppVeyor; liquidctl-d6b3ed1a054b-bin-windows-x86_64)
[INFO] liquidctl.driver.nzxt_smart_device: setting fan2 duty to 100%
[DEBUG] liquidctl.driver.nzxt_smart_device: write 62 01 02 00 64 00 (and 58 padding bytes)
[DEBUG] liquidctl.driver.usb: ensure interface is released
[DEBUG] liquidctl.driver.usb: ensure interface is released

C:\Users\casey\Downloads\liquidctl-d6b3ed1a054b-bin-windows-x86_64>liquidctl.exe --debug -d 1 status
[DEBUG] __main__: liquidctl v1.2.0 (AppVeyor; liquidctl-d6b3ed1a054b-bin-windows-x86_64)
Device 1, NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 28 00 2e 00 16 57 41 47 35 31 33 20 06 20 01 01 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 33 06 cd 04 00 00 00 00 00 00 00 00 00 00 3c 64 32 00 00 00 00 00 3c 64 32 00 00 00 00 00 5c 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 bc 2f fa 2e 65 2f 00 00 00 00 00 00 00 00 00 00 00 00 26 01 99 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 2f 06 cd 04 00 00 00 00 00 00 00 00 00 00 3c 64 32 00 00 00 00 00 3c 64 32 00 00 00 00 00 60 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 c6 2f 08 2f 73 2f 00 00 00 00 00 00 00 00 00 00 00 00 30 01 99 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 28 00 2e 00 16 57 41 47 35 31 33 20 02 04 0a 0a 00 00 00 0c 0c 0c 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.usb: ensure interface is released
Fan 2 duty                             100  %
Fan 2 speed                           1587  rpm
Fan 3 duty                              50  %
Fan 3 speed                           1229  rpm
Firmware version                     1.5.0
LED 1 accessory 1          HUE 2 LED Strip
LED 1 accessory 2    HUE 2 Underglow 200mm
LED 1 accessory 3    HUE 2 Underglow 200mm
LED 2 accessory 1          AER RGB 2 140mm
LED 2 accessory 2          AER RGB 2 140mm
LED 2 accessory 3          AER RGB 2 140mm
LED 2 accessory 4          AER RGB 2 120mm
Noise level                             92  dB
[DEBUG] liquidctl.driver.usb: ensure interface is released
```

CaseySJ, 2019-10-25:
And after setting `Fan 2` speed to 0%, the fan disappears from the list, but **it's still spinning** (but is very quiet)... Notice `Noise Level 48 dB`

```
C:\Users\casey\Downloads\liquidctl-d6b3ed1a054b-bin-windows-x86_64>liquidctl.exe --debug -d 1 set fan2 speed 0
[DEBUG] __main__: liquidctl v1.2.0 (AppVeyor; liquidctl-d6b3ed1a054b-bin-windows-x86_64)
[INFO] liquidctl.driver.nzxt_smart_device: setting fan2 duty to 0%
[DEBUG] liquidctl.driver.nzxt_smart_device: write 62 01 02 00 00 00 (and 58 padding bytes)
[DEBUG] liquidctl.driver.usb: ensure interface is released
[DEBUG] liquidctl.driver.usb: ensure interface is released

C:\Users\casey\Downloads\liquidctl-d6b3ed1a054b-bin-windows-x86_64>liquidctl.exe --debug -d 1 status
[DEBUG] __main__: liquidctl v1.2.0 (AppVeyor; liquidctl-d6b3ed1a054b-bin-windows-x86_64)
Device 1, NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 28 00 2e 00 16 57 41 47 35 31 33 20 06 20 01 01 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 00 00 c8 04 00 00 00 00 00 00 00 00 00 00 3c 00 32 00 00 00 00 00 3c 00 32 00 00 00 00 00 30 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 e8 2f 4c 00 83 2f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 b6 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 00 00 d0 04 00 00 00 00 00 00 00 00 00 00 3c 00 32 00 00 00 00 00 3c 00 32 00 00 00 00 00 30 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 de 2f 2b 00 7d 2f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c6 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 00 00 cd 04 00 00 00 00 00 00 00 00 00 00 3c 00 32 00 00 00 00 00 3c 00 32 00 00 00 00 00 30 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 e4 2f 00 00 a5 2f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c3 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 00 00 cd 04 00 00 00 00 00 00 00 00 00 00 3c 00 32 00 00 00 00 00 3c 00 32 00 00 00 00 00 30 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 de 2f 00 00 65 2f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c1 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 28 00 2e 00 16 57 41 47 35 31 33 20 02 04 0a 0a 00 00 00 0c 0c 0c 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.usb: ensure interface is released
Fan 3 duty                              50  %
Fan 3 speed                           1224  rpm
Firmware version                     1.5.0
LED 1 accessory 1          HUE 2 LED Strip
LED 1 accessory 2    HUE 2 Underglow 200mm
LED 1 accessory 3    HUE 2 Underglow 200mm
LED 2 accessory 1          AER RGB 2 140mm
LED 2 accessory 2          AER RGB 2 140mm
LED 2 accessory 3          AER RGB 2 140mm
LED 2 accessory 4          AER RGB 2 120mm
Noise level                             48  dB
[DEBUG] liquidctl.driver.usb: ensure interface is released
```

CaseySJ, 2019-10-25:
I should note that my tests in Windows were **not** done in a virtual environment `venv`.

jonasmalacofilho, 2019-10-25:
@CaseySJ 

Can you double check that Fan 2 is the DC one?  Fan 3 continues to report the same (approximate) speed as before, when it was the DC one.

Regardless, I really think we have a problem _somewhere._  flyweight's last comment still clearly showed the fan control vs. speed problem.  Maybe we're missing something in the initialization routine, which could in turn make the behavior hard to reproduce...

---

As soon as I can I'm going to dig through all the debug information with a fine comb.  It would also help if _either of you_ could provide some captures on CAM starting and then controlling DC fans.

@flyweight

To capture USB traffic the best software tool is [Wireshark](https://www.wireshark.org/) (see [Wireshark on Windows](https://wiki.wireshark.org/CaptureSetup/USB#Windows)), as it records a lot more information about each packet.  But if you can't get Wireshark to work, [Free USB Analyzer from HHD](https://freeusbanalyzer.com/) should be enough for dealing with this specific issue.

jonasmalacofilho, 2019-10-25:
When running the pre-build binaries from AppVeyor the Python version or the use of virtual environments should have no impact: those binaries already bundle liquidctl, Python, all Python dependencies, HIDAPI and libusb (only kernel drivers for non-HID devices are missing).

CaseySJ, 2019-10-25:
After a good night's sleep I think I found the problem. The device initialization sequence needs a second command `\0x60\0x03\`. This seems to properly enable fan speed control. It also has the effect of setting all fans to 50% duty (default). Now we can control fan speeds properly on all channels for both DC and PWM fans. 

I've made the change to a new patch branch, "Fix-Device-Initialization". Will issue pull request momentarily.

jonasmalacofilho, 2019-10-25:
@flyweight 

Can you give eef945dc8775a8ef7fb807ee391ef564bf5f3477 ([build #311](https://ci.appveyor.com/project/jonasmalacofilho/liquidctl/builds/28378810)) a try and report back?

---

@CaseySJ 

Because the PR had conflicts and couldn't be built, I cherry-picked your commit so AppVeyor runs and @flyweight can already test the fix.

If you want, I can help with some good git practices (for example, I see that our forks have diverged, even though the new driver was merged into my branch... while this is fine, it's a lot more work for you) and with fixing that pull request.  Otherwise, just let me know and (assuming flyweight confirms the patch fixes the issue for him) I'll simply merge the cherry-picked commit.

CaseySJ, 2019-10-25:
@jonasmalacofilho,

It would be good to know how to rebase a forked repository from the original. GitHub is not exactly intuitive.

CaseySJ, 2019-10-25:
Also, let’s ignore the sample results posted last night. I retested all of those conditions this morning before leaving for work (I wake up between 2 and 3am every day so I have a few hours of free time each morning) and the fan behavior was looking good. I’m not sure we can trust rpm for DC fans, but speed control is working correctly.

jonasmalacofilho, 2019-10-25:
@CaseySJ 

> It would be good to know how to rebase a forked repository from the original. GitHub is not exactly intuitive. It’s rather infuriating for beginners.

I don't think that's possible at all.  GitHub is a good code sharing site, the issue tracking is flexibile, and the pull requests work for actually requesting code to be pulled in and for simple code reviews... but GitHub is not where you should be developing your code.

Instead, you should have your code in a local copy of the repository, and manage it all using `git`.  There are also GUIs for `git`, but I personally haven't found one that I really like; and most people I know only open a GUI for solving complex merge conflicts.

Some stuff in Git is complicated (like non-trivial rebases), but most of it is really simple once you get the fundamentals: for example, each commit is a _snapshot_ of the entire project and of its history (think snapshots and history, not files and deltas).

CaseySJ, 2019-10-25:
@flyweight 

Glad to hear it! 

Incidentally, the **Input/Output error** that you initially reported (which is the title of this "Issue"), will occur if CAM is active -- not just the CAM GUI, but also the CAM background processes in Windows.

jonasmalacofilho, 2019-10-25:
Fixed with d6b3ed1 and eef945d.

closed 2019-10-25 by jonasmalacofilho

sgtcoder, 2021-03-04:
This is still not being detected. When will this be fixed?

jonasmalacofilho, 2021-03-04:
@sgtcoder

Please open a new issue, following our bug report template.  Specifically, we need to understand *exactly* what you expected liquidctl to do, and what it is doing.

It will help if you provide as much information as you can, like `--debug` information from liquidctl, the output from `lsusb -v -d<vid>:<pid>` (since you apparently have a device detection issue), and information about you operating system and kernel.

---

Additionally,

> This is still not being detected.

this issue was not about detection issues,

> When will this be fixed?

and the problem was fixed... in 2019.

CaseySJ, 2021-03-04:
@sgtcoder 

Are you referring to this issue I logged just a moment ago? I discovered this a few days ago but forgot to log it until your message reminded me...

https://github.com/liquidctl/liquidctl/issues/303#issue-822595027

CaseySJ, 2021-03-04:
Okay, after fixing a wiring problem on my Hackintosh, the Smart Device V2 is operational again. With `liquidctl 1.5.1` there is no problem accessing and configuring the device.

```bash
% liquidctl list
Device ID 0: NZXT Smart Device V2
Device ID 1: NZXT Kraken X (X42, X52, X62 or X72)

% liquidctl -d 0 status
NZXT Smart Device V2
├── Fan 2 duty       50  %
├── Fan 2 speed    1038  rpm
└── Noise level      80  dB

% liquidctl -d 0 set led1 color spectrum-wave
```

sgtcoder, 2021-03-05:
@jonasmalacofilho

You are right, sorry about my frustration. I went ahead and opened up a proper ticket.

Here is the link: https://github.com/liquidctl/liquidctl/issues/304

Thank you!

jonasmalacofilho, 2021-03-05:
@sgtcoder no problem, and thanks for opening the issue.