liquidctl/liquidctl #61: Support for NXZT AER 2 120mm and NZXT HUE 2.

David-Frick, 2019-10-30:
We talked in the other issue about adding support for these devices. I am super willing to run commands to get this on the move. I did see this project on the side: https://github.com/kusti8/hue-plus which covers the hue + and has a GUI. I do not know if y'all have that on a roadmap. I would certainly be willing to help code that. 

EDIT: Will be adding all the commands I am running and what they return next after I review where the other issue left off.

CaseySJ, 2019-10-31:
@David-Frick 

As a Linux user, have you installed the latest commit of liquidctl? Or do you only have the last official release, `liquidctl 1.2.0`?

If you have the latest commit, please try modifying `SUPPORTED_DEVICES` and running the three `liquidctl` commands.

jonasmalacofilho, 2019-10-31:
@David-Frick 

I'm glad to have you back.  Let's get started!

The very first thing you can try is modify the `SmartDeviceDriverV2` to bind to the HUE 2, and see what works.

You can follow Casey's instructions for that, but I would start checking the following (order of) commands:

```
# liquidctl list -v
# liquidctl status            # (before initialize) could fail or produce weird results
# liquidctl initialize all    # initialize the HUE 2 (it's a no-op for the X62)
# liquidctl status            # (after initialize)
# liquidctl --device <id> set led1 color fixed ff0000
# liquidctl --device <id> set led2 color fixed 00ff00
# liquidctl --device <id> set led3 color fixed 0000ff
# liquidctl --device <id> set led4 color fixed ff00ff
```

A few notes:

 - always start on top of the latest HEAD of the master branch
 - make sure to take note of the base commit and mention it with any results (as there might have been updates to 'master' since)
 - when a command fails or produces unexpected output, run it again with `--debug` and post the output

jonasmalacofilho, 2019-10-31:
You need to add it to `SmartDeviceDriverV2`, not `SmartDevice` ; )

David-Frick, 2019-10-31:
Opposie. Let me hop on that!

jonasmalacofilho, 2019-10-31:
Despite the error, it seems close enough.  Try the color setting commands next.

David-Frick, 2019-10-31:
Okay, fair enough. I went ahead and ran the following commands and they work! It looks great:
```
david@Debian-1:~/liquidctl$ sudo liquidctl --device 0 set led1 color fixed ff0000
david@Debian-1:~/liquidctl$ sudo liquidctl --device 0 set led2 color fixed ff0000
david@Debian-1:~/liquidctl$ sudo liquidctl --device 0 set led2 color fixed 00ff00
david@Debian-1:~/liquidctl$ sudo liquidctl --device 0 set led1 color fixed 00ff00
david@Debian-1:~/liquidctl$ sudo liquidctl --device 0 set led1 color fixed 0000ff
david@Debian-1:~/liquidctl$ sudo liquidctl --device 0 set led2 color fixed 0000ff
```
My note is this: led1 changes 1 led and led2 change the other 5. This is because I have the 5 on one channel and the leftover 1 on its own. Is there a way to "subset" so to speak a channel?

David-Frick, 2019-10-31:
Let me just say that this is awesome. Thanks so much! I am so happy to have this working without that nasty GB data mining CAM software :)

jonasmalacofilho, 2019-10-31:
That's great news!

Back to...

> Running liquidctl status after that makes the Aer 2 fans blink for a second (blink off/on) and then the command gets hung up.

I modified the status loop to handle this case.  That is, assuming I haven't messed it up (I don't have anything to test that driver with).

Please give commit 2f537119cbc01427e83d114ff63614c3d526932c from the new 'add-support-for-hue2' branch a try.

```
# liquidctl status --debug
```

> My note is this: led1 changes 1 led and led2 change the other 5. This is because I have the 5 on one channel and the leftover 1 on its own. Is there a way to "subset" so to speak a channel?

On the Smart Device V1 (from the HUE+ generation) modes were set per channel.  Some modes allowed you to address each individual LED IC separately, and a few other modes could adapt to the type of the accessories (e.g. spectrum-wave with fans vs. strips), but it was still just one mode per channel.

Now with the new generation I'm not sure if that's still the case.  On one hand the idea behind having channels is that they are the partition.  On the other, the ability to mix many different devices already suggests that the channel might not be that important anymore.

Maybe you or @CaseySJ can check this with CAM.

---

P.S. The new branch is being tracked as PR #64 ("Add support for the NZXT HUE 2").

jonasmalacofilho, 2019-11-01:
Hi @David-Frick 

I would like to include the support for the HUE 2 in v1.3.0, but for that #64 would need to be stable(ish) by this weekend's release candidate.

CaseySJ, 2019-11-01:
> On the Smart Device V1 (from the HUE+ generation) modes were set per channel. Some modes allowed you to address each individual LED IC separately, and a few other modes could adapt to the type of [...]

I believe the behavior is still the same with HUE 2 and Smart Device V2, namely one color mode per channel. I'll grab some screenshots in Windows this evening and post them here in a ZIP file.

CaseySJ, 2019-11-02:
Here's a zip with 2 screenshots of CAM showing that individual components on an LED channel cannot be controlled independently. Instead, the channel is assigned a color mode and that mode is applied to all components on that channel.

[NZXT-CAM.zip](https://github.com/jonasmalacofilho/liquidctl/files/3800010/NZXT-CAM.zip)

David-Frick, 2019-11-06:
Sorry for the long silence. I have been working on some website code. I will get back to y'all first thing tomorrow when I am up. Do y'all happen to have any tips for python libraries to control Asus Aura Sync RGB lights (my mobo has it) and EVGA X10 Precision (I think it is called) RGB lights. I hope to be able to get this small minor light on the mobo that just cycles through rainbow colors to be similar in functionality and my EVGA GPU as well, but my guess is those would be like opening another can of worms.

@CaseySJ I figured. I watched some CAM YT's and saw people demonstrating the modes. Odd that it works that way, but I can not say I understand how channels work, but each channel is 5 devices I think so it is improving in some ways at least!

jonasmalacofilho, 2019-11-06:
@David-Frick 

That's fine, and I'm glad you're back.

Please give the code in #64/add-support-for-hue2 a try and let me know what works and what doesn't... including the output of `status`.

---

For ASUS motherboards you can take a look at [CalcProgrammer1/OpenAuraSDK](https://gitlab.com/CalcProgrammer1/OpenAuraSDK).  Be sure to read the complete README as you might need a patched driver.

On the other hand, the only thing I know about Nvidia graphics cards is that they expose a `GPULogoBrightness` attribute in `nvidia-settings`.  Unfortunately, cards from EVGA don't honor it and, even if they did, you still would have no control over the LED color or animation.

David-Frick, 2019-11-06:
So I was getting an issue about too many devices found after I had changed to the branch suggested, which I think had something to do with my uninstall process. I am at a weird issue where pip3 uninstall liquidctl does not find the files to uninstall but liquidctl command still works from the CLI, and pip3 install editable . for the new branch does seem to work. But then when I go to run sudo liquidctl initialize all, it is not found again. Not really sure how I managed to do this. Could you guide me through cleaning the install. Sorry for the trouble.

David-Frick, 2019-11-06:
Report:
`liquidctl status --debug` returns the following:
```[DEBUG] liquidctl.cli: running liquidctl v1.2.0 (editable)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2001
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xa88
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xa87
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc33c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc33c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc539
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc539
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc539
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc539
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc33c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xa87
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xa88
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2001
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.cli: device: NZXT HUE 2 (experimental)
NZXT HUE 2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 55 ff 73 06 78 71 57 51 22 33 22 67 01 20 01 02 05 00 31 33 39 39 31 38 34 32 32 30 30 31 32 35 32 38 ff ff ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 55 ff 73 06 78 71 57 51 22 33 22 67 04 0b 00 00 00 00 00 0b 0b 0b 0b 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
├── Firmware version               2.5.0  
├── LED 1 accessory 1    AER RGB 2 120mm  
├── LED 2 accessory 1    AER RGB 2 120mm  
├── LED 2 accessory 2    AER RGB 2 120mm  
├── LED 2 accessory 3    AER RGB 2 120mm  
├── LED 2 accessory 4    AER RGB 2 120mm  
└── LED 2 accessory 5    AER RGB 2 120mm  

[DEBUG] liquidctl.cli: device: NZXT Kraken X (X42, X52, X62 or X72)
NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.kraken_two: received 04 1c 01 00 00 09 fa 00 00 00 ff 04 00 00 02 1e 00 00 15 59 a0 9b 6a a7 46 4d 7f 71 08 8e 7e e5 c4 22 f9 88 45 24 7d 55 38 d8 8d ae 76 f5 0f 51 94 35 df 52 9f bc ab 5a 45 59 c7 f4 be 2a 65 e8
├── Liquid temperature     28.1  °C
├── Fan speed                 0  rpm
├── Pump speed             2554  rpm
└── Firmware version      4.0.2  
```

the command does cause the LED fans to blink then returns this output. All commands seem to work other than, I do not seem to be able to set the fan speed of each of the AER fans.

CaseySJ, 2019-11-06:
@David-Frick 

The HUE 2 is a lighting controller only. Each AER RGB 2 fan has, of course, two connectors: One for fan speed and the other for RGB lighting effects. If the fan speed connector on all AER fans is connected to the Kraken, then all AER fans can be controlled through the Kraken `Fan Speed` channel. However, this is a single channel that will set all fans to the same speed instead of controlling each fan individually.

The Kraken `status` reply posted above shows `0 rpm` so perhaps the AER fans are connected directly to the motherboard? Or to a different fan controller?

David-Frick, 2019-11-06:
So the Kraken fan header has a 4 fan split. I was wondering why it was 0 rpm so I checked the header and the fans skipped over the first header, so I guess it was only seeing the first one. I changed the order and reran the command for this:

```
NZXT HUE 2 (experimental)
├── Firmware version               2.5.0  
├── LED 1 accessory 1    AER RGB 2 120mm  
├── LED 2 accessory 1    AER RGB 2 120mm  
├── LED 2 accessory 2    AER RGB 2 120mm  
├── LED 2 accessory 3    AER RGB 2 120mm  
├── LED 2 accessory 4    AER RGB 2 120mm  
└── LED 2 accessory 5    AER RGB 2 120mm  

NZXT Kraken X (X42, X52, X62 or X72)
├── Liquid temperature     28.9  °C
├── Fan speed               766  rpm
├── Pump speed             2573  rpm
```

CaseySJ, 2019-11-06:
@David-Frick 

That's more like it! As you saw, we cannot skip a fan header on that cable. In my NZXT H510 Elite, I've connected all 4 AER RGB 2 fans to the Kraken fan splitter cable. I cannot control them individually, but at least I can control them! (I'm running macOS.)

David-Frick, 2019-11-07:
Report:
```
liquidctl --device 1 set fan speed 100
```
does work, I believe for the two fans connected to the pump which makes sense. Great! 

Side note: the liquidctl status command does make the lights flicker off then on. Is this as it should be?

Do you know how I could control the fans for the fan header (also, I do have one plugged into the EVGA GPU header)? The fan header came with the Phanteks case so I assume I will have to figure out how to code it myself or dig around.

Also, I saw a issue request for help with ubuntu, debian, and some other distro for making linux packages. What does this entail? I do have debian, so if y'all need me to do something to help with that, let me know.

My next steps are:
1) See if I can find out about the EVGA RGB for the GPU.
2) See if I can get the other fan speeds controlled.
3) See if I can get some RGB RAM as well ;) and if that would work (the linked Asus Aura did mention some), but I know some redditors complained CAM did not like their RGB RAM and messed it up somehow.
4) Incorporate the Asus Aura opensource software linked above. 
5) Make a GUI to incorporate them all, maybe Tkinter. Should be fun ;)

jonasmalacofilho, 2019-11-07:
> Side note: the liquidctl status command does make the lights flicker off then on. Is this as it should be?

@CaseySJ, does this happen with the SDV2 as well, or is it specific to the HUE 2?

We need to consider that some users may be calling `status` (or `get_status`) from a loop (e.g. using a custom bash script or something like `extra/liquiddump`), potentially making the lights flicker every few seconds.

One solution that comes to mind is to move the firmware and accessory listing code to `initialize`.  Does the device respond to 0x1001, x2003 immediately or do we really need to go through the loop if we're only interested in those two responses?

For this to work `initialize` would also need to return a list of tuples similar to `get_status`, and the CLI code would have to render it as well.

jonasmalacofilho, 2019-11-07:
@David-Frick 

> Do you know how I could control the fans for the fan header (also, I do have one plugged into the EVGA GPU header)? The fan header came with the Phanteks case so I assume I will have to figure out how to code it myself or dig around.

Do you mean a fan hub from Phanteks?  Does it connect to the motherboard via USB or does it simply duplicate a PWM signal (i.e. is connected to the motherboard using a standard fan connector as well)?
 
> Also, I saw a issue request for help with ubuntu, debian, and some other distro for making linux packages. What does this entail? I do have debian, so if y'all need me to do something to help with that, let me know.

It would be awesome if you could help with that!

Since you're running Debian, I think the best route is to request and then work on a package for Debian, that is, a package that will eventually be part of Debian (and, as a result, Ubuntu and other Debian compatible or based distros).

I think you can start by:

 - opening a [Debian Request for Package (RFP)](https://wiki.debian.org/RFP)
 - taking a look at [Debian/Packaging](https://wiki.debian.org/Packaging) and [Debian/Python Policy for Python developers](https://wiki.debian.org/Python#Debian_Python_Policy_for_Python_developers)

> My next steps are:
>
> 1. See if I can find out about the EVGA RGB for the GPU.
> 2. See if I can get the other fan speeds controlled.
> 3. See if I can get some RGB RAM as well ;) and if that would work [...]

Nice!  Please keep me updated on your progress on the GUI.

David-Frick, 2019-11-07:
So, let's tackle the Phanteks thing first if that can help you! It simply duplicates the PWM signal so it connects to a 4 pin on the motherboard.

jonasmalacofilho, 2019-11-07:
> So, let's tackle the Phanteks thing first if that can help you! It simply duplicates the PWM signal so it connects to a 4 pin on the motherboard.

In this case you'll need to figure out how to control the motherboard fan headers.

What board do you have, and is there support for its Super I/O chip in the kernel?  Have tried running `sensors-detect` (from lm-sensors)?

CaseySJ, 2019-11-07:
> @CaseySJ, does this happen with the SDV2 as well, or is it specific to the HUE 2?
>
> We need to consider that some users may be calling `status` (or `get_status`) from a loop (e.g. using a custom bash [...]

My H510 Elite system is currently offline, but should be returned to duty tomorrow. As I recall, however, the RGB lights on the Smart Device V2 do **not** blink when `status` is called. Let me double check this tomorrow.

jonasmalacofilho, 2019-11-07:
Thanks @CaseySJ!

(You probably already considered it, but) I forgot to mention: to avoid the loop we need to know if `0x1001` and `0x2003` _always_ produce immediate replies.

So it might be necessary to check that a few times (at least 10)...

David-Frick, 2019-11-07:
running sensors-detect returned the following:
```
Now follows a summary of the probes I have just done.
Just press ENTER to continue:  

Driver `k10temp' (autoloaded):
  * Chip `AMD Family 17h thermal sensors' (confidence: 9)

Driver `to-be-written':
  * ISA bus, address 0x290
    Chip `ITE IT8665E Super IO Sensors' (confidence: 9)

Note: there is no driver for ITE IT8665E Super IO Sensors yet.
Check https://hwmon.wiki.kernel.org/device_support_status for updates.

No modules to load, skipping modules configuration.

Unloading i2c-dev... OK
Unloading cpuid... OK
```

jonasmalacofilho, 2019-11-07:
@David-Frick 

Oh, a board with an ITE chip. Unfortunately it seems that ITE, unlike other chip manufacturers, refuses to provide datasheets even to kernel maintainers.

Try taking a look at [`asus-wmi-sensors`](https://github.com/electrified/asus-wmi-sensors), but AFAIK it doesn't support fan control.

David-Frick, 2019-11-08:
So I am going to play around with a GUI for this for now and get back to y'all!

CaseySJ, 2019-11-08:
Quick update: Weekday commitments have delayed my testing of the `liquidctl status` command on Smart Device V2 to determine whether it causes lights to blink. Will test that within the next day or two.

David-Frick, 2019-11-08:
@jonasmalacofilho I went ahead and made a repo for the gui while I hammer it out. I will keep in touch and get feed back from y'all.

CaseySJ, 2019-11-09:
Just confirmed that Smart Device V2 also blinks when `liquidctl status` is invoked.

CaseySJ, 2019-11-09:
- When we issue command `0x1001` we receive an immediate `1101` reply.
- But when we issue command `0x2003` we do not receive an immediate `2103` reply. Instead, the reply arrives after three sets of `0x6702 / 0x6704` replies. In other words, the `2103` reply is in the 7th `device.read` command.
- This behavior is repeatable. I tried it 10 times with exactly the same order of replies.
```
NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 28 00 2e 00 16 57 41 47 35 31 33 20 06 20 01 01 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 08 04 00 00 00 00 00 00 00 00 00 00 00 00 32 32 32 00 00 00 00 00 32 32 32 00 00 00 00 00 42 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 c6 2f 8a 2f da 2f 00 00 00 00 00 00 00 00 00 00 00 00 31 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 08 04 00 00 00 00 00 00 00 00 00 00 00 00 32 32 32 00 00 00 00 00 32 32 32 00 00 00 00 00 42 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 da 2f 6f 2f d0 2f 00 00 00 00 00 00 00 00 00 00 00 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 0a 04 00 00 00 00 00 00 00 00 00 00 00 00 32 32 32 00 00 00 00 00 32 32 32 00 00 00 00 00 42 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 e8 2f 76 2f c6 2f 00 00 00 00 00 00 00 00 00 00 00 00 3f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 28 00 2e 00 16 57 41 47 35 31 33 20 02 04 0a 0a 00 00 00 0c 0c 0c 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 06 04 00 00 00 00 00 00 00 00 00 00 00 00 32 32 32 00 00 00 00 00 32 32 32 00 00 00 00 00 42 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 da 2f 7d 2f d7 2f 00 00 00 00 00 00 00 00 00 00 00 00 6f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 28 00 2e 00 16 57 41 47 35 31 33 20 03 00 00 02 00 00 00 00 00 00 00 00 06 04 00 00 00 00 00 00 00 00 00 00 00 00 32 32 32 00 00 00 00 00 32 32 32 00 00 00 00 00 42 00 00 00 00 00 00 00
└── Nothing    to  display
```

- I modified the `status` function as follows in order to conduct this test. Namely, the first command always returns the correct reply, hence after issuing `0x1003` we can invoke `device.read`. But after issuing `2003` we still need a loop.
```Python
        # Get configuration information from device
        self._write([0x10, 0x01])  # get firmware info
        msg = self.device.read(self._READ_LENGTH)
        LOGGER.debug('received %s', ' '.join(format(i, '02x') for i in msg))
    
        self._write([0x20, 0x03])  # get lighting info
        for x in range(10):
            msg = self.device.read(self._READ_LENGTH)
            LOGGER.debug('received %s', ' '.join(format(i, '02x') for i in msg))
        self.device.release()
        return sorted(status)
```

David-Frick, 2019-11-09:
Would you like me to run and duplicate this?

CaseySJ, 2019-11-09:
> Would you like me to run and duplicate this?
> […](#)

No need to duplicate. It's just a confirmation that both HUE 2 Lighting Controller and Smart Device V2 (and presumably the HUE 2 Ambient V2) exhibit the same behavior. It has already been resolved by @jonasmalacofilho in the `rebalance-sdv2-initialize-status` branch.

jonasmalacofilho, 2019-11-10:
Done : )

closed 2019-11-10 by jonasmalacofilho

David-Frick, 2019-11-12:
@jonasmalacofilho I have a question. Is it possible the RGB on the AER 2 fans could be controlled if the AER 2 fans are connected to the mobo and not the NZXT Hue 2 or is that the issue we talked about above with the ISA Chip? I am hoping to possible get rid of the Hue 2 and the Internal USB hub as they take up so much space in my case ;/  Hope all is well!

jonasmalacofilho, 2019-11-12:
@David-Frick 

All good here, hope you're fine too!

Regarding connecting AER 2 fans to motherboard headers...

Well, I'm not 100% sure, but given that HUE 2 is backwards compatible with HUE+, it probably still uses the same WS2812B LED ICs as the previous generation.  And that's the same IC most **addressable** LED systems (Aura, Mystic, Corsair) use.

So yeah, it should be compatible, but you'll need to adjust the pinout: you can leave FD unconnected (or whatever the pin other than +5V, DATA and GND is called on HUE 2), and just match the other pins by name/function.

_Also, obviously, make sure to connect them to addressable/5V headers, not 12V ones!_

And for software control you could take a look at [CalcProgrammer1/OpenAuraSDK](https://gitlab.com/CalcProgrammer1/OpenAuraSDK).

---

That said you can reduce some of the bulk by removing the internal USB hub.

The HUE 2 and Kraken each only use one of the two physical ports in a USB 2.0 header.  So you can simply transfer the pins from one the cables to the other, making both devices share a single header (see [video](https://youtu.be/BbvfiWY4xWc?t=255)).

: )

David-Frick, 2019-11-12:
I will definitely combine the USB cables. OpenAuraSDK does not seem to
support my mobo yet. I have seen some other interesting projects that are
related to that here:
https://rog.asus.com/forum/showthread.php?95652-AURA-SDK-Projects-and-Apps-List