liquidctl/liquidctl #62: Package for Debian/Ubuntu

jonasmalacofilho, 2019-10-31:
Help get liquidctl packaged for Debian, Ubuntu and distros derived from them.

jonasmalacofilho, 2019-11-10:
The best route is probably to request and then work on a package for Debian, that is, a package that will eventually be part of Debian (and, as a result, Ubuntu and other Debian compatible or based distros).

You can start by:

 - opening a [Debian Request for Package (RFP)](https://wiki.debian.org/RFP)
 - taking a look at [Debian/Packaging](https://wiki.debian.org/Packaging) and [Debian/Python Policy for Python developers](https://wiki.debian.org/Python#Debian_Python_Policy_for_Python_developers)

---

An alternative would be for someone to setup and maintain a Ubuntu PPA for liquidctl.

keton, 2019-11-10:
@jonasmalacofilho is there a dependency list for liquidctl? It will be super helpful for checking if given python package/library is already packaged for given distribution or should we package it as well.

jonasmalacofilho, 2019-11-10:
The Python dependencies are specified in setup.py

```
    install_requires=['docopt', 'pyusb', 'hidapi', 'appdirs'],
```

but Linux packages may also need `setuptools`.

This is a common Python packaging problem when `script` entries are used, and is caused by `setup.py install` generating the executable entry point with a dependency to the `pkg_resources` Python module:

```
$ cat `which liquidctl`
#!/usr/bin/python
# EASY-INSTALL-ENTRY-SCRIPT: 'liquidctl==1.3.0rc1','console_scripts','liquidctl'
__requires__ = 'liquidctl==1.3.0rc1'
import re
import sys
from pkg_resources import load_entry_point

if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
    sys.exit(
        load_entry_point('liquidctl==1.3.0rc1', 'console_scripts', 'liquidctl')()
    )
```

Anyway, this is all standard Python packaging stuff, so if this extra dependency is needed on Debian should™ be documented somewhere.  Alternatively you could check if `pkg_resources` is already part of the Debian `python3` distribution.

---

The runtime dependencies are libusb and HIDAPI, though whether or not these need to be explicitly required can vary:

 - does the pyusb package already depend on libusb?
 - does the cython-hidapi package (the one for the Python wrapper of HIDAPI) already depend _or_ include HIDAPI?

jonasmalacofilho, 2019-11-10:
Update:

 - Python `appdirs` is available as [`python3-appdirs`](https://packages.debian.org/bullseye/python3-appdirs)
 - Python `docopt` is available as [`python3-docopt`](https://packages.debian.org/bullseye/python3-docopt)
 - Python `pyusb` is available as [`python3-usb`](https://packages.debian.org/bullseye/python3-usb), already depends on libusb
 - Python `pkg_resources` **is required** and available as [`python3-pkg-resources`](https://packages.debian.org/bullseye/python3-pkg-resources)
 - **Python `hidapi` is not available (upstream: [trezor/cython-hidapi](https://github.com/trezor/cython-hidapi))**

For the HIDAPI wrapper there is [`python3-hidapi`](https://packages.debian.org/bullseye/python3-hidapi), but it's a different and less known module (upstream: [jbaiter/hidapi-cffi](https://github.com/jbaiter/hidapi-cffi)).

Assuming it works, liquidctl could be adapted to support it (or _maybe_ even switched to it), but it would be good if we had the option of sticking with cython-hidapi.

---

Even though the Debian Python packaging docs don't mention it, `pkg_resources` does appear to an additional dependency.<sup>[1](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=917271)</sup>  You could also make contribute this fix to the Debian Wiki.

keton, 2019-11-10:
For Ubuntu 19.10:
* docopt: https://packages.ubuntu.com/eoan/python3-docopt
* pyusb: https://packages.ubuntu.com/eoan/python3-usb
* hidapi: https://packages.ubuntu.com/eoan/python3-hidapi
* appdirs: https://packages.ubuntu.com/eoan/python3-appdirs

Hidapi is the correct package see [Homepage entry](http://archive.ubuntu.com/ubuntu/pool/universe/p/python-hidapi/python-hidapi_0.7.99.post21-1build2.dsc). 
It seems like a better target for the first attempts.

jonasmalacofilho, 2019-11-10:
You're right, it's the correct one.  I just wish the packages had more consistent names...

For example:

 - the source package for python3-hid is... python-hidapi
 - the source package for python3-hidapi is... hidapi-cffi
 - python-hidapi is also the name of the binary package for hidapi-cffi × python2

keton, 2019-11-14:
I've got to the point where .deb builds on Ubuntu eoan. Same metadata should be good for any recent Ubuntu and Debian versions.

```sh
git clone -b add-debian-packaging git@github.com:keton/liquidctl.git
cd liquidctl
# install build dependencies
sudo apt install $(dpkg-checkbuilddeps 2>&1 | grep error | cut -d':' -f 4 )
# do build
dpkg-buildpackage -g --no-sign
#install resulting package
sudo apt install ../python3-liquidctl_1.3.0-1_all.deb
```
now `sudo  liquidctl initialize` should work.

I've also set up test PPA here: https://launchpad.net/~keton22-gmail/+archive/ubuntu/liquidctl-test-ppa
```
sudo add-apt-repository ppa:keton22-gmail/liquidctl-test-ppa
sudo apt update
sudo apt install python3-liquidctl
```

Now I need to figure out how to setup similar repo for Debian.

jonasmalacofilho, 2019-11-14:
That's great, thanks!

Can you use `DIST_NAME` and `DIST_PACKAGE` to customize the output from `--version`?  It might help me when dealing with bug reports...
 - `DIST_NAME` can be something like `<Ubuntu|Debian> <version>` (`/etc/os/release` might help)
 - `DIST_PACKAGE` can be the complete distro package name, including version, release, etc. (e.g. `python3-liquidctl_1.3.0-1_all`)
 - you can use the [PKGBUILD](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=python-liquidctl-git#n25) for ArchLinux/git as a reference

You also probably need to include a man page; you find it in `liquidctl-1.3.0rc1.tar.gz/liquidctl.8`.  I've made some improvements to it since, but if you still find any errors please let me know (and/or open a pull request).

Thanks again!

jonasmalacofilho, 2019-11-14:
Oh, and I think you might need to split the package into two:

 - `python3-liquidctl`: Python library
 - `liquidctl`: `/usr/bin/liquidctl`, man page

I didn't see this requirement in the docs, but take a look at [flake8](https://packages.ubuntu.com/disco/flake8).

jonasmalacofilho, 2019-11-17:
I just released 1.3.0; note that `appdirs` is no longer a dependency.

David-Frick, 2019-12-01:
Okay, so I got OpenAuraSDK to work thanks to the maintainers help so thanks a lot for pointing that out to me. I am adding more stuff slowly to the liquidctlgui. Do y'all need anything to help with making the Debian package?

jonasmalacofilho, 2019-12-02:
Besides the changes/fixes listed above (which will probably be required for the package to be accepted), you can look into opening a [RFP](https://wiki.debian.org/RFP) (and possibly also a [ITP](https://wiki.debian.org/ITP), since you'd be working on it).

David-Frick, 2019-12-02:
Okay, I am reading up a little on it. Let me know when you are ready for the next steps and tell me what to do :)

jonasmalacofilho, 2019-12-03:
I'm available/ready to assist you.

But besides the few changes above that I know (or at least think) are necessary, I'm not sure what else has to be done.  This is why I think it's time to open the RFP (and the ITP), and get some proper feedback.

I haven't taken charge on either because I'm not a Debian user on the desktop.¹  I would either ask them to create a package I wouldn't use, or endlessly bother them with stupid questions because I'm not familiar with the distro.

---

¹ I've been running Arch on the desktop for several years.  I run Ubuntu on servers and on containers, but mostly either using official packages or stuff we build/deploy with external tools.

David-Frick, 2019-12-03:
Okay gotcha. Let me go ahead and contact them!

-------- Original Message --------

David-Frick, 2019-12-07:
Following up: This link on the debian website [wnpp process](https://www.debian.org/devel/wnpp/) walks through the process a little bit. Going to walk through this, this week. Seems pretty straight forward.

jonasmalacofilho, 2019-12-07:
Great, thanks!

jonasmalacofilho, 2019-12-23:
Hi @David-Frick, did you make any progress?

David-Frick, 2019-12-23:
Hey!

The Debian docs are not the best! I am actually reading over the arch linux docs regarding how they do it to get a better idea and come back to debian. I am sorry I have not kept in touch, it has been pouring rain (in winter of all times) and I got super sick :(

I am going to jot up notes and do it first thing tomorrow. One question: I upgraded to Debian unstable (sid). Would this affect how the package is maintained? I would assume so! I can always go back to good, regular Debian if need be.

-------- Original Message --------

jonasmalacofilho, 2019-12-23:
Hi!

> The Debian docs are not the best! I am actually reading over the arch linux docs regarding how they do it to get a better idea and come back to debian. I am sorry I have not kept in touch, it has been pouring rain (in winter of all times) and I got super sick :(

Yeah, I have to admit I still have a few questions myself about how Debian's WNNP bugs work.

> I am going to jot up notes and do it first thing tomorrow. One question: I upgraded to Debian unstable (sid). Would this affect how the package is maintained? I would assume so! I can always go back to good, regular Debian if need be.

Actually, I think that new packages have to go to unstable anyway, at least until the next Debian stable release.  But I could be _totally_ wrong about this.  It's just that I'm no used to seeing recent software packages on Debian stable.

David-Frick, 2019-12-23:
@jonasmalacofilho So I went ahead and submitted the ITP. It says I should receive an email shortly and that I will need to get in touch with a Debian mentor who will help me prepare it and become a mentor myself. So we will see how it goes. I am looking at the next steps as I need to make a .deb file out of it (actually package). 

They have good documentation in terms of workflows and what need to be done. It is just when I go through step by step, I get links to other documents which then have links. I can not seem to get a straight answer on say the actual programs and the relative commands. I end up looking at Arch Docs for comparison, and then looking at people posting about the debian related commands.
I will probably post on the Debian subreddit and see if they can help out!

jonasmalacofilho, 2019-12-26:
@David-Frick I still can't seem to find the ITP.  Did you get the email?

jonasmalacofilho, 2019-12-26:
Also, feel free to C/C when necessary.

David-Frick, 2019-12-26:
I did not :(
Let me try again.
I am having some issues with Debian overall as sddm of kde decided to not want to work with my monitor anymore even on stable (buster).

Can KDE Neon or even Ubuntu make Debian packages to be released on Debian because I mihht need to switch.

Sent from ProtonMail mobile

-------- Original Message --------

jonasmalacofilho, 2019-12-27:
> I am having some issues with Debian overall as sddm of kde decided to not want to work with my monitor anymore even on stable (buster).

You might have more luck with debian unstable (and more up-to-date software), assuming you're running new-ish hardware.

> Can KDE Neon or even Ubuntu make Debian packages to be released on Debian because I mihht need to switch.

I would assume so, though some care might be necessary when testing or debugging, since they might be running different versions of dependencies and have slightly different default configurations.

David-Frick, 2019-12-27:
Yeah, I figured. I am making a fresh install of sid now and am almost done.

-------- Original Message --------

Frick-David, 2019-12-30:
@jonasmalacofilho so a user on another thread says that  a Trusted User (TU) needs to pick up the package for it to be put into the package manager, and implied it is hard/takes a long time to become a Trusted User. When I filled in the ITP (which I will redo to make sure it goes through), it did talk about getting in touch with a TU to become one and finish the package by getting mentored. I am hoping to get in touch with one. I even emailed the debian listed email to request a mentor, but have not heard back. So I will try again on the ITP, and talk with some other Debian users that might know a little more to get the process under way than myself.

jonasmalacofilho, 2019-12-31:
Thanks for the update... I'll also see how I know that's already involved in Debian and can help.

jonasmalacofilho, 2020-01-25:
No news on my end.

@Frick-David, filling the ITP requires you to send an email right?  Maybe it has to be plain-text only?

Frick-David, 2020-01-27:
Let me look and see. The Debian website does seem to have some parts being cleaned up so here is to hoping!

I am going to continue work and see how much of the GUI I can power through tomorrow.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

Frick-David, 2020-01-28:
@jonasmalacofilho resubmitted the ITP through bugreport. Should get an email. If I do not, I will do it by email.

Frick-David, 2020-01-28:
Update: Did not hear anything back, nor did I see it in their database of requests so I sent an email. Hopefully that goes through,

Frick-David, 2020-01-28:
I went ahead and emailed the owner of the bug report address because it keeps sending back an email error about having no package tag when I can clearly see it. So frustrating, Hopefully I hear back from someone about it. I think the process probably needs to be tweaked on their end because it, itself, seems to need a bug report ;/

Frick-David, 2020-01-30:
@jonasmalacofilho have not heard back? What should I do next?

jonasmalacofilho, 2020-01-31:
I'm not sure, but I'll take a look.  Thanks.

Frick-David, 2020-02-04:
I saw a Debian team member commenting on the creator of gwe (who also made gkraken which relies on liquidctl) for permission to package his repo. I am going to message him and ask as well!

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

jonasmalacofilho, 2020-02-04:
That would be great!

Frick-David, 2020-02-14:
I went ahead and filing another ITP right now using a gmail account. I was having issues with another service due to my email so maybe the Debian thinks my email address is spam? Idk. I also messaged the Debian packager on another thread so I will check in on what he said.

Frick-David, 2020-02-24:
@jonasmalacofilho No response so I am not sure what to do :(
Real shame. This has made my opinion of Debian go down a little :(
Any ideas for debian packaging? 
I will keep plugging away at the gui on weekends.

Frick-David, 2020-02-27:
I cleaned up my comments for a potential future user who has more success to see the base issue and not my clutter. To summarize all my comments, to submit an intent to package, simply using the reportbug program and follow through with the prompts and options. The documentation for that is found here: https://www.debian.org/devel/wnpp/
Unfortunately, I do not seem to get any error messages or any emails back despite using multiple emails and different attempts with different names, etc etc, despite the form being filled out pretty thoroughly. Similarly, I filled out one directly by email with no response. I have even @ed a Debian team member without hearing back. 
I am unsure of why this is, and it is a bit of a let down, because I have had nothing but positive things to say about Debian up until now. I have no reason to believe it is an issue on my end, but it is possible. I hope someone can give this a shot in the near future, keep me posted, and hopefully they have more luck! 

The process is as follows:
Run `sudo apt-get install reportbug`
Run `reportbug --email username@domain.tld wnpp`
And walk through the prompts. We want an ITP (intent to package).
Do not remember the exact order of the prompts, but you will fill out
type of package (we want an ITP (intent to package)), a form that has the information of the package: It will ask for the same type data here:
https://aur.archlinux.org/packages/liquidctl/   (version number ,package name, etc)
and some other pretty straight forward questions.
It will probably ask if the package is similar to a bunch of others and give you a super long list. One the keys skips that part (none of them are).

Overall, it is pretty straightforward and should not take longer than 10, 15 minutes, if that.  After that a Debian team member (fingers crossed) should contact you about maintaining the package. It seems like a good topic to learn as it involves building the package, and keeping it up to date with the git repo for Debian Users. If you are looking to learn how Linux, Debian, and I would assume Systemd works, this is a great opportunity to learn some cool, neat stuff. You can even help out with other projects that you really like and want to help others with (Yay! Very nice!)

@jonasmalacofilho might I suggest deleting comments to get this more to the top for other people to see.

jonasmalacofilho, 2020-02-27:
Hi, @Frick-David 

Sorry for not replying to you sooner.  I was going to try `reportbug` myself to see if anything came to mind, but a few other things popped up and I haven't found the time for it yet.  I'll keep you updated on any progress I make.

---

Instead of deleting previous comments, I'll just link to that last one from the top post.

Frick-David, 2020-02-28:
No problem, just letting you know! Really hope this happens as this is a great repo! I will be chipping away at the GUI this weekend!

Best of luck!

jonasmalacofilho, 2020-11-26:
I would like to have liquidctl 1.5 in Debian 11 (Bullseye) and Ubuntu 21.04.

The schedule is a bit tight though, since both will start their feature freezes in early February.  So we should probably get 1.4.2 packaged before that, only requiring a quick update after 1.5.

@MarshallAsch, do you by any chance have any experience with this?

MarshallAsch, 2020-11-26:
Unfortunately I do not, I have never created a distro package before.

jonasmalacofilho, 2020-12-25:
RFP: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=977256

jonasmalacofilho, 2021-01-13:
RFP converted into an ITP (someone is working on it).  In fact, it should soon be in Debian:

> Packaged and going to upload it soon.
>—*László Böszörményi (GCS) <gcs@debian.org>*

closed 2021-01-13 by jonasmalacofilho

mattiasghodsian, 2021-01-31:
Update for ubuntu ?

jonasmalacofilho, 2021-01-31:
Ubuntu automatically takes packages from Debian unstable prior to the [Debian Import Freeze](https://wiki.ubuntu.com/DebianImportFreeze) date.  For Ubuntu 21.04 Hirsute hippo, [that date is 25 February](https://discourse.ubuntu.com/t/hirsute-hippo-release-schedule/18539).  So as long as `liquidctl` reaches Debian unstable by then, it should arrive in Ubuntu 21.04 automatically.

That said, you could try reaching out to some of Ubuntu's devs to see if there's another (more direct) way to get liquidctl into Ubuntu.

And you could also contact the developers [working on the Debian package](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=977256) and see if you can help in some way.   I'm also available to assist (you and any Debian or Ubuntu developers).

jonasmalacofilho, 2021-02-01:
Hurray, now available as:

- [liquidctl](https://packages.debian.org/sid/liquidctl) on the Debian sid/unstable repository
- [liquidctl](https://packages.ubuntu.com/hirsute/liquidctl) on the Ubuntu Hirsute Hippo (future 21.04) repository

Other useful links:

- [Debian Package Tracker - liquidctl](https://tracker.debian.org/pkg/liquidctl)
- [Debian Packages Search](https://www.debian.org/distrib/packages)
- [Ubuntu Packages Search](https://packages.ubuntu.com)