liquidctl/liquidctl #63: Package for macOS/Homebrew

jonasmalacofilho, 2019-10-31:
Help get liquidctl in homebrew (for macOS).

CaseySJ, 2019-10-31:
@jonasmalacofilho 

I read through [Python for Formula Authors](https://github.com/Homebrew/brew/blob/master/docs/Python-for-Formula-Authors.md) and looked at the `trezor-agent` example, which has resulted in the following **starting point** on macOS Catalina 10.15.1.

This is still incomplete. Some notes:

- In the `bottle` section there is currently no entry for `catalina`. Not sure how to obtain this.
- For `system` dependencies, I've listed both `python` and `libusb`, but not Xcode or other development tools.
- For `python` dependencies, I installed `liquidctl` in a virtualenv, installed `homebrew-pypi-poet` and ran the command `poet liquidctl`. This generated the various `resources` shown in the listing below.
- However, `poet` generated a `resource` entry that was identical to the sdist information for liquidctl, so I removed that block. Not sure if we need to keep it...
- There is currently no `test` section at the end of the file.

Input, feedback, and contributors to this code are welcome! 

```Ruby
class liquidctl < Formula
	include Language::Python::Virtualenv

	desc "Cross-platform CLI and Python drivers for AIO liquid coolers and other devices"
	homepage "https://github.com/jonasmalacofilho/liquidctl"
	url "https://files.pythonhosted.org/packages/f2/08/7e56efece1ed5083e82826a1734f6e3c452401602ca211448c4b2c9e4cc9/liquidctl-1.2.0.tar.gz"
	sha256 "ad8c03c0695620fedaec11e7a8286bb5d4da18ba0c71e55888bfa06f8f7d7529"

	bottle do
		cellar :any_skip_relocation
		sha256 "798de4490c97283280259ffc1dc39159bd0ded85edb47f3212ad5ec9a174289e" => :mojave
		sha256 "fdb2a72157445c33a462388f05580489c427f4f0d2a3d4cdc1b7867ef69e7e53" => :high_sierra
		sha256 "4ef3590e26c5ac96d64dc985b035ec7055f215c84d31dfb09542d958f6ec4e77" => :sierra
	end

	depends_on "libusb"
	depends_on "python"

	resource "appdirs" do
		url "https://files.pythonhosted.org/packages/48/69/d87c60746b393309ca30761f8e2b49473d43450b150cb08f3c6df5c11be5/appdirs-1.4.3.tar.gz"
		sha256 "9e5896d1372858f8dd3344faf4e5014d21849c756c8d5701f78f8a103b372d92"
	end

	resource "docopt" do
		url "https://files.pythonhosted.org/packages/a2/55/8f8cab2afd404cf578136ef2cc5dfb50baa1761b68c9da1fb1e4eed343c9/docopt-0.6.2.tar.gz"
		sha256 "49b3a825280bd66b3aa83585ef59c4a8c82f2c8a522dbe754a8bc8d08c85c491"
	end

	resource "hidapi" do
		url "https://files.pythonhosted.org/packages/c1/86/89df0e8890f96eeb5fb68d4ccb14cb38e2c2d2cfd7601ba972206acd9015/hidapi-0.7.99.post21.tar.gz"
		sha256 "e0be1aa6566979266a8fc845ab0e18613f4918cf2c977fe67050f5dc7e2a9a97"
	end

	resource "pyusb" do
		url "https://files.pythonhosted.org/packages/5f/34/2095e821c01225377dda4ebdbd53d8316d6abb243c9bee43d3888fa91dd6/pyusb-1.0.2.tar.gz"
		sha256 "4e9b72cc4a4205ca64fbf1f3fff39a335512166c151ad103e55c8223ac147362"
	end

	def install
		virtualenv_install_with_resources
	end
	
end
```

jonasmalacofilho, 2019-10-31:
Hi Casey, thanks for taking a look at this!

> In the bottle section there is currently no entry for catalina. Not sure how to obtain this.

I think the bottle will be automagically available at some time in the future (see: [Bottles/Creation](https://github.com/Homebrew/brew/blob/master/docs/Bottles.md#creation)).

> For system dependencies, I've listed both python and libusb, but not Xcode or other development tools.

That seems in line with all other formulas.  I think homebrew itself already requires all basic development tools.

> For python dependencies, I installed liquidctl in a virtualenv, installed homebrew-pypi-poet and ran the command poet liquidctl. This generated the various resources shown in the listing below.

Looks mostly fine to me, except that `setuptools` is also a dependency for the CLI.  It's implicitly available on a developer's machines, but I'm not sure if it would be required or not for end users installing through bottles (binary packages).

Also, does poet check more stuff than what's in `setup.py`?  If that's the case, it would be nice to periodically compare its results with what we specify explicitly in setup.py and what our other packages require (e.g. the [Arch Linux (git) package](https://aur.archlinux.org/packages/python-liquidctl-git/)).

> However, poet generated a resource entry that was identical to the sdist information for liquidctl, so I removed that resource block. Not sure if it's necessary to keep it...

Can you show me this entry?

> There is currently no test section at the end of the file.

My packages for ArchLinux currently have no tests either, but on AppVeyor I have the equivalent to

```
liquidctl list --verbose
```

which at least checks that Hidapi and PyUSB are able to enumerate devices.

You could also have it run the doctests.  There are some in the sources for the more complex yet still pure functions (e.g. in `liquidctl/util.py` and `liquidctl/pmbus.py`).

jonasmalacofilho, 2019-10-31:
Just edited the above: I suppose we can also pass `--debug` to the test `list`.

CaseySJ, 2019-10-31:
The presumably extraneous `liquidctl` resource entry looks like this:

```Ruby
        resource "liquidctl" do
                url "https://files.pythonhosted.org/packages/f2/08/7e56efece1ed5083e82826a1734f6e3c452401602ca211448c4b2c9e4cc9/liquidctl-1.2.0.tar.gz"
                sha256 "ad8c03c0695620fedaec11e7a8286bb5d4da18ba0c71e55888bfa06f8f7d7529"
        end
```

jonasmalacofilho, 2019-10-31:
Yeah, that's probably unnecessary : )

CaseySJ, 2019-10-31:
> I think the bottle will be automagically available at some time in the future (see: [Bottles/Creation](https://github.com/Homebrew/brew/blob/master/docs/Bottles.md#creation)).
> 
Yes, it seems `bottle` section is generated when we build a bottle. And then the `bottle` section can be manually added to the Formula.
> Looks mostly fine to me, except that `setuptools` is also a dependency for the CLI. It's implicitly available on a developer's machines, but I'm not sure if it would be required or not for end users installing through bottles (binary packages).

If it helps, the output from `brew search setuptools` is:
```
$ brew search setuptools
No formula or cask found for "setuptools".
Open pull requests:
python 3.7.5 (https://github.com/Homebrew/homebrew-core/pull/46020)
awscli 1.16.270 (https://github.com/Homebrew/homebrew-core/pull/46001)
snapcraft: Fix build for Linux (https://github.com/Homebrew/linuxbrew-core/pull/14123)
```
> Also, does poet check more stuff than what's in `setup.py`? If that's the case, it would be nice to periodically compare its results with what we specify explicitly in setup.py and what our other [...]

`Setup.py` lists these dependencies:
```Python
    install_requires=['docopt', 'pyusb', 'hidapi', 'appdirs'],
```
...and `poet` only added these. 
> My packages for ArchLinux currently have no tests either, but on AppVeyor I have the equivalent to
> 
> ```
> liquidctl list --verbose --debug
> ```
> 
> which at least checks that Hidapi and PyUSB are able to enumerate devices.

I've added this temporarily:
```Ruby
        test do
                output = shell_output("#{bin}/liquidctl list --verbose --debug 2>&1", 1)
        end
```

CaseySJ, 2019-10-31:
I just tested the Homebrew Formula and it seems to work. Others should test it as well, but here's the entire output:
```
casey's-MacBook-Air:Formula casey$ brew install liquidctl
Warning: Your Xcode (11.0) is outdated.
Please update to Xcode 11.1 (or delete it).
Xcode can be updated from the App Store.

==> Downloading https://files.pythonhosted.org/packages/f2/08/7e56efece1ed5083e82826a1734f6e3c452401602ca211448c4b2c9e4cc9/liquidctl-1.2.0.tar.gz
Already downloaded: /Users/casey/Library/Caches/Homebrew/downloads/39be550eec74c5d7cdb7cf0c68086b84735f115f9ff092f4771745a51348499e--liquidctl-1.2.0.tar.gz
==> Downloading https://files.pythonhosted.org/packages/11/74/2c151a13ef41ab9fb43b3c4ff9e788e0496ed7923b2078d42cab30622bdf/virtualenv-16.7.4.tar.gz
Already downloaded: /Users/casey/Library/Caches/Homebrew/downloads/507d2087bf24df82641b681a5fe4da778ef50ebe819a86ea1b584f70788f0f63--virtualenv-16.7.4.tar.gz
==> python3 -c import setuptools... --no-user-cfg install --prefix=/private/tmp/liquidctl--homebrew-virtualenv-20191031-13821-lu97fc/target --install-scripts=/private/tmp/liquidctl--homebrew-vi
==> python3 -s /private/tmp/liquidctl--homebrew-virtualenv-20191031-13821-lu97fc/target/bin/virtualenv -p python3 /usr/local/Cellar/liquidctl/1.2.0/libexec
==> Downloading https://files.pythonhosted.org/packages/48/69/d87c60746b393309ca30761f8e2b49473d43450b150cb08f3c6df5c11be5/appdirs-1.4.3.tar.gz
Already downloaded: /Users/casey/Library/Caches/Homebrew/downloads/ed8751b12aab49b1efdadc6d550c8953ab56e5d254247350f1f99135833be5b0--appdirs-1.4.3.tar.gz
==> /usr/local/Cellar/liquidctl/1.2.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/liquidctl--appdirs-20191031-13821-1fwnonk/appdirs-1.4.3
==> Downloading https://files.pythonhosted.org/packages/a2/55/8f8cab2afd404cf578136ef2cc5dfb50baa1761b68c9da1fb1e4eed343c9/docopt-0.6.2.tar.gz
Already downloaded: /Users/casey/Library/Caches/Homebrew/downloads/cae543cf1748375c2e56c2a4984cfa49a5858a0a55e6b55a819dfe969ce443d9--docopt-0.6.2.tar.gz
==> /usr/local/Cellar/liquidctl/1.2.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/liquidctl--docopt-20191031-13821-io17b8/docopt-0.6.2
==> Downloading https://files.pythonhosted.org/packages/c1/86/89df0e8890f96eeb5fb68d4ccb14cb38e2c2d2cfd7601ba972206acd9015/hidapi-0.7.99.post21.tar.gz
Already downloaded: /Users/casey/Library/Caches/Homebrew/downloads/1821f3108e754e54378ae15679b4d0d09805ad48d87e08bc60ed86388cc83ef5--hidapi-0.7.99.post21.tar.gz
==> /usr/local/Cellar/liquidctl/1.2.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/liquidctl--hidapi-20191031-13821-1foqk81/hidapi-0.7.99.post21
==> Downloading https://files.pythonhosted.org/packages/5f/34/2095e821c01225377dda4ebdbd53d8316d6abb243c9bee43d3888fa91dd6/pyusb-1.0.2.tar.gz
Already downloaded: /Users/casey/Library/Caches/Homebrew/downloads/bed999b3be90efb898cd6c543e0e7eb329f71fa7626edd10cb7bdad98a7a7865--pyusb-1.0.2.tar.gz
==> /usr/local/Cellar/liquidctl/1.2.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/liquidctl--pyusb-20191031-13821-h5fy4h/pyusb-1.0.2
==> /usr/local/Cellar/liquidctl/1.2.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/liquidctl-20191031-13821-slb62y/liquidctl-1.2.0
🍺  /usr/local/Cellar/liquidctl/1.2.0: 1,090 files, 11.4MB, built in 37 seconds
casey's-MacBook-Air:Formula casey$ liquidctl --debug list
[DEBUG] liquidctl.cli: liquidctl v1.2.0
```

CaseySJ, 2019-10-31:
Here's the Formula file (as tested above). We just copy it to this folder:
`/usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula`
... and then type:
`brew install liquidctl`

```Ruby
class Liquidctl < Formula
        include Language::Python::Virtualenv

        desc "Cross-platform CLI and Python drivers for AIO liquid coolers and other devices"
        homepage "https://github.com/jonasmalacofilho/liquidctl"
        url "https://files.pythonhosted.org/packages/f2/08/7e56efece1ed5083e82826a1734f6e3c452401602ca211448c4b2c9e4cc9/liquidctl-1.2.0.tar.gz"
        sha256 "ad8c03c0695620fedaec11e7a8286bb5d4da18ba0c71e55888bfa06f8f7d7529"

        depends_on "libusb"
        depends_on "python"

        resource "appdirs" do
                url "https://files.pythonhosted.org/packages/48/69/d87c60746b393309ca30761f8e2b49473d43450b150cb08f3c6df5c11be5/appdirs-1.4.3.tar.gz"
                sha256 "9e5896d1372858f8dd3344faf4e5014d21849c756c8d5701f78f8a103b372d92"
        end

        resource "docopt" do
                url "https://files.pythonhosted.org/packages/a2/55/8f8cab2afd404cf578136ef2cc5dfb50baa1761b68c9da1fb1e4eed343c9/docopt-0.6.2.tar.gz"
                sha256 "49b3a825280bd66b3aa83585ef59c4a8c82f2c8a522dbe754a8bc8d08c85c491"
        end

        resource "hidapi" do
                url "https://files.pythonhosted.org/packages/c1/86/89df0e8890f96eeb5fb68d4ccb14cb38e2c2d2cfd7601ba972206acd9015/hidapi-0.7.99.post21.tar.gz"
                sha256 "e0be1aa6566979266a8fc845ab0e18613f4918cf2c977fe67050f5dc7e2a9a97"
        end

        resource "pyusb" do
                url "https://files.pythonhosted.org/packages/5f/34/2095e821c01225377dda4ebdbd53d8316d6abb243c9bee43d3888fa91dd6/pyusb-1.0.2.tar.gz"
                sha256 "4e9b72cc4a4205ca64fbf1f3fff39a335512166c151ad103e55c8223ac147362"
        end

        def install
                virtualenv_install_with_resources
        end

        test do
                system "liquidctl list --verbose --debug"
        end
end
```

CaseySJ, 2019-10-31:
Regarding `setuptools` we can see this in the output quoted above (line #5):
```
==> python3 -c import setuptools... --no-user-cfg install ...
```

jonasmalacofilho, 2019-10-31:
> ```
> $ brew search setuptools
> No formula or cask found for "setuptools".
> ```

setuptools is a Python package, so _if necessary_ it would require another resource stanza.

---

We don't actually ship an executable script, and instead rely on `pip install` (or equivalents to create one.

It's that generated script that will usually depend on setuptools.  Here's mine on Linux:

```
$ cat `which liquidctl`
#!/usr/bin/python
# EASY-INSTALL-ENTRY-SCRIPT: 'liquidctl==1.2.0','console_scripts','liquidctl'
__requires__ = 'liquidctl==1.2.0'
import re
import sys
from pkg_resources import load_entry_point

if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
    sys.exit(
        load_entry_point('liquidctl==1.2.0', 'console_scripts', 'liquidctl')()
    )
```

After taking another look at what [`virtualenv_install_with_resources`](https://github.com/Homebrew/brew/blob/3944416c4602ae8618c4da399743fe96ddf0cf6a/Library/Homebrew/language/python.rb#L153) does, I'm starting to wonder if it's enough.

Did you remember to uninstall any/all liquidctl stuff before testing the Formula?  Like really making sure there's no executable `liquidctl` and that `from liquidctl import *` fails...

jonasmalacofilho, 2019-10-31:
>Regarding setuptools we can see this in the output quoted above (line # 5):
>```
>==> python3 -c import setuptools... --no-user-cfg install ...
>```

Hm, maybe that's why it's working... still, just to confirm, I'd make sure there's no trace left of liquidctl before testing the Formula.

CaseySJ, 2019-10-31:
Yes I completely removed `liquidctl` from `/usr/local/bin` and verified by typing `which liquidctl` (and also inspecting the directories themselves).

Let me repeat that and capture/post the outcome...one moment...

CaseySJ, 2019-10-31:
```
casey's-MacBook-Air:Formula casey$ brew remove liquidctl
Uninstalling /usr/local/Cellar/liquidctl/1.2.0... (1,090 files, 11.4MB)

casey's-MacBook-Air:Formula casey$ which liquidctl
(no output)

casey's-MacBook-Air:Formula casey$ ls /usr/local/bin/liquid*
ls: /usr/local/bin/liquid*: No such file or directory
```

CaseySJ, 2019-10-31:
And now let's install it again!  :)
```
casey's-MacBook-Air:Formula casey$ brew install liquidctl
Warning: Your Xcode (11.0) is outdated.
Please update to Xcode 11.1 (or delete it).
Xcode can be updated from the App Store.

==> Downloading https://files.pythonhosted.org/packages/f2/08/7e56efece1ed5083e82826a1734f6e3c452401602ca211448c4b2c9e4cc9/liquidctl-1.2.0.tar.gz
Already downloaded: /Users/casey/Library/Caches/Homebrew/downloads/39be550eec74c5d7cdb7cf0c68086b84735f115f9ff092f4771745a51348499e--liquidctl-1.2.0.tar.gz
==> Downloading https://files.pythonhosted.org/packages/11/74/2c151a13ef41ab9fb43b3c4ff9e788e0496ed7923b2078d42cab30622bdf/virtualenv-16.7.4.tar.gz
Already downloaded: /Users/casey/Library/Caches/Homebrew/downloads/507d2087bf24df82641b681a5fe4da778ef50ebe819a86ea1b584f70788f0f63--virtualenv-16.7.4.tar.gz
==> python3 -c import setuptools... --no-user-cfg install --prefix=/private/tmp/liquidctl--homebrew-virtualenv-20191031-14801-1vq6g20/target --install-scripts=/private/tmp/liquidctl--homebrew-vir
==> python3 -s /private/tmp/liquidctl--homebrew-virtualenv-20191031-14801-1vq6g20/target/bin/virtualenv -p python3 /usr/local/Cellar/liquidctl/1.2.0/libexec
==> Downloading https://files.pythonhosted.org/packages/48/69/d87c60746b393309ca30761f8e2b49473d43450b150cb08f3c6df5c11be5/appdirs-1.4.3.tar.gz
Already downloaded: /Users/casey/Library/Caches/Homebrew/downloads/ed8751b12aab49b1efdadc6d550c8953ab56e5d254247350f1f99135833be5b0--appdirs-1.4.3.tar.gz
==> /usr/local/Cellar/liquidctl/1.2.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/liquidctl--appdirs-20191031-14801-1y67h2f/appdirs-1.4.3
==> Downloading https://files.pythonhosted.org/packages/a2/55/8f8cab2afd404cf578136ef2cc5dfb50baa1761b68c9da1fb1e4eed343c9/docopt-0.6.2.tar.gz
Already downloaded: /Users/casey/Library/Caches/Homebrew/downloads/cae543cf1748375c2e56c2a4984cfa49a5858a0a55e6b55a819dfe969ce443d9--docopt-0.6.2.tar.gz
==> /usr/local/Cellar/liquidctl/1.2.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/liquidctl--docopt-20191031-14801-1gj723u/docopt-0.6.2
==> Downloading https://files.pythonhosted.org/packages/c1/86/89df0e8890f96eeb5fb68d4ccb14cb38e2c2d2cfd7601ba972206acd9015/hidapi-0.7.99.post21.tar.gz
Already downloaded: /Users/casey/Library/Caches/Homebrew/downloads/1821f3108e754e54378ae15679b4d0d09805ad48d87e08bc60ed86388cc83ef5--hidapi-0.7.99.post21.tar.gz
==> /usr/local/Cellar/liquidctl/1.2.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/liquidctl--hidapi-20191031-14801-rfrjnm/hidapi-0.7.99.post21
==> Downloading https://files.pythonhosted.org/packages/5f/34/2095e821c01225377dda4ebdbd53d8316d6abb243c9bee43d3888fa91dd6/pyusb-1.0.2.tar.gz
Already downloaded: /Users/casey/Library/Caches/Homebrew/downloads/bed999b3be90efb898cd6c543e0e7eb329f71fa7626edd10cb7bdad98a7a7865--pyusb-1.0.2.tar.gz
==> /usr/local/Cellar/liquidctl/1.2.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/liquidctl--pyusb-20191031-14801-nkhyuo/pyusb-1.0.2
==> /usr/local/Cellar/liquidctl/1.2.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/liquidctl-20191031-14801-1csw3cn/liquidctl-1.2.0
🍺  /usr/local/Cellar/liquidctl/1.2.0: 1,090 files, 11.4MB, built in 38 seconds
casey's-MacBook-Air:Formula casey$ which liquidctl
/usr/local/bin/liquidctl
casey's-MacBook-Air:Formula casey$ liquidctl --debug list
[DEBUG] liquidctl.cli: liquidctl v1.2.0
```

jonasmalacofilho, 2019-10-31:
Well, it certainly works : )

Can you cat `/usr/local/bin/liquidctl` for me?

CaseySJ, 2019-10-31:
My NZXT system is on a different machine, so I'll test this Formula on it later this evening. Would be nice if Linux users can also run some tests.

CaseySJ, 2019-10-31:
> Well, it certainly works : )
> 
> Can you cat `/usr/local/bin/liquidctl` for me?
Sure...
```
casey's-MacBook-Air:Formula casey$ cat /usr/local/bin/liquidctl 
#!/usr/local/Cellar/liquidctl/1.2.0/libexec/bin/python3.7
# EASY-INSTALL-ENTRY-SCRIPT: 'liquidctl==1.2.0','console_scripts','liquidctl'
__requires__ = 'liquidctl==1.2.0'
import re
import sys
from pkg_resources import load_entry_point

if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
    sys.exit(
        load_entry_point('liquidctl==1.2.0', 'console_scripts', 'liquidctl')()
    )
```

jonasmalacofilho, 2019-10-31:
> ```
> #!/usr/local/Cellar/liquidctl/1.2.0/libexec/bin/python3.7
> ```

Ok, that looks good.  It means that it's probably using the `setuptools` in that environment (which was automagically installed).

> Would be nice if Linux users can also run some tests.

I can do that!

CaseySJ, 2019-10-31:
It seems that the `test` block does not seem to work. I've intentionally introduced errors, but the `assert` statements don't seem to have any effect. 

In any case, here's the most up to date version of the Formula (file attached: `liquidctl.py.zip`):

```Ruby
class Liquidctl < Formula
        include Language::Python::Virtualenv

        desc "Cross-platform CLI and Python drivers for AIO liquid coolers and other devices"
        homepage "https://github.com/jonasmalacofilho/liquidctl"
        url "https://files.pythonhosted.org/packages/f2/08/7e56efece1ed5083e82826a1734f6e3c452401602ca211448c4b2c9e4cc9/liquidctl-1.2.0.tar.gz"
        sha256 "ad8c03c0695620fedaec11e7a8286bb5d4da18ba0c71e55888bfa06f8f7d7529"

        depends_on "libusb"
        depends_on "python"

        resource "appdirs" do
                url "https://files.pythonhosted.org/packages/48/69/d87c60746b393309ca30761f8e2b49473d43450b150cb08f3c6df5c11be5/appdirs-1.4.3.tar.gz"
                sha256 "9e5896d1372858f8dd3344faf4e5014d21849c756c8d5701f78f8a103b372d92"
        end

        resource "docopt" do
                url "https://files.pythonhosted.org/packages/a2/55/8f8cab2afd404cf578136ef2cc5dfb50baa1761b68c9da1fb1e4eed343c9/docopt-0.6.2.tar.gz"
                sha256 "49b3a825280bd66b3aa83585ef59c4a8c82f2c8a522dbe754a8bc8d08c85c491"
        end

        resource "hidapi" do
                url "https://files.pythonhosted.org/packages/c1/86/89df0e8890f96eeb5fb68d4ccb14cb38e2c2d2cfd7601ba972206acd9015/hidapi-0.7.99.post21.tar.gz"
                sha256 "e0be1aa6566979266a8fc845ab0e18613f4918cf2c977fe67050f5dc7e2a9a97"
        end

        resource "pyusb" do
                url "https://files.pythonhosted.org/packages/5f/34/2095e821c01225377dda4ebdbd53d8316d6abb243c9bee43d3888fa91dd6/pyusb-1.0.2.tar.gz"
                sha256 "4e9b72cc4a4205ca64fbf1f3fff39a335512166c151ad103e55c8223ac147362"
        end

        def install
                virtualenv_install_with_resources
        end

        test do
                assert_predicate bin/"liquidctl", :exist?
                assert_predicate bin/"liquidctl", :executable?

                version_output = shell_output("#{bin}/liquidctl --version 2>&1", 1)
                assert_match "liquidctl #{version}", version_output
        end
end
```
[liquidctl.rb.zip](https://github.com/jonasmalacofilho/liquidctl/files/3796080/liquidctl.rb.zip)

jonasmalacofilho, 2019-10-31:
After a lot of pain, but I was able to install the formula on Linux.

I think most issues were simply caused by me using linuxbrew with conservative privileges.

 - the formula for libusb requires systemd, which is huge (and in my machine failed to build)
 - cython-hidapi using the default superenv couldn't find libusb.h

But I was able to hack my way around those issues and then the formula seemed to work fine.

 ---

For the tests:

 - my bad, on 1.2.0 we can't yet run `liquidctl list --verbose --debug` without privileges
 - `shell_output(..., <errcode>)` should be enough, as we can't assume the user has or hasn't any particular device

So in my tests I ended up with:

```
        test do
                shell_output("#{bin}/liquidctl list", 0)
        end
```

I also don't think the exist/executable assertions on `bin/liquidctl` are needed, since that's setuptools's responsibility and already implicitly tested by any later test call to liquidctl.

jonasmalacofilho, 2019-11-01:
Stuff to add or change that's blocked by (or at least waiting) a new release:

 - [ ] run doctests (these are probably more suited to run at build time, not later)
 - [ ] patch² and install the man page
 - [ ] test with `shell_output("#{bin}/liquidctl list --verbose --debug", 0)`¹

---

¹ Yes, `--verbose` is redundant when `--debug` is passed.  But it shows what we _want_ to test.

² It's necessary to apply a patch that replaces  [`/run/liquidctl/*`](https://github.com/jonasmalacofilho/liquidctl/blob/master/liquidctl.8#L155-L159) with the correct path for macOS.  That can be figured out by executing
```
appdirs.site_data_dir('liquidctl', 'jonasmalacofilho')
```
in a `python3` REPL (on macOS, of course).

CaseySJ, 2019-11-01:
The result of the `appdirs` command is: `/Library/Application Support/liquidctl`
```
Caseys-iMac-Pro:driver casey_sj$ python3
Python 3.7.4 (v3.7.4:e09359112e, Jul  8 2019, 14:54:52) 
[Clang 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import appdirs
>>> appdirs.site_data_dir('liquidctl', 'jonasmalacofilho')
'/Library/Application Support/liquidctl'
>>> quit()
```

jonasmalacofilho, 2019-11-01:
By the way, there is actually an issue with the Linux formula: cython-hidapi has hardcoded include paths on Linux and BSD (but not macOS).  Working on fixing this now...

CaseySJ, 2019-11-01:
Just did a `pip3 uninstall liquidctl` on my Hackintosh containing NZXT devices, and verified that all traces of the previous installation were gone. Then I copied `liquidctl.rb` to the Home-brew Formula folder and issued `brew install liquidctl`. 

It first informed me of file permission issues on a man page, which was easy to fix with the suggestion provided in the error message.

Repeating the `brew install liquidctl` command then proceeded normally. It downloaded Python sources and compiled them. Although I already have Python 3 installed from the python **pkg** installer for Mac (not through Homebrew), another copy was installed.

Homebrew went on to install other prerequisites and then finally it began the `liquidctl` installation, which was done successfully. I was immediately able to run the application. This is the standard 1.2.0, so I can only control my Kraken, but this was a more exhaustive test of `brew`.

The attached zip file contains the entire log.
[liquidctl brew install log.txt.zip](https://github.com/jonasmalacofilho/liquidctl/files/3796391/liquidctl.brew.install.log.txt.zip)

jonasmalacofilho, 2019-11-01:
I made a tap from your recipe and also figured out how to patch cython-hidapi in the recipe to fix the build issue on Linux/BSD.

```
$ brew tap jonasmalacofilho/liquidctl
$ brew install liquidctl
```

The repository for the tap is: [jonasmalacofilho/homebrew-liquidctl](https://github.com/jonasmalacofilho/homebrew-liquidctl).

---

P.S. I took the liberty to commit on your behalf when adding the initial recipe.

CaseySJ, 2019-11-01:
Very nice!

jonasmalacofilho, 2019-11-01:
> The output from `appdirs` command is: `/Library/Application Support/liquidctl`

Thanks.  I'll take a look at the updates for v1.3.0 in the next few days.

jonasmalacofilho, 2019-11-03:
I added CI testing to [jonasmalacofilho/homebrew-liquidctl](https://github.com/jonasmalacofilho/homebrew-liquidctl), with both homebrew/macOS and linuxbrew, as well as updated the formula to support `--HEAD` (latest 'master') and `--devel` (currently v1.3.0rc1).

---

@CaseySJ do you think that liquidctl would fit in homebrew-core?

CaseySJ, 2019-11-03:
> @CaseySJ do you think that liquidctl would fit in homebrew-core?

Yes I think `liquidctl` should be submitted for inclusion in the Homebrew Core tap. Do you still need a third party (such as myself) to submit it?

jonasmalacofilho, 2019-11-03:
Ok.

I can open the draft pull request to start the discussion but yes, having the support of actual macOS users will probably be important.

On the technical side I think the remaining issues are:
 - patching cython-hidapi (that can already be skipped on macOS, but the best it to fix upstream)
 - conditionals on head/devel (unnecessary after 1.3.0 final, thus the draft pull request for now)
 - non-obvious customization of `liquidctl --version` (will add some comments explaining that)

jonasmalacofilho, 2019-11-04:
@CaseySJ 

Actually I just saw this in a [reply to a recent pull request](https://github.com/Homebrew/homebrew-core/pull/46052#issuecomment-548687388):

> You're also submitting your own software which isn't really the way this should work. Pull requests should be made by happy users of the project

So I think it will be best if you open the pull request.  You can use this already [cleaned up formula for homebrew-core](https://github.com/jonasmalacofilho/homebrew-liquidctl/blob/e55d858bb26902df01fcd1d67266fd6dc37dd131/Formula/liquidctl.rb) (e.g. no conditionals for Linux).

It [passes](https://github.com/jonasmalacofilho/homebrew-liquidctl/commit/e55d858bb26902df01fcd1d67266fd6dc37dd131/checks?check_suite_id=294317672) `brew audit --new-formula` on the CI I've setup, but you should probably retest that on your mac, while going through the rest of the checklist for homebrew-core.

CaseySJ, 2019-11-04:
@jonasmalacofilho 

I should be able to test the latest formula this evening, after which I can open a pull request for homebrew-core. I'll provide some justification as well.

CaseySJ, 2019-11-05:
@jonasmalacofilho 

The homebrew Formula for liquidctl works very well. Tested it last night on a different Mac -- that did not have Homebrew previously installed. When installing Homebrew, Xcode Command Line Tools are automatically installed.

I'll issue a pull request for homebrew-core. Having your own tap for liquidctl is not a bad idea if the Homebrew team ends up rejecting my pull request. For fellow Mac users, the one extra command to set up the tap is trivial. Anyway, let me issue the request and see what happens.

jonasmalacofilho, 2019-11-06:
@CaseySJ 

I'm glad to hear that.

Let me known when you open the pull request.  Regardless if it's integrated or not, we can still maintain the custom tap to test the more bleeding edge features and even the packaging process itself.

By the way, would you accept to be the maintainer of that tap?

CaseySJ, 2019-11-06:
@jonasmalacofilho 

I may be able to issue the pull request this evening. Will certainly let you know.

Regarding the question of being a maintainer of the tap, what responsibilities would that entail?

CaseySJ, 2019-11-07:
@jonasmalacofilho 

I've just submitted the [pull request](https://github.com/Homebrew/homebrew-core/pull/46442). Regarding maintainer responsibilities, I just found that information in the `brew.sh` site. Let me read it over more carefully...

jonasmalacofilho, 2019-11-07:
@CaseySJ 

That's awesome, thanks!

By the way, I think you forgot to use the cleaned up version of the formula I linked in one of my previous commits.  It removes the `OS.mac?` conditionals and the patching of cython-hidapi (which is not need on macOS), both of which are changes that will likely be requested to you.

[diff master..clean-up-for-homebrew-core](https://github.com/jonasmalacofilho/homebrew-liquidctl/compare/master..clean-up-for-homebrew-core?expand=1)

CaseySJ, 2019-11-07:
Alas, that's what happens during weekdays as I'm always pressed for time! Okay, will give them the cleaned-up version once they respond!

jonasmalacofilho, 2019-11-07:
Regarding maintaining the custom tap,

> what responsibilities would that entail?

on my part, I would mostly expect you to respond to issues, keep it and the dependencies up-do-date, and periodically check for issues (and fix them, if possible).

Honestly, some of this commitment you're already showing here (like following up on the SDV2 driver)... thanks, by the way!

> I just found that information in the brew.sh site. Let me read it over more carefully...

Can you point me to that document?

CaseySJ, 2019-11-07:
> Can you point me to that document?

There's `Maintainers` section [here](https://docs.brew.sh).
[image]

jonasmalacofilho, 2019-11-07:
Oh, I think that's more for homebrew-core maintainers.  And [How to Create and Maintain a Tap](https://docs.brew.sh/How-to-Create-and-Maintain-a-Tap#maintaining-a-tap) doesn't list anything special for us to do.

CaseySJ, 2019-11-07:
Please see [the failures](https://jenkins.brew.sh/job/Homebrew%20Core%20Pull%20Requests/51369/) reported by the homebrew automated checker. One of the issues is to avoid OS.Mac, which has been done. But the other issue to to avoid the `devel` spec.

jonasmalacofilho, 2019-11-07:
Right, only `head` is allowed on homebrew-core.

At first I didn't understood why, but I guess it makes sense: `devel` can give a lot of headache to users and maintainers since it looks like a normal version, but may require a lot of updates.  Alternatively, `head` obviously comes with no assurances.

---

Removing the devel block is obviously required, as well as the `|| build.devel?` condition.

Additionally, I don't think the conditionals for man page and test are necessary if 1.3.0rc1 wont be available through that channel.

I think it's better to just support 1.2.0 for now (keeping `head`, but not doing anything special for it), and in less than two weeks we can update to 1.3.0 and add back the man page and tests with verbose&debug without those conditionals.

CaseySJ, 2019-11-08:
I've download the latest change from `clean-up-for-homebrew-core` and getting ready to update the pull request. However, when I run `brew test liquidctl` I get the following (on macOS):
```
$ brew test liquidctl
Testing liquidctl
==> /usr/local/Cellar/liquidctl/1.2.0/bin/liquidctl list
```
But if I run the actual operation manually, I get this (which is correct):
```
$ liquidctl list
Device 0, NZXT Kraken X (X42, X52, X62 or X72)
```
Is this still okay or should `shell_output` pipe the full output to the console?

CaseySJ, 2019-11-08:
After committing the change to my GitHub branch, `homebrew-core` automatically processed it and this time, [All checks have passed](https://github.com/Homebrew/homebrew-core/pull/46442)!

jonasmalacofilho, 2019-11-08:
@CaseySJ 

> Is this still okay or should shell_output pipe the full output to the console?

I hadn't considered that, and it seems that only stderr is automatically redirected to the terminal.

I think it's fine for now, since without `--debug` the output isn't that useful anyway.  The most important check is that `list` doesn't exit with a non-zero status.

Once we revisit this for 1.3.0 and re-enable `--debug`, then it might make sense to forward stdout as well, just so users aren't surprised.

And even then I doubt many users run `brew test` on core formulas.

---

> After committing the change to my GitHub branch, homebrew-core automatically processed it, and this time All checks have passed!

Good job, and thanks again!

:rocket: 

By the way, I'm giving you write access to the jonasmalacofilho tap...

Assuming liquidctl (stable) will soon be in homebrew-core, the ideia is to keep the custom formula with all features (e.g. including devel releases and experimental stuff) on the 'master' branch.

From it other branches can be created when slimmed down versions are required prior to submission of updates to homebrew-core or linuxbrew-core.

Oh, and automated tests run on macOS and Linux for every push or pull request, and also once a week.

jonasmalacofilho, 2019-11-17:
@CaseySJ 

I just released 1.3.0 and updated our tap.

I also updated my reference [cleaned-up version for homebrew-core](https://github.com/jonasmalacofilho/homebrew-liquidctl/blob/clean-up-for-homebrew-core/Formula/liquidctl.rb), which now also includes the man page and a slightly better test.

Can you update the PR formula (and the PR title as well)?

CaseySJ, 2019-11-18:
@jonasmalacofilho 

I've submitted the 1.3.0 commit to the same pull request, located [here](https://github.com/Homebrew/homebrew-core/pull/46442). It's still being 'built', though it's been over 5 minutes...

jonasmalacofilho, 2019-11-18:
@CaseySJ 

Thank you!  And it's done building now, and all looks well.

jonasmalacofilho, 2019-11-18:
Hm, I'm going to close this issue: we have the custom tap and an up-to-date pull request to homebrew-core...

Thanks again!

closed 2019-11-18 by jonasmalacofilho