liquidctl/liquidctl #65: Device identifiers on subsequent calls on macOS

jonasmalacofilho, 2019-11-01:
The listing order changes on subsequent calls on macOS, making the device ID unusable.

```
$ liquidctl list
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)
Device ID 1: NZXT Smart Device V2 (experimental)

$ liquidctl list
Device ID 0: NZXT Smart Device V2 (experimental)
Device ID 1: NZXT Kraken X (X42, X52, X62 or X72)
```

<details><summary>With `--debug`</summary>

```
$ liquidctl --debug list
[DEBUG] liquidctl.cli: running liquidctl v1.2.0
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceDriverV2])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x256
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x256
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2006
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0x82d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x256
Device ID 0: NZXT Smart Device V2 (experimental)
├── Vendor ID: 0x1e71
├── Product ID: 0x2006
├── Release number: 0x0200
├── Serial number: 00000000001A
├── Bus: hid
├── Address: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/BXBR@1,2/IOPP/BYUP@0/IOPP/BYD8@8/IOPP/XHC0@0,3/XHC0@61000000/PRT6@61500000/USB2.0 Hub@61500000/AppleUSB20Hub@61500000/AppleUSB20HubPort@61510000/NZXT USB Device@61510000/IOUSBHostInterface@0/IOUSBHostHIDDevice@61510000,0
└── Driver: SmartDeviceDriverV2 using module hid
[DEBUG] liquidctl.cli: hierarchy: CommonSmartDeviceDriver, UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice

Device ID 1: NZXT Kraken X (X42, X52, X62 or X72)
├── Vendor ID: 0x1e71
├── Product ID: 0x170e
├── Release number: 0x0200
├── Serial number: 6D73078C495
├── Bus: hid
├── Address: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/BXBR@1,2/IOPP/BYUP@0/IOPP/BYD8@8/IOPP/XHC0@0,3/XHC0@61000000/PRT6@61500000/USB2.0 Hub@61500000/AppleUSB20Hub@61500000/AppleUSB20HubPort@61540000/NZXT USB Device@61540000/IOUSBHostInterface@0/IOUSBHostHIDDevice@61540000,0
└── Driver: KrakenTwoDriver using module hid
[DEBUG] liquidctl.cli: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice

$ liquidctl --debug list
[DEBUG] liquidctl.cli: running liquidctl v1.2.0
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceDriverV2])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x256
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x256
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2006
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0x82d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5ac:0x256
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)
├── Vendor ID: 0x1e71
├── Product ID: 0x170e
├── Release number: 0x0200
├── Serial number: 6D73078C495
├── Bus: hid
├── Address: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/BXBR@1,2/IOPP/BYUP@0/IOPP/BYD8@8/IOPP/XHC0@0,3/XHC0@61000000/PRT6@61500000/USB2.0 Hub@61500000/AppleUSB20Hub@61500000/AppleUSB20HubPort@61540000/NZXT USB Device@61540000/IOUSBHostInterface@0/IOUSBHostHIDDevice@61540000,0
└── Driver: KrakenTwoDriver using module hid
[DEBUG] liquidctl.cli: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice

Device ID 1: NZXT Smart Device V2 (experimental)
├── Vendor ID: 0x1e71
├── Product ID: 0x2006
├── Release number: 0x0200
├── Serial number: 00000000001A
├── Bus: hid
├── Address: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/BXBR@1,2/IOPP/BYUP@0/IOPP/BYD8@8/IOPP/XHC0@0,3/XHC0@61000000/PRT6@61500000/USB2.0 Hub@61500000/AppleUSB20Hub@61500000/AppleUSB20HubPort@61510000/NZXT USB Device@61510000/IOUSBHostInterface@0/IOUSBHostHIDDevice@61510000,0
└── Driver: SmartDeviceDriverV2 using module hid
[DEBUG] liquidctl.cli: hierarchy: CommonSmartDeviceDriver, UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice
```
</details>

This appears to be a regression, and is probably related to the new implementation of device discovery.

Reported by: @CaseySJ (in [#64/comment](https://github.com/jonasmalacofilho/liquidctl/pull/64#issuecomment-548764666))

jonasmalacofilho, 2019-11-01:
@CaseySJ can you please run the following test script on any (or all) of your macs?

```python
import sys

if sys.platform.startswith('linux'):
    try:
        import hidraw
        hid = hidraw
    except:
        pass
else:
    import hid

def check_iters(iters):
    if not iters:
        raise ValueError('Not defined for iters == []')
    return iters[1:] == iters[:-1]

def print_hid_device_line(info):
    vid = info['vendor_id']
    pid = info['product_id']
    path = info['path'].decode()
    print(f'hid {vid:04x}:{pid:04x} on {path}')

if __name__ == "__main__":
    errors = 0
    args = sys.argv[1:]
    niters = int(args[0]) if args else 3
    print(f'Iteration count is {niters}')
    print()

    print('Checking: enumerate')
    iters = []
    for i in range(niters):
        print(f'iteration: {i}')
        order = []
        for info in hid.enumerate():
            print_hid_device_line(info)
            order.append(info['path'])
        iters.append(order)
    if check_iters(iters):
        print('passed')
    else:
        errors += 1
        print('FAILED')
    print()

    nhids = len(iters[0])
    print(f'HID count is {nhids}')
    print()

    print(f'Checking: enumerate, break after {nhids//2} HIDs')
    iters = []
    for i in range(niters):
        print(f'iteration: {i}')
        order = []
        cnt = 0
        for info in hid.enumerate():
            print_hid_device_line(info)
            order.append(info['path'])
            cnt += 1
            if cnt == nhids // 2:
                break
        iters.append(order)
    if check_iters(iters):
        print('passed')
    else:
        errors += 1
        print('FAILED')
    print()

    print(f'Checking: enumerate and open/close, break after {nhids//2} HIDs')
    iters = []
    for i in range(niters):
        print(f'iteration: {i}')
        order = []
        cnt = 0
        for info in hid.enumerate():
            print_hid_device_line(info)
            path = info['path']
            dev = hid.device()
            dev.open_path(path)
            dev.close()
            order.append(path)
            cnt += 1
            if cnt == nhids // 2:
                break
        iters.append(order)
    if check_iters(iters):
        print('passed')
    else:
        errors += 1
        print('FAILED')
    print()

    errors += 1

    if not errors:
        print('all ok')
    else:
        print('SOME ERRORS WHERE FOUND')
        sys.exit(1)
```

CaseySJ, 2019-11-01:
Sure; I can run it shortly on my MacBook Air (no NZXT devices) and later this evening on my Hackintosh with two NZXT controllers.

CaseySJ, 2019-11-01:
Attached is the full output from `check-hid-enumerate-order.py` on my 2012 MacBook Air.

Last line reads `SOME ERRORS WERE FOUND`

[check-hid-enumerate.txt](https://github.com/jonasmalacofilho/liquidctl/files/3799482/check-hid-enumerate.txt)

jonasmalacofilho, 2019-11-01:
Thanks!

So on macOS `hid.enumerate()` returns the devices in random/unstable order.  I'll take a look to see if this is something I can fix in hidapi, otherwise I'll simple sort them in liquidctl.

---

In the mean time a workaround is to only use `--vendor`, `--product` and other stable filter options.

jonasmalacofilho, 2019-11-01:
Closing the issue as it should already be fixed in master with 31404de.

When you have a chance please give it a try with a few successive calls to `liquidctl list`.

closed 2019-11-01 by jonasmalacofilho

CaseySJ, 2019-11-02:
Now it works consistently:

```
Caseys-iMac-Pro:driver casey_sj$ liquidctl list
Device ID 0: NZXT Smart Device V2 (experimental)
Device ID 1: NZXT Kraken X (X42, X52, X62 or X72)

Caseys-iMac-Pro:driver casey_sj$ liquidctl list
Device ID 0: NZXT Smart Device V2 (experimental)
Device ID 1: NZXT Kraken X (X42, X52, X62 or X72)

Caseys-iMac-Pro:driver casey_sj$ liquidctl -d 0 initialize

Caseys-iMac-Pro:driver casey_sj$ liquidctl -d 0 status
NZXT Smart Device V2 (experimental)
├── Fan 2 duty                              50  %
├── Fan 2 speed                           1020  rpm
├── Firmware version                     1.5.0  
├── LED 1 accessory 1          HUE 2 LED Strip  
├── LED 1 accessory 2    HUE 2 Underglow 200mm  
├── LED 1 accessory 3    HUE 2 Underglow 200mm  
├── LED 2 accessory 1          AER RGB 2 140mm  
├── LED 2 accessory 2          AER RGB 2 140mm  
├── LED 2 accessory 3          AER RGB 2 140mm  
├── LED 2 accessory 4          AER RGB 2 120mm  
└── Noise level                             66  dB

Caseys-iMac-Pro:driver casey_sj$ liquidctl -d 0 status
NZXT Smart Device V2 (experimental)
├── Fan 2 duty                              50  %
├── Fan 2 speed                           1045  rpm
├── Firmware version                     1.5.0  
├── LED 1 accessory 1          HUE 2 LED Strip  
├── LED 1 accessory 2    HUE 2 Underglow 200mm  
├── LED 1 accessory 3    HUE 2 Underglow 200mm  
├── LED 2 accessory 1          AER RGB 2 140mm  
├── LED 2 accessory 2          AER RGB 2 140mm  
├── LED 2 accessory 3          AER RGB 2 140mm  
├── LED 2 accessory 4          AER RGB 2 120mm  
└── Noise level                             68  dB
```