liquidctl/liquidctl #68: Support for NZXT Hue 2 Ambient V2

keton, 2019-11-09:
I have NZXT Hue 2 Ambient kit V2. 
I played around with code from #61 and #64 and it's very straightforward to add my device to liquidctl. 

Protocol is the same as for HUE2. The only difference seems to be in USB id, it's 1e71:2002 instead of 1e71:2001 for HUE2. 

So if you add code below to https://github.com/jonasmalacofilho/liquidctl/blob/add-support-for-hue2/liquidctl/driver/nzxt_smart_device.py#L320 it just works at least for basic LED color settings.

```python
       (0x1e71, 0x2002, None, 'NZXT HUE 2 Ambient (experimental)', {
            'speed_channel_count': 0,
            'color_channel_count': 2
        }),
```

I don't want to open pull request to ongoing pull request so I'm posting it here. I can help with testing experimental code on Windows and Linux.

jonasmalacofilho, 2019-11-09:
That's great, thanks!

Do you [also](https://github.com/jonasmalacofilho/liquidctl/issues/61#issuecomment-550560096) see the LEDs blinking (or the animations breaking in other ways) when executing `liquidctl status`?

Can I also ask you to test [rebalance-sdv2-initialize-status](https://github.com/jonasmalacofilho/liquidctl/tree/rebalance-sdv2-initialize-status)?  It's supposed to fix the issue above, assuming you can reproduce it.

But if you can't, I would still like to know if that code works, as it could be a general improvement to the entire SD/HUE 2 family.

jonasmalacofilho, 2019-11-09:
I'm also guessing `status` doesn't yet know about the Ambient accessories.

By analyzing the output with `--debug` infos you may be able to figure out what needs to be added to `_ACCESSORY_NAMES`.

keton, 2019-11-09:
[rebalance-sdv2-initialize-status](https://github.com/jonasmalacofilho/liquidctl/tree/rebalance-sdv2-initialize-status) doesn't blink LEDs bui to doesn't output anything either. LED color setting works ok.

On add-support-for-hue2 branch shows 3 out of 4 LED strips connected

Debug output (add-support-for-hue2 branch) is as follows:
```
[DEBUG] liquidctl.cli: running liquidctl v1.2.0 (ea4d75fd2b1e)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xe0f:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 Ambient (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xe0f:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xe0f:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x1
[DEBUG] liquidctl.cli: device: NZXT HUE 2 Ambient (experimental)
NZXT HUE 2 Ambient (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 51 ff 71 06 78 71 51 54 09 44 14 67 02 20 01 02 06 00 31 33 39 39 31 39 31 31 32 30 30 32 33 35 37 37 ff ff ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 51 ff 71 06 78 71 51 54 09 44 14 67 02 04 05 05 00 00 00 04 04 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
├── Firmware version               2.6.0  
├── LED 1 accessory 1    HUE 2 LED Strip  
├── LED 1 accessory 2            Unknown  
├── LED 1 accessory 3            Unknown  
├── LED 2 accessory 1    HUE 2 LED Strip  
├── LED 2 accessory 2    HUE 2 LED Strip  
└── LED 2 accessory 3            Unknown  

```

keton, 2019-11-09:
I've noticed that on rebalance-sdv2-initialize-status branch `liquidctl --debug initialize` shows accessory status, output below

```
root@ubuntu:~# liquidctl --debug initialize
[DEBUG] liquidctl.cli: running liquidctl v1.3.0rc1 (2538baec0bfc)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xe0f:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Hue 2 Ambient (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2002
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xe0f:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xe0f:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x1
[DEBUG] liquidctl.cli: device: NZXT Hue 2 Ambient (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 60 02 01 e8 03 01 e8 03 (and 56 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 60 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 51 ff 71 06 78 71 51 54 09 44 14 67 02 20 01 02 06 00 31 33 39 39 31 39 31 31 32 30 30 32 33 35 37 37 ff ff ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 51 ff 71 06 78 71 51 54 09 44 14 67 02 04 05 05 00 00 00 04 04 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
NZXT Hue 2 Ambient (experimental)
├── Firmware version               2.6.0  
├── LED 1 accessory 1    HUE 2 LED Strip  
├── LED 1 accessory 2            Unknown  
├── LED 1 accessory 3            Unknown  
├── LED 2 accessory 1    HUE 2 LED Strip  
├── LED 2 accessory 2    HUE 2 LED Strip  
└── LED 2 accessory 3            Unknown  

```

CaseySJ, 2019-11-09:
This shows that `HUE 2 Ambient` has the ID `0x05`, which means we can modify `_ACCESSORY_NAMES` as follows:
```Python
    _ACCESSORY_NAMES = {
        0x04: "HUE 2 LED Strip",
        0x05: "HUE 2 Ambient"
        0x08: "HUE 2 Cable Comb",
        0x0a: "HUE 2 Underglow 200mm",
        0x0b: "AER RGB 2 120mm",
        0x0c: "AER RGB 2 140mm"
    }
```

keton, 2019-11-09:
I have only LED strips. There are 2 types of LED strip in this kit 25 and 30 cm long. It looks like accessories labeled as unknown are long LED strips.

CaseySJ, 2019-11-09:
> @CaseySJ
> 
> I have only LED strips. There are 2 types of LED strip in this kit 25 and 30 cm long. It looks like accessories labeled as unknown are long LED strips.

Thanks, that's very helpful. I actually have the 30cm LED Strip and its ID is 0x04, so ID 0x05 must be for your 25cm LED Strip. Hence, we can change `_ACCESSORY_NAMES`  from this:
```Python
    _ACCESSORY_NAMES = {
        0x04: "HUE 2 LED Strip",
        0x05: "HUE 2 Ambient"
        0x08: "HUE 2 Cable Comb",
        0x0a: "HUE 2 Underglow 200mm",
        0x0b: "AER RGB 2 120mm",
        0x0c: "AER RGB 2 140mm"
    }
```

to this:
```Python
    _ACCESSORY_NAMES = {
        0x04: "HUE 2 LED Strip 300mm",
        0x05: "HUE 2 LED Strip 250mm"
        0x08: "HUE 2 Cable Comb",
        0x0a: "HUE 2 Underglow 200mm",
        0x0b: "AER RGB 2 120mm",
        0x0c: "AER RGB 2 140mm"
    }
```

jonasmalacofilho, 2019-11-10:
Thank you both for the additional infos.

---

@keton 

> On add-support-for-hue2 branch shows 3 out of 4 LED strips connected

> I have only LED strips. There are 2 types of LED strip in this kit 25 and 30 cm long. It looks like accessories labeled as unknown are long LED strips.

We're reading 6 accessories in total, but it seems you only have 4.

Can you confirm the accessories connected to each channel, and in which order?

keton, 2019-11-10:
@jonasmalacofilho 

I'm sorry writing that I was mistaken about what channel and accessory mean. I have 2 channels connected with 3 accessories on each channel. All accessories are LED strips.

jonasmalacofilho, 2019-11-10:
@keton 

Oh, that's good to hear.

So the following output for your system would look right to you?

```
NZXT HUE 2 Ambient (experimental)
├── Firmware version                     2.6.0  
├── LED 1 accessory 1    HUE 2 LED Strip 300mm  
├── LED 1 accessory 2    HUE 2 LED Strip 250mm  
├── LED 1 accessory 3    HUE 2 LED Strip 250mm  
├── LED 2 accessory 1    HUE 2 LED Strip 300mm  
├── LED 2 accessory 2    HUE 2 LED Strip 300mm  
└── LED 2 accessory 3    HUE 2 LED Strip 250mm  
```

keton, 2019-11-10:
@jonasmalacofilho 

Almost but it looks like short and long IDs are swapped. On LED1 channel I have 2 long 1 short strip, on LED2 2 short 1 long. 

To be clear I've measured strips lengths in my kit again and not including connectors (in between gray plastic pieces holding the metal connector pins) short one is about 24,7cm and long strip is about 30,2cm long.

keton, 2019-11-10:
accessory table that shows my setup correctly is as follows:

```python
_ACCESSORY_NAMES = {
        0x04: "HUE 2 LED Strip (short)",
        0x05: "HUE 2 LED Strip  (long)",
        0x08: "HUE 2 Cable Comb",
        0x0a: "HUE 2 Underglow 200mm",
        0x0b: "AER RGB 2 120mm",
        0x0c: "AER RGB 2 140mm"
    }
```

jonasmalacofilho, 2019-11-10:
@keton 

That's weird, because it would suggest a conflict with the rest of the devices in the family, even though everything else is shared.

Can I ask you to double check the strips in each channel, and also, the channel identification itself?  (maybe it's the channel number that we're getting wrong?)

Specifically, I would like you to do the following tests:

 - test #1: connect only longs strips to channel 1, and no strips to channel 2... run `liquidctl initialize --debug`
 - test #2: connect only short strips to channel 2, and no strips to channel 1... run `liquidctl initialize --debug`

You can just send the relevant debug line from each test.

```
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 ...
```

keton, 2019-11-10:
@jonasmalacofilho I'm sorry but LED strips are glued to back of my monitor and disconnecting them is not a reversible operation (adhesive will be ruined in the process). 

I've triple checked and for sure I have 2 long 1 short in one channel and 2 short 1 long in another one. 

On my control box channels are labeled (1 and 2). Liquidctl seems to behave according to those labels. Ergo the setup is as I've described above.

I guess this is a miscommunication problem between us. Most likely you measure your LED strips with connectors which will add about 0.5-1cm to length at least.

According to the [manufacturer](https://www.nzxt.com/products/hue-2-ambient-v2) my kit shipped as follows:

LED Strip Length | Kits include suggested strip configuration for monitor sizes, 34"-35" 26"-32" - 300mm x4, 250mm x4
-- | --

So "long strip" in my kit is designated as 300mm, short as 250mm. There's also 200mm variant in version of the kit for ultrawide so you may be referring to that.

If we take into account that 200mm strip exists wouldn't be logical that they are named as follows:
0x03 - 200mm
0x04 - 250mm
0x05 - 300mm

Honestly I don't care about the naming. If it's more compatible to other users we can label accessory 0x05 as 250mm and 0x04 as 300. For me the most important is that liquidctl just works unlike CAM :)

jonasmalacofilho, 2019-11-10:
@keton 

I'm so sorry, I didn't take into account that reconnecting the strips in arbitrary order is not easy/reversible on an Ambient setup.

> I guess this is a miscommunication problem between us. Most likely you measure your LED strips with connectors which will add about 0.5-1cm to length at least.

When I or @CaseySJ refer to 250 mm vs. 300 mm strips, we're using the standard names, which mostly only reflects the number of LEDs in each strip.

> If we take into account that 200mm strip exists wouldn't be logical that they are named as follows: [...]

Yeah, I thought so too at first.  However the case that @CaseySJ has only comes with 300 mm (or 10 LED strips), so he couldn't have gotten that wrong (for the Smart Device V2).

> For me the most important is that liquidctl just works

Thanks! : )

---

> On my control box channels are labeled (1 and 2). Liquidctl seems to behave according to those labels. Ergo the setup is as I've described above.

Hm, so after

```
liquidctl set led1 color fixed ffffff
liquidctl set led2 color off
```

you end up with 2 long and 1 short strips lighten?

keton, 2019-11-10:
It seems that I missed my morning coffee. You're of course right. My strips are mounted on surface with complicated shape (not flat) that makes measuring difficult.

the correct map is:
```python
_ACCESSORY_NAMES = {
        0x04: "HUE 2 LED Strip 300mm",
        0x05: "HUE 2 LED Strip 250mm",
        0x08: "HUE 2 Cable Comb",
        0x0a: "HUE 2 Underglow 200mm",
        0x0b: "AER RGB 2 120mm",
        0x0c: "AER RGB 2 140mm"
    }
```

report:
```
NZXT HUE 2 Ambient (experimental)
├── Firmware version                     2.6.0
├── LED 1 accessory 1    HUE 2 LED Strip 300mm
├── LED 1 accessory 2    HUE 2 LED Strip 250mm
├── LED 1 accessory 3    HUE 2 LED Strip 250mm
├── LED 2 accessory 1    HUE 2 LED Strip 300mm
├── LED 2 accessory 2    HUE 2 LED Strip 300mm
└── LED 2 accessory 3    HUE 2 LED Strip 250mm
```

Moreover the accessory number matches connection order - accessory 1 is directly connected to the controlled, 2 is in the middle, 3 at the end.

I've checked and rechecked it couple of times and I'm 100% sure that this reflects my setup - 0x04 long strip 0x05 short strip. 

Again sorry for the confusion.

jonasmalacofilho, 2019-11-10:
I'll close this in a sec, just making some fixes to the documentation and updating the man page as well.

jonasmalacofilho, 2019-11-10:
Done in 8ddd80b.

Thanks @keton!

closed 2019-11-10 by jonasmalacofilho

jonasmalacofilho, 2019-11-10:
> I can help with testing experimental code on Windows and Linux.

What Linux distro do you run?

keton, 2019-11-10:
@jonasmalacofilho I have a couple of VMs, Debian 10 and various versions of Ubuntu. So far I've tested on Ubuntu Mate 19.10.

jonasmalacofilho, 2019-11-10:
> I have a couple of VMs, Debian 10 and various versions of Ubuntu.

Any chance you'd be interested in working on #62 ("Package for Debian/Ubuntu")?

> So far I've tested on Ubuntu Mate 19.10.

Good to know, thanks.

keton, 2019-11-10:
> Any chance you'd be interested in working on #62 ("Package for Debian/Ubuntu")?

I'd need to teach myself about how .deb building works. Probably adapting recipe from another python3 module will be a good starting point. 
Problem is that I'll have time no sooner than next weekend. So if that's urgent you may look for someone else. Otherwise it sounds like fun little project.

jonasmalacofilho, 2019-11-10:
It's awesome that you're interested.  Thanks again!

And there's no urgency whatsoever : )