liquidctl/liquidctl #70: 1.3.0 - no longer able to run as a regular user

aerosol, 2019-11-18:
Hey, after upgrading from 1.2.0 to 1.3.0 via AUR I'm unable to run `liquidctl` as a regular user anymore. 

It doesn't look like the udev rules need an update, but maybe they do?

Here are my settings:

# lsusb

```
(yharnam)  ~ $ lsusb | grep NZXT
Bus 001 Device 002: ID 1e71:170e NZXT NZXT USB Device
```

```
(yharnam)  ~ $ lsusb -d 1e71:170e -v

Bus 001 Device 002: ID 1e71:170e NZXT NZXT USB Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x170e
  bcdDevice            2.00
  iManufacturer           1 NZXT.-Inc.
  iProduct                2 NZXT USB Device
  iSerial                 3 49933649343
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     143
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

# udev rules

```
(yharnam)  ~ $ </etc/udev/rules.d/kraken.rules
SUBSYSTEM=="usb", ATTRS{idVendor}=="1e71", ATTRS{idProduct}=="170e", MODE="0666"
```

# liquidctl list

```
(yharnam)  ~ $ liquidctl list -v
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)
├── Vendor ID: 0x1e71
├── Product ID: 0x170e
├── Release number: 0x0200
├── Serial number: 49933649343
├── Bus: hid
├── Address: /dev/hidraw0
└── Driver: KrakenTwoDriver using module hidraw
```

```
(yharnam)  ~ $ liquidctl status
OSError: open failed
```

```
(yharnam)  ~ $ sudo liquidctl status
NZXT Kraken X (X42, X52, X62 or X72)
├── Liquid temperature     28.6  °C
├── Fan speed               938  rpm
├── Pump speed             2826  rpm
└── Firmware version      4.0.2
```

jonasmalacofilho, 2019-11-18:
HID devices on Linux are now normally accessed through hidraw, which is a bit faster and avoids unloading kernel drivers (like the [hwmon](https://github.com/jonasmalacofilho/liquidtux) ones I'm working on).

You can change the rule to apply for all subdevices of the parent USB device:

```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1e71", ATTRS{idProduct}=="170e", MODE="0666"
         ^
```

or add an additional rule for the hidraw device specifically:

```
  SUBSYSTEM=="usb", ATTRS{idVendor}=="1e71", ATTRS{idProduct}=="170e", MODE="0666"
> SUBSYSTEM=="hidraw", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1e71", ATTRS{idProduct}=="170e", MODE="0666"
```

aerosol, 2019-11-19:
Many thanks @jonasmalacofilho that helped!

closed 2019-11-19 by aerosol