liquidctl/liquidctl #71: Smart Device V2 fan status issue

CaseySJ, 2019-12-11:
Hello @jonasmalacofilho 

Line **451** of the _NZXT Smart Device V2_ driver is currently this:
```Python
        self._read_until({b'\x67\x04': parse_fan_info})
```
But this is incorrect. It results in:
```
NZXT Smart Device V2 (experimental)
├── Fan 1 duty         0  %
├── Fan 1 speed    12254  rpm
├── Fan 2 duty         0  %
├── Fan 2 speed    12150  rpm
├── Fan 3 duty        71  %
├── Fan 3 speed    12237  rpm
└── Noise level        0  dB
```
It should be changed to:
```Python
        self._read_until({b'\x67\x02': parse_fan_info})
```
Which results in the correct status:
```
NZXT Smart Device V2 (experimental)
├── Fan 2 duty      20  %
├── Fan 2 speed    626  rpm
└── Noise level     50  dB
```

jonasmalacofilho, 2019-12-11:
Thanks for reporting this.

Already fixed it in the master branch.  Will do a release later today.

closed 2019-12-11 by jonasmalacofilho

jonasmalacofilho, 2019-12-11:
@CaseySJ 

I just released 1.3.2 with this fix.  Can you update our custom tap, as well as the under-review formula for homebrew-core?

CaseySJ, 2019-12-11:
Certainly, will do so this evening. Thanks for the quick turnaround on this!!

commit 1f76c5ca (2019-12-11): Parse fan info from correct status message

The Smart Device V2 regularly sends both b'6702' and b'6704' prefixed
status messages, and the fan (speed) information is in the first one.

Commit 2538baec0bfc accidentally changed the code to wait for (and try
to parse as fan information) the incorrect message; this fixes it.

Fixes: 2538baec0bfc ("Move the reporting of SDV2 firmware and LED infos
to initialize")
Fixes: #71 ("Smart Device V2 fan status issue")
Reported-by: @CaseySJ
Suggested-by: @CaseySJ
commit 0f5188f7 (2020-02-02): Parse fan info from correct status message

commit 1f76c5caef9b323439bde8bd66c56c4ed7a741c1 on 'master'.

The Smart Device V2 regularly sends both b'6702' and b'6704' prefixed
status messages, and the fan (speed) information is in the first one.

Commit 2538baec0bfc accidentally changed the code to wait for (and try
to parse as fan information) the incorrect message; this fixes it.

Fixes: 2538baec0bfc ("Move the reporting of SDV2 firmware and LED infos
to initialize")
Fixes: #71 ("Smart Device V2 fan status issue")
Reported-by: @CaseySJ
Suggested-by: @CaseySJ