liquidctl/liquidctl #72: CPU temperature

sebikolo, 2019-12-19:
How to control pump and fans based on CPU temperature, not Liquid temperature?

jonasmalacofilho, 2019-12-19:
liquidctl only exposes native device features (through a CLI and a set of APIs).

The supported coolers only have a liquid temperature sensor, and there's no way for them to _directly_ access or support CPU temperature sensors.  
But you can use the liquidctl APIs + a suitable sensor API to create a script that adjusts the speeds according to the algorithm of your choice.

There's already an example of this in [`extra/krakencurve-poc`](https://github.com/jonasmalacofilho/liquidctl/blob/master/extra/krakencurve-poc), which supports fan/pump curves based on any sensor available through the Linux hwmon subsystem.

That example can be adapted to support complex algorithms (e.g. including  hysteresis), access to sensors on platforms other than Linux, or the use of a different cooler.

One final note: as this will be software-based fan/pump speed control, the script will need to be kept running as if it was a service.  This differs from the liquidctl CLI, that manipulates hardware-based features and doesn't have to run continuously.

Please let me know if you have any other questions.

sebikolo, 2019-12-19:
Thank you for the quick reply. Unfortunately, I'm not a programmer :-(. I'm looking for a ready solution.

jonasmalacofilho, 2019-12-19:
What OS are you running?

sebikolo, 2019-12-19:
macOS Catalina 10.15.2

jonasmalacofilho, 2019-12-20:
I don't have a mac to work on this, but there should be _some_ way of getting the CPU temperature programmatically.  Maybe through the BSD layer, maybe using a library or CLI program.

If you can figure that out and point me to how it works, I should be able to help.  The example script I linked already is functional, it just lacks access to external sensors on macOS.

---

@CaseySJ do you think you can help us here?

CaseySJ, 2019-12-20:
@jonasmalacofilho 

One of the command line tools we can use is a Ruby app called `iStats`. We can install it via the command `sudo gem install iStats`. Here is some sample CLI output:

```
$ istats --help
 
  - iStats: help ---------------------------------------------------

  istats --help                        This help text
  istats --version                     Print current version

  # Commands
  istats all                           Print all stats
  istats cpu                           Print all CPU stats
  istats cpu [temp | temperature]      Print CPU temperature
  istats fan                           Print all fan stats
  istats fan [speed]                   Print fan speed
  istats battery                       Print all battery stats
  istats battery [health]              Print battery health
  istats battery [time | remain]       Print battery time remaining
  istats battery [cycle_count | cc]    Print battery cycle count info
  istats battery [temp | temperature]  Print battery temperature
  istats battery [charge]              Print battery charge
  istats battery [capacity]            Print battery capacity info

  istats scan                          Scans and print temperatures
  istats scan [key]                    Print single SMC temperature key
  istats scan [zabbix]                 JSON output for Zabbix discovery
  istats enable [key | all]            Enables key
  istats disable [key | all]           Disable key
  istats list                          List available keys

  # Arguments
  --no-graphs                          Don't display sparklines graphs
  --no-labels                          Don't display item names/labels
  --no-scale                           Display just the stat value
  --value-only                         No graph, label, or scale
  -f, --fahrenheit                     Display temperatures in fahrenheit

  for more help see: https://github.com/Chris911/iStats
```

And more interestingly:

```
$istats cpu

CPU temp:               43.63°C     ▁▂▃▅▆
```

If we don't supply any arguments, we get (on my 2012 MacBook Air laptop):
```
$istats

--- CPU Stats ---
CPU temp:               45.63°C     ▁▂▃▅▆▇

--- Fan Stats ---
Total fans in system:   1           
Fan 0 speed:            2001 RPM    ▁▂▃▅▆▇

--- Battery Stats ---
Battery health:         unknown     
Cycle count:            71          ▁▂▃▅▆▇  7.1%
Max cycles:             1000        
Current charge:         5789 mAh    ▁▂▃▅▆▇  100%
Maximum charge:         5789 mAh    ▁▂▃▅▆▇  86.4%
Design capacity:        6700 mAh    
Battery temp:           31.89°C     
```

A macOS script that can regularly monitor CPU temps and adjust fan speeds on Kraken or Smart Device would be wonderful!

jonasmalacofilho, 2019-12-20:
@CaseySJ thanks!

I'll take a look at its source code and see how it does that.

: )

jonasmalacofilho, 2019-12-23:
Quick update: it seems a native extension is needed to read the data from the SMC.

It should be possible to port the extension from iStats to Python, but it's a bit too much work for me right now (besides other work, I'd need to set up a Mac system to test this).

---

Other/simpler alternatives:

 - execute iStats from krakencurve, but that isn't very efficient
 - call iStats methods using something like [rb_call](https://github.com/yohm/rb_call)

jwmann, 2020-01-21:
I'm also here looking for a solution to algorithmically / temp curve control my pump and fans
Also using macOS 10.15+ 

Just adding my +1

jonasmalacofilho, 2020-01-25:
> Maybe this can be useful in some way
> https://github.com/kentwait/loadlog

It is, thanks.  It shows me that executing `iStats` is a viable alternative, and shows how to parse the output (which is important for someone write/testing the code without access to a Mac).

But, err... I don't have time to work on this right now : \

I'll leave the issue open and add it to my queue... maybe some of you can get started?

_(a first step could be to adapt krakencurve-poc to use `iStats` if it's available and the platform is darwin)_

jonasmalacofilho, 2020-01-30:
Can you please test the refreshed [`extra/krakencurve-poc`](https://github.com/jonasmalacofilho/liquidctl/blob/master/extra/krakencurve-poc) (v0.0.2)?

```
$ extra/krakencurve-poc --help
Adjust Kraken X42/X52/X62/X72 speeds dynamically, with software.

Periodicatly adjusts the pump and fan speeds according to user-specified
profiles.

Different sensors can be used for the pump and fans; when no sensor
is specified, the coolant temperatue is used.  Use show-sensors to view the
available sensors.

Usage:
  krakencurve-poc [options] show-sensors
  krakencurve-poc [options] control --pump <profile> --fan <profile>
  krakencurve-poc --help
  krakencurve-poc --version

Options:
  --pump <profile>        Profile to use for the pump
  --fan <profile>         Profile to use for the fan
  --pump-sensor <sensor>  Select alternate sensor for pump speed
  --fan-sensor <sensor>   Select alternate sensor for fan speed
  --interval <seconds>    Update interval in seconds [default: 2]
  -v, --verbose           Output additional information
  -g, --debug             Show debug information on stderr
  --version               Display the version number
  --help                  Show this message

Examples:
  krakencurve-poc show-sensors
  krakencurve-poc control --pump '(25,50),(35,100)' --fan '(25,40),(45,100)'
  krakencurve-poc control --pump '(25,50),(35,100)' --fan '(20,25),(60,100)' --fan-sensor 'istats.cpu'

Requirements:
  all platforms  liquidctl, including the Python APIs (pip install liquidctl)
  Linux/FreeBSD  psutil (pip install psutil)
  macOS          iStats (gem install iStats)
  Windows        none, system sensors not yet supported

Changelog:
  0.0.2  MacOS support for iStats; breaking refresh of the CLI and sensor names
  0.0.1  Initial proof-of-concept; system sensors only supported on Linux
```

jonasmalacofilho, 2020-01-30:
Well, let me know if you find any issues on macOS, as I cant' test that.

But assuming you have `iStats` (and the liquidctl APIs) and that I haven't messed up, something like the following should work:

```
krakencurve-poc control --pump '(25,50),(35,100)' --fan '(20,25),(60,100)' --fan-sensor 'istats.cpu'
```

CaseySJ, 2020-01-30:
Wow, is this what I think it is??!!

Will be happy to test it on MacOS later this evening.

jonasmalacofilho, 2020-01-30:
@CaseySJ I'm also working on generalizing this for the other devices : )

jonasmalacofilho, 2020-02-03:
Hi again!

You might as well skip ahead and test [`extra/yoda`](https://github.com/jonasmalacofilho/liquidctl/blob/master/extra/yoda), which is the beginning of a more refined script for fan/pump dynamic control.  It already supports all liquidctl devices, has some fault tolerance, and uses a low-pass filter to avoid jerks in pump/fan speeds.

Device selection works like in liquidctl,¹ and the script's requirements are the same as of the last krakencurve proof-of-concept: you must have access to the liquidctl APIs² and, on macOS, [iStats](https://github.com/Chris911/iStats); on Linux [psutil](https://github.com/giampaolo/psutil) is used instead.

For usage and options, run `yoda --help`.  Here's an example session with `--verbose` output enabled:

```
$ yoda --match kraken show-sensors
Sensor identifier                                                    Temperature
--------------------------------------------------------------------------------
_internal.liquid                                                          28.0°C
istats.cpu                                                                31.0°C

$ yoda --match kraken control pump with '(20,50),(50,100)' on istats.cpu and fan with '(20,25),(34,100)' on _internal.liquid --verbose
INFO: device: NZXT Kraken X (X42, X52, X62 or X72) on bus hid and address [...]
INFO: channel: pump following profile [(20, 50), (50, 100), (100, 100)] on istats.cpu
INFO: channel: fan following profile [(20, 25), (34, 100), (100, 100)] on _internal.liquid
INFO: update interval: 2 s; cutoff frequency (low-pass): 0.05 Hz; ema alpha: 0.27
INFO: starting...
INFO: pump control: lpf(istats.cpu) = lpf(30.0°C) = 30.0°C => duty := 67%
INFO: fan control: lpf(_internal.liquid) = lpf(28.0°C) = 28.0°C => duty := 68%
INFO: pump control: lpf(istats.cpu) = lpf(29.0°C) = 29.7°C => duty := 66%
INFO: fan control: lpf(_internal.liquid) = lpf(28.0°C) = 28.0°C => duty := 68%
INFO: pump control: lpf(istats.cpu) = lpf(32.0°C) = 30.3°C => duty := 67%
INFO: fan control: lpf(_internal.liquid) = lpf(28.0°C) = 28.0°C => duty := 68%
INFO: pump control: lpf(istats.cpu) = lpf(29.0°C) = 30.0°C => duty := 67%
INFO: fan control: lpf(_internal.liquid) = lpf(28.0°C) = 28.0°C => duty := 68%
INFO: stopped by user.
```

Again, I haven't tested any of this on macOS, only Linux, so the iStats part could be completely broken (though it should be easy to fix).

Additionally, some things ~~might~~ will change... for starters, the script will get renamed to something else other than Yoda.  I'm also concerned about how to distribute this; currently it uses tricky dependencies, so I'm looking into ways to reduce/replace them.  Finally, I'm still refining the CLI.  Check back regularly for updates.

---

_¹ The only exception is that the `--device` option is not supported.  You can use `--pick` instead, if you really need to differentiate between identical devices; but for a quick way to select devices, just use `--match`._

_² I'm mentioning this again because I'm not sure if those are exposed by default in homebrew installations; please let me know if this is indeed the case._

CaseySJ, 2020-02-03:
@jonasmalacofilho 

Added to my to-do list. This was a busy weekend. Not much gets done when your team is in the Super Bowl! Will test this on Mac this week.

jonasmalacofilho, 2020-02-07:
@sebikolo, @jwmann 

Can you give it a try and give me some feedback?

augustopaulogh, 2020-02-07:
Hi @jonasmalacofilho,
Thanks for your hard work / Obrigado pelo teu trabalho fantástico!
I've installed Liquidctl, iStats & docopt (pip3 install docopt) on Catalina 10.5.3 and get the following liquidctl output:
```
pauloaugusto@Paulos-iMac-Pro Downloads % liquidctl status    
WARNING: reading the +12V OCP mode is an experimental feature
Corsair HX1000i (experimental)
├── Current uptime                     8:24:57  
├── Total uptime            559 days, 17:24:57  
├── Temperature 1                         43.0  °C
├── Temperature 2                         37.0  °C
├── Fan control mode                  Hardware  
├── Fan speed                                0  rpm
├── Input voltage                       230.00  V
├── Total power                          88.00  W
├── +12V OCP mode                   Multi rail  
├── +12V output voltage                  12.16  V
├── +12V output current                   3.75  A
├── +12V output power                    44.00  W
├── +5V output voltage                    5.11  V
├── +5V output current                    6.88  A
├── +5V output power                     35.00  W
├── +3.3V output voltage                  3.33  V
├── +3.3V output current                  1.62  A
└── +3.3V output power                    5.00  W

NZXT Kraken X (X42, X52, X62 or X72)
├── Liquid temperature     30.6  °C
├── Fan speed               501  rpm
├── Pump speed             1909  rpm
└── Firmware version      6.0.2  
```
And following iStat output:

```
pauloaugusto@Paulos-iMac-Pro Shared % iStats                       
--- CPU Stats ---
CPU temp:               34.0°C      ▁▂▃▅▆▇

--- Fan Stats ---
Total fans in system:   5           
Fan 0 speed:            1896 RPM    ▁▂▃▅▆▇
Fan 1 speed:            0 RPM       ▁▂▃▅▆▇
Fan 2 speed:            0 RPM       ▁▂▃▅▆▇
Fan 3 speed:            0 RPM       ▁▂▃▅▆▇
Fan 4 speed:            0 RPM       ▁▂▃▅▆▇

--- Battery Stats ---
No battery on system

--- Extra Stats ---
TC0c  temp:             31.0°C      ▁▂▃▅▆▇
TC0p  temp:             34.0°C      ▁▂▃▅▆▇
TC0C CPU 0 Core temp:   31.0°C      ▁▂▃▅▆▇
TC0D CPU 0 Die temp:    34.0°C      ▁▂▃▅▆▇
TC0E CPU 0 ?? temp:     34.0°C      ▁▂▃▅▆▇
TC0F CPU 0 ?? temp:     34.0°C      ▁▂▃▅▆▇
TC0H CPU 0 Heatsink temp: 34.0°C      ▁▂▃▅▆▇
TC0P CPU 0 Proximity temp: 34.0°C      ▁▂▃▅▆▇
TC1c  temp:             32.0°C      ▁▂▃▅▆▇
TC1C Core 1 temp:       32.0°C      ▁▂▃▅▆▇
TC2c  temp:             34.0°C      ▁▂▃▅▆▇
TC2C Core 2 temp:       34.0°C      ▁▂▃▅▆▇
TC3c  temp:             32.0°C      ▁▂▃▅▆▇
TC3C Core 3 temp:       32.0°C      ▁▂▃▅▆▇
TC4c Unknown temp:      33.0°C      ▁▂▃▅▆▇
TC4C Core 4 temp:       33.0°C      ▁▂▃▅▆▇
TC5c Unknown temp:      33.0°C      ▁▂▃▅▆▇
TC5C Core 5 temp:       33.0°C      ▁▂▃▅▆▇
TC6c Unknown temp:      34.0°C      ▁▂▃▅▆▇
TC6C Core 6 temp:       34.0°C      ▁▂▃▅▆▇
TC7c Unknown temp:      33.0°C      ▁▂▃▅▆▇
TC7C Core 7 temp:       33.0°C      ▁▂▃▅▆▇
```
But get the _following error when executing yoda_:
```
pauloaugusto@Paulos-iMac-Pro Downloads % python3 yoda.py
Traceback (most recent call last):
  File "yoda.py", line 68, in <module>
    import liquidctl.cli as _borrow
ModuleNotFoundError: No module named 'liquidctl'
pauloaugusto@Paulos-iMac-Pro Downloads %
```

jonasmalacofilho, 2020-02-07:
@augustopaulogh 

I was worried about that.... homebrew installs liquidctl within a virtual environment, so the APIs are not exposed.

For the time being, can you instead try [installing liquidctl manually](https://github.com/jonasmalacofilho/liquidctl#installing-on-macos), with `pip3`?

---

Once the script has seen a bit more testing I'll start to look into how to package it.  The main issues on macOS are that iStats also needs to be packaged, and may bring too many indirect requirements (if the liquidctl formula is made to depend on it).

augustopaulogh, 2020-02-09:
Hi @jonasmalacofilho,
Believe that I was able to manually install liquidctl manually, but throws out other errors starting from line 217 of "yoda.py".
Please have a look below:
```
pauloaugusto@Paulos-iMac-Pro Downloads % brew uninstall liquidctl       
Uninstalling /usr/local/Cellar/liquidctl/HEAD-e30f25e... (1,135 files, 11.6MB)
pauloaugusto@Paulos-iMac-Pro Downloads % liquidctl set fan speed 20 -d 1 

zsh: command not found: liquidctl
pauloaugusto@Paulos-iMac-Pro Downloads % brew install libusb
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> New Formulae
gh                                               pueue                                            wasm3
==> Updated Formulae
activemq          antlr             aws-sdk-cpp       faiss             gradle            maven@3.3         pipx              subversion
akka              antlr@2           ballerina         gimme-aws-creds   hapi-fhir-cli     maven@3.5         rav1e             terraformer
allure            apache-archiva    bazel             git-delta         manticoresearch   netlify-cli       spotbugs
ammonite-repl     apache-ctakes     bazelisk          glslviewer        maven             opa               ssh-vault
ant               artifactory       beast             golo              maven@3.2         phpunit           starship

Warning: libusb 1.0.23 is already installed and up-to-date
To reinstall 1.0.23, run `brew reinstall libusb`
pauloaugusto@Paulos-iMac-Pro Downloads % brew reinstall libusb
==> Reinstalling libusb 
==> Downloading https://homebrew.bintray.com/bottles/libusb-1.0.23.catalina.bottle.tar.gz
######################################################################## 100.0%
==> Pouring libusb-1.0.23.catalina.bottle.tar.gz
🍺  /usr/local/Cellar/libusb/1.0.23: 29 files, 531.7KB
pauloaugusto@Paulos-iMac-Pro Downloads % pip3 install liquidctl
Collecting liquidctl
  Downloading liquidctl-1.3.2.tar.gz (63 kB)
     |████████████████████████████████| 63 kB 1.9 MB/s 
Requirement already satisfied: docopt in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from liquidctl) (0.6.2)
Collecting pyusb
  Downloading pyusb-1.0.2.tar.gz (54 kB)
     |████████████████████████████████| 54 kB 9.1 MB/s 
Collecting hidapi
  Downloading hidapi-0.9.0.post2.tar.gz (55 kB)
     |████████████████████████████████| 55 kB 16.6 MB/s 
Requirement already satisfied: setuptools>=19.0 in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from hidapi->liquidctl) (41.2.0)
Installing collected packages: pyusb, hidapi, liquidctl
    Running setup.py install for pyusb ... done
    Running setup.py install for hidapi ... done
    Running setup.py install for liquidctl ... done
Successfully installed hidapi-0.9.0.post2 liquidctl-1.3.2 pyusb-1.0.2
pauloaugusto@Paulos-iMac-Pro Downloads % 
pauloaugusto@Paulos-iMac-Pro Downloads % liquidctl status
WARNING: possibly uninitialized device
ERROR: Unexpected error with Corsair HX1000i (experimental)
ValueError: 0 is not a valid OCPMode

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
ValueError: 0 is not a valid OCPMode
pauloaugusto@Paulos-iMac-Pro Downloads % liquidctl initialize -d 0       
pauloaugusto@Paulos-iMac-Pro Downloads % liquidctl initialize -d 1
pauloaugusto@Paulos-iMac-Pro Downloads % liquidctl status         

WARNING: reading the +12V OCP mode is an experimental feature
Corsair HX1000i (experimental)
├── Current uptime                     1:10:14  
├── Total uptime            560 days, 11:10:14  
├── Temperature 1                         42.0  °C
├── Temperature 2                         35.5  °C
├── Fan control mode                  Hardware  
├── Fan speed                                0  rpm
├── Input voltage                       230.00  V
[... 58 lines cut ...]
pauloaugusto@Paulos-iMac-Pro Downloads % python3 yoda.py                 
Traceback (most recent call last):
  File "yoda.py", line 217, in <module>
    args = docopt(__doc__, version='yoda v{}'.format(VERSION))
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/docopt.py", line 560, in docopt
    pattern = parse_pattern(formal_usage(DocoptExit.usage), options)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/docopt.py", line 373, in parse_pattern
    result = parse_expr(tokens, options)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/docopt.py", line 387, in parse_expr
    seq = parse_seq(tokens, options)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/docopt.py", line 396, in parse_seq
    atom = parse_atom(tokens, options)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/docopt.py", line 413, in parse_atom
    result = pattern(*parse_expr(tokens, options))
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/docopt.py", line 381, in parse_expr
    seq = parse_seq(tokens, options)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/docopt.py", line 396, in parse_seq
    atom = parse_atom(tokens, options)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/docopt.py", line 421, in parse_atom
    return parse_long(tokens, options)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/docopt.py", line 311, in parse_long
    raise tokens.error('%s is not a unique prefix: %s?' %
docopt.DocoptLanguageError: --pick is not a unique prefix: --pick, --pick?
pauloaugusto@Paulos-iMac-Pro Downloads %  
```

jonasmalacofilho, 2020-02-09:
Ops... should be fixed in the latest `yoda` (0.0.3).

_(apparently the docopt version I have doesn't complain about the duplicated/ambiguous option definition)_

augustopaulogh, 2020-02-09:
It works!
```
pauloaugusto@Paulos-iMac-Pro Downloads % python3 yoda.py  --match kraken show-sensors 
Sensor identifier                                                    Temperature
--------------------------------------------------------------------------------
_internal.liquid                                                          31.8°C
istats.cpu                                                                42.0°C
pauloaugusto@Paulos-iMac-Pro Downloads % 
```

jonasmalacofilho, 2020-02-09:
Great!

Let me know any other issues you find.

stasivanov, 2020-02-11:
Hi guys. Is there a way to configure it once under Windows, and just save (I assume the information is stored in the NZXT controller). Just to not install full Windows on a real HDD/SDD, I launched Bootable Live CD from my flash drive, installed NZXT Cam app and configured everything how I want. However, once I restart to the Mac OS, it seems like it just forgot everything... Is that because I was configuring under Live CD? Is there a way to configure it under Windows once so that it works after restart in Mac OS

jonasmalacofilho, 2020-02-12:
@stasivanov 

TL; DR: not sanely, no.  That's why this project exists.

The device does save settings to the features _it_ supports (i.e. that are not implemented in software) as long as it's powered on from the 5VSB rail.  Generally this means until the power supply is physically turned off, but ErP settings can affect this.

Because these are short lived, using CAM to configure the device before booting to another OS isn't very convenient.  Additionally, at the firmware level the Smart Device V2 only supports _static_ fan speeds.

And that second part relates to this particular issue: CPU/GPU based fan profiles across all devices supported by liquidctl, including the SDV2.  The core problem is that no USB cooling device has access to any external (to it) temperature sensor, so software (either running in userspace or a kernel driver) has to send that information (or, more commonly, the corresponding PWM duties per channel) to the device.

---

P.S. Please be aware that this isn't a support forum for CAM.

jwmann, 2020-02-13:
@jonasmalacofilho How do I install this yoda thing?

jonasmalacofilho, 2020-02-15:
There are no installation steps for yoda, you can copy it somewhere and run it.

But before that (for now) you'll need to [manually install liquidctl](https://github.com/jonasmalacofilho/liquidctl#installing-on-macos), with `pip3`.  Otherwise yoda won't find the liquidctl APIs.

In the future this will change, but I'm just not ready yet to ship yoda as part of liquidctl.  It doesn't yet support Windows and it depends on unpackaged dependencies on both Linux and macOS; on macOS specifically, `iStats` would first need to be made available through homebrew.

jonasmalacofilho, 2020-04-13:
Quick update: I've decided to integrate (the currently named) yoda into liquidctl, to deal with all the issues with distributing and running it.

This will involve a bit of work, and on mac OS will still require `iStats` to already be installed, but that's something that can be improved upon later.

CaseySJ, 2020-04-13:
@jonasmalacofilho, Looking forward to it!

augustopaulogh, 2020-04-13:
@jonasmalacofilho, Fantástico, fico à espera para implementar!
@jonasmalacofilho Fantastic, I'll wait to try it our!

imikejackson, 2020-05-06:
Just came upon this thread but another idea would be to use Intel's PowerGadget framework that gets installed when Intel Power Gadget is installed. It gives you native access to CPU temperatures through a C++ API call (which is trivial to wrap with Python). Just another thought.

joansbou, 2020-10-03:
Sorry, I'm trying to run both krakencurve and yoda on mi manjaro and there is no way. yoda output:
./yoda --match kraken show-sensors
TypeError: int() argument must be a string, a bytes-like object or a number, not 'list'

krakencurve output: 
./krakencurve-poc show-sensors
IndexError: list index out of range

are the scripts ready for linux?

Thanks

jonasmalacofilho, 2020-10-03:
@joansbou,

Try to enable `--debug` on both.  But if I had to venture a guess, you may have multiple liquidctl installations or for some other reason your `yoda`/`krakencurve-poc` scripts don't match the available liquidctl libraries.

P.S. The scripts have been developed on Linux (ArchLinux, btw).

jonasmalacofilho, 2020-11-21:
Quick update: integrating this into the CLI stalled, and I'm no longer working on that.

Users wanting software-based fan or pump speed profiles should use the [`extra/yoda`](https://github.com/jonasmalacofilho/liquidctl/blob/master/extra/yoda) script for now, which is fairly usable as is; contributions to improve it are also very welcome.

Finally, I'm closing this issue, but if someone wants to pick up the task of building the feature into the main CLI, please let me know.  Just be warned that this is not an easy problem (because of OS limitations)...

closed 2020-11-21 by jonasmalacofilho