liquidctl/liquidctl #73: "OSError: open failed" for Corsair HX1000i

augustopaulogh, 2020-01-06:
Hi, 
Thanks for this great tool in Catalina (10.15.2)!
Just works great with my Kraken NZXT X72 AIO cooler, but it won't initialise my Corsair HX1000i PSU.

```
pauloaugusto@Paulos-iMac-Pro Downloads % liquidctl list -v
Device ID 0: Corsair HX1000i (experimental)
├── Vendor ID: 0x1b1c
├── Product ID: 0x1c07
├── Release number: 0x0002
├── Bus: hid
├── Address: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC@14/XHC@14000000/HS13@14d00000/USB2.0 Hub@14d00000/AppleUSB20Hub@14d00000/AppleUSB20HubPort@14d10000/USB2.0 Hub@14d10000/AppleUSB20Hub@14d10000/AppleUSB20HubPort@14d12000/                      @14d12000/IOUSBHostInterface@0/AppleUserUSBHostHIDDevice
└── Driver: CorsairHidPsuDriver using module hid

Device ID 1: NZXT Kraken X (X42, X52, X62 or X72)
├── Vendor ID: 0x1e71
├── Product ID: 0x170e
├── Release number: 0x0200
├── Serial number: 6D83478F525
├── Bus: hid
├── Address: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC@14/XHC@14000000/HS13@14d00000/USB2.0 Hub@14d00000/AppleUSB20Hub@14d00000/AppleUSB20HubPort@14d10000/USB2.0 Hub@14d10000/AppleUSB20Hub@14d10000/AppleUSB20HubPort@14d14000/NZXT USB Device@14d14000/IOUSBHostInterface@0/AppleUserUSBHostHIDDevice
└── Driver: KrakenTwoDriver using module hid
```

```
pauloaugusto@Paulos-iMac-Pro Downloads % liquidctl initialize -d 0
OSError: open failed
```

```
pauloaugusto@Paulos-iMac-Pro Downloads % liquidctl status -v
Traceback (most recent call last):
  File "/usr/local/bin/liquidctl", line 11, in <module>
    load_entry_point('liquidctl==1.3.2', 'console_scripts', 'liquidctl')()
  File "/usr/local/Cellar/liquidctl/1.3.2/libexec/lib/python3.7/site-packages/liquidctl/cli.py", line 304, in main
    dev.connect(**opts)
  File "/usr/local/Cellar/liquidctl/1.3.2/libexec/lib/python3.7/site-packages/liquidctl/driver/usb.py", line 135, in connect
    self.device.open()
  File "/usr/local/Cellar/liquidctl/1.3.2/libexec/lib/python3.7/site-packages/liquidctl/driver/usb.py", line 400, in open
    self.hiddev.open_path(self.hidinfo['path'])
  File "hid.pyx", line 72, in hid.device.open_path
OSError: open failed
```
Can you please provide some pointers ?

Thanks.

augustopaulogh, 2020-01-06:
Sorry!
Please disregard my previous post and close this thread!
I just needed to reboot...

closed 2020-01-06 by augustopaulogh

jonasmalacofilho, 2020-01-07:
Hi!  I'm glad that everything worked out.  Let me know if you have any other issues...