liquidctl/liquidctl #74: Unable to get liquidctl to work on my windows machine.

Louiscar1, 2020-01-11:
As per title looking for an alternative to CAM I found this but after trying the command line tool the only thing that makes sense is:

```
liquidctl-1.3.2-bin-windows-x86_64>liquidctl list
Device ID 0: Asetek 690LC (assuming EVGA CLC)
```

I get my Kraken x31 shown but no status.
any other command results in "unimplimented" error like this:

```
liquidctl status
NotImplementedError: Operation not supported or unimplemented on this platform
[4864] Failed to execute script cli
```
Not sure why it's not working but some help would be appreciated.
Do I need to uninstall CAM .. it wasn't running when I attempted all this.

jonasmalacofilho, 2020-01-12:
Hi @Louiscar1 !

I think there are two things causing this..

First, you need to install the WinUSB driver on this device.  Unlike more modern coolers, the Kraken X31 normally uses a proprietary driver which is not compatible with (the) open-source libraries (we use).

The next thing is that you *must* pass the `--legacy-690lc`  on all invocations of liquidctl.  This is very important because there is no way for us to determinate in software if it's a Kraken X31 or an EVGA CLC 120, and while both devices are similar, their protocols are incompatible.  Asetek uses the same vendor and product IDs, and there's no way to query the device type through the protocol either.

_(this will make `liquidctl list --legacy-690lc` state that the device is assumed to be a NZXT Kraken, instead of an EVGA CLC)_

**You can find more information in our documentation for the [Asetek 690LC liquid coolers](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/asetek-690lc.md).**

Louiscar1, 2020-01-12:
Hi Jonas,

I was trawling through the documents and saw the driver thing. I had already tried the legacy-690lc and saw that the message did at least change to (assume Kraken x). I then downloaded Zadig and had after figuring how to use it, had stopped there as I wasn't sure of undoing something if it went wrong.

But thanks for the heads up as now I have confirmation that this is what needs to be done.

Louiscar1, 2020-01-12:
Great I got it working however, but have some questions . It appears to me that the pump speed is fixed or can't be altered. I got confused by the status report giving me either 1800rpm or 1680rpm when I changed fan speed.

However, as it transpires I don't have to do anything I just have to poll in succession so the result is not constant. Eg. I polled twice in quick succession to get these results.

```
liquidctl --legacy-690lc status

Asetek 690LC (assuming NZXT Kraken X) (experimental)
├── Liquid temperature       33.0  °C
├── Fan speed                 960  rpm
├── Pump speed               1800  rpm
└── Firmware version      0.7.0.0

liquidctl --legacy-690lc status

Asetek 690LC (assuming NZXT Kraken X) (experimental)
├── Liquid temperature       33.0  °C
├── Fan speed                 900  rpm
├── Pump speed               1680  rpm
└── Firmware version      0.7.0.0

```
I'm wondering if this model doesn't change the pump speed? Eventually it seems to settle always on 1800.

Also when trying to set a profile for the fan , this fails with the following.

```
liquidctl --legacy-690lc set fan speed 40 40 55 60 65 100
ERROR: Unexpected error with Asetek 690LC (assuming NZXT Kraken X) (experimental
)
Traceback (most recent call last):
NotImplementedError

liquidctl --legacy-690lc set fan speed 55 60 65 100
ERROR: Unexpected error with Asetek 690LC (assuming NZXT Kraken X) (experimental
)
Traceback (most recent call last):
NotImplementedError

```
So it seems I can only set a single speed
Am I doing this correctly?

jonasmalacofilho, 2020-01-12:
I'm glad you got it to work already : )

> I'm wondering if this model doesn't change the pump speed? Eventually it seems to settle always on 1800.

I remember reading that the X31 was more limited than the other coolers of the same family.  Maybe one of these limitations is that the pump only runs at 1800 rpm?

On the other hand the variance in _reported_ pump speed is normal (a bit high compared to my Asetek 690LC from EVGA, but still within the expected range).

> So it seems I can only set a single speed

It's possible that the cooler supports fan profiles, but we haven't been able to successfully implement that (much less without breaking other features).  The legacy protocol is not very friendly, and our knowledge about it is also somewhat limited.

Louiscar1, 2020-01-12:
Ok Thanks for the information Jonas,

Without the ability to set some kind of profile it would be a pain to switch to this as an alternative to CAM.

Right now it appears I'll have to tolerate  CAM's telemetry requirements until I find another solution.

Either way a big thanks to people like you who strive to make better software for us to use.

jonasmalacofilho, 2020-01-12:
Thanks! : )

You know, you can easily adapt `extra/krakencurve-poc` to work with these old Krakens, and have a software based fan profile.  You will need to set up Python, libusb and liquidctl _as a full Python package,_ but otherwise the code change should be very straightforward (if I'm not forgetting anything):

```diff
- from liquidctl.driver.kraken_two import KrakenTwoDriver
+ from liquidctl.driver.asetek import LegacyAsetekDriver

- device = KrakenTwoDriver.find_supported_devices()[0]
+ device = LegacyAsetekDriver.find_supported_devices(legacy_690lc=True)[0] 
```

The only thing is that you'll be limited to the liquid temperature, as there's no good API we can rely on in Windows for other sensors (we still need to figure that part out).

jonasmalacofilho, 2020-01-13:
Just remembered that you'd also need to change:

```diff
- cooler.set_instantaneous_speed('pump', pump_duty)
- cooler.set_instantaneous_speed('fan', fan_duty)
+ cooler.set_fixed_speed('pump', pump_duty)
+ cooler.set_fixed_speed('fan', fan_duty)
```

---

Well, I'm going to close this since there's not much more liquidctl can do (at least for now).  But please feel free to add more comments if you have more questions or suggestions...

Also, if you start messing with Python you could also consider resurrecting #35 ("Some Kraken X61 functionality still missing").  My newer 690LC from EVGA only allows me to go so far, the rest has to be done by someone who has one of these older Krakens.

Another related issue that needs help is #72 ("Software-based fan/pump control"), which essentially is about turning `extra/krakencurve-poc` into a first-class feature.  For now the discussion there has been focused on MacOS, but Windows has similar problems (i.e. lack of a way for us to query sensors).

closed 2020-01-13 by jonasmalacofilho