liquidctl/liquidctl #75: OSError: open failed with version 1.3.2 on Ubuntu 19.10

leinardi, 2020-01-18:
I've updated liquidctl from v1.1.0 to 1.3.2 but now when I try to run a `liquidctl status` I get this error:

```
$ liquidctl status --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.2
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x424:0x274a
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e7d:0x2e22
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e7d:0x2e22
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x24f0:0x140
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x24f0:0x140
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xb05:0x1872
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x812
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x424:0x5734
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x781:0x558c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xb05:0x1872
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x24f0:0x140
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x2109:0x2812
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e7d:0x2e22
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x424:0x274a
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x424:0x2734
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.cli: device: NZXT Kraken X (X42, X52, X62 or X72)
Traceback (most recent call last):
  File "/home/leinardi/.local/bin/liquidctl", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/liquidctl/cli.py", line 304, in main
    dev.connect(**opts)
  File "/usr/local/lib/python3.7/dist-packages/liquidctl/driver/kraken_two.py", line 125, in connect
    super().connect(**kwargs)
  File "/usr/local/lib/python3.7/dist-packages/liquidctl/driver/usb.py", line 135, in connect
    self.device.open()
  File "/usr/local/lib/python3.7/dist-packages/liquidctl/driver/usb.py", line 400, in open
    self.hiddev.open_path(self.hidinfo['path'])
  File "hidraw.pyx", line 72, in hidraw.device.open_path
OSError: open failed
```

Before running it I've installed all the Ubuntu dependencies listed in the README.md:
```
sudo apt install python3 libusb-1.0-0 python3-pkg-resources python3-docopt python3-usb python3-hid
```

I saw that #73 reported a similar issue and the solution in that case was to reboot: I tried that too but didn't work.

jonasmalacofilho, 2020-01-18:
If I remember correctly, you use an udev rule to run liquidctl as a non-priviledged user.

As the default backend used for HIDs on Linux has changed in 1.3.2, from libusb to hid(raw), you probably need to adjust your udev rule to not only work at the USB level.

I get into a bit more detail about this in [#70 (comment)](https://github.com/jonasmalacofilho/liquidctl/issues/70#issuecomment-554980902).

leinardi, 2020-01-18:
Hey thanks! Changing from `SUBSYSTEM` to `SUBSYSTEMS` did the trick :+1:

closed 2020-01-18 by leinardi