liquidctl/liquidctl #76: have installed, or think I have installed, but liquidctl does not see my aio

jwjennison, 2020-01-24:
lsusb result - Bus 001 Device 004: ID 1b1c:0c17 Corsair 

root@james-desktop:/home/james/liquidctl-master# liquidctl initialize
No devices matches available drivers and selection criteria

Using Corsair H115i Platinum (Fans pluged into pump, pump plugged into usb header and Satapower)

jwjennison, 2020-01-24:
root@james-desktop:/home/james/liquidctl-master# liquidctl initialize -g
[DEBUG] liquidctl.cli: running liquidctl v1.3.1
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
**[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0xc17**
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0x405e
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0x4062
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x8087:0x29
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0xc17
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
No devices matches available drivers and selection criteria

jonasmalacofilho, 2020-01-24:
Hi @jwjennison,

The H115i Platinum is not yet supported.  I would love to support it, but I don't have one of these: for the time being it will depend on someone stepping up to help write the driver for it.

By the way, there's a huge difference between the many H115i variants:

 - H115i: 5-th gen. Asetek design (supported)
 - H115i PRO: 6-th gen. Asetek design (not supported)
 - H115i Platinum: CoolIT design (not supported, but planned)
 - upcoming H115i PRO XT: CoolIT design again (not available yet, but planned)

jwjennison, 2020-01-24:
well lets get started.. what can I do to help

jonasmalacofilho, 2020-01-24:
> well lets get started.. what can I do to help

Great!

I'm on my way out now but I'll write a few notes of what's necessary as soon as I'm back...

_(renamed and reopened the issue)_

jonasmalacofilho, 2020-01-25:
Hi again!

The first step is to figure out the protocol.  Specifically, we need to know what has to be send and/or read in each of the base liquidctl methods:

 - `connect`: handshake sometimes necessary after connecting to the device
 - `initialize`: handshake only necessary (if that) after the device has just been powered on
 - `get_status`: get a status report (fan/pump speeds, liquid temperature, etc.)
 - `set_fixed_speed`: set fan and/or pump to a fixed speed
 - `set_speed_profile`: set fan and/or pump to follow a profile (based on liquid temperature)
 - `set_color`: set animations and colors to each LED channel
 - `disconnect`: handshake sometimes necessary before disconnecting from the device

Besides capturing the USB traffic, some experimentation can be helpful.  So I usually start with a skeleton driver (NOOP connect, initialize and disconnect) and start working on a simple method (usually `get_status`, or maybe `set_fixed_speed`).

Initially I pretty much copy the sections of the captured USB traffic that seem relevant, which I then start to separate into parameters, handshakes, and plainly useless traffic (it happens).  Then test, rinse and repeat...

For capturing the USB traffic I prefer to use Wireshark on Linux, running the vendor software running in a Windows VM, using USB pass through (technically: the VM will have control over the device, and Wireshark will capture the traffic at the root hub).

jwjennison, 2020-01-25:
I dont see my device in some obvious way but I captured 3 and 5, (only ones that produced data).and a screen shot

jwjennison, 2020-01-25:
Just wanted to make sure that the attachments I made via email landed ok, otherwise I can send directly

jonasmalacofilho, 2020-01-25:
They didn't.  Honestly, replying to issues directly from the notification emails doesn't work well at all: markdown is disabled (even if you later try to edit the comment), stuff that should be filtered isn't, other stuff can get silently filtered...

I'm guessing that the device is the one in Port 2.  But the way to really tell is from the captured data: search for the vendor and product IDs, since they are part of the USB device descriptor, which is fetched when the device is enumerated.  The HID interface will appear as a child device of this main USB device, and that is what we want to monitor in this case.

jwjennison, 2020-01-26:
https://www.dropbox.com/s/7oazou3bunrklhc/3.pcap?dl=0

jonasmalacofilho, 2020-01-26:
> https://www.dropbox.com/s/7oazou3bunrklhc/3.pcap?dl=0

Thanks, I'll take a look.

But I'll probably need you to figure out how the protocol works.  I've already seen a couple of generic captures for this device, and so far I haven't been able to advance much... it will probably require being able to _interactively_ test different iCUE parameters (later: driver code) and monitor changes.

jwjennison, 2020-01-26:
well I've done a few more captures, but have yet to find anything that jumps out at me.
Unfortunately to be entirely honest, I don't know a whole lot of coding and the like.,
I Started with OpenCorsairLink, and I wasn't able to get it to function either, so that's when I found your app.  Frustrating situation when I can only offer limited help.
During my Captures I have manipulated the fans and pump to different settings hoping to produce some "Gotcha" moment in the capture, but I am just not seeing it. (Frustrated)

jonasmalacofilho, 2020-01-26:
Yes, this is probably one of the most complex protocols I've seen yet.

I'll see if I can at least figure out the basic structure and operations.

jwjennison, 2020-02-04:
Just wondering what else I can do to assist the dev of this for my h115i Plat.. I am using Boinc on Ubuntu and hitting some pretty nasty temps (high of 96c)Seems like the pump is delayed on occasion and allows for some hot tempsnormaly its at 84c, but I am throttling the work load to keep the temps down.
You would think they could just put a switch on the block to set it (usb, Balanced, perf) settingsAnyways let me know what else I can do to support.James Jennison+

 
 

https://www.dropbox.com/s/7oazou3bunrklhc/3.pcap?dl=0

jonasmalacofilho, 2020-02-06:
So far I looked into the incoming stream of data and I think I managed to figure out the position of the most important fields.  I'll document and/or implement this soon.

The next step is to look into the data that's sent to the device.  Can you capture another session for me?  Specifically, I would like you to:

 - set all LEDs to off (or static black), to try to reduce the noise in the USB traffic
 - change the fans and pump to fixed 100%
 - change the fans and pump to the lowest possible fixed value (and let me know what that is)
 - change the fans and pump to the extreme mode
 - change the fans and pump to some custom curve _based on liquid temperature_ (and let me know what that is)
 - set all the LEDs to static red
 - (if possible) set the pump LEDs to some other static color (e.g. blue)
 - (if possible) set the LEDs of each of the fans to some other static color (e.g. green)

Finally, what additional information does iCue show for the device?  For example, is that a firmware version (and what is it)?

jonasmalacofilho, 2020-02-06:
Recognizable values in the incoming reports:

```py
liquid_temperature = data[8] + data[7] / 256  # [°C]
fan1_speed = data[15] << 8 | data[16]         # [rpm]
fan2_speed = data[22] << 8 | data[23]         # [rpm]
pump_speed = data[29] << 8 | data[30]         # [rpm]
```

_Note: this was also seen for the H100i Platinum (which is to be expected anyway)._

olielvewen, 2020-02-06:
@jwjennison 

```
The next step is to look into the data that's sent to the device. Can you capture another session for me? Specifically, I would like you to:
    set all LEDs to off (or static black), to try to reduce the noise in the USB traffic
    change the fans and pump to fixed 100%
    change the fans and pump to the lowest possible fixed value (and let me know what that is)
    change the fans and pump to the extreme mode
    change the fans and pump to some custom curve based on liquid temperature (and let me know what that is)
    set all the LEDs to static red
    (if possible) set the pump LEDs to some other static color (e.g. blue)
    (if possible) set the LEDs of each of the fans to some other static color (e.g. green)
```

It will be nice and less heavy if you could do them (= for all) in several files for sample one for  set all the LEDs to static red, one for  change the fans and pump to the extreme mode, etc....

It will be easiest for reading them. Thanks.

@jonasmalacofilho 

```
Finally, what additional information does iCue show for the device? For example, is that a firmware version (and what is it)?
```

Normally, if my memories is still good, when we have implanted the PSU RMi Series I have done a test with my dark core and see two things : one the PSU RMi series don't have a serial number/version firmware and two that was the bcdDevice which  give us the information. And ISerial for serie's number.
And jwjennison has already give us this information. 
However, it is to confirm in one of our long discussion... ^ _ ^.

jonasmalacofilho, 2020-02-06:
@jwjennison 

> It will be nice and less heavy if you could do them (= for all) in several files for sample one for set all the LEDs to static red, one for change the fans and pump to the extreme mode, etc....

Actually, this isn't the case _yet._ 

The issue is that there is a _lot_ of noise... we're talking about tens of reports being sent and received every second.  With so much noise there's less benefit from trying to isolate the steps (even isolated, there's still a lot of noise), and it needless complicate things for us both.

At this stage, that is.  More isolated captures will be surely be necessary once we start to fine tune or debug our implementation.

But for now, please ensure that all steps are performed in order and in a single iCue session.

---

@olielvewen 

You're right about the specific case of the RMi PSUs, but I don't know that it will _always_ the case; for example, on most devices the bcdDevice descriptor is used to identify _hardware_ revisions.

jwjennison, 2020-02-06:
here's another capture, I will try to do some smaller individual files as well.I ran all presets, custom presets will only work for fan, not for pump.

jonasmalacofilho, 2020-02-07:
Hi @jwjennison, I can't see the new capture... it probably got lost due to the limitations of email replies (in GitHub).  Can you send it again?  Maybe using the GitHub interface will help.

You can also submit it as a pull request to [jonasmalacofilho/liquidctl-device-data](https://github.com/jonasmalacofilho/liquidctl-device-data), or even send it to me directly by email; my address is listed in my GitHub profile.

jwjennison, 2020-02-07:
try this..icue individuals

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
icue individuals

Shared with Dropbox
 |

 |

 |

jonasmalacofilho, 2020-02-07:
Nope.

Directly replying to GitHub's email notifications doesn't work.

Personally, I always click on "view issue" and reply using the web UI.  I think you should start doing the same.

Anyway, if you prefer to stick to email, my address is:

jonas@protocubo.io

jwjennison, 2020-02-07:
ok, lets try this (from Github)

[icue individuals.zip](https://github.com/jonasmalacofilho/liquidctl/files/4170798/icue.individuals.zip)

olielvewen, 2020-02-07:
@jwjennison 

I got it and I've just take a quick look. What is the subject of the file ch03-02-06-2020 ? It is clear for me for others but not this one. Sorry.

@jonasmalacofilho 

I wonder if **[hidrd](https://github.com/DIGImend/hidrd)** (available on AUR) could not help us in this task. The Only thing that we must done is to understand how he works ( ^ __ ^ )

jonasmalacofilho, 2020-02-07:
@olielvewen 

You're right, we should definitively check the HID descriptors.  Though I'm not betting we'll find much (of practical use).

jwjennison, 2020-02-07:
on ch3 file was just a general all commands to the AIO, then I did individual files where I only did one change at a time and then closed the capture files

olielvewen, 2020-02-08:
@jwjennison 
Okay. 
BTW, could you make us a screenshot in ICue about the part which contains the firmware version, .... if they are available of course (to my mind at least the firmware version). Thanks.

@jonasmalacofilho 

```
You're right, we should definitively check the HID descriptors. Though I'm not betting we'll find much (of practical use).
```
Right too. However, we are not alone to trying to understand the CoolIt protocol. I've not have the time to do something today about but thinking at that, I've done a quick search and I've found another project which is trying to implement [an Hwmon driver for Linux ](https://github.com/rigred/mod_coolit)(you know what I said that !!!!) about the CoolIt pump. And we have even a base of information which be very very helpful for us. I think [at this one](https://github.com/rigred/mod_coolit/blob/master/Command%20Reference.md) and this[ one two. ](https://github.com/rigred/mod_coolit/blob/master/libusb_h110i.c)

But I've not add the time to ask him if he would like to help us. Together, we are stronger.

jwjennison, 2020-02-08:
[image]

jonasmalacofilho, 2020-02-09:
@jwjennison thanks again!

olielvewen, 2020-02-09:
@jwjennison 
Thanks. But I've forgotten to ask you too make this command in a terminal:

```
sudo lsusb -d 1b1c:0c17 -v
```
Sorry and thanks again.

@jonasmalacofilho 

Did you have read [his notes](https://github.com/rigred/mod_coolit/blob/master/Command%20Reference.md) about CoolIT pum ?

jonasmalacofilho, 2020-02-09:
@olielvewen 

You can find the output from lsusb in the first comment.

About the project you linked, I haven't had time to check it thoroughly, but it seems that it only handles the old CoolIT coolers, _which apparently are incompatible with these latest models.¹_  I also couldn't find under which license the code is distributed under.

_¹ According to discussions in the OpenCorsairLink repository._

audiohacked, 2020-02-10:
@jonasmalacofilho 
I'll just add that there are two packets associated with setting leds or fans. First packet to setup control pkt. Second packet contains data to send to device. The Host Pkt is usually 92 bytes (28 byte header + 64 byte payload). The Device Response Pkt is usually 91 bytes (27 byte header + 64 byte response payload).

## host to device
### Packet Structure
Byte 3 is the CMD
Where CMD is:
0x00 is SetLED
0x14 is SetFan
0xFF is GetStatus
Byte 4 - Byte 63 is the command data
Byte 64 is the packet checksum (??)

### SetLED packet
XX
XX
0x14
BB
GG
RR
....

## device to host
Byte 1 is CMD
Where CMD is
0xFF is Status Pkt

That's as far as I've been able to reverse engineer. I've been traveling a ton lately so I have not been able to do much work on this. Hopefully both our projects can add this support.

audiohacked, 2020-02-10:
For those of you trying to decipher your packet dumps; here are a few Wireshark filters to filter out packets.

- `usb.src == "2.3.1" or usb.dst == "2.3.0"`
- `frame.len == 91`
- `frame.len == 92`

where "2.3.0" was my host output port (endpoint 0x00) to the H100i Platinum device address;
and "2.3.1" was my host input port (endpoint 0x81) from the H100i Platinum device address;

jonasmalacofilho, 2020-02-10:
Hi @audiohacked,

Thanks, that helps.

On my end I also figured out that the last byte is indeed a CRC-8 byte; specifically, it's computed using same polynomial and settings as in SMBus.

> Hopefully both our projects can add this support.

Yes, we should definitively work more closely on these new devices.

---

> `usb.src == "2.3.1" or usb.dst == "2.3.0"`

I've been using a filter that looks like `usb.src like "^2\.3\." || usb.src like "^2\.3\."` (adjusted to the necessary values for a specific capture) as, in theory, there could be some devices where both HID and raw USB are used simultaneously.  But yes, it's overkill...

I also (only) keep the already filtered captures in [my archive](https://github.com/jonasmalacofilho/liquidctl-device-data).

And for others reading this: one trick I use to figure out which is the right device, when that info is missing for an unfiltered capture, is to look for all replies to GET DEVICE DESCRIPTOR, and see which device has the desired vendor/product IDs.

jonasmalacofilho, 2020-02-10:
After looking at how "set LEDs to blue" is sent to the cooler and analyzing a few caputres, I think that a portion of the type of outgoing message is encoded in the 3 least significant bits of byte offset 0x1:

 - data[1] == 0x4 (0b100): set lighting
 - data[1] == 0x5 (0b101): set lighting, continued
 - data[1] == 0x0 (0b000): set cooling (data[2] == 0x14) or just read status (data[2] == 0xff)

_(relevant captures, spreadsheets and notes in [liquidctl-device-data/Corsair H115i Platinum](https://github.com/jonasmalacofilho/liquidctl-device-data/tree/master/Corsair%20H115i%20RGB%20Platinum))_

---

Because the PEC byte doesn't consider the first byte of the data, I don't think that first byte is relevant even in the incoming reports.

olielvewen, 2020-02-10:
@jonasmalacofilho 

```About the project you linked, I haven't had time to check it thoroughly, but it seems that it only handles the old CoolIT coolers, which apparently are incompatible with these latest models.¹ I also couldn't find under which license the code is distributed.```

Okay. I've found this conversation. 

`You can find the output from lsusb in the first comment.`

Yes. I know but I would like to check if the firmware has been updated like we can see in the screenshot and if we can get the serial number. Now I'm not sure to get a positive answer contrary to Sunday morning. And the output has been done there are 17 days ago.

@audiohacked 

Thanks a lot for your help. That's a good start.

```
For those of you trying to decipher your packet dumps; here are a few Wireshark filters to filter out packets.

    usb.src == "2.3.1" or usb.dst == "2.3.0"
    frame.len == 91
    frame.len == 92

where "2.3.0" was my host output port (endpoint 0x00) to the H100i Platinum device address;
and "2.3.1" was my host input port (endpoint 0x81) from the H100i Platinum device address;
```
Useful tip.

jwjennison, 2020-02-14:
Bus 001 Device 004: ID 1b1c:0c17 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c17 
  bcdDevice            1.00
  iManufacturer           1 Corsair Inc.
  iProduct                2 H115i Platinum
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7
Device Status:     0x0001
  Self Powered

olielvewen, 2020-02-14:
@jwjennison 

Thanks a lot. I'm a bit disappointed. Because the bcdDevice is still at 1.00 and not at 1.01.15 like your screenshot and what I was thinking after an update (to 1.00 at 1.01.15) and the iSerial is still at 0. I was waiting the number like for my Corsair Dark Core SE RGB.

jwjennison, 2020-02-14:
I just want a way to turn fan and pump to 100%..

jonasmalacofilho, 2020-02-14:
@jwjennison 

> I just want a way to turn fan and pump to 100%.

Well get there.  Unfortunately we need to understand a lot of stuff just to be sure we aren't doing anything terribly stupid, even with the (presumably) simple things.  But we'll get there... soon.

@olielvewen 

Yeah, the bcdDevice isn't used for that on this device (and that's actually very common).  But it's ok, I'm pretty sure I found the firmware version in the replies from the cooler (see #82).

audiohacked, 2020-02-14:
Typically (99% of the time) the USB/HID data is fixed at manufacturing of the device and can't be upgraded via firmware.

olielvewen, 2020-02-17:
@jonasmalacofilho 

`Yeah, the bcdDevice isn't used for that on this device (and that's actually very common). But it's ok, I'm pretty sure I found the firmware version in the replies from the cooler (see #82).`

Oh. I've not seen. Thanks a lot Jonas. Good news.

@jwjennison 

>I just want a way to turn fan and pump to 100%.

Personally I am more greedy I want everything ...............like Jonas.

olielvewen, 2020-02-17:
@jonasmalacofilho 

I've been able to isolate several times hex dump. 
I've tried without no success hidrd.
I've seen that you have created an analyze python file. How does it work ? Can he decode them ?

jonasmalacofilho, 2020-02-17:
@olielvewen 

Is your cooler up and running?  Or can you at least power it outside your system and plug a USB cable into it?

I'm going to need a few more captures to really understand how the cooling settings are sent to the cooler.  The LEDs are easy, but the cooling isn't clear at all.

---

> I've been able to isolate several times hex dump.

What dumps exactly?

> I've seen that you have created an analyze python file. How does it work ? Can he decode them ?

It's nothing automatic, just a way to reprocess an entire capture file (exported in Wireshark to JSON) according to know or theorized rules for parsing fields (which you have to supply by modifying the code).

I've also used spreadsheets and very simple statistics to find some field boundaries.

olielvewen, 2020-02-17:
@jonasmalacofilho 

>Is your cooler up and running? Or can you at least power it outside your system and plug a USB cable into it?

>I'm going to need a few more captures to really understand how the cooling settings are sent to the cooler. The LEDs are easy, but the cooling isn't clear at all.

No. To be clear, I have removed and you know why. Hum I don't see how to do that without  to run it.

I make certainly an error with pcpap files. 

Here how I process. To avoid to have a ton of files? I use a usb filter but not like you. Any way it doesn't work. I took the usb device address of Jwjennison (see at the beginning the lsusb command). `Bus 001 device 004`. And Normally that should be the device and not the bus. I don't know why but I can isolate the usb dump with.

`I.E usb.device_address == 001`

I have not see something to export dump in json format. After, i use the right click and copy in as hex Dump. 

I get something like this (for the first the first host line (after filtering usb adress) request device).

```
0000   1c 00 00 00 00 00 00 00 00 00 00 00 00 00 0b 00
0010   00 01 00 01 00 80 02 08 00 00 00 00 80 06 00 01
0020   00 00 12 00
```

jonasmalacofilho, 2020-02-17:
@olielvewen 

> To be clear, I have removed and you know why.

Not that you mention it, it rings a bell... but I can't remember (or find it in our emails).  
Can you remind again by email?

> Bus 001 device 004.

Device enumeration and addresses can change from boot to boot, and even more so from OS to OS.

I just opened one of his captures and the cooler appears to be in bus 1, address 3 _(during the capture)._

You can also just use the capture files in the liqudctl-device-data repository, which should have already been properly filtered.

olielvewen, 2020-02-17:
@jonasmalacofilho 

>Now that you mention it, it rings a bell... but I can't remember (or find it in our emails).
Can you remind me again by email?

Done.

>Device enumeration and addresses can change from boot to boot, and even more so from OS to OS.

Yes I know since we have seen that together. ^ __ ^ the last year. But I'm sure that 's the good address. See the lsusb output that I've asked again at Jwjennison there are few days ago. 

>I just opened one of his captures and the cooler appears to be in bus 1, address 3 (during the capture).

`Bus 01 address 01` for me after filtering.

>You can also just use the capture files in the liqudctl-device-data repository, which should have already been properly filtered.

I'll take a look again but more deeply this time. Thoses were well structured one action = one capture

And I'm agree again with you we need more capture files still well structured.

olielvewen, 2020-02-17:
So we make like the last time. Github is a paintful each time that we want to share a file.

Look at here:
[Hex dump files](https://my.pcloud.com/publink/show?code=kZOhgskZN1KLrIaRkxX4nrMbSNi6xjpQ8Oek)

It is just for the ch03.xxxx and LED Blue files.

jonasmalacofilho, 2020-02-18:
@olielvewen 

Actually, (with this cooler) you can simply focus on the following fields in Wireshark: `usb.capdata` (from incoming reports) and `usb.data_fragment` (in outgoing reports).  Everything else is just low level USB/HID stuff.

But you don't have to share manually extracted data from the captures unless you're adding some new information with it (for example, the location of a previously unknown field).  We all have access to the same capture files.

One last thing: in the ch3 capture file bus 1, address 1 is a Genesys Hub, and not the cooler; the cooler is in bus 1, address 3.  You can check this for yourself by looking at the device descriptors sent by each device (GET DESCRIPTOR Response DEVICE), and finding the one with correct idVendor (0x1b1c) and idProduct (0x0c17) fields.

olielvewen, 2020-02-18:
@jonasmalacofilho 
>Actually, (with this cooler) you can simply focus on the following fields in Wireshark: usb.capdata (from incoming reports) and usb.data_fragment (in outgoing reports). Everything else is just low level USB/HID stuff.

Thanks a lot for the tip.

patricksyoussef, 2020-04-20:
Hi everyone! I was wondering if there is any way I could help, I am looking to order an H115i Platinum soon and I wanted to see where this could go before ordering.

jonasmalacofilho, 2020-04-26:
@patricksyoussef 

The initial captures only got me so far, but I was still making some progress using an emulated device (an Arduino with a custom firmware that tells the OS and iCUE that it's a H115i Platinum).  However I haven't been able to touch that since February.

IIRC the most significant issues left were:

 - apparently need to keep sequence numbers
 - some ambiguity between set cooling and set lighting messages
 - a bunch of (apparent) constants that may or may not need to be emulated
 - actually implementing set color, but that's pretty simple (only "static" supported at the hardware level)

olielvewen, 2020-04-28:
@jonasmalacofilho 
Take a look at your mail box. I've replied at our last and long mail and I've send you another mail with a big surprised. And a lot of work done and more a lot of work to do.
I hope to get you in touch quickly. I'm a bit available.

makk50, 2020-05-18:
Hello, thanks for your impressive work. I have this model and I would love to be able to control it from my linux pc, i'm not a developer but I would be more than happy to help you with terst or anything you may ask for

jonasmalacofilho, 2020-06-13:
Can I ask some of you to review (and, at your discretion, test) the code in #82?  I would like to known that the most basic stuff works before implementing the remaining features.

Note that they may be gross errors in the implementation since I don't have any way to test that code.

ollie1, 2020-06-13:
Thanks for your work on this - I'm really excited if I will soon be able to control the fans on my H100i Platinum when I'm running linux.  The status is now working for me:
```
$ sudo liquidctl status
Corsair H100i Platinum (experimental)
├── Liquid temperature      29.7  °C
├── Fan 1 speed             1969  rpm
├── Fan 2 speed             1447  rpm
├── Pump speed              2053  rpm
└── Firmware version      1.1.15  
```

However, unfortunately, setting the fan speed doesn't seem to have any effect
```
# liquidctl status
Corsair H100i Platinum (experimental)
├── Liquid temperature      30.1  °C
├── Fan 1 speed             2001  rpm
├── Fan 2 speed             1471  rpm
├── Pump speed              2055  rpm
└── Firmware version      1.1.15  
# sudo liquidctl set fan1 speed 95
# sudo liquidctl status
Corsair H100i Platinum (experimental)
├── Liquid temperature      30.1  °C
├── Fan 1 speed             2002  rpm
├── Fan 2 speed             1473  rpm
├── Pump speed              2053  rpm
└── Firmware version      1.1.15  
# liquidctl set fan1 speed 5
# sudo liquidctl status
Corsair H100i Platinum (experimental)
├── Liquid temperature      30.1  °C
├── Fan 1 speed             2002  rpm
├── Fan 2 speed             1472  rpm
├── Pump speed              2050  rpm
└── Firmware version      1.1.15  
```

As you can see, there seems to be no change in the speeds of the fans on the status command, and there is no audible volume change from the fans when I run the command.  I don't know if I'm doing something wrong?  Happy to run more tests.

See below for debug output:
```
# liquidctl -g set fan1 speed 5
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (f650d5eb3d54)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CoolitPlatinumDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 047d:8002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c18
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H100i Platinum (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:0a47
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4060
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c318
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c318
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0024
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 2109:0812
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c318
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:0826
[DEBUG] liquidctl.driver.usb: probing drivers for device 2109:2812
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:0a47
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0732
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 0a5c:21e8
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c18
[DEBUG] liquidctl.driver.usb: probing drivers for device 047d:8002
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0024
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: Corsair H100i Platinum (experimental)
[DEBUG] liquidctl.keyval: data in /var/run/liquidctl/1b1c_0c18
[DEBUG] liquidctl.keyval: stored fan1_duty=5 (in /var/run/liquidctl/1b1c_0c18/fan1_duty)
[DEBUG] liquidctl.keyval: loaded fan1_duty=5 (from cache)
[DEBUG] liquidctl.keyval: loaded fan2_duty=95 (from /var/run/liquidctl/1b1c_0c18/fan2_duty)
[DEBUG] liquidctl.driver.coolit_platinum: write 003f0814000000000000000002000000000c0200000100f20000000000000000000000000000000000000000000000000000000000000000000000000000000032
[DEBUG] liquidctl.driver.coolit_platinum: received ff20110f100100101e0000ade803add30700ade803adc10500760000760408000000000000000000000000000000000000000000000000000000000000000080
```

jonasmalacofilho, 2020-06-13:
Made a few adjustments already, let me know if they help in any way.

audiohacked, 2020-06-13:
@jonasmalacofilho I did notice that the `_PUMP_DATA_OFFSET` was wrong. Should be `0x17 - _SET_COOLING_DATA_OFFSET`

jonasmalacofilho, 2020-06-16:
Now asking you to specifically test the code.  Please take a look at [#82 (comment)](https://github.com/jonasmalacofilho/liquidctl/pull/82#issuecomment-644647183).

closed 2020-06-19 by jonasmalacofilho