liquidctl/liquidctl #77: installation issue

thathead, 2020-01-25:
guys,
i have a NZXT Kraken X62 280mm and the CAM software is constantly F'n with me. I only have like 6 days left to return the x62 by mail which is a hassle in itself.

i really want to install liquidctl.exe on windows 10 (1809) but i'm not sure what im doing wrong. I've replaced the driver for the device listed "NZXT USB Device" in zadag with WinUSB (v6/1/7600) but i'm not sure why I cant install liquidctl.exe. 
zadag lists (usb id) 1e71 & 170e and has a red x next to WCID. I've also tried libsub-win32 driver
dunno.
when I right click run as admin or i just double click the executable, runliquidctl.exe, it does the same thing, it flashes a blank cmd like box for half a second then nothing.   I don't know why it will not install.  do i need to make a .bat file or something like with mining software?  i keep thinking that maybe its a permissions thing but I'm not sure what else to try.

any suggestions or advice at this point would be greatly appreciated
TIA

jonasmalacofilho, 2020-01-25:
Hi @thathead,

The `liquidctl.exe` executable is the program itself, and it's a command-line interface.  You can call it directly from any shell  (Windows Command Prompt, PowerShell, etc.) opened in the same directory, or by using its full path.

You can also "install" it by placing it in a directory of your choosing and adding that directory to your `PATH` environment variable, or by placing the executable in a directory already in the `PATH` list.  This will allow you to call `liquidctl.exe` or even `liquidctl` from any native shell and any directory.

---

The driver replacement _is not necessary_ for USB devices that are exposed as Human Interface Devices (HIDs), like the X62.  While liquidctl can work with either driver (HID or WinUSB), I recommend you revert to the default/first.

It's the third time in two days that someone has the impression that a special driver is always necessary, when in fact only a handful of devices need this.  Can I ask if you're following a third-party tutorial (and which one)?

---

Thanks for any feedback here.  I want to improve the documentation and address these installation/setup issues.

jonasmalacofilho, 2020-01-25:
Related: #56

thathead, 2020-01-26:
not third party i dont think
https://github.com/jonasmalacofilho/liquidctl#installing-on-windows

> installing on Windows
> A pre-built executable for the last stable version is available in liquidctl-1.3.2-bin-windows-x86_64.zip.
> 
> Executables for previous releases can be found in the assets of the Releases tab, and development builds can be found in the artifacts on the AppVeyor runs.
> 
> Products that cannot use the generic Microsoft HID Driver require another driver that is compatible with libusb. In most cases Microsoft WinUSB is recommended, which can be easily set up for a device with Zadig.

i put my driver back to the original, its called HidUsb ( v 10.0.17763.719). I assumed that the exe, 
liquidctl.exe,  needed that driver mentioned because when i doubled and right clicked it as admin it did not run but just flashed a black a cmd box for a second.

i don't know what u mean by this:

> The liquidctl.exe executable is the program itself, and it's a command-line interface. You can call it directly from any shell (Windows Command Prompt, PowerShell, etc.) opened in the same directory, or by using its full 

you're  saying that double or right clicking the .exe is not the way to run it, right? if so, say i bring up the blue windows power shell (adim). what commands do i then need to type to gain control over my water cooler's (x68) two fans and pump? or can u point me to a quick tutorial on how to run programs like that?

 so far i just downloaded and extracted the zip to a folder that i use for installation files called apps. the path is E:\apps\liquidctl-1.3.2-bin-windows-x86_64\liquidctl-1.3.2-bin-windows-x86_64
but i can move the folder to program files or just my C drive so its less text

when i ran the CAM software i was able to get my cpu to 71C while using 100% cpu on the performance setting when the program was working.  But It would mostly freeze and crash my pc 2 -3 times a day .  since i removed the CAM software my cpu is very close to 80C when Im using 100% of it. I want to set up my cooler so i hear my fans click on every time my cpu hits like 72C or 73C  because they never seem to click on now that CAM is removed from my system. but the plus side is my pc no longer BSOD.  even when hardly running my cpu or it being idle, like it now at 47C the CAM software would would freeze or cause errors to my system like i could just be watching a flick and time to reboot again

jonasmalacofilho, 2020-01-26:
First, you probably want to follow these steps:

> You can also "install" it by placing it in a directory of your choosing and adding that directory to your PATH environment variable, or by placing the executable in a directory already in the PATH [...]

After this then, yes, open a Power Shell or Windows Command Prompt window and simply follow the [_specific documentation page for the Kraken X62 family_](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/nzxt-kraken-x-3rd-generation.md) (which you can otherwise find listed in the table of supported devices).

For example,

```
liquidctl set pump speed 90
```

will set your pump speed to 90%, fixed.

There are also some more general instructions about the CLI in the main README ([_The command-line interface_](https://github.com/jonasmalacofilho/liquidctl#introducing-the-command-line-interface) section).

thathead, 2020-01-26:
wish there was a gui version.  last time i messed with power shell to run and configure an application it game me a real headache - i dont know enough about the commands.  prob less work to just buy and install a new corsair cooler and send the x62 back to the retailer. 
thanks for taking the time and the info

jonasmalacofilho, 2020-01-26:
> i dont know enough about the commands.
> prob less work to just buy and install a new corsair cooler

Really, once you have `liquidctl.exe` in a folder listed in your `PATH` environment variable, _all_ you'll need to learn is a handful of liquidctl specific commands (which are in the X62-specific document I linked before):

```
liquidctl set fan speed 20 25 34 100
liquidctl set pump speed 24 50 34 100
liquidctl set sync color fixed ffffff
```

The above is my _actual_ setup: both fans and pump run on (different) profiles based on liquid temperature, and the LEDs are set to white (static).

---

I mean, it's totally your choice, and thanks anyway for letting me know some parts of the documentation weren't super clear.

But, really, it's certainly takes less time to learn the syntax of 3 commands above (which, as you can see, are very simple) than to buy and install a different cooler.

jonasmalacofilho, 2020-01-26:
_(for myself)_

Things to improve on the docs for Windows:

 - [ ] more clearly list the only cases where the driver needs to be replaced
 - [ ] add simple installation instructions for users of the pre-built liquidctl.exe
 - [ ] mention (or link to a guide on) how to make the created batch file into a task that runs at every boot

thathead, 2020-01-26:
the extracted files are in a folder called liquidctl
the path is C:\liquidctl

in windows power shell (admin)
PS C:\Windows\system32> dir  liquidctl

this shows me the list of the files. I think i kinda remember im supposed to be using this character - in front of something dunno. i need to figure out "path environment variable".
i got kinda sidetracked when the nhl allstar games came on for a while.

i think i need to figure out how to make this "PS C:\Windows\system32>" say what i need it to say before i can install  but I dont know what thats supposed to look like yet so i dont know where to start reading and then there's prob a run or install command followed by the name of the exe file,  but i think i need to be inside the C:\liquidctl directory where i quoted the system32 folder dunno,  and I'm not sure what that # is doing in front of everything in the link below. or even if i put the folder in the right place. how do i add C:\liquidctl to my PATH environment variable?

https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/nzxt-kraken-x-3rd-generation.md

jonasmalacofilho, 2020-01-26:
> how do i add C:\liquidctl to my PATH environment variable?

Just type environment variable on your taskbar and you should get to the relevant dialog for setting this up.  This allows you to add or edit all environment variables.  Technically `PATH` is a semi-colon (`;`) separated list of directories; but if I remember correctly, Windows 10 now has a GUI to manage the list.

Once that's done you will need to close and reopen any Windows Command Prompt/Power Shell windows.  Sometimes a restart can also be necessary, but this is not common anymore on Windows 10.

If that works, just typing `liquidctl --version` on a Command Promp/Power Shell should return a sensible result (for example: `liquidctl v1.3.1`).

> i think i need to figure out how to make this "PS C:\Windows\system32>" say what i need it to say before

That just tells you your current directory.  And if the liquidctl.exe is in a directory which is listed in the `PATH` variable, then the current directory doesn't matter, as `liquidctl` will be accessible everywhere.

> I'm not sure what that # is doing in front of everything in the link below

On Windows you can just take it as a meaningless prefix only used to differentiate between commands you type and their output.

jonasmalacofilho, 2020-01-26:
**Note:** you want to _append_ the `C:\liquidctl` directory to the `PATH` list (the directories are searched in the order they appear in the list, so the order matters).  Also, make sure you don't mess with existing paths in the `PATH` list, as that would break other programs.

thathead, 2020-01-26:
i see environmental variables. theres a box for user and a box for system. new edit delete

what box do i add the directory C:\liquidctl to? system right?

what is variable name? and variable value

jonasmalacofilho, 2020-01-26:
_Variable name:_ `PATH`  
_Variable value:_ a list of directories
_System or user variable:_ either,¹ but I personally use system

Perhaps you might want to find and follow a specific guide on how to manipulate environment variables on Windows (and, more specifically, the `PATH` variable).

The `PATH` variable is very important for your system to properly function, and if the Windows GUI for setting this up is no longer as clear as it was/should, I no longer feel very comfortable giving you remote advice on this.

---

_¹ If you choose a user variable then you have to run Command Prompt/Power Shell as your user.  I mention this because before you said you were running them as Administrator._

thathead, 2020-01-26:
yeah, i dont have the computer science pre req it takes to install liquidctl.  I gotta crash. 

maybe I'll look for a youtube tutorial on how to install apps with powershell command line tomorrow 

late

thathead, 2020-01-26:
i added C:\liquidctl to path. not sure why it wasnt clicking for me before

commands see to be working. i can hear the fans.

liquidctl set pump speed 90
liquidctl set fan speed 75

my temp says 75C with cpu 100%
fan at 85 seems better 74C
liquid temp at 32.5

be nice if i could keep the chip around 71C without having to max both the pump and fans.

i wonder if its ok to keep the pump at 100% all the time

jonasmalacofilho, 2020-01-26:
> be nice if i could keep the chip around 71C without having to max both the pump and fans.

The maximum performance you'll get from the cooler will be limited by the cooler itself, its fans and pump when both are maxed out, and your case airflow and ambient temperature.

But there's no reason to run your pump and fans at full speed _all the time._  You can use profiles to have them ramp up according to the liquid temperature:

```
liquidctl set pump speed 24 50 34 100
```

translates to "set the pump speed to follow the profile:"

 - bellow or at 24°C, set the pump speed to 50% duty
 - between 24°C and 34°C, interpolate¹ between 50% and 100% duty
 - at or above 34°C, set the pump speed to 100% duty

These profiles run on the cooler itself, so once they are set you don't have to worry about a software crash causing the cooler to get stuck at potentially too low speeds.²

And while the liquid temperature is not directly related to the CPU temperature, it has the advantage of dampening shorts bursts of activity.

> i wonder if its ok to keep the pump at 100% all the time

There's no statement from any manufacturer whether you should or shouldn't do that.

And the most important factor for reliability might be liquid temperature (at the impeller), not flow speed.

---

_¹ Technically liquidctl interpolates the profile before sending it to the cooler, as the cooler itself doesn't have that capability.  Essentially we send a much larger number of points, without requiring you to specify each one yourself._

_² Just remember that the cooler potentially resets all settings at every boot.  You need to automate your configuration at boot time, and on Windows the best way to do that is [_with a .bat batch set up to run on login with Windows Task Scheduler._](https://github.com/jonasmalacofilho/liquidctl#set-up-windows-using-task-scheduler)  As the lighting resets too, if you use a custom color and/or animation, that can also tell you if your profiles have reset._

thathead, 2020-01-27:
yeah i need a bat file for startup folder but the 1st three combos im not feeling at 100% CPU

once like this =76C
> liquidctl set pump speed 24 50 34 100
> liquidctl set fan speed  20 30  30 50  34 80  40 90  50 100

once like this =76C
> liquidctl set pump speed 90
> liquidctl set fan speed  20 30  30 50  34 80  40 90  50 100

and once like this =75C
iquidctl set pump speed 24 50 34 100
liquidctl set fan speed  90 (could hear fan on this try)

the third one maybe would of got lower if i wated a few more minutes.  but the fan would not click on to 90 or 100 with the first two. i couldn't  hear it.

this was 74C
> liquidctl set pump speed 90
> liquidctl set fan speed  90

this seems to work best. I just dont want to run the pump at 100 if itll increasing the chance of leakage. but i havent found anything saying anything like that yet
> liquidctl set pump speed 100
> liquidctl set fan speed  90

my colors are purple and blue
and the movement is one color chasing the other color around the rim. i forget what its called. im not really too worried about the colors

> Set up Windows using Task Scheduler

> The configuration of devices can be automated by writing a batch file and setting up a new scheduled task for (every) log on. The batch file can be really simple and consist of the invocations of liquidctl that would otherwise be done manually.

i need to figure out how to do that. is the bat just 4 lines?
pump
fan
color
logo

not sure about this

> Make sure that liquidctl is available in the context where the batch file will run: in short, liquidctl --version should work within a normal Command Prompt window.

is there any pics of what txt is in the bat file? im not sure how to use task scheduler?  can i just copy paste a bat file into the start or start up folder

jonasmalacofilho, 2020-01-27:
> the third one maybe would of got lower if i wated a few more minutes. but the fan would not click on to 90 or 100 with the first two. i couldn't hear it.

This is caused because 50°C (and even 40°C) isn't a realistic liquid temperature for a X62 (assuming mainstream processor, reasonable airflow and sensible ambient temperature).

You probably want to max out (or close to) your fans and pump somewhere between 35°C and 40°C to get the performance you want.  
_(or maybe even less... I max out at 34°C, and my ambient temperature is not even that low)_

> this seems to work best. I just dont want to run the pump at 100 if itll increasing the chance of leakage. but i havent found anything saying anything like that yet

Yeah, I also don't think that's a major factor in a leak.

> my colors are purple and blue
> and the movement is one color chasing the other color around the rim. i forget what its called. im not really too worried about the colors

Sure, just meant that you can use that as a visual indicator of whether the cooler is configured or not.

> not sure about this
> 
> > Make sure that liquidctl is available in the context where the batch file will run: in short, liquidctl --version should work within a normal Command Prompt window.

That's already working, since you're able to run `liquidctl` on any Command Prompt/Power Shell window and directory.
 
> is there any pics of what txt is in the bat file?

Just the commands you want to execute, just like you would otherwise type on the command line.

> im not sure how to use task scheduler?

It's fairly intuitive (only tip: you want to run your batch at "login").  And it tells you whether things ran or failed the last time.

> can i just copy paste a bat file into the start or start up folder

Sure, that's a different way to do it.

jonasmalacofilho, 2020-01-29:
Hope you've managed to fine tune your setup by now.

I'm going to close issue since I've now updated the documentation to address the problems you encountered.  Still, please feel free to continue the discussion here.

closed 2020-01-29 by jonasmalacofilho