liquidctl/liquidctl #78: Support for NZXT Kraken X63 and X73

jonasmalacofilho, 2020-01-29:
I imagine there will be a significant amount of interest in this new iteration of Kraken X coolers.  They are a replacement to the popular X42/X52/X62/X72 family, and I'm also guessing that NZXT will soon stop selling that previous generation.

https://www.nzxt.com/products/kraken-x63
https://www.nzxt.com/products/kraken-x73

On the technical side, I'm mostly certain that these devices use a new protocol, mostly to accommodate the added HUE 2 header and functionality.  Hopefully hardware-level fan & pump profiles are still supported, and maybe even improved with interpolation.

Please add a :+1: if you are interested in support for either of these devices.

Additionally, if you have one of these, please get in touch (a comment is fine, but you can also reach me by email) so we can work together on the new driver.

---

Status:

 - [ ] Collect basic USB data for both devices (at least vendor and product IDs)
 - [ ] Inspect and analyze the new protocol (monitoring, cooling and lighting)
 - [ ] Implement and test the new driver

jonasmalacofilho, 2020-02-07:
Hardware fan (speed) control is certainly *not* supported, since there are is no longer a fan (speed) controller onboard.

Users are expected to connect the fans to the motherboard or an external controller, loosing the ability to use the liquid temperature as the control sensor.

That's disappointing...

mijoe, 2020-02-15:
Had a X63 for testing purposes and dumped the output of `lsusb`:

[NZXT_Kraken_X63.txt](https://github.com/jonasmalacofilho/liquidctl/files/4209002/NZXT_Kraken_X63.txt)

Sidenote: I return the X63 due to the missing fan control. Also pump can not be controlled via CPU FAN header (its only connected for speed signal).

That's very disappointing...

jonasmalacofilho, 2020-02-15:
@mijoe 

Thanks!

> Sidenote: I returned the X63 due to the missing fan control. Also pump can not be controlled via CPU FAN header (its only connected for speed signal).
> 
> That's very disappointing...

The pump also wasn't controllable via the fan header in the X62, but at least you could control both it and the fans with software.

But the X63 is a mess: the pump can only be controlled from software, while the fans can only be controlled from the motherboard.  Plus it's no longer possible to control the fans based on the coolant's temperature.

yfuerst, 2020-02-20:
I am very interested in support for the newer models. I have the X53 installed and I am willing to provide logs. Just tell me what you need.

jonasmalacofilho, 2020-02-21:
@yfuerst 

Great!  Can you take a look at [this answer](https://github.com/jonasmalacofilho/liquidctl/issues/79#issuecomment-585456831) and post/send me some initial traffic captures?

yfuerst, 2020-02-23:
This is the output from lsusb: [x53.log](https://github.com/jonasmalacofilho/liquidctl/files/4241816/x53.log)

Here are my captures: [pcapng-files.zip](https://github.com/jonasmalacofilho/liquidctl/files/4241827/pcapng-files.zip)
First file: I stopped and started the CAM software
Second and third file: I changed the profile from fixed to silent and from silent to performance

This was the first time I used Wireshark. So errors on my part are possible ;)

jonasmalacofilho, 2020-02-24:
@yfuerst 

Thanks!  Oh, and the captures look fine : )

I don't have the time to analyze this in depth right now.  Do you want to give it a try?  I could tell you the strategies I use to figure out what's what.

yfuerst, 2020-02-24:
Sure, I can give it a try, but I definitely need instructions.

jonasmalacofilho, 2020-03-01:
@yfuerst 

Great!  I'll take some time to write a bit how I analyze these capture logs.

yfuerst, 2020-03-09:
@jonasmalacofilho Great, I am looking forward to it.

tristan-k, 2020-03-22:
I recently bought a X53. If there is anything I can do just holler.

tomaculum, 2020-03-25:
I bought a X73 and would be happy to help, too!

mimfgg, 2020-03-27:
just got a x53 as well, I can help :)

jonasmalacofilho, 2020-03-28:
Thank you all for volunteering to help here.

I'm sorry I still haven't been able to [write a bit about how to analyze these protocols and develop a new driver](https://github.com/jonasmalacofilho/liquidctl/issues/78#issuecomment-593090215)... but it's on the top of my to-do queue now, so please hang in there.

tomaculum, 2020-03-29:
I've started with the prototype driver and did use your open pull request for the Corsair CoolIt Platinum as reference. I am able to read the liquid temperature and pump speed: [changes](https://github.com/jonasmalacofilho/liquidctl/compare/master...tomaculum:master)

Unfortunately I found your [liquidctl-device-data](https://github.com/jonasmalacofilho/liquidctl-device-data/tree/master/NZXT%20Kraken%20X63) repo after spending my day with this but I can confirm that product id is the same on the X73, firmware version is in my case v2.1, your assumption regarding the coolant temperature is correct and pump speed is just flipped: 
`buf[0x12] << 8 | buf[0x11]`.

jonasmalacofilho, 2020-03-30:
@tomaculum That's great news!

tomaculum, 2020-03-31:
Here is a quick summary of the message types I've found:

### startup

> r = read, w = write

| data[0] | data[1] | direction  | type |  |
| :------------- |:-------------:| :-----:| :----- | :----- |
| `0x75` | `0x02` | r | device status | received periodically
| `0x10` | `0x01` | w | request firmware info |
| `0x20` | `0x03` | w | request lightning info |
| `0x70` | `0x02` | w | initialize 1 |
| `0x11` | `0x01` | r | firmware info response |
| `0x70` | `0x01` | w | initialize 2 |
| `0x71` | `0x01` | r | unknown response | only with initialize |
| `0x21` | `0x03` | r | lightning info response |

_+ set colors, set pump speed_

* in CAM the description label says ring is channel 2 and logo channel 3 and there seems to be no channel 1? But the startup sequence will set a color for three devices: `0x01`, `0x02`, `0x04`.

<details><summary>show examples: </summary>
<p>

```
# request firmware info
10010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
# request lightning info
20030000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
# initialize 1
700201b80b0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
# initialize 2
70010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
# firmware info response
1101180040000c574154333634200720020201000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
# unknown response 
7101180040000c574154333634200100010100000200000000000000000000000000000000000000000000000000000000000000000000000000000000000000
# led / device info
2103180040000c574154333634200300000000000010000000000011000000000000000000000000000000000000000000000000000000000000000000000000
```

</p>
</details>

### pump speed
| data[0] | data[1] | direction  | type |  |
| :------------- |:-------------:| :-----:| :----- | :----- |
| `0x75` | `0x02` | r | device status | received periodically
| `0x72` | `0x01` | w | set pump mode |

**set pump mode:**
* this is really strange, if you select performance, it will send many `0x46` values (= 70%) to the pump. If I select the silent profile, it will send `0x3c` values (= 60%). But if I select a fully custom curve, it still sends only one percent value instead of the curve. 
* in addition to that it seems to make no difference if I select the liquid or cpu temperature as source for the curve. 

<details open><summary>show examples: </summary>
<p>

```
# change mode to silent
720100003c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c0000000000000000000000000000000000000000
# change only temperature mode from cpu to liquid
720100003c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c3c0000000000000000000000000000000000000000
# change mode to performance
72010000464646464646464646464646464646464646464646464646464646464646464646464646464646460000000000000000000000000000000000000000
# change mode to custom
72010000474747474747474747474747474747474747474747474747474747474747474747474747474747470000000000000000000000000000000000000000
```

</p>
</details>

### color
| data[0] | data[1] | direction  | type |  |
| :------------- |:-------------:| :-----:| :----- | :----- |
| `0x75` | `0x02` | r | device status | received periodically
| `0x2a` | `0x04` | w | set color |
| `0xff` | `0x01` | r | color response |

**set color:**
* `data[2]`, `data[3]` could be the device address: `0x02`, `0x02` is ring and `0x04`, `0x04` is logo
* `data[7]`, `data[8]`, `data[9]` seems to be the RGB values of the selected color but in another order: GRB? 
At least if I set the color manually to `#FF9600` it results in: 
`data[7] = 96`, `data[8] = ff`, `data[9] = 00`
* `data[4]` is the color mode which seems to be equal to the smart device v2 and kraken X2 codes

<details open><summary>show examples </summary>
<p>

```
# ring to fixed color white:
2a040202003200ffffff000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000100080300000000
# ring fixed color brightness to low:
2a0402020032000f0f0f000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000100080300000000
# ring fixed color change to #FF9600:
2a04020200320096ff00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000100080300000000
# ring fixed to fading three colors:
2a04020201280000517aff00000000ff000000000000000000000000000000000000000000000000000000000000000000000000000000000308080300000000
# logo to fixed color white:
2a040404003200ffffff000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000100010300000000
# logo fixed color brightness to low:
2a0404040032000a0a0a000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000100010300000000
```

</p>
</details>

**color response:**
* `data[2-14]` is equal to the status response and `data[15]` seems to be always `0x04`, the rest is filled with `0x00`

<details><summary>show examples: </summary>
<p>

```
# ring to fixed color white:
ff01180040000c574154333634202a04000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
# logo fixed color brightness to low:
ff01180040000c574154333634202a04000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

</p>
</details>

### Questions
@jonasmalacofilho 
1. Does the pump speed write request look equally to those in X2 devices or do you know what this percentage value is about?
2. The _set color_ request seems very strange. Is it possible that this a similar structure like the X2 device _set_color_ request? If yes, I would start trying to understand what this _set_color_ method does.

jonasmalacofilho, 2020-03-31:
> @jonasmalacofilho
> 
> 1. ~Does the pump speed write request look equally to those in X2 devices or do you know what this percentage value is about?~

No, it's very different, particularly these "modes".

Note: CPU temperature isn't available at the hardware level.

> 
> _**Edit:**_
> I've found out that CAM sends the new percentage values dynamically. The pump doesn't know the whole curve and only knows its current percentage. Previously my PC pretty much just idled in Windows therefore I couldn't observe this.

Indeed, this looks like it's just sending many times a given duty value (e.g. `0x3c`, prob. meaning 60% though the encoding could be different).

However, the fact that it's repeated that many times suggests that the hardware does support some time of liquid vs. duty curve.  Unlike the X62, it seems that the temperature is implicit (i.e. preset at the controller), instead of explicitly set by CAM/the user.

I count 40 "points" (duty values).  This means that they also aren't using the same default temperature steps as in the X62 (5°C steps from 0°C to 100°C).  The fact that it's an even number is also a bit weird, but it could mean that at the critical temperature (previously 60°C) the duty is also implicitly 100%.

Long story short, _maybe_ the controller allows you individual duty points from 20°C to 59°C.

> 1. Is this something new or did the X2 devices did have a fixed pump speed outside of Windows, too?

CAM used to set a curve only when the user selected the profile to be based on the liquid temperature (because the controller doesn't have access to any other sensor), and regardless of that would send updates every 1 or 2 s.  This had probably to do with the X62 not interpolating and, by default, using that horrible 5°C steps (for no reason, as liquidctl can set it to use 1°C steps just fine).

Either way, reports from r/NZXT tell me that CAM has changed and no longer sets the curve; as you noticed, this can be tested by checking what happens after you close CAM.

> 2. The _set color_ request seems very strange. Is it possible that this a similar structure like the X2 device _set_color_ request? If yes, I would start trying to understand what this _set_color_ method does.

From your description, it is _not_ similar to the X62.

I haven't looked at it in depth, but I expect it to be similar to the HUE 2/Smart Device V2.  The initial messages seem to match (e.g. detection of connected accessories).  Take a look at the [SmartDeviceV2Driver](https://github.com/jonasmalacofilho/liquidctl/blob/70f6b899b82466a9c744e894f76b8916cd62b729/liquidctl/driver/nzxt_smart_device.py#L318).

---

P.S. When I write X62 I mean all third-generation/suffix-2 Kraken X coolers (X42, X52, X62 and X72).

tomaculum, 2020-04-01:
First of all, thank you for your quick response!

> Indeed, this looks like it's just sending many times a given duty value (e.g. 0x3c, prob. meaning 60% though the encoding could be different).

I've tested this with different curves in CAM and it is always in this format with the current duty value as hex value. I did implement this into the _set_fixed_speed_ method and it does work well so far. 

> Long story short, maybe the controller allows you individual duty points from 20°C to 59°C.

The _device status_ response does contain two bytes which represents the current pump duty value. Currently they are always equal but maybe with those individual duty points this would change and make more sense.

> Take a look at the SmartDeviceV2Driver.

This did help a bit but the _set color_ requests are completely different at least for the most common `0x2a 0x04` requests. 
The multipart request for individual led color changes do look the same. But its second request will never contain any color values since you can only assign eight colors and they will always fit into the first request. 

Today I did focus on the first four color modes but did not find a matching pattern yet. If I can't find a pattern tomorrow, I will post some tables and examples and ask for help again.

@mimfgg @tristan-k @yfuerst 
Does anyone use the RGB LED connector thing of the pump? Is this for the smart device or do you control like LED devices with this?

mimfgg, 2020-04-01:
No, I have the x53 on my 1080ti using a g12 ... so I wouldn't see the pump ...

tomaculum, 2020-04-02:
Ok, I think I did find out enough to implement the single message color changes:

| byte | purpose | |
| :--- | :--- | :--- |
| `0` | op code | always `0x2a` |
| `1` | op code | always `0x04` |
| `2` | address | `0x02` - ring <br>`0x04` - logo  |
| `3` | address | `0x02` - ring <br>`0x04` - logo  |
| `4` | mode | mostly as in SmartDevice V2 driver |
| `5` | speed value | see tables below
| `6` | speed value | 
| `7` | color 1 G value | 
| `8` | color 1 R value | 
| `9` | color 1 B value | 
| `10` | color 2 G value | 
| `11` - `54` | … | 
| `55` | backwards byte | some mode related byte<br>backwards mode: `value` + `0x02`<br>`0x01`: alternating moving, starry night<br>`0x04`: marquee, covering marquee<br>`0x00`: everything else
| `56` | number of colors in request | e.g. mode _fixed_ will pass one color: `0x01`
| `57` | some kind of mode related byte | always `0x00` <br> except  <br>`0x08` fading, pulse, breathing <br>`0x05` water cooler, tai chi  <br>`0x04` loading
| `58` |  static value - maybe available leds? | `0x08`- ring, `0x01`- logo |
| `59` |  _marquee_ or _alternating_ LED size | always `0x03`<br>except in marquee and alternating mode for LED size:<br>`0x03` - 3 LEDs<br>`0x04` - 4 LEDs<br>`0x05` - 5 LEDs<br>`0x06` - 6 LEDs

### Question:
Only thing I don't like much are the speed bytes. I kind of hope to find a pattern to avoid hardcoding all those values. Does anyone find a pattern or grouping which would reduce this?

**scale 1:** fixed, candle
| speed | `data[5]` | `data[6]` | base10 |
| :--- | :---: | :---: | :---: |
| slowest | - | - |  |
| slow | - | - |  |
| normal | `0x32` | `0x00` | `50` |
| fast | - | - |  |
| fastest | - | - |  |

**scale 2:** fading
| speed | `data[5]` | `data[6]` | base10 |
| :--- | :---: | :---: | :---: |
| slowest |`0x50` | `0x00` | `80` |
| slow | `0x3c` | `0x00` | `60` |
| normal | `0x28` | `0x00` | `40` |
| fast | `0x14` | `0x00` | `20` |
| fastest | `0x0a` | `0x00` | `10` |

**scale 3:** spectrum wave, marquee, covering marquee, rainbow flow, rainbow pulse
| speed | `data[5]` | `data[6]` | base10 |
| :--- | :---: | :---: | :---: |
| slowest |`0x5e` | `0x01` | `350` |
| slow | `0x2c` | `0x01` | `300` |
| normal | `0xfa` | `0x00` | `250` |
| fast | `0x96` | `0x00` | `150` |
| fastest | `0x50` | `0x00` | `80` |

**scale 4:** alternating
| speed | `data[5]` | `data[6]` | base10 |
| :--- | :---: | :---: | :---: |
| slowest |`0x40` | `0x06` | `1600` |
| slow | `0x14` | `0x05` | `1300` |
| normal | `0xe8` | `0x03` | `1000` |
| fast | `0x20` | `0x03` | `800` |
| fastest | `0x58` | `0x02` | `600` |

**scale 5:** alternating moving
| speed | `data[5]` | `data[6]` | base10 |
| :--- | :---: | :---: | :---: |
| slowest |`0x20` | `0x03` | `800` |
| slow | `0xbc` | `0x02` | `700` |
| normal | `0xf4` | `0x01` | `500` |
| fast | `0x90` | `0x01` | `400` |
| fastest | `0x2c` | `0x01` | `300` |

**scale 6:** pulse, starry night
| speed | `data[5]` | `data[6]` | base10 |
| :--- | :---: | :---: | :---: |
| slowest |`0x19` | `0x00` | `25` |
| slow | `0x14` | `0x00` | `20` |
| normal | `0x0f` | `0x00` | `15` |
| fast | `0x07` | `0x00` | `7` |
| fastest | `0x04` | `0x00` | `4` |

**scale 7:** breathing, water cooler
| speed | `data[5]` | `data[6]` | base10 |
| :--- | :---: | :---: | :---: |
| slowest |`0x28` | `0x00` | `40` |
| slow | `0x1e` | `0x00` | `30` |
| normal | `0x14` | `0x00` | `20` |
| fast | `0x0a` | `0x00` | `10` |
| fastest | `0x04` | `0x00` | `4` |

**scale 8:** tai chi
| speed | `data[5]` | `data[6]` | base10 |
| :--- | :---: | :---: | :---: |
| slowest |`0x32` | `0x00` | `50` |
| slow | `0x28` | `0x00` | `40` |
| normal | `0x1e` | `0x00` | `30` |
| fast | `0x14` | `0x00` | `20` |
| fastest | `0x0a` | `0x00` | `10` |

yfuerst, 2020-04-02:
@tomaculum I am not using any LED RGB feature. The AIO is installed in a Ghost S1.

jonasmalacofilho, 2020-04-19:
> I did a quick test with five 50% values and 35 100% values and the pump speed adjusted as you predicted after passing 25°C to 100%. Afterwards I did set the first six values to 50% and it did adjust the pump speed after passing 26°C.

That's great news!

> Do you know a way to check higher temperatures or is it safe to assume this would apply all the way up to 59°C?

With the cooler unmounted from the CPU you could artificially (and safely) increase the coolant temperature with an external heat source (air dryer, heat gun, etc.).  But I know that's a considerable hassle...

Installed, the best you can do if check if this still works with slightly higher coolant temperatures, by stressing your CPU (and, if the radiator is mounted as case exhaust, the GPU as well) while keeping the fan speeds low.

With this you might be able to check that our hypothesis still holds up to a 35°C-40°C coolant temperature; while that's still only a small part of the full profile, it would show that the behavior is indeed linear, even across larger variations in temperature (at least up to that point).

ardevd, 2020-04-23:
I have an X63. Is there any info required that I can provide?

wamphlett, 2020-04-25:
@tomaculum I have just installed an X53 with two RGB HUE 2 strips connected directly to it. I've got this installed on a linux box - defaults to all white lights and it would be great to be able to adjust the lighting (even if its just on and off). Im new to this so my knowledge is limited but Im keen to help out if I can so if there is anything I can provided, just give me a shout and if you have any resources on how I can get started, that would be appreciated!

jonasmalacofilho, 2020-04-26:
@wamphlett 

Can you start by installing the latest code in #100 (fork/branch ['tomaculum/master'](https://github.com/tomaculum/liquidctl/tree/master))?

Let me know if you need any assistance.

wamphlett, 2020-04-27:
@jonasmalacofilho All installed and working. 

```
liquidctl list
Device ID 0: NZXT Kraken X3 Pump (X53, X63 or X73) (experimental)
```
```
liquidctl initialize
NZXT Kraken X3 Pump (X53, X63 or X73) (experimental)
├── Firmware version                   1.8.0
├── LED 1 accessory 1         HUE+ LED Strip
├── LED 1 accessory 2         HUE+ LED Strip
├── LED 2 accessory 1    Kraken X3 Pump Ring
└── LED 3 accessory 1    Kraken X3 Pump Logo
```
```
liquidctl status
NZXT Kraken X3 Pump (X53, X63 or X73) (experimental)
├── Liquid temperature    26.1  °C
├── Pump speed             888  rpm
└── Pump duty               25  %
```

@tomaculum looking at this comment [kraken_three_x.py#L175](https://github.com/tomaculum/liquidctl/blame/master/liquidctl/driver/kraken_three_x.py#L175) - I have 2 strips connected so if there is anything I can do or any info I can provide, let me know  - happy to help.

mimfgg, 2020-04-27:
seems to work for me as well!

`
liquidctl list  
Device ID 0: NZXT Kraken X3 Pump (X53, X63 or X73) (experimental)

liquidctl initialize
NZXT Kraken X3 Pump (X53, X63 or X73) (experimental)
├── Firmware version                   2.1.0  
├── LED 2 accessory 1    Kraken X3 Pump Ring  
└── LED 3 accessory 1    Kraken X3 Pump Logo  

liquidctl status    
NZXT Kraken X3 Pump (X53, X63 or X73) (experimental)
├── Liquid temperature    33.1  °C
├── Pump speed            1061  rpm
└── Pump duty               30  %
`

jonasmalacofilho, 2020-04-27:
@wamphlett 

Thanks.

I notice that you have an older firmware version (1.8.0).  Are you on Linux?  Can you provide the output of `lsusb -d 1e71:2007`?  If you're are on another platform all I want to see is the `bcdDevice` number/revision, which you should be able to find in the properties of the USB device.

---

@mimfgg 

Thanks for testing this as well.

wamphlett, 2020-04-27:
@jonasmalacofilho Yeah Ubuntu 18.04.4 LTS. 

`lsusb -d 1e71:2007` gives me `Bus 003 Device 002: ID 1e71:2007 NZXT`

jonasmalacofilho, 2020-04-27:
@wamphlett , sorry, forgot to mention the `-v` flag is necessary too.

```
lsusb -d 1e71:2007 -v
```

wamphlett, 2020-04-27:
Thought thats what you might have wanted, should have done it anyway

```
Bus 003 Device 002: ID 1e71:2007 NZXT
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x2007
  bcdDevice            2.00
  iManufacturer           1
  iProduct                2
  iSerial                 3
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     412
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

yfuerst, 2020-04-28:
Works for me as well. I just changed the pump speed to 50%. Thanks to everyone involved.
```
liquidctl status           
NZXT Kraken X3 Pump (X53, X63 or X73) (experimental)
├── Liquid temperature    32.1  °C
├── Pump speed            1570  rpm
└── Pump duty               50  %
```

tomaculum, 2020-04-28:
@wamphlett Thank you for your help! Are you able to boot into windows and record the communication between the pump and cam software with wireshark?

If yes, did you ever use wireshark or should I also create a step by step guide?

wamphlett, 2020-04-28:
@tomaculum no problem. I dont have Windows installed at the moment as the box is just running as a home server but I do have a spare SSD lying around so Ill put that in tonight after work and install Windows. 

I've only ever used Wireshark for WAN traffic, never to record data to USB devices so any pointers would be good 👍

mimfgg, 2020-04-28:
I have windows 10 running for games that don't run on Linux, I can do it. 
What do you want to see? Any specific thing I should do with Cam?

inglor, 2020-04-28:
> If yes, did you ever use wireshark or should I also create a step by step guide?

That would be useful in general @tomaculum and maybe commit in `docs`. Personally when I did it I kinda like wing it :rofl: so I'm not sure If I did it correctly.

tomaculum, 2020-04-28:
@wamphlett 
That would be really great! 

@mimfgg 
The part with the additional LED accessories would be the most interesting part since I cannot record this myself. But there are some other things which are worth to validate with the X53 and X63 devices. 

@inglor 
Great idea, maybe this is something @jonasmalacofilho could use as base to accomplish https://github.com/jonasmalacofilho/liquidctl/issues/78#issuecomment-593090215

Alright I will create this guide along with a list of all interesting operations in cam tomorrow.

mimfgg, 2020-04-28:
ah, sorry, I didn't get that ... I don't have the LED part :)

wamphlett, 2020-04-28:
@tomaculum Okay, leave it with me, Ill do that tonight. I actually have some of the compatible [AER RGB 2 fans](https://www.nzxt.com/products/aer-rgb-2) in a box somewhere too so I could plug those in and record whats sent to them as well if thats of interest. 

Worth noting that @jonasmalacofilho also mentioned [here](https://github.com/jonasmalacofilho/liquidctl/pull/100#issuecomment-620508462) about my 1.8.0 firmware. Once I get CAM up and running, Im assuming it will upgrade the firmware so if there is anything you need me to get on the old firmware, I will need to do that first.

tomaculum, 2020-04-28:
# Install Wireshark

1. Download Wireshark: https://www.wireshark.org/download.html
2. Be sure to select **"install USBPcap"** in setup wizard.

#### Question:
_I don't remember if there is anything else worth mentioning?_

# Prepare CAM

1. Start / open CAM.
2. Download the attached _info.txt_ file and start filling out the first three lines: [info.txt](https://github.com/jonasmalacofilho/liquidctl/files/4547632/info.txt) 
For example:
```
kraken device: X73
CAM version: 4.6.0
firmware version: 2.01
```

You can find the CAM version in `CAM -> settings -> bottom left corner`. And the kraken firmware version in `CAM -> settings -> NZXT Firmware`.

3. Reset the cooling mode to performance or silent and edit this line in _info.txt_.
```
cooling mode: performance
```

4. Reset the lighting for example to mode fixed and color white and edit this line as well. If you have any additional accessories, please add those as well.
```
logo lighting mode: fixed
logo colors: #FFFFFF
ring lighting mode: fixed
ring colors: #FFFFFF
```

# Prepare Wireshark

1. Open Wireshark and double click on "USB Pcap 1" in section "Capture".
[image]
2. After accepting the admin popup you should see your usb traffic.

[image]

3. Now apply a filter to just see the traffic of your pump:

   1. Search for a device which have device leftover data starting with `7502`

   2. Apply following filter: `usb.src == "1.8.1" || usb.dst == "1.8.1"` and replace `1.8.1` with your own usb device address.
   
[image]

   3. Add the usb device address from your filter to the _info.txt_ file:
```
usb device address: 1.8.1
```

#### Question:
_I am sure there is a better way to determine the usb device address? And is the `Leftover Capture Data` column enabled by default?_

# Capture

1. Open Wireshark and CAM.
2. Reset CAM (and wireshark) to preparation settings.
3. Start capturing your USB traffic with the "blue shark fin" icon.
4. Wait until you see the the first `7502` message.
5. Perform one operation from the table below.
6. Wait until you see two consecutive `7502` messages and stop the capture.
7. Select `file -> save as` and enter filename from the table as well.

# Operation Table

| filename | operation | info |
| :--- | :--- | :--- |
| startup | close CAM <br>start wireshark capture <br>open CAM | first and only operation which requires to start CAM while capturing
| ring_fixed_blue | set ring color to fixed with blue preset color |
| **todo** | |

wamphlett, 2020-04-28:
@tomaculum `usb.capdata` is a custom column. Im set up and ready to go so once you're finished, Ill run through the operations and get them back over to you

tomaculum, 2020-04-28:
@wamphlett 
Great, thanks!

> Okay, leave it with me, Ill do that tonight. I actually have some of the compatible AER RGB 2 fans in a box somewhere too so I could plug those in and record whats sent to them as well if thats of interest.

I would assume that they are using the same protocol as with your led strips. Therefore it is not worth the effort I think.

> Worth noting that @jonasmalacofilho also mentioned here about my 1.8.0 firmware. Once I get CAM up and running, Im assuming it will upgrade the firmware so if there is anything you need me to get on the old firmware, I will need to do that first.

There is a "automatically update firmware" checkbox in CAM settings but I don't know if it updates the firmware without asking on first start of CAM. But since he asked for it, I would not take the chance and wait until he responds. 

After he responded and you start capturing, could you please send me a screenshot of your lighting section as the one below? This would be helpful to know which channel and LED stuff is hidden in the captured messages. 

[image]

jonasmalacofilho, 2020-04-29:
@wamphlett 

> Worth noting that @jonasmalacofilho also mentioned here about my 1.8.0 firmware. Once I get CAM up and running, Im assuming it will upgrade the firmware so if there is anything you need me to get on the old firmware, I will need to do that first.

Until recently at least the firmware can be updated without user consent, just by opening CAM.

I think you should wait for @tomaculum to enable the [guessed 'accessory' and 'sync' channels](https://github.com/jonasmalacofilho/liquidctl/pull/100#discussion_r415420483) before opening CAM.  It would be useful to know if we managed to get all lighting channels working well even on 1.8.0 firmware.

In the mean time, can you tell us if liquidctl can already control the ring/logo lighting as expected?

---

@tomaculum 

Great work writing these instructions on capturing USB traffic.  Thanks!

P.S. Last I checked there was indeed a checkbox for automatic updates, but it could not be disabled because... reasons.

jonasmalacofilho, 2020-04-29:
Can ask some of you to check which pins are populated in the female fan-style connector used to connect the pump to CPU_FAN on motherboard?  Please also let us know which model you have.

We're interested in checking whether it follows the previous generation and only has the sense and ground pins populated.

P.S. You can also post a picture of the cable right where it reaches the connector.

iH8c0ff33, 2020-04-30:
Hi @jonasmalacofilho,
I have a Kraken X73 and it has a 3 pin connector (i.e. no PWM control), only speed reporting and the middle 12V power pin is not populated.
By the way, I also have an X63 but I haven't installed it yet (still in the box), should I check also the connector on that one?

wamphlett, 2020-04-30:
@tomaculum / @jonasmalacofilho sorry, its taken me slightly longer to get this done. So long story short, I dont have a GPU in the box with the X53 so I ended up connecting it to my other computer - when I did, CAM was open and updated the firmware before I had a chance to stop it so unfortunately, I dont have `1.8.0` anymore!

In better news, I have got all the recordings you need, you can [download them here](https://www.dropbox.com/sh/qu232v5qt1v9ojv/AADzVkBVeJDq3Ppp-sz6at4Ga?dl=0).

**CAM**
[image]

**Fixed**
[image]

**Breathing**
[image]

If there is anything else I can do or something isn't quite right, just let me know 👍

jonasmalacofilho, 2020-05-01:
@iH8c0ff33 Thanks!  And it would be great if you could also check the X63.  We don't expect any differences, but it's good to check these things.

@wamphlett Thanks for all the data!

iH8c0ff33, 2020-05-01:
@jonasmalacofilho 
I just checked the x63 and it only has, as the x73, the ground and speed sense pin.

jonasmalacofilho, 2020-05-01:
@iH8c0ff33 Thanks again!

tomaculum, 2020-05-01:
@wamphlett Thank you very much! I will start to analyse those files today. 

@jonasmalacofilho In the screenshot above there are 20 LEDs in channel 1 available. If I remember correctly there are 40 leds for the `super-*` modes in the source files of the smart device. Is there any way that there are more than 20 leds connected to the kraken device?

jonasmalacofilho, 2020-05-01:
That figure – 40 LEDs – is the maximum number of LEDs supported by a HUE 2 (or Hue+) channel.

In theory that should still apply to the Kraken's HUE 2 header.

---

Can't remember if you implemented the "super-*" modes, and I'm on my phone now, so not easy to check... they should probably still exist, even on the ring and logo channels.

"super-*" is just the name I used for the modes where we have control of each individual LED (this used to live in a "custom" tab back in CAM 3x).

So the maximum number of colors for the super modes will be 8 (?) for ring, 1 for logo, and 40 for accessory and sync.

This reminds me of something else I should mention about these super modes: on the other drivers I established that unspecified colors would default to #000000, effectively keeping those unspecified LEDs turned off.

tomaculum, 2020-05-01:
Ok makes sense! I implemented all modes including the super modes. But I used 8 as a maximum of colors since the ring has 8 LEDs. With the additional accessories and the potential sync mode I will set it to 40 instead of 29 (20 strip + 8 ring + 1 logo)?

> This reminds me of something else I should mention about these super modes: on the other drivers I established that unspecified colors would default to #000000, effectively keeping those unspecified LEDs turned off.

Ok, already changed!

@wamphlett 
Did you record the startup process in row 1 of the table above?

wamphlett, 2020-05-01:
@tomaculum I thought I did but I cant see it in the files. I'll have a look for it and if not, I'll do it again and add it for you. 

I forgot to also mention, there is no recording for `logo_super-fixed*_red` as the option wasn't available for me.

tomaculum, 2020-05-01:
Differences I noticed so far:

For readability reasons I refer to my device recordings with **A** and to the recordings with accessories as **B**.

#### Status response:
**A:** `7502180040000c5741543336342001160653073c3c000[..]`
**B:** `750232004d000c574155303936200119064d073c3c000[..]`

jonasmalacofilho, 2020-05-01:
@tomaculum 

> With the additional accessories and the potential sync mode I will set it to 40 instead of 29 (20 strip + 8 ring + 1 logo)?

Set it to 40, as it's the maximum number of possible LEDs per channel.  Since the channels are (or at least seem to be) synced with the use of a bit mask, we don't need to sum the maximums of each channel (otherwise we would need 49).

> Status response

Lots of differences, but that's to be expected when we have lots of non-zero bytes before the fields we know/care about it.  There's really probably something there, and not just a constant picked at random.  But who knows what it means, or even if it's used by CAM.  If our `get_status` works and returns the information that we care about and that is displayed by CAM to the user, then we're probably fine.

The only thing I think of and that you should check is if there's a pattern resembling a sequence number (but not necessarily counting from 0 to 255), as that could be something that we could need to emulate.  Then again, if the current WIP driver works on your machines for long periods of time, it's very unlikely that there is such thing or, at the very least, that it's required.

Also, non-zero bytes, especially those after the fields of interest of a particular reply, can also be data left over in a buffer that's shared by many functions.  Not super likely in this case, since we only ever see a single reply (IIRC), but it's something I've seen in another device.

> Color requests

What values are you comparing for A/B?

tomaculum, 2020-05-01:
> What values are you comparing for A/B?

I should have added a WIP comment somewhere since I am not done yet (;

> The only thing I think of and that you should check is if there's a pattern resembling a sequence number (but not necessarily counting from 0 to 255), as that could be something that we could need [...]

No, those values are static and just the pump speed does change.

> Set it to 40, as it's the maximum number of possible LEDs per channel. Since the channels are (or at least seem to be) synced with the use of a bit mask, we don't need to sum the maximums of each channel (otherwise we would need 49).

Ok! I will add a comment where you first mentioned the bit mask since I did not understand this fully. And I added a new comment regarding a static led value which could be a bit complicated to solve.

tomaculum, 2020-05-01:
@jonasmalacofilho 
I updated the post above.

@wamphlett 
> I forgot to also mention, there is no recording for logo_super-fixed*_red as the option wasn't available for me.

Yes makes sense (; If there is only one LED, why should you need to set the LEDs individually.

Did you install liquidctrl in your linux environment already? Besides the missing setup file it would be interesting to know, if you can set the marquee or starry night modes even tough they were kind of different in your recordings. But before you try this, lets wait for a response from Jonas, maybe there is an explanation, why those are different.

wamphlett, 2020-05-01:
@tomaculum already installed on linux and they both work as exepted. 

I have added a V2 directory to the [dropbox folder](https://www.dropbox.com/sh/qu232v5qt1v9ojv/AADzVkBVeJDq3Ppp-sz6at4Ga?dl=0) - in there is the startup file and I also did the marquee and starry night files again just incase I made a mistake the first time round 👍

tomaculum, 2020-05-02:
@wamphlett thank you again!

Startup differences:

For some reason some of the requests are made 2x.

request firmware info - same
request lightning info - same
response firmware info - somehow there are two different responses to exactly the same request:
**A:**
```
1101180040000c57415433363420072002020100
```
**B:**
```
110132004d000c57415530393620072002020100

110151ff6d06787151541544246701200102060031333939313931313230303137393531ffffffff00
```

initialize 1: same

initialize 2: same

unknown response 1: 

**A:**
```
7101180040000c574154333634200100010100000200000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

**B:**
```
710132004d000c574155303936200100010100000200000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

led info response:
**A:**
```
2103180040000c574154333634200300000000000010000000000011000000000000000000000000000000000000000000000000000000000000000000000000
```

**B:**
```
210351ff6d0678715154154424670409090000000000000000000004040000000004040000000000000000000000000000000000000000000000000000000000
210332004d000c574155303936200301010000000010000000000011000000000000000000000000000000000000000000000000000000000000000000000000
```

set color: 
[`0x2a`, `0x04`] seem to be quite similar except bytes at the end?

```
TODO: add example
```

new unknown set color:
__I assume this sets some kind of color since it gets the same 0xff response as the set color above__

**B:**
```
w: 2803011e000000000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
r: ff0132004d000c574155303936202a04000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
w: 2803011e07020000010000517a000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
r: ff0151ff6d0678715154154424672803000800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
r: ff0151ff6d0678715154154424672803000900000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
w: 28030200000000000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
r: ff0151ff6d0678715154154424672803000a00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
w: 28030200000000000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
w: 28030414000000000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
r: ff0151ff6d0678715154154424672803000b00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
r: ff0151ff6d0678715154154424672803000c00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
w: 2803041409020000010000517a000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
w: 28030814000000000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
r: ff0151ff6d0678715154154424672803000d00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
r: ff0151ff6d0678715154154424672803000e00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
w: 2803081409020000010000517a000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
r: ff0151ff6d0678715154154424672803000f00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

```

set pump mode: same

[8 more replies]

closed 2020-05-29 by jonasmalacofilho