liquidctl/liquidctl #79: Support for Z63 and Z73

jonasmalacofilho, 2020-01-29:
The new Kraken Z series adds an LCD to the pump/block top.

https://www.nzxt.com/products/kraken-z63
https://www.nzxt.com/products/kraken-z73

While these certainly require a new protocol to handle the LCD graphics, it might end up being an extension to (or part of) the (presumably) already new protocol for the X63/X73 (see #78).

I'm not sure if there will be a lot of interest in these devices (based on pricing and adoption of the existing competitors), much less of using them on platforms other than Windows or with software other than CAM.

Still, please add a :+1: if you are interested in support for either of these devices in liquidctl.

Additionally, if you have one of these, please get in touch (a comment is fine, but you can also reach me by email) so we can talk about working together on a driver.

---

Status:

 - [ ] Collect basic USB data for both devices (at least vendor and product IDs)
 - [ ] Inspect and analyze the new protocol (monitoring, cooling and LCD)
 - [ ] Implement and test the new driver

astowe1080, 2020-02-09:
Hi! New here, heard about your work from CaseySJ’s remarkable (and popular) Z390 hackintosh build guide over at tonymacx86. 

I don’t have the Z63 yet but it will arrive next week and I’m going to be interested in getting some sort of control over it within OSX 10.15. The build will also boot Win10 where I expect to simply use CAM.

I’m not much of a coder myself, but just want to throw my hat in the ring here as someone interested in supporting the Z63. Please let me know how I can help!

Thanks!

inglor, 2020-02-11:
Hi @jonasmalacofilho i have a z63 and running Linux. How can I grab what I do is required for supporting new device?

jonasmalacofilho, 2020-02-12:
Hi @astowe1080 and @inglor,

Thank you both for expressing interest in helping out here.

@inglor, can you start by running `lsusb` and figuring out the USB vendor and product IDs used by the Z63?  After that, please post the output of `lsusb -d <vendor>:<product> -v`.

---

Afterwards, @astowe1080 and @inglor, we need some USB traffic captures.  First I would like to ask you to disable as most dynamic features as possible (use a static animation, fixed pump and fan speeds, etc.), to reduce the amount of noise in the captured traffic.

To capture USB traffic I recommend [Wireshark](https://wiki.wireshark.org/CaptureSetup/USB), and you can post the pcap files here or add them to [liquidctl-device-data](https://github.com/jonasmalacofilho/liquidctl-device-data).

We would need to start with captures that show:

 - a general picture of what messages compose the protocol
 - the initial exchange of packets when CAM first connects to the device
 - how pump and fan speeds and temperature are read from the device

Later:

 - setting pump and fan speeds to fixed values
 - profiles based on liquid temperature, and later CPU/GPU temperatures¹
 - configuring simple GIFs
 - configuring more complex animations
 - temperature display modes

I would also like to ask you to describe how these are configured in CAM, and how they work on the cooler (since these are substantially new/different devices).

Some screenshots can also help.  For example: showing the temperature and rpms for the corresponding traffic captures.

---

_¹ One way of supporting features based on CPU/GPU temperature is to send that value to the cooler, and let it handle after that.  This is **not** how previous Krakens worked, but since the Z63/Z73 have the temperature display mode, it would make sense to use the embedded processor to handle that instead of relying on the main CPU to render new GIFs every second.  TL;DR: there's a chance that the protocol supports something like this, which would be interesting._

inglor, 2020-02-13:
Hi @jonasmalacofilho see below for `lsusb -v` output of the device
```
$ sudo lsusb -d 1e71:3008 -v
Bus 003 Device 003: ID 1e71:3008 NZXT ASM107x
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x3008 
  bcdDevice            5.00
  iManufacturer           1 NZXT Inc.
  iProduct                2 NZXT KrakenZ Device
  iSerial                 3 326835463438
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0039
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          4 In Application
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               5
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     598
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0001
can't get debug descriptor: Resource temporarily unavailable
  Self Powered
```

jonasmalacofilho, 2020-02-13:
Thank you, @inglor.

Can you confirm the firmware version of your Z63, as reported by CAM?

---

It seems that this device simultaneously uses two interfaces, so when filtering captures it's important to use something like:

```
usb.src matches "^1\.3\." || usb.dst matches "^1\.3\."
```

_(in this example: device is on bus 1, address 3, but we match all interfaces)_

I'm guessing that the bulk endpoint 2 on interface 0 is used to transfer the GIFs, while the hid/interrupt endpoints on interface 1 are used for status reporting and configuring cooling settings.

inglor, 2020-02-13:
> Can you confirm the firmware version of your Z63, as reported by CAM?

Not really CAM software is so buggy that it doesn't work. I'm hit by the infamous "white screen" [issue]. I'm trying to resolve it with NZXT support but it's gonna take some time. Also I'm away for some days so I'll report back after back. Hopefully they would have released a new version of NZXT CAM as the 4.2.3 is buggy.

[issue]: https://www.google.com/search?client=firefox-b-d&q=nzxt+cam+white+screen

jonasmalacofilho, 2020-02-13:
> Not really CAM software is so buggy that it doesn't work.

Effective anti-analysis technique :slightly_smiling_face:

astowe1080, 2020-02-13:
Received mine, build looking good but I’m also out of town for a few days so I will install OS’s next week.

Maybe dumb question, but are the command line tools used for inspecting the Z63 USB traffic useful regardless of OS? In theory we’re trying to figure out how non-Windows OS’s interact with the Z63 but can Windows be used to gather the useful USB data? Or is the whole point needing to see the USB data from within Linux/OSX?

Thanks!

jonasmalacofilho, 2020-02-14:
@astowe1080 

> command line tools used for inspecting the Z63 USB traffic

Which tools do you mean?

> In theory we’re trying to figure out how non-Windows OS’s interact with the Z63 but can Windows be used to gather the useful USB data?

Actually, we want to know how the software and/or kernel drivers supplied by manufacturer _(usually only for Windows)_ interact with the device, so we can figure out how the protocol works and add support for it in our own cross-platform tool, liquidctl.

Without suitable software or kernel drivers, there is not much interaction with the device (or any USB device, really).  While the HID interface is standard (so Linux and macOS will bind a standard HID driver to it), it's not a mouse or keyboard and (AFAIK) the kernel won't automatically start to poll the device for reports.  And no (practical) data will be sent to the device (other than very low level USB management stuff).

astowe1080, 2020-02-14:
Yep, was a dumb question 😅

I just meant lsusb and Wireshark. You definitely answered my question, Windows/CAM makes sense as starting point to pick apart NZXT’s drivers.

Fwiw, I got CAM v4.2.3 installed last night and it reported my Z63’s firmware as “5.0” — can dig in deeper with it next week.

jonasmalacofilho, 2020-02-15:
@astowe1080 

Don't worry about it : )  
_(also, it wasn't a dumb question)_

> Fwiw, I got CAM v4.2.3 installed last night and it reported my Z63’s firmware as “5.0” — can dig in deeper with it next week.

Thanks!

inglor, 2020-02-26:
Hi @jonasmalacofilho 
I finally managed to run NZXT CAM software - I had to basically install it while ethernet was not plugged in, block it from firewall and now it's running. A couple of interesting points
- Only latest NZXT CAM 4.2.3 works for Kraken Z63, previous versions don't (i.e. 4.1.1 or 4.0.1 which I tried)
- If you don't have installed the NZXT CAM version or you start from a cold start (shutdown, power-off the PSU for 10 seconds and power on and start windows/linux) the Kraken Z63 displays the *current* Liquid temperature in white text (default orientation).
- Once you start the NZXT CAM in windows it updates the LCD screen with what you've configured in the CAM settings and adds the colors.
- The default "profile" of the Z63 without CAM is to have both pump and fans set to idle and it doesn't spin up or down while stressing the CPU. The Liquid temperature after a few cinebenches goes up to 31 C from 29 C and max CPU temp from motherboard is around 70 C for my setup (AMD 3950x).
- If you shut down the CAM in windows the LCD screen stays *static*. It doesn't update but just shows the latest update from when CAM was running, which leads to me believe the LCD is pretty dump and just calculates the next *frame* and shows it.
- If you restart of do a soft shutdown (no power down of the PSU) the Kraken Z63 shows the latest static image from the last time CAM was running - I assume this will be the same if you dual boot to linux after windows, but haven't tried it yet.
- If you start CAM the fans and pump seems to be monitoring CPU temperature as I see them spin up when running a bentchmark and spin down after it finishes. It's not ideal and I'd rather monitor the Liquid temperature 
- After various attempts on the Wireshark to capture the USB data I've finally got some data.
[nzxt_kraken_z63_staring_CAM.zip](https://github.com/jonasmalacofilho/liquidctl/files/4255550/nzxt_kraken_z63_staring_CAM.zip). This is from a cold reboot to starting and closing CAM software so I presume the initiallization process and a few updates on CPU temp.

Personally I'm a little disapointed as I was hoping to be able to monitor water temperature instead of CPU for spinning up pump and fans of the AIO.

jonasmalacofilho, 2020-03-01:
@inglor 

Thank you.  I'll take a look as soon as I can.

astowe1080, 2020-03-03:
I’ve got Windows and OS X squared away now, looks like OS X can detect a couple things at surface level. Just in case you see something helpful in it, attached a screen grab from System Information within OS X.

I’ll give Wireshark a shot later today too — 

Thanks!

jonasmalacofilho, 2020-03-07:
@astowe1080 the screenshot is missing.  AFAIK all attachments are scrubbed when you reply to a issue directly from the email notification.

richardvonrueden, 2020-03-07:
I use the NZXT Z63 (CPU: Ryzen 3900x) with Arch Linux and Windows 10. It only shows me the liquid temperature. As I use Linux most of the time, I really would appreciate support for this version.

astowe1080, 2020-03-08:
@jonasmalacofilho 
My mistake, here's the screenshot I mentioned and the CAM startup data I got from Wireshark -- 

[CAMdata.zip](https://github.com/jonasmalacofilho/liquidctl/files/4303782/CAMdata.zip)

jonasmalacofilho, 2020-03-08:
@astowe1080 

Thanks, I'll take a look as soon as I can.

ddv2005, 2020-03-18:
Z3 devices has 2 Endpoints. And looks like #1 used for control/status and #2 for upload LCD images. I guess that CAM software render and upload LCD images on demand. Hardware can store several images and then software can reuse old images by switching it via endpoint #1.

inglor, 2020-04-16:
Hi @jonasmalacofilho 

How is this progressing? Do the Z devices get any similarities with the latest generation X ones and would it help the PR#100 once merged? In terms of CAM software they released a new version which seems a new UI is in a place so maybe something changed under the hood as well (communication with the pump).

Let me know if I can assist anyhow with trying to reverse engineer the API.

jonasmalacofilho, 2020-04-26:
Hi @inglor,

It looks like the pump speed is set the same way between your Z63 and a X63.  And the fan speed follows a very obvious/similar pattern.

I haven't checked incoming status messages, but they also seem similar.  So, as expected, the #100 should kick start the X63/X73 driver quite nicely.

> In terms of CAM software they released a new version which seems a new UI is in a place so maybe something changed under the hood as well (communication with the pump).

A new UI _yet again?_  Because CAM 3.x never got to support the new coolers, right?

Well, thanks for letting me know.  But if there are changes in the protocol, I expect them to be restricted to the LCD; or at most they could explain the differences we're seeing between the actual HUE 2 and the X63.  Anyway, I'll keep this in mind and if necessary I'll ask you for a few additional captures.

inglor, 2020-05-25:
Hi @jonasmalacofilho,

> It looks like the pump speed is set the same way between your Z63 and a X63. And the fan speed follows a very obvious/similar pattern.
> 
> I haven't checked incoming status messages, but they also seem similar. So, as expected, the #100 should kick start the X63/X73 driver quite nicely.
>  
I tried  @tomaculum's `3232347` commit against my `Z63` and doesn't detect something even though it's searching for it (looking at the USB device ID). I'm not sure if I need to modify the code to expect something though..
```
inglor@tiamat ~/.local/bin$ ./liquidctl --version
liquidctl v1.3.3 (32323472fdb0)
inglor@tiamat ~/.local/bin$ ./liquidctl --debug list
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (32323472fdb0)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenThreeXDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4086
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0140
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0140
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b0e:0348
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:3008
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 174c:3074
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:3008
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 174c:2074
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 2109:0812
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0610
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b0e:0348
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:2514
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0140
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 2109:2812
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
inglor@tiamat ~/.local/bin$ sudo lsusb | grep NZXT
[sudo] password for inglor:
Bus 003 Device 005: ID 1e71:3008 NZXT ASM107x
```

> A new UI _yet again?_ Because CAM 3.x never got to support the new coolers, right?
>
> Well, thanks for letting me know. But if there are changes in the protocol, I expect them to be restricted to the [...]

Yeah I assume they upgrade it to add more support for the LCD screen.

jonasmalacofilho, 2020-05-25:
@inglor 

I just branched off a Z driver from the in-progress X one.

Can you try [`fourth-generation-krakens`](https://github.com/jonasmalacofilho/liquidctl/commits/fourth-generation-krakens) @ 9c96a1dc4657fa6734cf6de7e64176bfe4185687?

---

```
# lsusb | grep NZXT
Bus 003 Device 005: ID 1e71:3008 NZXT ASM107x
```

This is unrelated to liquidctl, but it's now the [second time](https://github.com/jonasmalacofilho/liquidctl/issues/114#issuecomment-632975480) a see a weird product description in lsusb.  I'm thinking there might be a bug with a particular hub and/or in udev.

Can you show me the `lsusb` output with all devices (to see if, I expect, `ASM107x` belongs to some other device)?

inglor, 2020-05-26:
I think I might have an answer for that. The Kraken Z series have an incompatibility list published recently from NZXT (see [support post]) with some motherboards and unfortunately mine is listed these. The "official" workaround is to get an internal USB hub for the motherboard. I think that's why it's appearing as this. Personally if I was given an option to RMA I'd choose it and maybe even go back to X series - disappointed with the issues I've had so far.

[support post]: https://support.nzxt.com/hc/en-us/articles/360045926294-Kraken-Z-USB-Compatibility-Issue

jonasmalacofilho, 2020-05-26:
> The "official" workaround is to get an internal USB hub for the motherboard. I think that's why it's appearing as this.

A non-compliant device and/or root hub (not sure what's causing the incompatibility) could explain what _triggers_ the issue, but I still think this is showing that there's a buffer reuse bug in udev (or in lsusb and I missed that).

Can you let me know what versions of libudev, systemd and lsusb you have?  I would like to take a second look at this once I have some free time.

---

> But status gets a stacktrace

Were you running with root privileges? Or, alternatively, have you set up a udev rule that grants your user access to the hidraw device?

Anyway, I messed up and that particular commit is useless: I never made KrakenZ3Driver a subclass of KrakenX3Driver, so right now it's just an abstract driver that will raise `NotImplementedError`if you try to do anything.

Fixing this now, please update the code before running again.  Also, you'll need to `initialize` the device before calling `status`.

inglor, 2020-05-26:
> Can you let me know what versions of libudev, systemd and lsusb you have? I would like to take a second look at this once I have some free time.
> 
Sure, it's Archlinux so I think the relevant packages are the ones in [here](http://ix.io/2nmV)

> Were you running with root privileges? Or, alternatively, have you set up a udev rule that grants your user access to the hidraw device?
> 
I've made this udev rule, I think they are OK with the `MODE=0666` not sure abour the `SYMLINK` or `TAG`, probably need to read a bit more about udev rules :confused: 
```
$ cat /etc/udev/rules.d/70-kraken.rules
# udev rules file
ACTION=="add", SUBSYSTEM=="hidraw", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1e71", \
	  ATTRS{idProduct}=="3008", MODE="0666", SYMLINK+="kraken", TAG+="systemd"
ACTION=="add", SUBSYSTEM=="hidraw", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1e71", \
	  ATTRS{idProduct}=="3008", MODE="0666", SYMLINK+="smart_device", TAG+="systemd"
```
> 
> ~Fixing this now,~ Just fixed with [da19151](https://github.com/jonasmalacofilho/liquidctl/commit/da19151d73df318fb93def7df0e39584802ee448), please update the code before running again. Also, you'll need to `initialize` the device before calling `status`.

Now the latest commit fails to construct the class as it misses some arguments
```
$ sudo liquidctl --debug status
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (Arch Linux; liquidctl-git 1.3.3.r87.da19151-1)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4086
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0140
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0140
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b0e:0348
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:3008
Traceback (most recent call last):
  File "/bin/liquidctl", line 11, in <module>
    load_entry_point('liquidctl==1.3.3', 'console_scripts', 'liquidctl')()
  File "/usr/lib/python3.8/site-packages/liquidctl/cli.py", line 277, in main
    selected = list(find_liquidctl_devices(**opts))
  File "/usr/lib/python3.8/site-packages/liquidctl/driver/__init__.py", line 65, in find_liquidctl_devices
    for dev in  bus_cls().find_devices(**kwargs):
  File "/usr/lib/python3.8/site-packages/liquidctl/driver/usb.py", line 600, in find_devices
    yield from drv.probe(handle, vendor=vendor, product=product, **kwargs)
  File "/usr/lib/python3.8/site-packages/liquidctl/driver/usb.py", line 138, in probe
    dev = cls(handle, description, **consargs)
TypeError: __init__() missing 2 required positional arguments: 'speed_channels' and 'color_channels'
```

jonasmalacofilho, 2020-05-26:
> I've made this udev rule, I think they are OK with the `MODE=0666` not sure abour the `SYMLINK` or `TAG`, probably need to read a bit more about udev rules 
> 
> ```
> $ cat /etc/udev/rules.d/70-kraken.rules
> # udev rules file
> ACTION=="add", SUBSYSTEM=="hidraw", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1e71", \
> 	  ATTRS{idProduct}=="3008", MODE="0666", SYMLINK+="kraken", TAG+="systemd"
> ```

This one seems fine.  But you should also restart the system, because getting the permissions to change is not super trivial: IIRC you need to unbind rebind the USB device, after reloading the rules with `udevadm control --reload` (`udevadm trigger` doesn't refresh permissions).

> ```
> ACTION=="add", SUBSYSTEM=="hidraw", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1e71", \
> 	  ATTRS{idProduct}=="3008", MODE="0666", SYMLINK+="smart_device", TAG+="systemd"
> ```

Now on this one the USB product ID doesn't match the apparently targeted device.

> Now the latest commit fails to construct the class as it misses some arguments

Oh, I had made yet another mistake.  Sorry about that, please try again with 9470111.

And if you get another error on open can you also try with `sudo liquidctl [...]`?

---

> Sure, it's Archlinux so I think the relevant packages are the ones in here

Thanks.

inglor, 2020-05-26:
> > I've made this udev rule, I think they are OK with the `MODE=0666` not sure abour the `SYMLINK` or `TAG`, probably need to read a bit more about udev rules
> > ```
> > $ cat /etc/udev/rules.d/70-kraken.rules
> > # udev rules file
> > ACTION=="add", SUBSYSTEM=="hidraw", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1e71", \
> > 	  ATTRS{idProduct}=="3008", MODE="0666", SYMLINK+="kraken", TAG+="systemd"
> > ```
> 
> This one seems fine. But you should also restart the system, because getting the permissions to change (immediately, with an udev rule) is not super trivial: IIRC you need to unbind and then rebind the USB device, after reloading the rules with `udevadm control --reload` (`udevadm trigger` doesn't refresh permissions).
OK, restart on the way :+1: 
> 
> > ```
> > ACTION=="add", SUBSYSTEM=="hidraw", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1e71", \
> > 	  ATTRS{idProduct}=="3008", MODE="0666", SYMLINK+="smart_device", TAG+="systemd"
> > ```
> 
> Now on this one the USB product ID doesn't match the apparently targeted device.
Isn't it the same vendor:product as above? What's different ?
> Oh, I had made yet another mistake. Sorry about that, please try again with [9470111](https://github.com/jonasmalacofilho/liquidctl/commit/9470111456b404abd2d4c482955584922cbae2ab).
>
> And if you get [...]
That's better with `list` and `initialize` finishing successfully and no errors, but status stuck..
```
$ sudo liquidctl --debug list
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (Arch Linux; liquidctl-git 1.3.3.r88.9470111-1)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4086
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0140
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0140
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b0e:0348
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:3008
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken Z (Z63 or Z73) (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 174c:3074
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:3008
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 174c:2074
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 2109:0812
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0610
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b0e:0348
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:2514
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0140
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 2109:2812
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
Device ID 0: NZXT Kraken Z (Z63 or Z73) (experimental)
├── Vendor ID: 0x1e71
├── Product ID: 0x3008
├── Release number: 0x0500
├── Serial number: 326835463438
├── Bus: hid
├── Address: /dev/hidraw3
└── Driver: KrakenZ3Driver using module hid
[DEBUG] liquidctl.cli: hierarchy: KrakenX3Driver, UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice

$ sudo liquidctl --debug initialize all
[... 61 lines cut ...]
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 174c:3074
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:3008
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 174c:2074
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 2109:0812
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18f3
[... cut ...] for device 05e3:0610
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b0e:0348
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:2514
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0140
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 2109:2812
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: NZXT Kraken Z (Z63 or Z73) (experimental)

```

jonasmalacofilho, 2020-05-26:
> That's better with list and initialize finishing successfully and no errors, but status stuck..

Stuck as in hangs indefinitely?  That's weird.  No timeout, not even after a while (for example: 5 seconds)?

Maybe I messed something up when making some changes to Tom's code.  Or maybe there's a subtle but relevant difference in how `status` works on Z coolers.  I'll take a look at this...

inglor, 2020-05-26:
> Stuck as in hangs indefinitely? That's weird. No timeout, not even after a while (for example: 5 seconds)?
>
> Maybe I messed something up when making some changes to Tom's code. Or maybe there's a [...]

Yeah just stuck, I extracted a thread dump if it's useful with `# kill -SIGABRT <PID>`:
```
# PYTHONFAULTHANDLER=true liquidctl --debug status
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (Arch Linux; liquidctl-git 1.3.3.r88.9470111-1)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[...]
[DEBUG] liquidctl.cli: device: NZXT Kraken Z (Z63 or Z73) (experimental)
Fatal Python error: Aborted

Current thread 0x00007f48ea18b740 (most recent call first):
  File "/usr/lib/python3.8/site-packages/liquidctl/driver/usb.py", line 504 in read
  File "/usr/lib/python3.8/site-packages/liquidctl/driver/kraken_three_x.py", line 265 in _read
  File "/usr/lib/python3.8/site-packages/liquidctl/driver/kraken_three_x.py", line 222 in get_status
  File "/usr/lib/python3.8/site-packages/liquidctl/cli.py", line 309 in main
  File "/usr/bin/liquidctl", line 11 in <module>
Aborted (core dumped)
```

jonasmalacofilho, 2020-05-26:
@inglor 

That does help a bit.  It seems that the device has not been fully initialized, so it isn't automatically sending status reports.  And in line 221 we clear any already enqueued reports (to avoid stale data), so `read` will block forever as there are no more messages.

I'll need to study the captures in more detail...

jonasmalacofilho, 2020-06-09:
Looking at the captures it seems that, unlike the X model, a specific message is sent before the status is read.  There are a lot of extra packets though, so what I implemented (and just pushed to the master branch) might not to be sufficient.  Still, give it a try and let me know.

---

On the GIF side, it seems that a frame consists of 102'400 (320*320) pixels, seemingly encoded in RGBA.  Each frame is sent in 800 512-byte long chunks plus a "header" packet (all bulk transfers).

[image]

I'm still not sure how the frames are stitched together (to form the animation), how the rotation is specified (is it possible to tell the firmware to apply a rotation?), nor what is in the "header" packet.  And then there's the matter of all the extra interrupt transfers (in comparison with the X model)...

inglor, 2020-06-09:
Hi @jonasmalacofilho 

This is working great with commit `84fa6c69cb7c` - see logs!
```
$ liquidctl --unsafe kraken_z3_driver --debug list
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (84fa6c69cb7c)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, KrakenZ3GuardDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4086
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0140
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0140
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b0e:0348
[DEBUG] liquidctl.driver.usb: probing drivers for device 043e:9a39
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:3008
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken Z (Z63 or Z73) (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 174c:3074
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:3008
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 174c:2074
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 2109:0812
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0610
[DEBUG] liquidctl.driver.usb: probing drivers for device 043e:9a39
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b0e:0348
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0140
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 2109:2812
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
Device ID 0: NZXT Kraken Z (Z63 or Z73) (experimental)
├── Vendor ID: 0x1e71
├── Product ID: 0x3008
├── Release number: 0x0500
├── Serial number: 326835463438
├── Bus: hid
├── Address: /dev/hidraw3
└── Driver: KrakenZ3Driver using module hid
[DEBUG] liquidctl.cli: hierarchy: KrakenX3Driver, UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice
$ liquidctl --unsafe kraken_z3_driver --debug list
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 2109:0812
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0610
[DEBUG] liquidctl.driver.usb: probing drivers for device 043e:9a39
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b0e:0348
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0140
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 2109:2812
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: NZXT Kraken Z (Z63 or Z73) (experimental)
[DEBUG] liquidctl.driver.kraken_gen4: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.kraken_gen4: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.kraken_gen4: write 70 02 01 b8 01 (and 59 padding bytes)
[DEBUG] liquidctl.driver.kraken_gen4: write 70 01 (and 62 padding bytes)
[... 29 lines cut ...]
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 174c:2074
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 2109:0812
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0610
[DEBUG] liquidctl.driver.usb: probing drivers for device 043e:9a39
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b0e:0348
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0140
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 2109:2812
[... cut ...]
[DEBUG] liquidctl.driver.kraken_gen4: write 74 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.kraken_gen4: received 75 01 16 00 37 00 0a 51 38 34 30 35 31 32 01 1e 04 3b 07 32 32 01 00 a2 03 28 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
NZXT Kraken Z (Z63 or Z73) (experimental)
├── Liquid temperature    30.4  °C
├── Pump speed            1851  rpm
├── Pump duty               50  %
├── Fan speed              930  rpm
└── Fan duty                40  %
```

inglor, 2020-06-09:
> I'm still not sure how the frames are stitched together (to form the animation), how the rotation is specified (is it possible to tell the firmware to apply a rotation?), nor what is in the "header" [...]

Within CAM software you can rotate by 90&deg; clockwise _only_ (so one need to make sure there's  enough clearance from the RAM sticks on the side of the cable sticking out (right hand side)). For the rest could it be related to the pump / fan curves? If this is only related to the GIF image then I can get you a screenshot of the CAM configuration for the GIF.

jonasmalacofilho, 2020-06-09:
I see that you have "rotate display" checked, and the frame I reconstructed from one of your capture files (the first one) was also rotated.  Unless you only checked the rotation option after that particular capture session, it seems that rotation is implemented in CAM _alone_ (not in firmware), which means that in liquidctl we really only need to rotate the image ourselves.

Also glad that `status` is working now.  How about fan/pump speed control?

> For the rest could it be related to the pump / fan curves?

That's something the X model already supports, which is why fan/pump speed control may already be working on the same commit.  But besides the packets necessary for fan/pump control, there are are a lot of not-seen-before packets that start with a `0x3?` prefix (IIRC).

inglor, 2020-06-10:
Pump and fan speed setting works fine as you expected
```
inglor@tiamat ~$ liquidctl --unsafe kraken_z3_driver status
NZXT Kraken Z (Z63 or Z73) (experimental)
├── Liquid temperature    29.1  °C
├── Pump speed            1578  rpm
├── Pump duty               40  %
├── Fan speed              896  rpm
└── Fan duty                38  %

inglor@tiamat ~$ liquidctl --unsafe kraken_z3_driver set pump speed 50
inglor@tiamat ~$ liquidctl --unsafe kraken_z3_driver set fan speed  20 20  30 35  34 50  40 80  50 100
inglor@tiamat ~$ liquidctl --unsafe kraken_z3_driver status
NZXT Kraken Z (Z63 or Z73) (experimental)
├── Liquid temperature    29.1  °C
├── Pump speed            1840  rpm
├── Pump duty               50  %
├── Fan speed              815  rpm
└── Fan duty                34  %
```
I tried the logo wave-spectrum but as expected this is not working:
```
$ liquidctl --unsafe kraken_z3_driver set logo color spectrum-wave
ERROR: Unexpected error with NZXT Kraken Z (Z63 or Z73) (experimental)
Traceback (most recent call last):
KeyError: 'logo'
```

inglor, 2020-06-19:
@jonasmalacofilho Is there a plan to release this without GIF support as the main bits are there and working? I've been using it with `systemd` and I can head the fans spin up faster during compilation and slowing down a bit later when water temp is down again. Also the top issue comment needs a bit update to reflect the current state.

```
$ cat /etc/systemd/system/liquidcfg.service
[Unit]
Description=AIO startup service

[Service]
Type=oneshot
ExecStart=liquidctl --unsafe kraken_z3_driver set pump speed 20 40  30 45  35 70  40 80  45 100
ExecStart=liquidctl --unsafe kraken_z3_driver set fan  speed 20 25  30 35  35 60  40 75  50 100
#ExecStart=liquidctl set ring color fading 350017 ff2608
#ExecStart=liquidctl set logo color spectrum-wave

[Install]
WantedBy=default.target
```

jonasmalacofilho, 2020-06-19:
Yes, absolutely.  In fact, I already enabled it in the main branch, _without_ requiring `--unsafe...`.

anishg24, 2020-06-29:
Hello! I wish to help in this development as well. I have a Kraken Z73 attached to my Ryzen 9 3950X to keep it cool. I dual boot Ubuntu 20.04 and Windows 10. If you need anything that might vary (from the Z63 to the Z73) I'll gladly help.

wesleyelder, 2020-07-18:
Hi All, i also have a Z73 - interested to help

lagerfeuer, 2020-07-23:
Hello, I am also planning on getting a Z73, willing to help out.

jonasmalacofilho, 2020-07-23:
Thank you all for the offering to help out!

Essentially, what's left here is to implement support for sending static and animated images to the cooler.

I was able to confirm the [basic encoding of a instantaneous frame](https://github.com/jonasmalacofilho/liquidctl/issues/79#issuecomment-641030310).  Now someone, preferably with access to one of these devices, needs to figure out exactly how the header packet works, how GIF animations are sent,¹ and if there's anything important in the extra interrupt transfers.

Basically, a lot of experimenting with different settings and watching the USB traffic.

_¹ It's not clear whether GIF animations are actually sent to the cooler as GIFs, or as sequences of these "frames"._

anishg24, 2020-07-23:
Sounds good so far.

When you said that it might stream the frames to the Kraken I was skeptical because that would mean that the gifs would depend on the performance of the Windows and how it treats the CAM software. To test this, I set my Kraken Z73 to GIF mode, and their default GIF came up and was rendering like a GIF would. I then restarted the computer and during the BIOS and POST screen, the GIF still was behaving like it did when I was booted into Windows. Hell, I booted into Ubuntu (where I haven't enabled liquidctl yet) and the animation is still playing beautifully.

I haven't sniffed anything in the USB side, but I strongly believe that NZXT CAM just sends the 20 MB file over to the Kraken, then in the Kraken it is stashed somewhere where it will play on loop forever.

TL;DR, GIF animations are sent by sending the entire gif file over to the cooler and making the cooler play it without the CAM software interfering. From what I can tell, the CAM software is only speaking with the cooler when it sends the CPU/GPU temps, otherwise, when it comes to GIF support or the liquid temps, the Kraken is does as expected in both Windows and Ubuntu.

I have no clue how to sniff USB traffic on a Windows machine, as I am familiar only to Mac and Ubuntu at best. If you have any software recommendations I should use please do let me know. Likewise, if you have any more questions do let me know.

Love the progress so far being made, but I am curious as to if you have pushed a version without GIF support but does support the temperature readings and how I would go about using that version. (it's referenced above but I never understood how to make it work)

inglor, 2020-07-23:
> I have no clue how to sniff USB traffic on a Windows machine, as I am familiar only to Mac and Ubuntu at best. If you have any software recommendations I should use please do let me know. Likewise, if you have any more questions do let me know.

@anishg24 For windows the one I used on post above [comment] is Wireshark. It used to be for network packages but with an extension it supports USB traffic also. (You might need a few installs / uninstalls) to make it work successfully in Windows 10 as Win 10 doesn't like messing with drivers and intercepting the traffic). Once you have it installed you can use filters to capture only the traffic based on the device ID (again see first comments here).

On the analysis @jonasmalacofilho mentioned and try a few different things it's quite hard for someone who doesn't know what to look for. So even though you can open and see the traffic of a package sent from and to the Kraken in Wireshark personally I'm not sure what I'm looking for 😕 . This is where maybe a Wiki page would come handy or 1 or 2 examples to get the idea.

For me the pump is well hidden in the case (no glass panel or something) so as long I can control the pump and fan curves I'm happy, and with the curves I used I can barely hear it rumping up when compiling a kernel and all 16 cores are maxed. The CPU temp display is an added bonus if / when it's supported and not interested in GIFs (one would argue why I even bought the one with the LED screen, and honestly with all the [issues] I had with this one I wonder it as well).

[comment]: https://github.com/jonasmalacofilho/liquidctl/issues/79#issuecomment-591395980
[issues]: https://support.nzxt.com/hc/en-us/articles/360045926294-Kraken-Z-USB-Compatibility-Issue

jonasmalacofilho, 2020-07-23:
I go over a few different strategies for understanding protocols in [this comment](https://github.com/jonasmalacofilho/liquidctl/issues/142#issuecomment-650568291).  But the truth is that this is a somewhat different feature, so new techniques may need to be developed...

As long as it's just standard GIFs or sequence of these simple "frames" (to be stored in and replayed by the cooler), plus some headers, we should be able to, _collectively,_ figure it out.

msigut, 2020-07-26:
Hi,

please look at this message: **280.5 °C** . I hope that temperature is incorrect ... 👍 
Please, fix this ...

```
NZXT Kraken X (X53, X63 or X73) (experimental)
├── Liquid temperature    280.5  °C
├── Pump speed             2739  rpm
└── Pump duty               100  %
```

debug info:
```
sudo liquidctl status --debug
[DEBUG] liquidctl.cli: running liquidctl v1.4.0a0 (0d2cdad0f223)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1781:0e39
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0ffc:0026
[DEBUG] liquidctl.driver.usb: probing drivers for device 258f:0b04
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2006
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 043e:9a11
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:0411
[DEBUG] liquidctl.driver.usb: probing drivers for device 043e:9a10
[DEBUG] liquidctl.driver.usb: probing drivers for device 1781:0e39
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:5411
[DEBUG] liquidctl.cli: device: NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.kraken_gen4: received 75 02 20 00 36 00 0b 51 53 58 34 35 33 20 01 ff ff cc 0a 64 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
NZXT Kraken X (X53, X63 or X73) (experimental)
├── Liquid temperature    280.5  °C
├── Pump speed             2764  rpm
└── Pump duty               100  %

[DEBUG] liquidctl.cli: device: NZXT Smart Device V2
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 2f 00 39 00 07 51 56 56 39 30 30 20 03 00 02 02 02 00 00 00 00 00 d4 02 11 02 c7 02 00 00 00 00 00 00 00 00 00 00 32 32 32 00 00 00 00 00 32 32 32 00 00 00 00 00 30 00 00 00 00 00 00 00
NZXT Smart Device V2
├── Fan 1 duty      50  %
├── Fan 1 speed    724  rpm
├── Fan 2 duty      50  %
├── Fan 2 speed    529  rpm
├── Fan 3 duty      50  %
├── Fan 3 speed    711  rpm
└── Noise level     48  dB
```

himekifee, 2020-09-22:
From what I can see from the usb traffic, the lcd operates on different modes. The gif one is simple, a bulk out composed with a magic number, a size, and the data, and 2 or 3 interrupts, it is set. However, then it becomes complex. First, the gifs are choosable in CAM and there is no bulk out to transfer img. It means previous images were stored inside some ram and can be configured to use that later. Second, when it is set to smth like cpu load img, the mode is said to be 0x02, gif is 0x01. It uses some unknown format. A bulk out no surprise, but when cpu load change from 1 to 2, no bulk, then to a larger number, say 30, it bulked. I don't think I can handle that easily, is anyone interested in and want to help?

jonasmalacofilho, 2020-09-23:
> However, then it becomes complex. First, the gifs are choosable in CAM and there is no bulk out to transfer img.

Since this will (probably) get cleared once the device is powered off (which may require turning off the PSU, but that's besides the point), we can probably do with uploading a single GIF at a time.  Maybe that helps?

> A bulk out no surprise, but when cpu load change from 1 to 2, no bulk, then to a larger number, say 30, it bulked.

Perhaps there's some threshold bellow which the display isn't updated.  CPU load varies a lot and very quickly, and users might only be interested in rough numbers.  On the other hand, there may be some maximum interval after which an update is always send (even if small).

himekifee, 2020-09-23:
Another wired thing is, the setup interrupt after bulk gif contains something varies between 0x03 - 0x0e. I guess its a index thing. But I don't see where does that come from. No extra info after bulk, nor begin.

himekifee, 2020-09-24:
Even after a power off of PSU, the lcd still keeps its gif. I believe it saved it in some sort of rom.

inglor, 2020-09-24:
@himekifee That's weird. I had mine revert to the default white letters showing water temperature after a PSU power off (and waiting for the capacitors to drain, or try to power up when PSU is off). Also the behaviour during my testing (months ago admittedly) was that the gif remained static if CAM was shutdown or running in Linux.

MacGyver27, 2020-10-03:
Hi, owning X63 on my Debian v10. Trying to set CPU+GPU utilization on the OLED display - but still encountering `KeyError` issues..

liquidctl v1.4.1

```
liquidctl list --verbose
Device ID 0: Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
├── Vendor ID: 0x048d
├── Product ID: 0x8297
├── Release number: 0x0003
├── Bus: hid
├── Address: /dev/hidraw0
└── Driver: RgbFusion2 using module hidraw

Device ID 1: Corsair HX1000i
├── Vendor ID: 0x1b1c
├── Product ID: 0x1c07
├── Release number: 0x0002
├── Bus: hid
├── Address: /dev/hidraw1
└── Driver: CorsairHidPsu using module hidraw

Device ID 2: NZXT Kraken Z (Z63 or Z73) (experimental)
├── Vendor ID: 0x1e71
├── Product ID: 0x3008
├── Release number: 0x0500
├── Serial number: 3058344F3338
├── Bus: hid
├── Address: /dev/hidraw2
└── Driver: KrakenZ3 using module hidraw
```

```
liquidctl --match kraken initialize
NZXT Kraken Z (Z63 or Z73) (experimental)
└── Firmware version    5.0.0 
```
- here I am missing similar output as stated in [docs](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/kraken-x3-z3-guide.md#initialization). It doesnt list all the accessories: 
```
├── LED accessory 1    HUE 2 LED Strip 300 mm
├── LED accessory 1          AER RGB 2 140 mm
├── LED accessory 2          AER RGB 2 140 mm
├── Pump Logo LEDs                   detected
└── Pump Ring LEDs                   detected
```

This then most likely results into `KeyErrors`: 
```
liquidctl --match kraken set sync color fixed af5a2f
ERROR: Unexpected error with NZXT Kraken Z (Z63 or Z73) (experimental)
Traceback (most recent call last):
KeyError: 'sync'
liquidctl --match kraken set ring color fading 350017 ff2608
ERROR: Unexpected error with NZXT Kraken Z (Z63 or Z73) (experimental)
Traceback (most recent call last):
KeyError: 'ring'
```

Anybody has any clue what I missed or problem might be? Appreciate, Michal

jonasmalacofilho, 2020-10-03:
@MacGyver27 OLED screen of the Z63 replaces the logo and ring LEDs of the X63.  And support for the OLED screen is still under ~~development~~ investigation.

---

> Hi, owning **X63** on my Debian v10. Trying to set CPU+GPU utilization on the **OLED display** [...]

You clearly meant to write Z63.

MacGyver27, 2020-10-03:
> You clearly meant to write Z63.

Yeah, sorry.

> OLED screen of the Z63 replaces the logo and ring LEDs of the X63. And support for the OLED screen is still under development investigation.

So then this:
```
├── LED accessory 1    HUE 2 LED Strip 300 mm
├── LED accessory 1          AER RGB 2 140 mm
├── LED accessory 2          AER RGB 2 140 mm
├── Pump Logo LEDs                   detected
└── Pump Ring LEDs                   detected
```
applies only for X3? Thanks, Michal

jonasmalacofilho, 2020-10-03:
This part only applies if you have these specific accessories connected to the HUE2 header:

> ```
> ├── LED accessory 1    HUE 2 LED Strip 300 mm
> ├── LED accessory 1          AER RGB 2 140 mm
> ├── LED accessory 2          AER RGB 2 140 mm
> ```

This part only applies to the X3 coolers:

> ```
> ├── Pump Logo LEDs                   detected
> └── Pump Ring LEDs                   detected
> ```

kebbleytea, 2020-10-26:
Due to my setup using Unraid, I had to make a docker container based off the `centos/systemd` image. But I'm happy to switch into a real environment if needed.

```bash
[root@eea9f6c6f82c /]# LC_ALL=en_US.utf8 liquidctl status
NZXT Kraken Z (Z63 or Z73) (experimental)
├── Liquid temperature    39.8  °C
├── Pump speed            1834  rpm
├── Pump duty               50  %
├── Fan speed             1834  rpm
└── Fan duty                88  %
```

I won't be good with debugging any USB / Wireshark stuff for the GIFs etc, but let me know how I can help out at all, my Z73 is at your disposal. Am more than happy to help test it.

voltage321, 2020-12-01:
Hi Everyone. I would be glad to help, but I only a web developer. I just got my nzxt kraken z73, please, let me know if you need anything.

Also, can I sync my CPU temp in nzxt display with MacOS?

Webxorcist, 2020-12-15:
Hello,

I just bought and installed the NZXT Kraken Z73. This is my output:

:~> lsusb -d 1e71:3008 -v

Bus 001 Device 003: ID 1e71:3008 NZXT NZXT KrakenZ Device
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x3008 
  bcdDevice            5.00
  iManufacturer           1 NZXT Inc.
  iProduct                2 NZXT KrakenZ Device
  iSerial                 3 3464337E3239
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0039
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               5
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     598
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5

Webxorcist, 2020-12-15:
I got liquitctl working on my Suse Tumbleweed system. Thanks for your efforts!
Why would I need to put in my own settings as for fan/pump speeds? I can already see that if the temp rises, the pump will work harder. Why would I need to tinker with this?

I am really happy I can now read my temperature and can do some sort of display design.

inglor, 2020-12-15:
@Webxorcist you don't _need_ to. But now you have the option to adjust the curve of pump/fan.

I think the pump / fans are using an internal hardcoded curve which is the default if you don't provide anything. This is like a failsafe for the AIO to not sit idle when you start the PC or in unsupported OSes. You can tell there's also there's also a failsafe for the screen to show the liquid temp in white background when you initially start the AIO (after a cold shutdown) and before loading CAM software in windows.

[35 more replies]

closed 2022-04-05 by jonasmalacofilho