liquidctl/liquidctl #80: Support for Corsair H100i, H115i and H150i PRO XT coolers

jonasmalacofilho, 2020-01-29:
Corsair has launched a new PRO XT line, very unlike the previous PRO series: these devices use CoolIT pumps (PRO series used a Asetek 6<sup>th</sup> gen. design) and the non-RGB are fast now (PRO series used low-speed fans).

[H100i PRO XT](https://www.corsair.com/us/en/Categories/Products/Liquid-Cooling/iCUE-RGB-PRO-XT-Coolers/p/CW-9060043-WW)
[H115i PRO XT](https://www.corsair.com/us/en/Categories/Products/Liquid-Cooling/iCUE-RGB-PRO-XT-Coolers/p/CW-9060044-WW)
[H150i PRO XT](https://www.corsair.com/us/en/Categories/Products/Liquid-Cooling/iCUE-RGB-PRO-XT-Coolers/p/CW-9060045-WW)

Corsair claims these are the most performant coolers they make so far, specifically stating that the CoolIT pump is a significant improvement over the Asetek one.  If that's indeed true (at reasonable fan speeds/noise levels), the H150i PRO XT might end up being quite popular.

I also imagine that the protocol will be the same (or share a lot with the one) used in the RGB Platinum line (see #76).

---

Status:

 - [ ] Collect basic USB data for both devices (at least vendor and product IDs)
 - [ ] Inspect and analyze the new protocol (monitoring, cooling and lighting)
 - [ ] Implement and test the new driver

audiohacked, 2020-03-12:
I just ordered the H115i PRO XT. As soon as I have time, I'll get some protocol dumps.

jonasmalacofilho, 2020-03-25:
@audiohacked that's great, thanks.

viggy96, 2020-04-23:
Does this apply to the non-XT models as well, like the [H150i Pro](https://www.corsair.com/us/en/Categories/Products/Liquid-Cooling/Hydro-Series%E2%84%A2-PRO-RGB-Liquid-CPU-Coolers/p/CW-9060031-WW)? Or is that device already supported?

jonasmalacofilho, 2020-04-23:
@viggy96 for the non-XT Pros you should follow #106.

audiohacked, 2020-06-05:
@jonasmalacofilho I think the protocol for the RGB Pro XT is the same as the Platinum.

jonasmalacofilho, 2020-06-05:
@audiohacked 

That's great to know/confirm.  But how is the additional fan of the H150i handled?

In the Platinum responses we have distinct fields for each fan speed, and the pump speed is in the way of naively adding a third fan to that response.

audiohacked, 2020-06-05:
I have both the H100i RGB Platinum and the H115i RGB Pro XT (installed), and both only control 2 fans. I would assume for the H150i that the 3rd fan is inserted into where the pump is on the H100i and H115i. I've seen the pattern that the physical fans are then followed by the pump (across any/majority Corsair AIO). 

Also, I've been mapping the LED control and FAN control.

jonasmalacofilho, 2020-06-05:
That makes sense.  When I have some time I'll use a mock USB device to confirm this (and will let you know).

Another question: how are you handling the sequence numbers in each command?  I remember reading somewhere that ignoring them (i.e. sending a constant value) eventually caused issues like making the device unresponsive.  How have you implemented that, and what has been your experience so far?

audiohacked, 2020-06-05:
I haven't implemented anything other than what's in liquidctl already. I've been using Wireshark and capturing packets.

audiohacked, 2020-06-05:
I was thinking of retiring OpenCorsairLink and redirecting users to use liquidctl.

jonasmalacofilho, 2020-06-06:
If you do (or even if you don't), I would be very happy to have you here, maybe focusing on new drivers or involved in some other way that interests you.

audiohacked, 2020-06-06:
It's getting harder to work on OCL, since fewer people are able to program in C, and with my job I'm always traveling. So, I think it might be best if OCL is retired.

audiohacked, 2020-06-06:
I think I've pretty much nailed down the Fan Control. https://github.com/audiohacked/Corsair-RE/blob/master/Corsair/AIO/RGB_Pro_XT/Fan_control.txt

jonasmalacofilho, 2020-06-11:
@audiohacked

Thanks, I'm taking a look.

jonasmalacofilho, 2020-06-13:
Can I ask some of you to review (and, at your discretion, test) the code in #82?  I would like to known that the most basic stuff works before implementing the remaining features.

Note that they may be gross errors in the implementation since I don't have any way to test that code.

jonasmalacofilho, 2020-06-16:
Now asking you to specifically test the code.  Please take a look at [#82 (comment)](https://github.com/jonasmalacofilho/liquidctl/pull/82#issuecomment-644647183).

closed 2020-06-19 by jonasmalacofilho

ShaunTheQuietGamer, 2020-07-03:
My H100i RGB PRO XT is initialized, the fan control works, but lighting does not seem to be working at all. (No idea how this works on the backend, if you need me to paste the output of any commands just ask, basically, i'm useless with any of the important end of this.)

jonasmalacofilho, 2020-07-03:
@ShaunTheQuietGamer 

I'll try to take a look this weekend, but can you please open a new issue to remind me?  And yes, please include the arguments and output of the commands that fail (adding the `--debug` flag for additional information).

Steamkraft, 2020-08-08:
Hi - I just received my H150i PRO XT and would love to help getting this to work :)

jonasmalacofilho, 2020-08-08:
@Steamkraft 

A few changes are needed to support the H150i.  You can take a look at #138 and see if you're able to figure out/implement the necessary modifications to the [`hydro_platinum`](https://github.com/jonasmalacofilho/liquidctl/blob/master/liquidctl/driver/hydro_platinum.py) module.

There's some additional documentation specific for developers in [`docs/developer/`](https://github.com/jonasmalacofilho/liquidctl/tree/master/docs/developer), and all previous traffic captures are archived in [liquidctl-device-data](https://github.com/jonasmalacofilho/liquidctl-device-data) (but since you have one of these coolers, you're probably better off doing your own captures).

---

Locking this issue because it doesn't make sense to continue discussions here.